openwrt/mt76 #471: mt76x2u: could not get hardware semaphore for ROM PATCH

paul-humanitas, 2020-11-05:
Hi,
we have a device with a HC-M7662BU1.

When we switch off and on the device, the driver is correctly loaded and we can use the device properly.
Then, if we issue a reboot, it will work fine but only one time. For any subsequent reboot, an error will happen and the Wi-Fi interface will not be recognized.

Initial boot or reboot:
> [   10.063318] usb 2-1: reset high-speed USB device number 2 using ehci-platform
> [   10.097507] mt76x2u: probe of 2-1:1.0 failed with error -22
> [   10.235282] usb 2-1: reset high-speed USB device number 2 using ehci-platform
> [   10.269454] mt76x2u: probe of 2-1:1.1 failed with error -22
> [   10.407280] usb 2-1: reset high-speed USB device number 2 using ehci-platform
> [   10.441657] mt76x2u 2-1:1.2: ASIC revision: 76620044
> [   10.502163] mt76x2u 2-1:1.2: ROM patch build: 20141115060606a
> [   10.696531] mt76x2u 2-1:1.2: Firmware Version: 0.0.00
> [   10.701602] mt76x2u 2-1:1.2: Build: 1
> [   10.705271] mt76x2u 2-1:1.2: Build Time: 201507311614____
> [   10.922154] mt76x2u 2-1:1.2: error: MCU resp evt:9 seq:1-0
> [   10.927778] mt76x2u 2-1:1.2: error: MCU resp evt:9 seq:1-0
> [   10.933402] mt76x2u 2-1:1.2: error: MCU resp evt:9 seq:1-0
> [   11.550495] usbcore: registered new interface driver mt76x2u

2+ reboot: 
> [   10.075449] usb 2-1: reset high-speed USB device number 2 using ehci-platform
> [   10.109687] mt76x2u: probe of 2-1:1.0 failed with error -22
> [   10.247413] usb 2-1: reset high-speed USB device number 2 using ehci-platform
> [   10.281503] mt76x2u: probe of 2-1:1.1 failed with error -22
> [   10.419410] usb 2-1: reset high-speed USB device number 2 using ehci-platform
> [   10.453955] mt76x2u 2-1:1.2: ASIC revision: 76620044
> [   11.744222] **mt76x2u 2-1:1.2: could not get hardware semaphore for ROM PATCH**
> [   12.781663] mt76x2u: probe of 2-1:1.2 failed with error -110
> [   12.787463] usbcore: registered new interface driver mt76x2u

We also tried with a netgear A6210 dongle, and had no issue
> [   10.124718] mt76x2u 3-1:1.0: ASIC revision: 76120044
>[   10.161951] mt76x2u 3-1:1.0: ROM patch build: 20141115060606a
>[   10.299100] mt76x2u 3-1:1.0: Firmware Version: 0.0.00
>[   10.304161] mt76x2u 3-1:1.0: Build: 1
>[   10.307833] mt76x2u 3-1:1.0: Build Time: 201507311614____
>[   11.096936] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
>[   11.098759] usbcore: registered new interface driver mt76x2u

paul-humanitas, 2020-11-09:
After some testing, we noticed the following:
on the initial boot, a ROM patch is applied.
on first reboot, the ROM patch is reported as already applied.
on any  subsequent  reboot, the driver fails to get the semaphore for ROM patch.

We assumed that the action to check if the ROM patch is already patched might fail to free the hardware semaphore, but we are unfortunately unable to tell why.

For our own application, we chose to change the return value when the hardware semaphore failure happen. Instead of returning an error code, we return 0. We are then able to use the dongle without any issue.
Hopefully, the actual semaphore error will not happen for us.

FaarisAnsari, 2021-12-26:
I seem to be having this exact issue with my XBOX Wireless Adapter, with the same code: mt76x2u. For me, it only seems to occur when I am using the runit init system, it seems to be fine on SystemD.