openwrt/mt76 #472: TL MR3420 v5: Wifi lost at some time

phmarek, 2020-11-06:
Similar issues from the past: #188, #198 

Since some upgrade half a year ago or so my MR3420-v5 isn't "on air" sooner or later.
I can still get in via cable; there's nothing interesting in the `logread` or `dmesg` output (no kernel crash or similar), but no clients can see the ESSID.

Rebooting or, as #198 states, changing the channel, makes Wifi come up again.

Is there something I can do to help debugging the issue? Currently I'm running r14808-499924adf0.

As the other tickets are closed I'm opening a new one.

Thank you!

phmarek, 2020-11-06:
Seems that was a lucky coincidence... right now neither changing the channel nor en/disabling the interface makes the ESSID visible again.

phmarek, 2020-11-06:
But just now changing the mode succeeded in giving back my Wifi:

```
uci set wireless.default_radio0.mode=ad-hoc ; reload_config ; sleep 1 ; uci set wireless.default_radio0.mode=ap ; reload_config
```

phmarek, 2020-11-06:
No, next time this didn't help again.

Is there some command I can use to reset the Wifi interface? `ifconfig wlan0 down | up` isn't enough; rebooting takes much too long and loses all conntrack data.

phmarek, 2020-11-07:
`echo 10300000.wmac > /sys/bus/platform/drivers/mt76_wmac/unbind` disrupts Wifi all right... but doing a `bind` that way reboots OpenWRT.

stre425, 2020-11-19:
The same Issue is also present on a TP-Link WR841N v13 on the latest stable release 19.07.04, which uses the same hardware. It can be triggered by transferring data with rates of ~5 Mbyte/s via wifi for 1-2 Minutes (e.g by downloading a larger file from a reasonably fast server) via wifi. After this time, the AP drops all connections and the SSID is not visible any more. As phmarek mentioned there are not suspcious entrys in the logs. Connection via cable is still present. 
Trying to scan for wifi cells from the router in this state using
`iwinfo wlan0 scan`
 does not show any neighboring cells and triggers a couple of kernel errors
'[154633.629472] mt76_wmac 10300000.wmac: MCU message 8 (seq 6) timed out
[154638.749666] mt76_wmac 10300000.wmac: MCU message 8 (seq 7) timed out
[154643.869555] mt76_wmac 10300000.wmac: MCU message 8 (seq 8) timed out
[154648.989521] mt76_wmac 10300000.wmac: MCU message 8 (seq 9) timed out
[154654.109636] mt76_wmac 10300000.wmac: MCU message 8 (seq 10) timed out
[154659.230076] mt76_wmac 10300000.wmac: MCU message 8 (seq 11) timed out
[154664.349981] mt76_wmac 10300000.wmac: MCU message 8 (seq 12) timed out
[154669.469086] mt76_wmac 10300000.wmac: MCU message 8 (seq 13) timed out
[154674.589239] mt76_wmac 10300000.wmac: MCU message 8 (seq 14) timed out
[154679.709615] mt76_wmac 10300000.wmac: MCU message 8 (seq 15) timed out
[154684.828882] mt76_wmac 10300000.wmac: MCU message 8 (seq 1) timed out'
However after 5-10 minutes the AP recovers on it's own, notices that all clients are inactive, disassociates them and then the SSID of the router is visible again and all clients can reconnect.

phmarek, 2020-11-19:
Can you tell which part needs a heatsink? ;)