openwrt/mt76 #473: secondary AP not running if in STA mode

alex-eri, 2020-11-14:
STA works, but without WDS support

OpenWrt 19.07.4, r11208-ce6496d796

in log I have many lines

```
daemon.err hostapd: Failed to set beacon parameters
daemon.notice hostapd: handle_probe_req: send failed

```

lukasz1992, 2020-11-16:
Don't count on WDS: https://patchwork.kernel.org/project/linux-wireless/patch/20201113101148.25268-1-johannes@sipsolutions.net/

kofec, 2020-11-16:
But this is some kind of old version of WDS:
https://patchwork.kernel.org/project/linux-wireless/patch/1476779337-3520-1-git-send-email-johannes@sipsolutions.net/
_The old WDS 4-addr frame support is very limited, e.g.
 * no encryption is possible on such links
 * it cannot support rate/HT/VHT negotiation
 * management APIs are very restricted

These make the WDS legacy mode useless in practice.

**All of these are resolved by the 4-addr AP/client support,
so there's also no reason to improve WDS in the future.**

Therefore, add a Kconfig option to disable legacy WDS.
This gives people an "emergency valve" while they migrate
to the better-supported 4-addr AP/client option; we plan
to remove it (and the associated cfg80211/mac80211 code,
which is the ultimate goal) in the future._

alex-eri, 2020-11-18:
Problem exists with WDS off. I tryed pseudobridge with relayd too.