openwrt/mt76 #474: MT7615 DBDC config via luci web not work property

ptpt52, 2020-11-16:
1. cannot change channel
2. cannot scan 2.4g

radio0 should be 11g mode but default is 11ac
radio1 should be 11ac mode but default is not.

I have to change radio0 to 11g and radio1 to 11ac, so that it works

the ubus resault:
```
[{
	"id": 380,
	"jsonrpc": "2.0",
	"result": {
		"hostapd.wlan0": {
			"wps_status": [],
			"list_bans": [],
			"switch_chan": {
				"center_freq1": "unknown",
				"center_freq2": "unknown",
				"ht": "unknown",
				"freq": "unknown",
				"sec_channel_offset": "unknown",
				"bandwidth": "unknown",
				"vht": "unknown",
				"block_tx": "unknown",
				"bcn_count": "unknown"
			},
			"rrm_nr_list": [],
			"del_client": {
				"deauth": "unknown",
				"addr": "object",
				"ban_time": "unknown",
				"reason": "unknown"
			},
			"get_clients": [],
			"bss_mgmt_enable": {
				"neighbor_report": "unknown",
				"bss_transition": "unknown",
				"beacon_report": "unknown"
			},
			"rrm_nr_set": {
				"list": "unknown"
			},
			"set_vendor_elements": {
				"vendor_elements": "object"
			},
			"rrm_nr_get_own": [],
			"reload": [],
			"rrm_beacon_req": {
				"addr": "object",
				"ssid": "object",
				"op_class": "unknown",
				"channel": "unknown",
				"bssid": "object",
				"duration": "unknown",
				"mode": "unknown"
			},
			"notify_response": {
				"notify_response": "unknown"
			},
			"wps_cancel": [],
			"wps_start": [],
			"get_features": [],
			"update_beacon": [],
			"wnm_disassoc_imminent": {
				"duration": "unknown",
				"neighbors": "unknown",
				"abridged": "unknown",
				"addr": "object"
			}
		},
		"hostapd.wlan1": {
			"wps_status": [],
			"list_bans": [],
			"switch_chan": {
				"center_freq1": "unknown",
				"center_freq2": "unknown",
				"ht": "unknown",
				"freq": "unknown",
				"sec_channel_offset": "unknown",
				"bandwidth": "unknown",
				"vht": "unknown",
				"block_tx": "unknown",
				"bcn_count": "unknown"
			},
			"rrm_nr_list": [],
			"del_client": {
				"deauth": "unknown",
				"addr": "object",
				"ban_time": "unknown",
				"reason": "unknown"
			},
			"get_clients": [],
			"bss_mgmt_enable": {
				"neighbor_report": "unknown",
				"bss_transition": "unknown",
				"beacon_report": "unknown"
			},
			"rrm_nr_set": {
				"list": "unknown"
			},
			"set_vendor_elements": {
				"vendor_elements": "object"
			},
			"rrm_nr_get_own": [],
			"reload": [],
			"rrm_beacon_req": {
				"addr": "object",
				"ssid": "object",
				"op_class": "unknown",
				"channel": "unknown",
				"bssid": "object",
				"duration": "unknown",
				"mode": "unknown"
			},
			"notify_response": {
				"notify_response": "unknown"
			},
			"wps_cancel": [],
			"wps_start": [],
			"get_features": [],
			"update_beacon": [],
			"wnm_disassoc_imminent": {
				"duration": "unknown",
				"neighbors": "unknown",
				"abridged": "unknown",
				"addr": "object"
			}
		}
	}
}]
```

ptpt52, 2020-11-16:
```
iw phy
Wiphy phy1
	wiphy index: 1
	max # scan SSIDs: 4
	max scan IEs length: 543 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	max # scan plans: 1
	max scan plan interval: 10
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 2:
		Capabilities: 0x1ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		VHT Capabilities (0x338001b1):
			Max MPDU length: 7991
			Supported Channel Width: neither 160 nor 80+80
			RX LDPC
			short GI (80 MHz)
			TX STBC
			RX antenna pattern consistency
			TX antenna pattern consistency
		VHT RX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT RX highest supported: 0 Mbps
		VHT TX MCS set:
			1 streams: MCS 0-9
			2 streams: MCS 0-9
			3 streams: not supported
			4 streams: not supported
			5 streams: not supported
			6 streams: not supported
			7 streams: not supported
			8 streams: not supported
		VHT TX highest supported: 0 Mbps
[... 179 lines cut ...]
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 16,
		   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }

	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	Supported extended features:
		* [ VHT_IBSS ]: VHT-IBSS
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ DEL_IBSS_STA ]: deletion of IBSS station support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support

```

porentak, 2020-12-15:
Since it is not possible to use 5GHz bands on first phy interface in DBDC mode, I changed the code to disable it:

`--- a/mt7615/eeprom.c`
`+++ b/mt7615/eeprom.c`
`@@ -127,7 +127,8 @@ mt7615_eeprom_parse_hw_band_cap(struct mt7615_dev *dev)`
` 		break;`
` 	case MT_EE_DBDC:`
` 		dev->dbdc_support = true;`
`-		/* fall through */`
`+		dev->mphy.cap.has_2ghz = true;`
`+		break;`
` 	default:`
` 		dev->mphy.cap.has_2ghz = true;`
` 		dev->mphy.cap.has_5ghz = true;`

Is it safe to do it this way?

ryderlee1110, 2021-01-09:
I think this is not related to mt76. You can manually enable it.

syb999, 2021-02-21:
Firest, use the patch(https://github.com/openwrt/mt76/pull/476) to fix macaddress duplication.
Modify /etc/config/wireless,  and use option macaddr instead of option path.
Now mt7615 dbdc can config on Luci wireless webpage.

ivanich, 2021-03-22:
Workaround https://github.com/openwrt/mt76/issues/483#issuecomment-801163614

ptpt52, 2021-04-02:
> May works when using `phy` instead of `path`. Not sure, because i don't have 5ghz devices.
> But after this all luci settings and scanning works fine.
> 
> Working example:
> 
> ```
> config wifi-device 'radio0'
> 	option type 'mac80211'
> 	option phy 'phy0'
> 	option htmode 'HT40'
> 	option serialize '1'
> 	option country 'US'
> 	option cell_density '0'
> 	option hwmode '11g'
> 	option channel '1'
> 
> config wifi-device 'radio1'
> 	option type 'mac80211'
> 	option phy 'phy1'
> 	option serialize '1'
> 	option country 'US'
> 	option cell_density '0'
> 	option hwmode '11a'
> 	option htmode 'VHT80'
> 	option channel '36'
> 	option txpower '20'
> ```

this really works

ryderlee1110, 2021-04-13:
note: there's a strict order for dbdc, so I don't see any issues here (including #483 (comment)). Just need to serialize interfaces.