openwrt/mt76 #475: mt7603e: MCU message -1 (seq 1) timed out

nekromant, 2020-11-19:
There seems to be a weird issue with a miniPCIe card and mt76 on a big endian system. Here's the relevant pieces of the log:

```
[    2.529610] nwl-pcie 1018000000.pcie: Link is UP
[    2.530145] nwl-pcie 1018000000.pcie: host bridge /plb/xhsif1/pcie@1018000000 ranges:
[    2.530959] nwl-pcie 1018000000.pcie:      MEM 0x101a020000..0x101fffffff -> 0x001a020000
[    2.531786] nwl-pcie 1018000000.pcie:      MEM 0x1300000000..0x13ffffffff -> 0x0400000000
[    2.532644] nwl-pcie 1018000000.pcie:   IB MEM 0x0100000000..0x01ffffffff -> 0x0000000000
[    2.533470] nwl-pcie 1018000000.pcie:   IB MEM 0x0200000000..0x02ffffffff -> 0x0200000000
[    2.534668] nwl-pcie 1018000000.pcie: PCI host bridge to bus 0000:00
[    2.535258] pci_bus 0000:00: root bus resource [bus 00-0f]
[    2.535747] pci_bus 0000:00: root bus resource [mem 0x101a020000-0x101fffffff] (bus address [0x1a020000-0x1fffffff])
[    2.536823] pci_bus 0000:00: root bus resource [mem 0x1300000000-0x13ffffffff] (bus address [0x400000000-0x4ffffffff])
[    2.537946] pci 0000:00:00.0: [19aa:e00c] type 01 class 0x060400
[    2.538562] pci 0000:00:00.0: enabling Extended Tags
[    2.539086] pci 0000:00:00.0: supports D1 D2
[    2.543559] pci 0000:01:00.0: [14c3:7603] type 00 class 0x028000
[    2.544169] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff]
[    2.545023] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
[    2.549387] pci 0000:00:00.0: PCI bridge to [bus 01-0c]
[    2.549943] pci 0000:00:00.0: BAR 8: assigned [mem 0x101a100000-0x101a1fffff]
[    2.550626] pci 0000:01:00.0: BAR 0: assigned [mem 0x101a100000-0x101a1fffff]
[    2.551311] pci 0000:00:00.0: PCI bridge to [bus 01-0c]
[    2.551770] pci 0000:00:00.0:   bridge window [mem 0x101a100000-0x101a1fffff]
....
[ 1004.931655] mt76: loading out-of-tree module taints kernel.
[ 1013.580432] mt7603e 0000:01:00.0: ASIC revision: 76030010
[ 1013.590463] mt7603e 0000:01:00.0: Firmware Version: ap_pcie
[ 1013.590981] mt7603e 0000:01:00.0: Build Time: 20160107100755
[ 1016.649794] mt7603e 0000:01:00.0: MCU message -1 (seq 1) timed out
[ 1016.650383] mt7603e 0000:01:00.0: Download request failed
[ 1016.651125] mt7603e: probe of 0000:01:00.0 failed with error -110
```

It looks like the firmware loading doesn't work. The system is a big endian PowerPC SoC (ppc476fp, to be precise).
The same miniPCIe adapter seems to work on a little endian x86 laptop. I'm not sure whether it's a problem with mt76 or our system, because ath9k and other PCIe devices seem to work with no issue. 

Versions of mt76 tested so far: mainline as of v5.5.0 and commit b54157df7c272df0186cad4efc4edf976da5f96f

ptpt52, 2021-03-25:
it is likely issue in mt76 with wrong endian handle?

nekromant, 2021-03-26:
For some reason it works on big endian mips with no issue.