openwrt/mt76 #477: Warning on CPU

gregd72002, 2020-11-24:
Openwrt 19.07.4
Router: ZBT-WE1326
SoC Type: MediaTek MT7621 

The router seemed to have recovered.

```
[62003.671208] ------------[ cut here ]------------
[62003.675906] WARNING: CPU: 2 PID: 19 at backports-4.19.137-1/net/mac80211/tx.c:4292 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[62003.686885] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base
[62003.758052]  usb_common
[62003.760558] CPU: 2 PID: 19 Comm: ksoftirqd/2 Not tainted 4.14.195 #0
[62003.766881] Stack : 00000000 00000000 00000000 8e899030 00000000 00000000 00000000 00000000
[62003.775214]         00000000 00000000 00000000 00000000 00000000 00000001 8fc83b70 ac07f5da
[62003.783551]         8fc83c08 00000000 00000000 00005d70 00000038 8049c858 00000007 00000000
[62003.791888]         00000000 80550000 000b9aee 00000000 8fc83b50 00000000 00000000 8f1e7fd0
[62003.800224]         8f1acbc4 000010c4 8fc13480 8e899030 00000003 802ad210 00000008 806b0008
[62003.808558]         ...
[62003.810999] Call Trace:
[62003.811059] [<8049c858>] 0x8049c858
[62003.816967] [<8f1acbc4>] 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[62003.822865] [<802ad210>] 0x802ad210
[62003.826344] [<8000c1a0>] 0x8000c1a0
[62003.829819] [<8000c1a8>] 0x8000c1a8
[62003.833298] [<804856c4>] 0x804856c4
[62003.836773] [<8f1acbc4>] 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[62003.842663] [<80072a84>] 0x80072a84
[62003.846134] [<8002e608>] 0x8002e608
[62003.849613] [<8f1acbc4>] 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[62003.855519] [<8002e6f0>] 0x8002e6f0
[62003.858992] [<80053b7c>] 0x80053b7c
[62003.862467] [<8f1acbc4>] 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[62003.868374] [<8f1af408>] 0x8f1af408 [mac80211@8f180000+0x6cca0]
[62003.874298] [<8e907b90>] 0x8e907b90 [mt76x02_lib@8e900000+0x96c0]
[62003.880374] [<8f1af86c>] 0x8f1af86c [mac80211@8f180000+0x6cca0]
[62003.886283] [<8e907bd0>] 0x8e907bd0 [mt76x02_lib@8e900000+0x96c0]
[62003.892362] [<8f1b65e0>] 0x8f1b65e0 [mac80211@8f180000+0x6cca0]
[62003.898272] [<8f1b66bc>] 0x8f1b66bc [mac80211@8f180000+0x6cca0]
[62003.904173] [<8e904a94>] 0x8e904a94 [mt76x02_lib@8e900000+0x96c0]
[62003.910247] [<8f1f5000>] 0x8f1f5000 [mt76@8f1f0000+0x79c0]
[62003.915719] [<800329f4>] 0x800329f4
[62003.919194] [<804a3658>] 0x804a3658
[62003.922664] [<8004fd10>] 0x8004fd10
[62003.926139] [<8004fd10>] 0x8004fd10
[62003.929615] [<80032cc8>] 0x80032cc8
[62003.933092] [<8004feb8>] 0x8004feb8
[62003.936560] [<8049ebd8>] 0x8049ebd8
[62003.940044] [<8004bf38>] 0x8004bf38
[62003.943520] [<8004be08>] 0x8004be08
[62003.947006] [<8004be08>] 0x8004be08
[62003.950493] [<8004be08>] 0x8004be08
[62003.953969] [<80006f78>] 0x80006f78
[62003.957448] 
[62003.959125] ---[ end trace 76264dc1c9d8024b ]---
root@OpenWrt:~# 
```

gregd72002, 2020-11-25:
It happened again just now. I lost wifi for a while and then the router recovered.

```
[ 7573.591644] ------------[ cut here ]------------
[ 7573.596327] WARNING: CPU: 2 PID: 19 at backports-4.19.137-1/net/mac80211/tx.c:4292 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[ 7573.607285] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base
[ 7573.678404]  usb_common
[ 7573.680889] CPU: 2 PID: 19 Comm: ksoftirqd/2 Not tainted 4.14.195 #0
[ 7573.687214] Stack : 00000000 00000000 00000000 8e249d30 00000000 00000000 00000000 00000000
[ 7573.695552]         00000000 00000000 00000000 00000000 00000000 00000001 8fc83b70 ac07f5da
[ 7573.703890]         8fc83c08 00000000 00000000 00005cc8 00000038 8049c858 00000007 00000000
[ 7573.712229]         00000000 80550000 000a63b9 00000000 8fc83b50 00000000 00000000 8f1e7fd0
[ 7573.720565]         8f1acbc4 000010c4 8fd51480 8e249d30 00000003 802ad210 00000008 806b0008
[ 7573.728904]         ...
[ 7573.731342] Call Trace:
[ 7573.731390] [<8049c858>] 0x8049c858
[ 7573.737292] [<8f1acbc4>] 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[ 7573.743189] [<802ad210>] 0x802ad210
[ 7573.746668] [<8000c1a0>] 0x8000c1a0
[ 7573.750139] [<8000c1a8>] 0x8000c1a8
[ 7573.753609] [<804856c4>] 0x804856c4
[ 7573.757081] [<8f1acbc4>] 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[ 7573.762973] [<80072a84>] 0x80072a84
[ 7573.766460] [<8002e608>] 0x8002e608
[ 7573.769944] [<8f1acbc4>] 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[ 7573.775870] [<8002e6f0>] 0x8002e6f0
[ 7573.779347] [<80053b7c>] 0x80053b7c
[ 7573.782824] [<8f1acbc4>] 0x8f1acbc4 [mac80211@8f180000+0x6cca0]
[ 7573.788732] [<8f1af408>] 0x8f1af408 [mac80211@8f180000+0x6cca0]
[ 7573.794651] [<80067ca8>] 0x80067ca8
[ 7573.798133] [<80017f6c>] 0x80017f6c
[ 7573.801634] [<8e927b90>] 0x8e927b90 [mt76x02_lib@8e920000+0x96c0]
[ 7573.807707] [<8f1af86c>] 0x8f1af86c [mac80211@8f180000+0x6cca0]
[ 7573.813625] [<8e927bd0>] 0x8e927bd0 [mt76x02_lib@8e920000+0x96c0]
[ 7573.819712] [<8f1b65e0>] 0x8f1b65e0 [mac80211@8f180000+0x6cca0]
[ 7573.825627] [<8f1b66bc>] 0x8f1b66bc [mac80211@8f180000+0x6cca0]
[ 7573.831531] [<8e924a94>] 0x8e924a94 [mt76x02_lib@8e920000+0x96c0]
[ 7573.837606] [<800329f4>] 0x800329f4
[ 7573.841087] [<804a3658>] 0x804a3658
[ 7573.844560] [<8004fd10>] 0x8004fd10
[ 7573.848034] [<8004fd10>] 0x8004fd10
[ 7573.851506] [<80032cc8>] 0x80032cc8
[ 7573.854983] [<8004feb8>] 0x8004feb8
[ 7573.858451] [<8049ebd8>] 0x8049ebd8
[ 7573.861944] [<8004bf38>] 0x8004bf38
[ 7573.865417] [<8004be08>] 0x8004be08
[ 7573.868892] [<8004be08>] 0x8004be08
[ 7573.872370] [<8004be08>] 0x8004be08
[ 7573.875847] [<80006f78>] 0x80006f78
[ 7573.879329] 
[ 7573.880994] ---[ end trace 02d9eb71de4d95d3 ]---
root@OpenWrt:~# cat /proc/uptime 
8611.21 33651.55
```

gregd72002, 2020-11-28:
And again
```
[89544.419153] ------------[ cut here ]------------
[89544.423841] WARNING: CPU: 2 PID: 19 at backports-4.19.137-1/net/mac80211/tx.c:4292 0x8f22cbc4 [mac80211@8f200000+0x6cca0]
[89544.434793] Modules linked in: pppoe ppp_async pppox ppp_generic pl2303 nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE ftdi_sio cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT usbserial slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug
[89544.506722]  usbcore nls_base usb_common
[89544.510659] CPU: 2 PID: 19 Comm: ksoftirqd/2 Not tainted 4.14.195 #0
[89544.516984] Stack : 00000000 00000000 00000000 8ea14c30 00000000 00000000 00000000 00000000
[89544.525321]         00000000 00000000 00000000 00000000 00000000 00000001 8fc83b70 ac07f5da
[89544.533660]         8fc83c08 00000000 00000000 00006340 00000038 8049c858 00000007 00000000
[89544.541997]         00000000 80550000 0007cac3 00000000 8fc83b50 00000000 00000000 8f267fd0
[89544.550332]         8f22cbc4 000010c4 8e31b480 8ea14c30 00000003 802ad210 00000008 806b0008
[89544.558673]         ...
[89544.561113] Call Trace:
[89544.561168] [<8049c858>] 0x8049c858
[89544.567077] [<8f22cbc4>] 0x8f22cbc4 [mac80211@8f200000+0x6cca0]
[89544.572971] [<802ad210>] 0x802ad210
[89544.576448] [<8000c1a0>] 0x8000c1a0
[89544.579920] [<8000c1a8>] 0x8000c1a8
[89544.583390] [<804856c4>] 0x804856c4
[89544.586868] [<8f22cbc4>] 0x8f22cbc4 [mac80211@8f200000+0x6cca0]
[89544.592761] [<80072a84>] 0x80072a84
[89544.596234] [<8002e608>] 0x8002e608
[89544.599715] [<8f22cbc4>] 0x8f22cbc4 [mac80211@8f200000+0x6cca0]
[89544.605625] [<8002e6f0>] 0x8002e6f0
[89544.609100] [<80053b7c>] 0x80053b7c
[89544.612581] [<8f22cbc4>] 0x8f22cbc4 [mac80211@8f200000+0x6cca0]
[89544.618487] [<8f22f408>] 0x8f22f408 [mac80211@8f200000+0x6cca0]
[89544.624404] [<80067ca8>] 0x80067ca8
[89544.627882] [<80017f6c>] 0x80017f6c
[89544.631365] [<8f287b90>] 0x8f287b90 [mt76x02_lib@8f280000+0x96c0]
[89544.637443] [<8f22f86c>] 0x8f22f86c [mac80211@8f200000+0x6cca0]
[89544.643351] [<8f287bd0>] 0x8f287bd0 [mt76x02_lib@8f280000+0x96c0]
[89544.649431] [<8f2365e0>] 0x8f2365e0 [mac80211@8f200000+0x6cca0]
[89544.655340] [<8f2366bc>] 0x8f2366bc [mac80211@8f200000+0x6cca0]
[89544.661255] [<8f284a94>] 0x8f284a94 [mt76x02_lib@8f280000+0x96c0]
[89544.667343] [<800329f4>] 0x800329f4
[89544.670822] [<804a3658>] 0x804a3658
[89544.674293] [<8004fd10>] 0x8004fd10
[89544.677774] [<8004fd10>] 0x8004fd10
[89544.681245] [<80032cc8>] 0x80032cc8
[89544.684720] [<8004feb8>] 0x8004feb8
[89544.688191] [<8049ebd8>] 0x8049ebd8
[89544.691675] [<8004bf38>] 0x8004bf38
[89544.695148] [<8004be08>] 0x8004be08
[89544.698628] [<8004be08>] 0x8004be08
[89544.702108] [<8004be08>] 0x8004be08
[89544.705579] [<80006f78>] 0x80006f78
[89544.709055] 
[89544.710715] ---[ end trace b9c76b428c96aa63 ]---
```

xavier83, 2020-11-29:
Hi @gregd72002 , I have the same device (512 MB version). Is the 5GHz radio working for you? for me it isn't and the latest Openwrt snapshots for it is going in a boot loop.

gregd72002, 2020-11-29:
@xavier83, I'm on v19.07.4 and both 2.4 and 5ghz work fine. Just occasionally I'm getting these kernel warnings.
I've never tried snapshot.

xavier83, 2020-11-29:
@gregd72002 I am not seeing this error though, do you have 512MB ram version or the 256MB ram version. 
Also can you please share the following to compare against mine even I am running 19.07.4 since that is the latest working version but 5ghz is still dead in it. I am looking to figure out if it is because of some 
'eeprom' values had been overwritten/updated by my experiments with one of the snapshot packages, If you can help me will be very much appreciated, I'm stuck with this issue since Oct 2018 https://github.com/openwrt/mt76/issues/199 . Thanks
```
root@OpenWrt:~# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00030000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00010000 00010000 "factory"
mtd3: 00fb0000 00010000 "firmware"
mtd4: 001c9c2c 00010000 "kernel"
mtd5: 00de63d4 00010000 "rootfs"
mtd6: 00b50000 00010000 "rootfs_data"
```
and the firmware info from my dmesg log contains
```
[   13.970568] bus=0x2, slot = 0x1, irq=0xff
[   13.974828] mt7603e 0000:02:00.0: ASIC revision: 76030010
[   15.013904] mt7603e 0000:02:00.0: Firmware Version: ap_pcie
[   15.019478] mt7603e 0000:02:00.0: Build Time: 20160107100755
[   15.062496] mt7603e 0000:02:00.0: firmware init done
[   15.232930] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   15.250525] bus=0x1, slot = 0x0, irq=0xff
[   15.254826] mt76x2e 0000:01:00.0: ASIC revision: 76120044
[   15.967234] mt76x2e 0000:01:00.0: ROM patch build: 20141115060606a
[   16.036855] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[   16.042338] mt76x2e 0000:01:00.0: Build: 1
[   16.046506] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[   16.072500] mt76x2e 0000:01:00.0: Firmware running!
[   16.080400] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
```

```
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 2.8M      2.8M         0 100% /rom
tmpfs                   250.6M    452.0K    250.2M   0% /tmp
/dev/mtdblock6           11.3M      1.4M      9.9M  13% /overlay
overlayfs:/overlay       11.3M      1.4M      9.9M  13% /
tmpfs                   512.0K         0    512.0K   0% /dev
```

gregd72002, 2020-11-29:
@xavier83, I think this is unrelated to this issue. Happy to post my values in the other thread. I've got 256MB version.