openwrt/mt76 #478: MT7615 pcie card

zhangsd20, 2020-11-27:
When I use the mt7615 pcie module. I found that the driver is abnormal and it looks like the verification data cannot be read from efuse.
I see that many dts files are defined to read data from the factory partition
 (mediatek,mtd-eeprom = <&factory 0x0000>)

Can the mt7615 driver support pcie module?

Hurricos, 2020-12-20:
I don't know which board you're using, but it sounds like you need to correct your dts to point to the eeprom on the PCIe module if you think it should be read from the module.

zhangsd20, 2020-12-21:
https://www.alibaba.com/product-detail/MTK-MT7615N-4x4-802-11ac-Wave2_60674301864.html

I use the u7615n-l3 module with a PCI card slot.
How do I specify to read parity data on u7615n?
I tried to remove “MediaTek, and MTD EEPROM = < & factory 0x0000 >” still failed to read the calibration data

ryderlee1110, 2021-01-09:
Are you should there's a factory partition on you board?

zhangsd20, 2021-01-09:
You may have misunderstood my meaning.

This MT7615N should read its own EEPROM data, rather than read the effect data from my motherboard's factory partition.

I want to know how to read the EEPROM of MT7615N？

ryderlee1110, 2021-01-09:
If the efuse isn't empty the driver should read it directly.

zhangsd20, 2021-01-12:
yes，you are right.
I tested it. When I specify reading data from a factory in a DTS (I populate other factories into my motherboard factory partition) the MT7615N works and the MAC address is correct.

When I don't specify in DTS to read data from factory, it should read data from efuse,but it doesn't read it, and prompts that the MAC address is invalid。

ryderlee1110, 2021-01-15:
can we close this ticket?

zhangsd20, 2021-03-31:
HI,
    The device I tested is U7621-06 + mt7615N pcie card.
   But it seems to work abnormally .
  
  dmesg:
[   12.837712] mt7615e 0000:02:00.0: Invalid MAC address, using random address 9e:1b:ba:4e:73:8a
[   12.860242] mt7615e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a

zhangsd20, 2021-04-13:
I am using U7615N-L3 pcie card, it should read calibration data from eeprom or efuse, not from flash.
(I bought U7621 and U7615N from different manufacturers)

From the results of my test, we can see that 7615 can only read calibration data from the flash (factory partition).

Can 7615 read calibration data from pcie card?