openwrt/mt76 #480: USB dongle that start ad CDROM with windows driver

Menion2k, 2020-11-30:
Hi all

I have just received a chinese USB dongle based on Mediatek MT7612U that is quite particular...

When plugged in it start with VID_0E8D&PID_2870 showing itseld as a SCSI CDROM, which can be mounted and contains the windows driver. In windows, once you install such driver it switch to the well known VID_0E8D&PID_7612 detected as a Mediatek adapter and working ok.

Unfortunately it seems not possible, or at least I do not know how make it to swith to "WiFi mode" in linux. Is it possible to do so?

Bye

mrkiko, 2020-12-05:
Hi!!

Have you tried ejecting the USB storage disk with the "eject" utility?
What happens? Assuming something happens...

koast18, 2021-01-20:
Hi，you can add a package named usbmodeswitch or some thing with dash in it.And you will find everything ok but you just can't run it at SuperSpeed.Using 1000Mps Wireless Card only at 480Mbps is so annoying....
I am a Chinese regreting to buy this card.....fxxxxxk.

Menion2k, 2021-01-20:
usbmodeswitch works but only at physical plug-unplug. eject utility is an idea but openwrt does not create cdrom entry under dev for some reason...

dangowrt, 2021-01-20:
Try installing `kmod-scsi-cdrom` which is needed in addition to `kmod-usb-storage` for USB CD-ROM drivers.

Menion2k, 2021-01-21:
I did install kmod-scsi-cdrom, now in dmesg I see the device attached:

`[    8.646995] sr 0:0:0:0: Attached scsi CD-ROM sr0`

But the device is not created under /dev, I think some issue with procd. makedev is also not available as package, so I am not sure what to do