openwrt/mt76 #487: MT7603E has terrible wifi latency

remysaville, 2020-12-30:
When running 
flent rrul -p all_scaled -l 60 -H netperf.bufferbloat.net -o mt76.png
Or equivalent to a local server wired into the lan the wifi performance is unusable.  When on the 5GHz mt7612 radio on the same Netgear r6220 device the performance is much more consistent.

See last three images in https://forum.openwrt.org/t/how-to-write-a-meaningful-mt76-lag-bug/82515/19 for a sample.

remysaville, 2021-09-14:
I still have the same problem with the recently released 21.02.  Maybe the bug needs to specify under load?  Maybe it is related to my home environment with ~50 smart devices?  I'm testing with an Intel 9560 in my laptop on today's latest drivers.

@JFtico are you able to try flent to reproduce?
`flent rrul -p all_scaled -l 60 -H netperf.bufferbloat.net -o mt76.png`

I get these cut up results.
[image]

With 19.07 5GHz I was able to get a nice graph like this:
[image]

trailblazergt, 2022-05-10:
I can still confirm the same issue with Archer C50 V3 and OpenWrt version  22.03.0-rc1.
[image]