openwrt/mt76 #492: Alot of interrupt errors - Netgear R6350 and Netgear R6220 with MT7621

podtor, 2021-01-15:
Alot of interrupt errors in Netgear R6350 MT7621

MediaTek MT7621 ver:1 eco:3
Netgear R6350
OpenWrt SNAPSHOT r15505-c40006e0af / LuCI Master git-21.009.66148-9e4391e

> root@OpenWrt:~ **uptime**
> 17:08:05 up 37 min,  load average: 0.07, 0.04, 0.02
> root@OpenWrt:~ **cat /proc/interrupts**
>            CPU0       CPU1       CPU2       CPU3
>   8:     561329     561394     561051     561356  MIPS GIC Local   1  timer
>   9:     211428          0          0          0  MIPS GIC  63  IPI call
>  10:          0      47607          0          0  MIPS GIC  64  IPI call
>  11:          0          0     108062          0  MIPS GIC  65  IPI call
>  12:          0          0          0      33572  MIPS GIC  66  IPI call
>  13:     206854          0          0          0  MIPS GIC  67  IPI resched
>  14:          0     134286          0          0  MIPS GIC  68  IPI resched
>  15:          0          0      64419          0  MIPS GIC  69  IPI resched
>  16:          0          0          0     129219  MIPS GIC  70  IPI resched
>  17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
>  19:         12          0          0          0  MIPS GIC  33  ttyS0
>  20:          0          0          0          0  MIPS GIC  29  xhci-hcd:usb1
>  21:     124040          0          0          0  MIPS GIC  10  1e100000.ethernet
>  23:         17          0     240310          0  MIPS GIC  11  mt7615e
>  24:         23          0          0          0  MIPS GIC  31  mt7603e
>  26:          0          0          0          0  1e000600.gpio   7  keys
>  27:          0          0          0          0  1e000600.gpio  14  keys
> **ERR:        519**

I have also two Netgear R6220 and they have the same problem with alot of interrupt errors.
MediaTek MT7621 ver:1 eco:3
Netgear R6220
OpenWrt SNAPSHOT r15469-425e392466 / LuCI Master git-21.009.66148-9e4391e

> **uptime**
>  18:16:46 up 2 days, 23:38,  load average: 0.00, 0.00, 0.00
> root@OpenWrt:~ **cat /proc/interrupts**
>            CPU0       CPU1
>   8:   64472251   64472439  MIPS GIC Local   1  timer
>   9:      89382          0  MIPS GIC  67  IPI call
>  10:          0     361362  MIPS GIC  68  IPI call
>  11:    4090332          0  MIPS GIC  69  IPI resched
>  12:          0    2911477  MIPS GIC  70  IPI resched
>  13:          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
>  15:         12          0  MIPS GIC  33  ttyS0
>  16:          0          0  MIPS GIC  29  xhci-hcd:usb1
>  17:         17          0  MIPS GIC  10  1e100000.ethernet
>  19:          2   11615122  MIPS GIC  11  mt76x2e
>  21:         23          0  MIPS GIC  32  mt7603e
>  22:          0          0  1e000600.gpio   7  keys
>  23:          0          0  1e000600.gpio   8  keys
>  24:          0          0  1e000600.gpio  14  keys
> **ERR:        169**

podtor, 2021-01-17:
Interrupt errors increase when I check wifi speed connection using Meteor application on android smartphone.

MeIsReallyBa, 2021-01-18:
What I mean is that the wan will work even if it switch is  not loaded properly.In fact after modifying the dts, the switch was successfully loaded

JFtico, 2021-04-19:
I saw constantly growing interrupt errors across the entire 19.07.x releases on a ZBT-WE3525 (7621/7612/7603).
Hoping for something better from 21.02, I built one on on 4/15/21, and while many aspects have improved, power being the most noticeable, the proc/interrupt errors continue to grow. And stations get into connect/disconnect loops if the strength is marginal.  Errors seem to climb mostly from 5Ghz radio activity, as the same box running as a router-only (no wifi) sees single digits worth of errors after a week.