openwrt/mt76 #493: MT7621 - wifi ac and n are twice as slow

podtor, 2021-01-15:
I updated Netgear R6350 onto latest snapshot OpenWrt SNAPSHOT r15505-c40006e0af / LuCI Master git-21.009.66148-9e4391e and now transfers on wifi ac and n are twice as slow.
On ac wifi previous was 120Mbit after updating is 60Mbit.
On N wifi previous was 60Mbit after updating is 33Mbit.

nbd168, 2021-01-15:
What were you running before?

podtor, 2021-01-15:
Previous I was mean b837534f029da10abbd1069392867e0700134ace
Now I have f841855f10535529e9cba99469fbbe7f4d73cbda
https://git.openwrt.org/?p=openwrt%2Fopenwrt.git&a=search&h=HEAD&st=commit&s=mt76:

nbd168, 2021-01-15:
What configuration are you using? Did you test 2.4 GHz and 5 GHz, or just 5 GHz?

nbd168, 2021-01-15:
Bonus question: how are you testing speed?

podtor, 2021-01-15:
I was testing using Samsung A70 smartphone and Meteor application. Smartphone is in the same place. Configuration wifi previous and now is the same.
On original firmware and testing in the same way I have 300Mbit/s

`
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'HT20'
	option channel 'auto'
	option legacy_rates '0'
	option country 'PL'
	option txpower '16'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option wpa_disable_eapol_key_retries '1'
	option key 'passtest22'
	option ssid 'Wi-Fi-2'
	option encryption 'psk2+ccmp'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'VHT80'
	option legacy_rates '0'
	option country 'PL'
	option channel '36'
	option txpower '16'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option wpa_disable_eapol_key_retries '1'
	option key 'passtest11'
	option encryption 'psk2+ccmp'
	option ssid 'Wi-Fi'
`

nbd168, 2021-01-15:
Please do this while running the test:
`cat /sys/kernel/debug/ieee80211/phy*/netdev:*/stations/*/rc_stats`
and paste the output here:

nbd168, 2021-01-15:
If possible, I'd like to see the same with the old firmware where you got higher speeds

nbd168, 2021-01-15:
I just ran some tests with a 7615 card and I can't reproduce any speed issues with the current version

podtor, 2021-01-15:
`              best   ____________rate__________    ____statistics___    _____last____    ______sum-of________
mode guard #  rate  [name   idx airtime  max_tp]  [avg(tp) avg(prob)]  [retry|suc|att]  [#success | #attempts]
VHT20 LGI 1      P  MCS0/1  180    1477     4.8       4.8     100.0       2     0 0             9   9
VHT20 LGI 1         MCS1/1  181     738     9.7       9.7     100.0       3     0 0             1   1
VHT20 LGI 1         MCS2/1  182     492    17.0      17.0     100.0       0     0 0             1   1
VHT20 LGI 1         MCS3/1  183     369    21.9      21.9     100.0       0     0 0            12   12
VHT20 LGI 1         MCS4/1  184     246    34.1      34.1     100.0       0     0 0            18   18
VHT20 LGI 1         MCS5/1  185     185    46.3      46.3     100.0       0     0 0            25   25
VHT20 LGI 1         MCS6/1  186     164    51.2      51.2     100.0       0     0 0            28   28
VHT20 LGI 1         MCS7/1  187     148    56.1      56.1     100.0       0     0 0            31   31
VHT20 LGI 1         MCS8/1  188     123    68.3      68.3     100.0       0     0 0            37   37
VHT20 SGI 1         MCS0/1  220    1329     4.8       4.8     100.0       0     0 0             3   3
VHT20 SGI 1         MCS1/1  221     665    12.2      12.2     100.0       2     0 0             3   3
VHT20 SGI 1         MCS2/1  222     443    19.5      19.5     100.0       0     0 0             1   1
VHT20 SGI 1         MCS3/1  223     332    24.4      24.4     100.0       0     0 0            12   12
VHT20 SGI 1         MCS4/1  224     222    39.0      39.0     100.0       0     0 0            18   18
VHT20 SGI 1         MCS5/1  225     166    51.2      51.2     100.0       0     0 0             1   1
VHT20 SGI 1         MCS6/1  226     148    56.1      56.1     100.0       0     0 0            55   55
VHT20 SGI 1         MCS7/1  227     133    63.4      63.4     100.0       0     0 0            31   31
VHT20 SGI 1         MCS8/1  228     111    75.6      75.6     100.0       0     0 0            11   11
VHT40 LGI 1         MCS0/1  260     711    12.2      12.2     100.0       0     0 0             3   3
VHT40 LGI 1         MCS1/1  261     356    24.4      24.4     100.0       0     0 0             6   6
VHT40 LGI 1         MCS2/1  262     237    36.6      36.6     100.0       2     0 0             1   1
VHT40 LGI 1         MCS3/1  263     178    46.3      46.3     100.0       0     0 0            12   12
VHT40 LGI 1         MCS4/1  264     119    70.8      70.8     100.0       0     0 0            18   18
VHT40 LGI 1         MCS5/1  265      89    92.7      92.7     100.0       3     0 0            25   25
VHT40 LGI 1         MCS6/1  266      79   102.5     102.5     100.0       0     0 0            28   28
VHT40 LGI 1         MCS7/1  267      71   114.7     114.7     100.0       0     0 0            22   22
VHT40 LGI 1         MCS8/1  268      59   134.2     134.2     100.0       0    38 38         3110   3116
VHT40 LGI 1         MCS9/1  269      53   148.9       0.0       0.0       0     0 0             0   0
VHT40 SGI 1         MCS0/1  300     640    12.2      12.2     100.0       0     0 0             1   1
VHT40 SGI 1         MCS1/1  301     320    26.8      26.8      99.3       2     0 0            20   21
VHT40 SGI 1         MCS2/1  302     214    39.0      39.0     100.0       3     0 0             1   1
VHT40 SGI 1         MCS3/1  303     160    53.7      53.7     100.0       0     0 0            12   12
VHT40 SGI 1         MCS4/1  304     107    78.1      78.1     100.0       0     0 0            18   18
VHT40 SGI 1         MCS5/1  305      80   102.5     102.5     100.0       3     0 0            49   49
VHT40 SGI 1         MCS6/1  306      71   114.7     114.7     100.0       0     0 0            16   16
VHT40 SGI 1         MCS7/1  307      64   124.5     124.5     100.0       0     0 0            58   58
VHT40 SGI 1         MCS8/1  308      54   148.9     148.9     100.0       0     0 0          3735   3744
VHT40 SGI 1         MCS9/1  309      48   163.5       0.0       0.0       0     0 0             0   0
VHT80 LGI 1         MCS0/1  340     328    26.8      26.8     100.0       0     0 0             3   3
VHT80 LGI 1         MCS1/1  341     164    51.2      51.2      96.8       2     0 0           159   162
VHT80 LGI 1         MCS2/1  342     109    75.6      75.6      97.9       2     0 0            50   51
VHT80 LGI 1         MCS3/1  343      82   100.0     100.0     100.0       0     0 0            12   12
VHT80 LGI 1         MCS4/1  344      55   144.0     144.0     100.0       0     0 0            18   18
VHT80 LGI 1         MCS5/1  345      41   185.5     185.5     100.0       4     0 0           521   521
VHT80 LGI 1         MCS6/1  346      36   207.5     207.5     100.0       5     0 0            28   28
VHT80 LGI 1     D   MCS7/1  347      33   224.6     224.6     100.0       5     0 0             1   1
VHT80 LGI 1   B     MCS8/1  348      27   261.2     261.2      96.7       5     0 0          3534   3549
VHT80 LGI 1         MCS9/1  349      25   285.6       0.0       0.0       0     0 0             0   0
VHT80 SGI 1         MCS0/1  380     295    29.2      29.2     100.0       0     0 0             2   2
VHT80 SGI 1         MCS1/1  381     148    56.1      56.1      98.2       3     0 0            13   14
VHT80 SGI 1         MCS2/1  382      99    83.0      83.0     100.0       3     0 0             5   5
[... cut ...]
VHT80 SGI 2         MCS5/2  395      19   351.5     183.1      47.2       5     0 0           410   566
VHT80 SGI 2         MCS6/2  396      16   385.7     209.9      48.9       5     0 0           673   951
VHT80 SGI 2         MCS7/2  397      15   415.0     268.5      58.6       5    78 125        4362   6453
VHT80 SGI 2  A      MCS8/2  398      12   468.7     270.9      52.3       5     0 0          2139   3340
VHT80 SGI 2         MCS9/2  399      11   498.0     131.8      23.9       5     0 76          453   1067

Total packet count::    ideal 20609      lookaround 344`

podtor, 2021-01-15:
`              best   ____________rate__________    ____statistics___    _____last____    ______sum-of________
mode guard #  rate  [name   idx airtime  max_tp]  [avg(tp) avg(prob)]  [retry|suc|att]  [#success | #attempts]
VHT20 LGI 1      P  MCS0/1  180    1477     4.8       4.8     100.0       2     0 0            11   11
VHT20 LGI 1         MCS1/1  181     738     9.7       9.7     100.0       3     0 0             1   1
VHT20 LGI 1         MCS2/1  182     492    17.0      17.0     100.0       0     0 0             1   1
VHT20 LGI 1         MCS3/1  183     369    21.9      21.9     100.0       0     0 0             2   2
VHT20 LGI 1         MCS4/1  184     246    34.1      34.1      96.2       0     0 0            36   37
VHT20 LGI 1         MCS5/1  185     185    46.3      46.3     100.0       0     0 0             4   4
VHT20 LGI 1         MCS6/1  186     164    51.2      51.2     100.0       0     0 0             1   1
VHT20 LGI 1         MCS7/1  187     148    56.1      56.1     100.0       0     0 0            31   31
VHT20 LGI 1         MCS8/1  188     123    68.3      68.3     100.0       0     0 0            41   41
VHT20 SGI 1         MCS0/1  220    1329     4.8       4.8     100.0       0     0 0             1   1
VHT20 SGI 1         MCS1/1  221     665    12.2      12.2     100.0       0     0 0             1   1
VHT20 SGI 1         MCS2/1  222     443    19.5      19.5     100.0       0     0 0             1   1
VHT20 SGI 1         MCS3/1  223     332    24.4      24.4     100.0       0     0 0             5   5
VHT20 SGI 1         MCS4/1  224     222    39.0      39.0     100.0       0     0 0            18   18
VHT20 SGI 1         MCS5/1  225     166    51.2      51.2     100.0       0     0 0            25   25
VHT20 SGI 1         MCS6/1  226     148    56.1      56.1     100.0       0     0 0            28   28
VHT20 SGI 1         MCS7/1  227     133    63.4      63.4     100.0       0     0 0            22   22
VHT20 SGI 1         MCS8/1  228     111    75.6      75.6     100.0       0     0 0            70   70
VHT40 LGI 1         MCS0/1  260     711    12.2      12.2     100.0       0     0 0             3   3
VHT40 LGI 1         MCS1/1  261     356    24.4      24.4     100.0       3     0 0             1   1
VHT40 LGI 1         MCS2/1  262     237    36.6      36.6     100.0       0     0 0             9   9
VHT40 LGI 1         MCS3/1  263     178    46.3      46.3     100.0       0     0 0            12   12
VHT40 LGI 1         MCS4/1  264     119    70.8      70.8     100.0       0     0 0            19   19
VHT40 LGI 1         MCS5/1  265      89    92.7      92.7     100.0       0     0 0            25   25
VHT40 LGI 1         MCS6/1  266      79   102.5     102.5     100.0       0     0 0            28   28
VHT40 LGI 1         MCS7/1  267      71   114.7     114.7     100.0       0     0 0            32   32
VHT40 LGI 1         MCS8/1  268      59   134.2     134.2     100.0       0     1 1          3490   3500
VHT40 LGI 1         MCS9/1  269      53   148.9       0.0       0.0       0     0 0             0   6
VHT40 SGI 1         MCS0/1  300     640    12.2      12.2     100.0       0     0 0             3   3
VHT40 SGI 1         MCS1/1  301     320    26.8      26.8     100.0       2     0 0            25   25
VHT40 SGI 1         MCS2/1  302     214    39.0      39.0     100.0       0     0 0             1   1
VHT40 SGI 1         MCS3/1  303     160    53.7      53.7     100.0       0     0 0            12   12
VHT40 SGI 1         MCS4/1  304     107    78.1      78.1     100.0       0     0 0             1   1
VHT40 SGI 1         MCS5/1  305      80   102.5     102.5     100.0       0     0 0            25   25
VHT40 SGI 1         MCS6/1  306      71   114.7     114.7     100.0       0     0 0            28   28
VHT40 SGI 1         MCS7/1  307      64   124.5     124.5     100.0       0     0 0             1   1
VHT40 SGI 1         MCS8/1  308      54   148.9     148.9     100.0       0     0 0          3357   3361
VHT40 SGI 1         MCS9/1  309      48   163.5       0.0       0.0       0     0 0             0   0
VHT80 LGI 1         MCS0/1  340     328    26.8      26.8     100.0       0     0 0             1   1
VHT80 LGI 1         MCS1/1  341     164    51.2      51.2      98.9       2     0 0            30   32
VHT80 LGI 1         MCS2/1  342     109    75.6      75.6     100.0       2     0 0             1   1
VHT80 LGI 1         MCS3/1  343      82   100.0     100.0     100.0       3     0 0            84   84
VHT80 LGI 1         MCS4/1  344      55   144.0     144.0     100.0       0     0 0            18   18
VHT80 LGI 1         MCS5/1  345      41   185.5     185.5     100.0       0     0 0            25   25
VHT80 LGI 1         MCS6/1  346      36   207.5     207.5     100.0       5     0 0            27   27
VHT80 LGI 1     D   MCS7/1  347      33   224.6     224.6     100.0       5     0 0             9   9
VHT80 LGI 1   B     MCS8/1  348      27   261.2     261.2     100.0       5    38 38         3093   3102
VHT80 LGI 1         MCS9/1  349      25   285.6       0.0       0.0       0     0 0             0   0
VHT80 SGI 1         MCS0/1  380     295    29.2      29.2     100.0       0     0 0             5   5
VHT80 SGI 1         MCS1/1  381     148    56.1      56.1     100.0       2     0 0            87   87
VHT80 SGI 1         MCS2/1  382      99    83.0      83.0     100.0       3     0 0           393   395
[... cut ...]
VHT80 SGI 2         MCS5/2  395      19   351.5     351.5     100.0       5     0 0           474   546
VHT80 SGI 2         MCS6/2  396      16   385.7     385.7      99.0       5     0 0          1641   2081
VHT80 SGI 2     D   MCS7/2  397      15   415.0     415.0      98.8       5     0 0          2765   3311
VHT80 SGI 2   B     MCS8/2  398      12   468.7     468.7      93.8       5     0 0          3781   4963
VHT80 SGI 2  A      MCS9/2  399      11   498.0     498.0      91.2       5    55 56         3621   4668

Total packet count::    ideal 24291      lookaround 313`

podtor, 2021-01-15:
During upload but i have only 20Mbit and always is the same value

`              best   ____________rate__________    ____statistics___    _____last____    ______sum-of________
mode guard #  rate  [name   idx airtime  max_tp]  [avg(tp) avg(prob)]  [retry|suc|att]  [#success | #attempts]
VHT20 LGI 1      P  MCS0/1  180    1477     4.8       4.8     100.0       2     0 0             9   9
VHT20 LGI 1         MCS1/1  181     738     9.7       9.7     100.0       3     0 0             1   1
VHT20 LGI 1         MCS2/1  182     492    17.0      17.0     100.0       0     0 0             1   1
VHT20 LGI 1         MCS3/1  183     369    21.9      21.9     100.0       0     0 0            12   12
VHT20 LGI 1         MCS4/1  184     246    34.1      34.1     100.0       0     0 0            18   18
VHT20 LGI 1         MCS5/1  185     185    46.3      46.3     100.0       0     0 0            25   25
VHT20 LGI 1         MCS6/1  186     164    51.2      51.2     100.0       0     0 0            28   28
VHT20 LGI 1         MCS7/1  187     148    56.1      56.1     100.0       0     0 0            31   31
VHT20 LGI 1         MCS8/1  188     123    68.3      68.3     100.0       0     0 0            37   37
VHT20 SGI 1         MCS0/1  220    1329     4.8       4.8     100.0       0     0 0             3   3
VHT20 SGI 1         MCS1/1  221     665    12.2      12.2     100.0       2     0 0             3   3
VHT20 SGI 1         MCS2/1  222     443    19.5      19.5     100.0       0     0 0             1   1
VHT20 SGI 1         MCS3/1  223     332    24.4      24.4     100.0       0     0 0            12   12
VHT20 SGI 1         MCS4/1  224     222    39.0      39.0     100.0       0     0 0            18   18
VHT20 SGI 1         MCS5/1  225     166    51.2      51.2     100.0       0     0 0             1   1
VHT20 SGI 1         MCS6/1  226     148    56.1      56.1     100.0       0     0 0            55   55
VHT20 SGI 1         MCS7/1  227     133    63.4      63.4     100.0       0     0 0            31   31
VHT20 SGI 1         MCS8/1  228     111    75.6      75.6     100.0       0     0 0            11   11
VHT40 LGI 1         MCS0/1  260     711    12.2      12.2     100.0       0     0 0             3   3
VHT40 LGI 1         MCS1/1  261     356    24.4      24.4     100.0       0     0 0             6   6
VHT40 LGI 1         MCS2/1  262     237    36.6      36.6     100.0       2     0 0             1   1
VHT40 LGI 1         MCS3/1  263     178    46.3      46.3     100.0       0     0 0            12   12
VHT40 LGI 1         MCS4/1  264     119    70.8      70.8     100.0       0     0 0            18   18
VHT40 LGI 1         MCS5/1  265      89    92.7      92.7     100.0       3     0 0            25   25
VHT40 LGI 1         MCS6/1  266      79   102.5     102.5     100.0       0     0 0            28   28
VHT40 LGI 1         MCS7/1  267      71   114.7     114.7     100.0       0     0 0            22   22
VHT40 LGI 1         MCS8/1  268      59   134.2     134.2     100.0       0     0 0          5587   5636
VHT40 LGI 1         MCS9/1  269      53   148.9       0.0       0.0       0     0 0             0   0
VHT40 SGI 1         MCS0/1  300     640    12.2      12.2     100.0       0     0 0             1   1
VHT40 SGI 1         MCS1/1  301     320    26.8      26.8      99.3       2     0 0            20   21
VHT40 SGI 1         MCS2/1  302     214    39.0      39.0     100.0       3     0 0             1   1
VHT40 SGI 1         MCS3/1  303     160    53.7      53.7     100.0       0     0 0            12   12
VHT40 SGI 1         MCS4/1  304     107    78.1      78.1     100.0       0     0 0            18   18
VHT40 SGI 1         MCS5/1  305      80   102.5     102.5     100.0       3     0 0            49   49
VHT40 SGI 1         MCS6/1  306      71   114.7     114.7     100.0       0     0 0            16   16
VHT40 SGI 1         MCS7/1  307      64   124.5     124.5     100.0       0     0 0            58   58
VHT40 SGI 1         MCS8/1  308      54   148.9     148.9      98.6       0     3 3          6599   6629
VHT40 SGI 1         MCS9/1  309      48   163.5       0.0       0.0       0     0 0             0   0
VHT80 LGI 1         MCS0/1  340     328    26.8      26.8     100.0       0     0 0             3   3
VHT80 LGI 1         MCS1/1  341     164    51.2      51.2      96.8       2     0 0           159   162
VHT80 LGI 1         MCS2/1  342     109    75.6      75.6      97.9       2     0 0            50   51
VHT80 LGI 1         MCS3/1  343      82   100.0     100.0     100.0       0     0 0            12   12
VHT80 LGI 1         MCS4/1  344      55   144.0     144.0     100.0       0     0 0            18   18
VHT80 LGI 1         MCS5/1  345      41   185.5     185.5     100.0       4     0 0           521   521
VHT80 LGI 1         MCS6/1  346      36   207.5     207.5     100.0       5     0 0            28   28
VHT80 LGI 1     D   MCS7/1  347      33   224.6     224.6     100.0       5     0 0             1   1
VHT80 LGI 1   B     MCS8/1  348      27   261.2     261.2     100.0       5     0 0          6655   6673
VHT80 LGI 1         MCS9/1  349      25   285.6       0.0       0.0       0     0 0             0   0
VHT80 SGI 1         MCS0/1  380     295    29.2      29.2     100.0       0     0 0             2   2
VHT80 SGI 1         MCS1/1  381     148    56.1      56.1      98.2       3     0 0            13   14
[... cut ...]
VHT80 SGI 2     D   MCS5/2  395      19   351.5     317.3      81.2       5     0 0           935   1270
VHT80 SGI 2         MCS6/2  396      16   385.7     251.4      58.8       5     0 0          1176   1890
VHT80 SGI 2         MCS7/2  397      15   415.0     300.2      65.1       5    39 89         7667   11588
VHT80 SGI 2  A      MCS8/2  398      12   468.7     336.9      65.0       5     1 1          3506   5454
VHT80 SGI 2    C    MCS9/2  399      11   498.0     319.8      58.1       5     0 0          1295   2310

Total packet count::    ideal 35723      lookaround 603`

nbd168, 2021-01-15:
Hmm, rate selection looks normal

musashino205, 2021-02-04:
Is this issue solved?

podtor, 2021-02-04:
Yes

closed 2021-03-26 by nbd168