openwrt/mt76 #495: mt76x2u for MediaTek Inc with USB 3.0 Port doesn't run at SuperSpeed(Only at High-Speed 480Mbps )

koast18, 2021-01-20:
It seems that the device was recognized as a USB 2.0 device and plugged on a USB 3.0 Port.But on my computer ,it can run at 5Gbps at SuperSpeed.I googled it and find some familiar problem on other wireless card,they solve it just modify a driver parameter,so I think maybe it is an issue of mt76 module.

Here is the lsusb result.
`
root@OpenWrt:~# lsusb
Bus 001 Device 002: ID 0627:0001 QEMU QEMU USB Tablet
Bus 001 Device 001: ID 1d6b:0001 Linux 5.4.86 uhci_hcd UHCI Host Controller
Bus 002 Device 002: ID 0e8d:7612 MediaTek Inc. 802.11ac WLAN
Bus 002 Device 001: ID 1d6b:0002 Linux 5.4.86 xhci-hcd xHCI Host Controller
Bus 003 Device 001: ID 1d6b:0003 Linux 5.4.86 xhci-hcd xHCI Host Controller
`

`
root@OpenWrt:~# lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=mt76x2u, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=uhci_hcd/2p, 12M
    |__ Port 1: Dev 2, If 0, Class=, Driver=usbhid, 12M
`

`
root@OpenWrt:~# lsusb -v -d 0e8d:7612

Bus 002 Device 002: ID 0e8d:7612 MediaTek Inc. 802.11ac WLAN
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0e8d
  idProduct          0x7612
  bcdDevice            1.00
  iManufacturer           1 MediaTek Inc.
  iProduct                2 802.11ac WLAN
  iSerial                 3 000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x004a
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          2 802.11ac WLAN
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              200mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           8
      bInterfaceClass       255
      bInterfaceSubClass    255
      bInterfaceProtocol    255
      iInterface              2 802.11ac WLAN
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x07  EP 7 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f41e
      BESL Link Power Management (LPM) Supported
    BESL value     1024 us
    Deep BESL value    61440 us
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         180 micro seconds
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
`

It says Device can operate at SuperSpeed (5Gbps),but it runs at 480Mbps,any advice to fix this?
Due to this,the 5GHz AP handles by this tested by iperf3 only runs at 7MBps,so awkard......

nbd168, 2021-01-20:
I don't think this has anything to do with the mt76 driver. USB speed negotiation is not handled by the USB device drivers, it happens before the device drivers even attach.
One more thing: in your lsusb output, the root hub that the device is attached to is also shown operating on 480Mbps.

closed 2021-01-20 by nbd168

koast18, 2021-01-21:
@nbd168 I tried to plug a usb 2.0 device on the usb 3.0 port ,it just shows as mt7612u.But when a usb3.0 drive plugged at the same physical port shows in the bus 03 and runs at 5Gbps.So,maybe the bus 02 is an alias of the usb3.0 for the purpose of usb2.0 compatibility.

Anyway，thanks for your addition information.I will look for other way to fix the problem.