openwrt/mt76 #498: mt76x2u:Speed of AP mode at 5GHz seems to be too low

koast18, 2021-01-29:
When I configure the usb wireless card from MediaTek Inc. AP mode at 5GHz and connect to it with my phone,it show the connect speed is 866Mbps on my phone.But from the luci interface,the connection speed is not stable,the RX/TX speed usually shows like ```
`
6.0 Mbit/s, 20MHz
216.0 Mbit/s, 40MHz, VHT-MCS 5, VHT-NSS 2
`
but sometimes it flash to something like
`
 -44 / 0 dBm	650.0 Mbit/s, 80MHz, VHT-MCS 7, VHT-NSS 2, Short GI
86.7 Mbit/s, 20MHz, VHT-MCS 8, VHT-NSS 1, Short GI
`
Here is the actual test result of iperf3
`
iperf3 -c 192.168.50.100
Connecting to host 192.168.50.100, port 5201
[  4] local 192.168.50.11 port 4393 connected to 192.168.50.100 port 5201
iperf3: getsockopt - Protocol not available
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  9.00 MBytes  75.4 Mbits/sec  1296506936   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   1.00-2.00   sec  8.00 MBytes  67.1 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   2.00-3.00   sec  8.38 MBytes  70.2 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   3.00-4.00   sec  7.00 MBytes  58.7 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   4.00-5.00   sec  7.88 MBytes  66.1 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   5.00-6.00   sec  8.12 MBytes  68.2 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   6.00-7.00   sec  8.75 MBytes  73.4 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   7.00-8.00   sec  8.75 MBytes  73.4 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   8.00-9.00   sec  8.25 MBytes  69.2 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   9.00-10.00  sec  7.88 MBytes  66.0 Mbits/sec  2998460360   0.00 Bytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  82.0 MBytes  68.8 Mbits/sec    0             sender
[  4]   0.00-10.00  sec  82.0 MBytes  68.8 Mbits/sec                  receiver

iperf Done.
`

`
iperf3 -c 192.168.50.100 --reverse
Connecting to host 192.168.50.100, port 5201
Reverse mode, remote host 192.168.50.100 is sending
[  4] local 192.168.50.11 port 5088 connected to 192.168.50.100 port 5201
iperf3: getsockopt - Protocol not available
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  7.52 MBytes  63.0 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   1.00-2.00   sec  4.29 MBytes  36.0 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   2.00-3.00   sec  3.31 MBytes  27.8 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   3.00-4.00   sec  3.60 MBytes  30.2 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   4.00-5.00   sec  6.28 MBytes  52.7 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   5.00-6.00   sec  6.98 MBytes  58.6 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   6.00-7.00   sec  8.22 MBytes  68.9 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   7.00-8.00   sec  7.35 MBytes  61.7 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   8.00-9.00   sec  7.03 MBytes  58.9 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   9.00-10.00  sec  8.10 MBytes  68.0 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  64.2 MBytes  53.9 Mbits/sec   46             sender
[  4]   0.00-10.00  sec  63.1 MBytes  52.9 Mbits/sec                  receiver

iperf Done.
`

some info about the AP configuration 

`
root@OpenWrt:~# wifi status
{
        "radio0": {
                "up": true,
                "pending": false,
                "autostart": true,
                "disabled": false,
                "retry_setup_failed": false,
                "config": {
                        "channel": "36",
                        "hwmode": "11a",
                        "path": "pci0000:00/0000:00:04.0/usb3/3-2/3-2:1.0",
                        "htmode": "VHT80",
                        "country": "US"
                },
                "interfaces": [
                        {
                                "section": "default_radio0",
                                "ifname": "wlan0",
                                "config": {
                                        "mode": "ap",
                                        "ssid": "OpenWrt",
                                        "encryption": "none",
                                        "network": [
                                                "lan"
                                        ],
                                        "mode": "ap"
                                },
                                "vlans": [

                                ],
                                "stations": [

                                ]
                        }
                ]
        }
}
`

some info about the openwrt software and the packages installed

`
 OpenWrt SNAPSHOT, r3050-b95ffef87
 -----------------------------------------------------
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.92 #0 SMP Wed Jan 27 09:12:30 2021 x86_64 GNU/Linux
root@OpenWrt:~# opkg list-installed |grep mt76
kmod-mt76-core - 5.4.92+2020-12-18-4a90fdf6-3
kmod-mt76-usb - 5.4.92+2020-12-18-4a90fdf6-3
kmod-mt76x02-common - 5.4.92+2020-12-18-4a90fdf6-3
kmod-mt76x02-usb - 5.4.92+2020-12-18-4a90fdf6-3
kmod-mt76x2 - 5.4.92+2020-12-18-4a90fdf6-3
kmod-mt76x2-common - 5.4.92+2020-12-18-4a90fdf6-3
kmod-mt76x2u - 5.4.92+2020-12-18-4a90fdf6-3
`

Using the same wireless card to establish a soft AP with the same band&channel on my windows laptop,and testing at the same distance,test device,the test result is about 780Mbps both RX and TX,so maybe that's an issue of the mt76 driver.

Anyway,thanks for your work in advance.

nbd168, 2021-01-29:
Please try the latest OpenWrt snapshot. I've added a number of performance/stability improvements since the version that you've tested.

koast18, 2021-01-29:
@nbd168 Hi,I have compiled the latest version of Openwrt ([This commit](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=7d12f29ae1d7337e4289baf9e6b4bf79445b0d59)), LUCI shows the firmware version is `OpenWrt SNAPSHOT r0-7d12f29 / LuCI Master git-21.027.56825-28496a1`.

But it seems that it doesn't help with the issue,and when I run `iperf3 -c 192.168.50.100 --reverse` from a phone connected the AP mode wireless card,I got a worse result.These commits doesn't help with the RX/TX rate of the wireless card,but make both TX/RX rate even worse.....

Here is the result from a client connected to the AP mode wireless card.
`
iperf3 -c 192.168.50.100 --reverse
Connecting to host 192.168.50.100, port 5201
Reverse mode, remote host 192.168.50.100 is sending
[  4] local 192.168.50.11 port 23175 connected to 192.168.50.100 port 5201
iperf3: getsockopt - Protocol not available
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   619 KBytes  5.06 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   1.00-2.00   sec   136 KBytes  1.11 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec
iperf3: getsockopt - Protocol not available
[  4]   3.00-4.05   sec   127 KBytes  1000 Kbits/sec
iperf3: getsockopt - Protocol not available
[  4]   4.05-5.00   sec   476 KBytes  4.08 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   5.00-6.01   sec   513 KBytes  4.19 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   6.01-7.00   sec   495 KBytes  4.07 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   7.00-8.00   sec   492 KBytes  4.03 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   8.00-9.00   sec   486 KBytes  3.98 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   9.00-10.00  sec   488 KBytes  4.00 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  4.38 MBytes  3.67 Mbits/sec   45             sender
[  4]   0.00-10.00  sec  3.87 MBytes  3.24 Mbits/sec                  receiver
`

`
iperf3 -c 192.168.50.100
Connecting to host 192.168.50.100, port 5201
[  4] local 192.168.50.11 port 30764 connected to 192.168.50.100 port 5201
iperf3: getsockopt - Protocol not available
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.01   sec  2.75 MBytes  22.9 Mbits/sec  1296506932   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   1.01-2.00   sec  2.50 MBytes  21.0 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   2.00-3.00   sec  2.62 MBytes  22.1 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   3.00-4.00   sec  2.62 MBytes  22.0 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   4.00-5.00   sec  2.50 MBytes  21.0 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   5.00-6.00   sec  2.62 MBytes  22.0 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   6.00-7.00   sec  1.25 MBytes  10.5 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   7.00-8.01   sec  0.00 Bytes  0.00 bits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   8.01-9.00   sec   640 KBytes  5.30 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   9.00-10.00  sec  2.38 MBytes  19.9 Mbits/sec  2998460364   0.00 Bytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  19.9 MBytes  16.7 Mbits/sec    0             sender
[  4]   0.00-10.00  sec  19.8 MBytes  16.6 Mbits/sec                  receiver
`

koast18, 2021-02-17:
**Update**:With the same settings and the latest version(`OpenWrt SNAPSHOT r0-482c9ff`),the problem still remains.

**Recv data speed from the clients**

```
iperf3 -c 192.168.50.12
Connecting to host 192.168.50.12, port 5201
[  4] local 192.168.50.11 port 20914 connected to 192.168.50.12 port 5201
iperf3: getsockopt - Protocol not available
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  9.75 MBytes  81.7 Mbits/sec  1296506929   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   1.00-2.00   sec  10.2 MBytes  86.0 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   2.00-3.00   sec  10.8 MBytes  90.2 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   3.00-4.00   sec  11.0 MBytes  92.3 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   4.00-5.00   sec  10.2 MBytes  85.9 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   5.00-6.00   sec  10.8 MBytes  90.2 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   6.00-7.00   sec  10.5 MBytes  88.1 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   7.00-8.00   sec  9.88 MBytes  82.9 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   8.00-9.00   sec  11.2 MBytes  94.3 Mbits/sec    0   0.00 Bytes
iperf3: getsockopt - Protocol not available
[  4]   9.00-10.00  sec  10.8 MBytes  90.3 Mbits/sec  2998460367   0.00 Bytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec   105 MBytes  88.2 Mbits/sec    0             sender
[  4]   0.00-10.00  sec   105 MBytes  88.2 Mbits/sec                  receiver

iperf Done.
```
**Send data speed from the clients**
```
iperf3 -c 192.168.50.12 --reverse
Connecting to host 192.168.50.12, port 5201
Reverse mode, remote host 192.168.50.12 is sending
[  4] local 192.168.50.11 port 21789 connected to 192.168.50.12 port 5201
iperf3: getsockopt - Protocol not available
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  5.33 MBytes  44.7 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   1.00-2.00   sec  4.69 MBytes  39.4 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   2.00-3.00   sec  4.98 MBytes  41.7 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   3.00-4.00   sec  4.49 MBytes  37.6 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   4.00-5.00   sec  4.43 MBytes  37.1 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   5.00-6.00   sec  4.73 MBytes  39.6 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   6.00-7.00   sec  4.62 MBytes  38.7 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   7.00-8.00   sec  4.19 MBytes  35.2 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   8.00-9.00   sec  4.65 MBytes  39.0 Mbits/sec
iperf3: getsockopt - Protocol not available
[  4]   9.00-10.00  sec  6.64 MBytes  55.7 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  49.3 MBytes  41.3 Mbits/sec  605             sender
[  4]   0.00-10.00  sec  48.9 MBytes  41.0 Mbits/sec                  receiver

iperf Done.
```

sanitariu, 2021-02-17:
Try getting closer to the router. I think mediatek 5Ghz still lacks correct tx-power support and if you just go 5 meters away signal is very low.

koast18, 2021-02-19:
@sanitariu Yes,you are right!
The RX speed of client can increase to about 130Mbps at the distance of zero.But it just have little help with the TX speed of clients(about 100Mbps).But at one metre away,the RX/TX speed will be like 88/20.

@nbd168 These new info may help.:)

Rising-Sun, 2021-05-03:
The driver runs fine for me
usb-adapter running @mt7621a, 
iperf odroid as sender and iphone as receiver
like 1.5m away
[  4] local 192.168.178.11 port 5001 connected with 192.168.178.27 port 52373
[  4]  0.0-22.9 sec   977 MBytes   358 Mbits/sec
[  4] local 192.168.178.11 port 5001 connected with 192.168.178.27 port 52374
[  4]  0.0-21.2 sec   977 MBytes   386 Mbits/sec
[  4] local 192.168.178.11 port 5001 connected with 192.168.178.27 port 52375
[  4]  0.0-22.3 sec   977 MBytes   367 Mbits/sec
like 7m away with a "wall" in between
[  4] local 192.168.178.11 port 5001 connected with 192.168.178.27 port 52376
[  4]  0.0-55.7 sec   977 MBytes   147 Mbits/sec
[  4] local 192.168.178.11 port 5001 connected with 192.168.178.27 port 52377
[  4]  0.0-33.2 sec   977 MBytes   247 Mbits/sec

The speeds are arond the same 6 months, increased like from 300->350mbit. 
I think the usb-adapter is able to do faster if connected to a faster host,

one thing I encountered when doing this test, my iphone was stuck at 20mhz till reconnecting. but could be a coincidence