openwrt/mt76 #500: MT7615 crashes on X86

Batbold-Gansukh, 2021-01-31:
Hello.

Build: OpenWrt SNAPSHOT, r15655-ddab795b37

Wifi card: BPI-MT7615 (http://wiki.banana-pi.org/BPI-R64_%2B_MT7615_function_test). Connected via mini-pci.

uname -a:
```Linux OpenWrt 5.4.93 #0 SMP Sat Jan 30 22:21:58 2021 x86_64 GNU/Linux```

cat /etc/config/wireless:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:1c.0/0000:01:00.0'
	option htmode 'VHT80'
	option cell_density '0'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option mode 'ap'
	option ssid 'OpenWrt'
	option network 'lan'
	option encryption 'psk-mixed'
	option key 'password'
```

dmesg | grep 7615:
```
[    1.885329] pci 0000:01:00.0: [14c3:7615] type 00 class 0x000280
[    8.520992] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[    8.558821] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
[    8.581635] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
```

lspci | grep 7615:
```01:00.0 Unclassified device [0002]: MEDIATEK Corp. Device 7615```

<details> 
  <summary>iw phy </summary>
<pre>
Wiphy phy0
	wiphy index: 0
	max # scan SSIDs: 4
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: 0
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Available Antennas: TX 0xf RX 0xf
	Configured Antennas: TX 0xf RX 0xf
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1ff
			RX LDPC
[... 896 lines cut ...]
[ 1582.909076] FS:  0000000000000000(0000) GS:ffff88845ed00000(0000) knlGS:0000000000000000
[ 1582.915963] mt7615e 0000:01:00.0: swiotlb buffer is full (sz: 3776 bytes), total 32768 (slots), used 32277 (slots)
[ 1582.935693] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1582.935694] CR2: 00007f71efe8d000 CR3: 0000000002208003 CR4: 00000000003606e0
[ 1582.935694] Call Trace:
[ 1582.935698]  dma_direct_map_page+0xde/0xf0
[ 1582.935702]  mt76_dma_cleanup+0x375/0x1450 [mt76]
[ 1582.948880] mt7615e 0000:01:00.0: swiotlb buffer is full (sz: 3776 bytes), total 32768 (slots), used 32419 (slots)
[ 1582.954606]  mt76_dma_cleanup+0x77b/0x1450 [mt76]
[ 1582.954608]  __napi_poll+0x2b/0x90
[ 1582.964471] mt7615e 0000:01:00.0: swiotlb buffer is full (sz: 3776 bytes), total 32768 (slots), used 32486 (slots)
[ 1582.972333]  ? napi_workfn+0x1e/0x80
[ 1582.972333]  napi_workfn+0x39/0x80
[ 1582.972335]  process_one_work+0x1e4/0x350
[ 1582.972337]  worker_thread+0x2f/0x410
[ 1583.120387]  kthread+0x10b/0x130
[ 1583.125201]  ? process_one_work+0x350/0x350
[ 1583.130907]  ? kthread_park+0x80/0x80
[ 1583.136145]  ret_from_fork+0x35/0x40
[ 1583.141313] ---[ end trace 489f839ee2203560 ]---
</pre>
</details>

At first, when web browsing etc... works fine.
When downloading file or performing iperf3 test, it crashes.

ryderlee1110, 2021-02-01:
Can you check if there is any memory leak on your system?

Batbold-Gansukh, 2021-02-02:
I think there is no memory leak.

<details>
<summary>dmesg | grep kmem</summary>
<pre>
[    3.650467] kmemleak: Kernel memory leak detector initialized (mem pool available: 15503)
[    3.650468] kmemleak: Automatic memory scanning thread started
</pre>
</details>

<details>
<summary>iperf3 -s -f K #result</summary>
<pre>
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.1.43, port 61740
[  5] local 192.168.1.100 port 5201 connected to 192.168.1.43 port 61742
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.04   sec  28.7 MBytes  28109 KBytes/sec
[  5]   1.04-2.00   sec  21.5 MBytes  22961 KBytes/sec
[  5]   2.00-3.00   sec  8.71 MBytes  8959 KBytes/sec
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 KBytes/sec
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 KBytes/sec
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 KBytes/sec
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 KBytes/sec
[  5]   7.00-8.00   sec  0.00 Bytes  0.00 KBytes/sec
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 KBytes/sec
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 KBytes/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.05  sec  58.9 MBytes  6005 KBytes/sec                  sender
WARNING:  Size of data read does not correspond to offered length
iperf3: error - unable to receive results: Bad file descriptor
</pre>
</details>

After the crash, detailed in first post, <b>cat /sys/kernel/debug/kmemleak</b> shows empty result.

<details>
<summary>ls -la /sys/kernel/debug/</summary>
<pre>
drwx------   25 root     root             0 Feb  2 03:37 .
drwxr-xr-x   10 root     root             0 Feb  2 03:37 ..
drwxr-xr-x    2 root     root             0 Feb  2 03:37 acpi
drwxr-xr-x   12 root     root             0 Feb  2 03:37 bdi
drwxr-xr-x    2 root     root             0 Feb  2 03:37 block
drwxr-xr-x    2 root     root             0 Feb  2 03:37 btt
--w-------    1 root     root             0 Feb  2 03:37 clear_warn_once
drwxr-xr-x    2 root     root             0 Feb  2 03:37 clk
drwxr-xr-x    2 root     root             0 Feb  2 03:37 device_component
-r--r--r--    1 root     root             0 Feb  2 03:37 devices_deferred
drwxr-xr-x    2 root     root             0 Feb  2 03:37 dma_buf
drwxr-xr-x    4 root     root             0 Feb  2 03:37 dri
drwxr-xr-x    2 root     root             0 Feb  2 03:37 extfrag
drwxr-xr-x    2 root     root             0 Feb  2 03:37 f2fs
-rw-r--r--    1 root     root             0 Feb  2 03:37 fault_around_bytes
-r--r--r--    1 root     root             0 Feb  2 03:37 gpio
drwxr-xr-x    7 root     root             0 Feb  2 03:37 hid
drwxr-xr-x    3 root     root             0 Feb  2 03:37 ieee80211
drwxr-xr-x    2 root     root             0 Feb  2 03:37 ixgbe
-rw-r--r--    1 root     root             0 Feb  2 03:37 kmemleak
drwxr-xr-x    2 root     root             0 Feb  2 03:37 mce
drwxr-xr-x    2 root     root             0 Feb  2 03:37 pinctrl
drwxr-xr-x    2 root     root             0 Feb  2 03:37 pkg_temp_thermal
drwxr-xr-x    2 root     root             0 Feb  2 03:37 pm_qos
drwxr-xr-x    2 root     root             0 Feb  2 03:37 ras
-r--r--r--    1 root     root             0 Feb  2 03:37 sleep_time
-r--r--r--    1 root     root             0 Feb  2 03:37 suspend_stats
drwxr-xr-x    2 root     root             0 Feb  2 03:37 swiotlb
drwxr-xr-x    4 root     root             0 Feb  2 03:37 usb
drwxr-xr-x    2 root     root             0 Feb  2 03:37 virtio-ports
-r--r--r--    1 root     root             0 Feb  2 03:37 wakeup_sources
drwxr-xr-x    2 root     root             0 Feb  2 03:37 x86
drwxr-xr-x    3 root     root             0 Feb  2 03:37 xen
</pre>
</details>

ryderlee1110, 2021-03-15:
can you try the latest mt76?

nbd168, 2021-03-26:
Fixed in 09a1befde4b729814ddaa3aa3de0ed0db7d4e1a8

closed 2021-03-26 by nbd168