openwrt/mt76 #502: iPhone stucks at 20Mhz / 5Ghz on MT7615E

rockenren, 2021-02-08:
Hello,

I have a problem with my iPhone Xs and 5Ghz Wi-Fi in OpenWrt, which drives me crazy.

iPhone Xs as only one 5Ghz client.
2.4Ghz works well (as it could).
Router (in my case this was TP-LINK Archer C6 first, then Xiaomi MI Router AC2100) in AP mode with 2.4/5Ghz wireless networks. One or separate SSIDs, no difference.
My iPhone works well at first, got 90/90 of my 100Mbit LAN via 5Ghz (40, 80Mhz channel wide). Then, after some random period of time (I suppose after iPhone goes to deep sleep mode), my iPhone "stucks" in 40/60Mbit via same 5Ghz wireless, same distance to AP, same channel and signal strength.
What does not help: Wi-Fi turn off/on on iPhone, any manipulations with wireless settings in OpenWrt (WMM off/on, disabling ACK timeout or diassoc, any settings), even router reboot does not help.
The only thing that working is rebooting iPhone completely. After reboot wireless speed returns to normal and problem repeats again.

Also tested on friend's iPhone 7 and he have the same problem with my router.

I've mentioned TP-LINK Archer C6 above, this problem was with this router as well as with my current AC2100 (I rid of C6 because of this).

But. I also have MikroTik hAP AC lite, and this problem does not occur on OpenWrt neither than RouterOS.

OpenWrt SNAPSHOT r15679-43ff6e641e.

I appreciate any help: how can I diagnose this strange behavior?

rockenren, 2021-02-08:
/etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option cell_density '0'
        option channel 'auto'
        option htmode 'HT20'
        option txpower '15'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid '<SSID>'
        option encryption 'psk2'
        option key '<password>'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11a'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option cell_density '0'
        option channel 'auto'
        option htmode 'VHT80'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid '<SSID>'
        option encryption 'psk2'
        option key '<password>'

ryderlee1110, 2021-04-04:
Can it be reproduced constantly? I tested against with my iPhone 7 (20Mhz) for couple hours, but it worked fine.

bitcoin1192, 2023-10-24:
Any update on this ? I have same problem as op describe and i have captured few probe response that i want to ask.

rockenren, 2023-10-24:
> Any update on this ? I have same problem as op describe and i have captured few probe response that i want to ask.

what versions of iOS and OpenWRT do you use now?

I didn't do anything, then just updated to 22.03.3 and iOS 16, issue has gone.

bitcoin1192, 2023-10-24:
i am on 23.05, both on ios 16.5.1 and 17.0.3 stuck on 20mhz channel width. other apple device such as iphone se on ios 15.7 were able to change channel width just fine, maybe device specific issue... false positive ?

rockenren, 2023-10-24:
Interesting. I have plans for upgrade to 23.05, but now i'm in doubt.

> maybe device specific issue

it can be. I also have Steam Deck, it have Realtek wi-fi chip and won't work with my APs Atheros wi-fi: it's just stuck at 54mpbs rate in any mode and won't go further no matter what. But it is working fine with MT7615E (Xiaomi MI Router AC2100).

But this does not explain why I don't have problems with my iPhone anymore.

bitcoin1192, 2023-12-21:
I might have found my problem. I recently stumbled upon ax200 card that doesn't connect to one of my AP, and that card have self managed (probably outdated) regdb in it, which ignored linux regdb from os. A little bit experiment of turning on my phone inside faraday cage, with my AP configured to selected country code, and i found out it connected at full bandwidth ! Check with "iw reg get" command to know if your realtek card have self managed regdb in it.