openwrt/mt76 #506: Kernel warning in MT7622 with mesh association - 21.02

jamespeng-wnc, 2021-03-11:
After build 21.02 and write to MT7622 device, when wlan2 join mesh, the warning message shown:

root@OpenWrt:/# [ 3621.458975] ------------[ cut here ]------------
[ 3621.463607] WARNING: CPU: 1 PID: 3104 at __skb_flow_dissect+0x27c/0x1290
[ 3621.470298] Modules linked in: pppoe ppp_async l2tp_ppp iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pptp pppox ppp_mppe ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc rfcomm nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ip_tables hidp hci_uart crc_ccitt compat btusb btintel bnep bluetooth sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred hid evdev input_core nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip_gre gre l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip_tunnel snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm snd_timer snd_mixer_oss snd_hwdep
[ 3621.470398]  snd_compress snd soundcore vfat fat ntfs autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha1_generic seqiv md5 md4 ghash_generic gf128mul gcm des_generic libdes ctr ccm arc4 usb_storage leds_gpio xhci_plat_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 crc32c_generic
[ 3621.588416] CPU: 1 PID: 3104 Comm: kworker/u5:0 Not tainted 5.4.99 #0
[ 3621.594845] Hardware name: MT7622_MT7531 RFB (DT)
[ 3621.599545] Workqueue: napi_workq napi_workfn
[ 3621.603894] pstate: 40000005 (nZcv daif -PAN -UAO)
[ 3621.608676] pc : __skb_flow_dissect+0x27c/0x1290
[ 3621.613284] lr : __skb_get_hash+0x74/0x220
[ 3621.617369] sp : ffffffc01234b8e0
[ 3621.620673] x29: ffffffc01234b8e0 x28: 0000000000000000 
[ 3621.625976] x27: 000000000000ffba x26: ffffff8019945418 
[ 3621.631279] x25: ffffffc010908748 x24: 0000000000000076 
[ 3621.636581] x23: ffffffc01234ba48 x22: ffffffc010909470 
[ 3621.641883] x21: 000000000000ffba x20: ffffff80199de000 
[ 3621.647184] x19: 0000000000000000 x18: 00000000009e0000 
[ 3621.652486] x17: 0000000000da0000 x16: 000000000000032d 
[ 3621.657788] x15: 000000000000011c x14: 81746efdff3a2000 
[ 3621.663090] x13: 00000060dd86ed03 x12: 0000000000000000 
[ 3621.668392] x11: 0000000000000006 x10: 0120f0b097592643 
[ 3621.673694] x9 : 0000000000000088 x8 : 0000000000000000 
[ 3621.678996] x7 : 0000000000000000 x6 : 0000000000000000 
[ 3621.684297] x5 : 0000000000000000 x4 : 0000000000000000 
[ 3621.689598] x3 : ffffffc01234ba48 x2 : 0000000000000000 
[ 3621.694900] x1 : 0000000000000000 x0 : 0000000000000000 
[ 3621.700201] Call trace:
[ 3621.702640]  __skb_flow_dissect+0x27c/0x1290
[ 3621.706900]  __skb_get_hash+0x74/0x220
[ 3621.710668]  ieee80211_txq_may_transmit+0x718/0x808 [mac80211]
[ 3621.716501]  ieee80211_tx_prepare_skb+0x1f4/0x250 [mac80211]
[ 3621.722159]  ieee80211_tx_pending+0x128/0x2c8 [mac80211]
[ 3621.727464]  tasklet_action_common.isra.15+0xac/0x150
[ 3621.732506]  tasklet_action+0x24/0x30
[ 3621.736159]  __do_softirq+0x124/0x260
[ 3621.739812]  do_softirq.part.14+0x60/0x68
[ 3621.743812]  __local_bh_enable_ip+0x80/0x88
[ 3621.747987]  napi_workfn+0x4c/0x80
[ 3621.751380]  process_one_work+0x1fc/0x390
[ 3621.755379]  worker_thread+0x48/0x4d0
[ 3621.759033]  kthread+0x120/0x128
[ 3621.762253]  ret_from_fork+0x10/0x1c
[ 3621.765819] ---[ end trace d73b5a27bfc81508 ]---

however, the mesh network are still connected.
this warning message was printed in the net/core/flow_dissector.c,
refer to patch: https://lists.openwall.net/netdev/2019/04/22/89

nbd168, 2021-11-25:
Fixed in OpenWrt

closed 2021-11-25 by nbd168

MauriceS-WTI, 2021-12-12:
Which commit was it? or can I apply the whole MT76 tree to a 5.11 kernel - for a raspberry pi?