openwrt/mt76 #507: MT7622 crash

graphine27, 2021-03-14:
The internet connectivity was lost, I tried to restart device from openwrt but then it crashed.

```
root@OpenWrt:/# [50843.336746] device wlan0 left promiscuous mode
[50843.341306] br-lan: port 2(wlan0) entered disabled state
[50843.382859] mt7622-wmac 18000000.wmac: Message 8 (seq 2) timeout
[50873.508743] ------------[ cut here ]------------
[50873.513381] Timeout waiting for MCU reset state 8
[50873.518171] WARNING: CPU: 0 PID: 6815 at mt7615_dma_reset+0x1ec/0x35c [mt7615_common]
[50873.526014] Modules linked in: pppoe ppp_async nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_ct nft_counter nft_chain_nat nf_tables mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_FLOWOFFLOAD ts_kmp ts_fsm ts_bm slhc nfnetlink nf_reject_ipv4 nf_log_ipv4 nf_flow_table libcrc32c iptable_nat iptable_mangle iptable_filter ip_tables crc_ccitt compat br_netfilter cryptodev ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos vfat fat
[50873.526181]  autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic authenc sdhci_pltfm sdhci leds_gpio xhci_plat_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug
[50873.631780] CPU: 0 PID: 6815 Comm: kworker/u4:2 Tainted: G S                5.10.23 #0
[50873.639685] Hardware name: Bananapi BPI-R64 (DT)
[50873.644301] Workqueue: phy0 mt7615_mac_reset_work [mt7615_common]
[50873.650388] pstate: 40000005 (nZcv daif -PAN -UAO -TCO BTYPE=--)
[50873.656388] pc : mt7615_dma_reset+0x1ec/0x35c [mt7615_common]
[50873.662127] lr : mt7615_dma_reset+0x1ec/0x35c [mt7615_common]
[50873.667861] sp : ffffffc014873ca0
[50873.671167] x29: ffffffc014873ca0 x28: ffffff8002933448 
[50873.676472] x27: ffffff80029332a8 x26: ffffffc010a09ee8 
[50873.681776] x25: 0000000000000000 x24: ffffff800292cfa8 
[50873.687080] x23: ffffff8002934fc8 x22: 0000000000000008 
[50873.692384] x21: ffffff80029326e0 x20: 0000000000000001 
[50873.697688] x19: 0000000000000000 x18: 000000000000010c 
[50873.702992] x17: 0000000000000002 x16: 0000000000000002 
[50873.708296] x15: ffffffc010a0bfd8 x14: 0000000000000324 
[50873.713600] x13: 000000000000010c x12: ffffffc014873978 
[50873.718905] x11: ffffffc010a63fd8 x10: 00000000fffff000 
[50873.724209] x9 : ffffffc010a63fd8 x8 : 0000000000000000 
[50873.729514] x7 : 80000000fffff000 x6 : 0000000000000001 
[50873.734818] x5 : 0000000000000000 x4 : 0000000000000000 
[50873.740122] x3 : 0000000000000000 x2 : ffffff803fdba3e0 
[50873.745426] x1 : ffffffc02f3d7000 x0 : 0000000000000025 
[50873.750730] Call trace:
[50873.753171]  mt7615_dma_reset+0x1ec/0x35c [mt7615_common]
[50873.758563]  mt7615_mac_reset_work+0x1b4/0x4e0 [mt7615_common]
[50873.764389]  process_one_work+0x1d4/0x370
[50873.768389]  worker_thread+0x178/0x4e0
[50873.772130]  kthread+0x120/0x124
[50873.775350]  ret_from_fork+0x10/0x18
[50873.778915] ---[ end trace ef518efa6544f714 ]---
```

ryderlee1110, 2021-03-15:
how to reproduce this?

graphine27, 2021-03-15:
> how to reproduce this?

Can't be reproduced constantly. Sometimes MT7622-wmac loses internet connectivity (this can happen after 1-5 days of uptime). 
When internet connectivity is lost,  pressing restart radio0 (MediaTek MT7622 802.11bgn) in LuCI causes that crash.

ryderlee1110, 2021-03-18:
Try to keep firmware log enabled after bootup to see if there's any error shows up when issue happens.

root@OpenWrt:/sys/kernel/debug/ieee80211/phy0/mt76# echo 1 > fw_debug
[   84.604743] ieee80211 phy0: N9: (00002536839)Thermo: new bounds=0x32(25), 0x3c(42)
....

graphine27, 2021-03-19:
```
root@OpenWrt:/sys/kernel/debug/ieee80211/phy0/mt76# [353442.199067] ieee80211 phy0: N9: (02990393757)EXT_CMD_OPT_BIT_0_ACK
[353442.199067] �+���iE��y�
�Hv�����O����qp�d0;��֏��_43����C�u�l��6����
[353504.900623] ieee80211 phy0: N9: (02992448198)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[353508.347594] ieee80211 phy0: N9: (02992561140)EXT_CMD_OPT_BIT_0_ACKMCQ] BMC in BMCQ pkt number=255, impossible
[353508.347594] HXee80211 phy0: N9: (02990590830)EXT_CMD_OPT_BIT_0_ACK
[353514.356372] ieee80211 phy0: N9: (02992758020)EXT_CMD_OPT_BIT_0_ACK���aNt�0#��'���b���Ita�,z�I�(F��ۦ'���&fIUߓO��!#�7"rɊD��7P]��X��z�(q�̼�99#Y��jd�*�AP�'fh�|��p��{Y;S��녩7^|w�&�
[353515.345413] ieee80211 phy0: N9: (02992790426)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[353520.363266] ieee80211 phy0: N9: (02992954839)EXT_CMD_OPT_BIT_0_ACKZ��y�B��-M2
[353520.363266] ��z�Z�=Ʋ�d
[353525.687864] ieee80211 phy0: N9: (02993129299)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[353526.373891] ieee80211 phy0: N9: (02993151779)EXT_CMD_OPT_BIT_0_ACK
[353526.373891] HX	
[353532.382671] ieee80211 phy0: N9: (02993348659)EXT_CMD_OPT_BIT_0_ACK
[353536.030214] ieee80211 phy0: N9: (02993468172)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[353538.389551] ieee80211 phy0: N9: (02993545477)EXT_CMD_OPT_BIT_0_ACKMCQ] BMC in BMCQ pkt number=255, impossible
[353538.389551] HXee80211 phy0: N9: (02991182452)EXT_CMD_OPT_BIT_0_ACK
[353544.398331] ieee80211 phy0: N9: (02993742357)EXT_CMD_OPT_BIT_0_ACK
[353544.398331] HXee80211 phy0: N9: (02991379335)EXT_CMD_OPT_BIT_0_ACK
[353546.372618] ieee80211 phy0: N9: (02993807045)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[353550.407130] ieee80211 phy0: N9: (02993939238)EXT_CMD_OPT_BIT_0_ACK
[353556.414023] ieee80211 phy0: N9: (02994136056)EXT_CMD_OPT_BIT_0_ACK
[353556.414023] HXee80211 phy0: N9: (02991770452)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[353556.817422] ieee80211 phy0: N9: (02994149273)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[353562.422881] ieee80211 phy0: N9: (02994332939)EXT_CMD_OPT_BIT_0_ACK
[353562.422881] 	
[353567.262239] ieee80211 phy0: N9: (02994491501)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[353568.431586] ieee80211 phy0: N9: (02994529816)EXT_CMD_OPT_BIT_0_ACK
[353568.431586] HX
[353572.325408] ieee80211 phy0: N9: (02994657393)EXT_CMD_OPT_BIT_0_ACK
[353572.325408] HX
[353572.356678] device wlan0 left promiscuous mode
[353572.361336] br-lan: port 2(wlan0) entered disabled state
[353572.381311] ieee80211 phy0: N9: (02994659230)StaRec2 Tag 0x6 
[353572.387253] ieee80211 phy0: N9: (02994659232)StaRec2 Tag 0xd 
[353572.393178] ieee80211 phy0: N9: (02994659233)WTBL# Idx 2 Op 2
[353572.399223] ieee80211 phy0: N9: (02994659818)StaRec2 Tag 0x6 
[353572.405189] ieee80211 phy0: N9: (02994659819)StaRec2 Tag 0xd 
[353572.411110] ieee80211 phy0: N9: (02994659821)WTBL# Idx 2 Op 2
[353572.417166] ieee80211 phy0: N9: (02994660406)StaRec2 Tag 0x6 
[353572.423125] ieee80211 phy0: N9: (02994660407)StaRec2 Tag 0xd 
[353572.429052] ieee80211 phy0: N9: (02994660409)WTBL# Idx 2 Op 2
[353572.435159] ieee80211 phy0: N9: (02994660995)StaRec2 Tag 0x6 
[353572.441120] ieee80211 phy0: N9: (02994660997)StaRec2 Tag 0xd 
[353572.447041] ieee80211 phy0: N9: (02994660998)WTBL# Idx 2 Op 2
[353572.453191] ieee80211 phy0: N9: (02994661586)StaRec2 Tag 0x6 
[353572.459158] ieee80211 phy0: N9: (02994661588)StaRec2 Tag 0xd 
[353572.465080] ieee80211 phy0: N9: (02994661589)WTBL# Idx 2 Op 2
[353572.472219] ieee80211 phy0: N9: (02994662209)StaRec2 Tag 0x6 
[353572.478182] ieee80211 phy0: N9: (02994662211)StaRec2 Tag 0xd 
[353572.484171] ieee80211 phy0: N9: (02994662212)WTBL# Idx 2 Op 2
[353572.490304] ieee80211 phy0: N9: (02994662802)StaRec2 Tag 0x6 
[353572.496268] ieee80211 phy0: N9: (02994662804)StaRec2 Tag 0xd 
[353572.502190] ieee80211 phy0: N9: (02994662805)WTBL# Idx 2 Op 2
[353572.509313] ieee80211 phy0: N9: (02994663424)StaRec2 Tag 0x6 
[353572.515246] ieee80211 phy0: N9: (02994663426)StaRec2 Tag 0xd 
[353572.521167] ieee80211 phy0: N9: (02994663427)WTBL# Idx 2 Op 2
[353572.527274] ieee80211 phy0: N9: (02994664014)StaRec1 Tag 0x6 
[353572.533235] ieee80211 phy0: N9: (02994664015)StaRec1 Tag 0xd 
[353572.539156] ieee80211 phy0: N9: (02994664017)WTBL# Idx 1 Op 2
[353572.548803] ieee80211 phy0: N9: (02994664718)StaRec1 Tag 0x6 
[353572.554779] ieee80211 phy0: N9: (02994664720)StaRec1 Tag 0xd 
[353572.560728] ieee80211 phy0: N9: (02994664721)WTBL# Idx 1 Op 2
[353572.566818] ieee80211 phy0: N9: (02994665309)StaRec1 Tag 0x6 
[353572.572785] ieee80211 phy0: N9: (02994665311)StaRec1 Tag 0xd 
[353572.578708] ieee80211 phy0: N9: (02994665312)WTBL# Idx 1 Op 2
[353572.584941] ieee80211 phy0: N9: (02994665903)StaRec1 Tag 0x0 
[353572.590917] ieee80211 phy0: N9: (02994665912)StaRec[1] is removed from staList
[353572.590917] ��z�Z�=Ʋ�d
[... 39 lines cut ...]
[353604.108468] x15: ffffffc010a0bfd8 x14: 0000000000000540 
[353604.113859] x13: 00000000000001c0 x12: ffffffc0113f3978 
[353604.119249] x11: ffffffc010a63fd8 x10: 00000000fffff000 
[353604.124641] x9 : ffffffc010a63fd8 x8 : 0000000000000000 
[353604.130031] x7 : 80000000fffff000 x6 : 0000000000000001 
[353604.135422] x5 : 0000000000000000 x4 : 0000000000000000 
[353604.140812] x3 : 0000000000000000 x2 : ffffff803fdce3e0 
[353604.146204] x1 : ffffffc02f3eb000 x0 : 0000000000000025 
[353604.151595] Call trace:
[... cut ...]
[353604.169600]  worker_thread+0x178/0x4e0
[353604.173429]  kthread+0x120/0x124
[353604.176736]  ret_from_fork+0x10/0x18
[353604.180392] Code: 2a1603e1 90000040 913e0000 95f277a2 (d4210000) 
[353604.186562] ---[ end trace af428c7c57da3457 ]---
[353604.191256] Kernel panic - not syncing: BRK handler: Fatal exception
[353604.197688] SMP: stopping secondary CPUs
[353604.201694] Kernel Offset: disabled
[353604.205260] CPU features: 0x0000002,04002004
[353604.209606] Memory Limit: none
[353604.212739] Rebooting in 3 seconds..

```

ryderlee1110, 2021-03-21:
could you please try this https://github.com/ryderlee1110/wireless/commit/3e05c2e6c144ab1d214c62460ce8246f61134185

graphine27, 2021-03-21:
Applied patch, I will let you know if it happens again.

ryderlee1110, 2021-03-22:
my bad. a quick patch without compiling.  btw please test the following change as well.

static void
mt7615_mcu_int_event_trigger(struct mt7615_dev *dev, u8 event)
{
	mt7622_trigger_hif_int(dev, true);

	mt76_wr(dev, MT_MCU_INT_EVENT, event);

	mt7622_trigger_hif_int(dev, false);
}

graphine27, 2021-03-22:
> could you please try this [ryderlee1110/wireless@3e05c2e](https://github.com/ryderlee1110/wireless/commit/3e05c2e6c144ab1d214c62460ce8246f61134185) ?

Internet connectivity still lost, when trying to reset the device it no longer crashes, but I'm unable to restart the wifi device without rebooting
```

[61182.340360] ieee80211 phy0: N9: (02004302161)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[61186.145138] ieee80211 phy0: N9: (02004426827)EXT_CMD_OPT_BIT_0_ACK
[61186.145138] HX
[61192.158820] ieee80211 phy0: N9: (02004623868)EXT_CMD_OPT_BIT_0_ACK
[61192.158820] HX
[61192.682748] ieee80211 phy0: N9: (02004641034)[appsMoveBMCFromACtoBMCQ] BMC in BMCQ pkt number=255, impossible
[61193.743334] ieee80211 phy0: N9: (02004675785)EXT_CMD_OPT_BIT_0_ACK
[61193.743334] HX
[61193.774366] device wlan0 left promiscuous mode
[61193.778926] br-lan: port 2(wlan0) entered disabled state
[61193.798805] ieee80211 phy0: N9: (02004677602)StaRec2 Tag 0x0 
[61193.804664] ieee80211 phy0: N9: (02004677611)StaRec[2] is removed from staList
[61193.804664] �.�8
[61193.813849] ieee80211 phy0: N9: (02004677614)StaRec2 Tag 0xd 
[61193.819682] ieee80211 phy0: N9: (02004677616)WTBL# Idx 2 Op 1
[61193.826766] ieee80211 phy0: N9: (02004678518)BSS0Info Tag 0x1
[61193.832645] ieee80211 phy0: N9: (02004678520)BSSInfo 0 Inactive
[61193.838667] ieee80211 phy0: N9: (02004678521)BSS0 PM update 
[61193.844419] ieee80211 phy0: N9: (02004678523)BSS0 Stop keep alive timer 
[61193.851486] ieee80211 phy0: N9: (02004679327)StaRec127 Tag 0x0 
[61193.859895] ieee80211 phy0: N9: (02004679339)StaRec127 Tag 0xd 
[61193.867199] ieee80211 phy0: N9: (02004679341)WTBL# Idx 127 Op 1
[61193.875495] ieee80211 phy0: N9: (02004679343)UMAC PLE HW error !!! (0x   1a000) 
[61193.884464] ieee80211 phy0: N9: (02004679345)SerL1ResetStep1: (A/B) Stop LMAC TxRx and Logic reset LMAC
[61193.901842] ieee80211 phy0: N9: (02004679380)(C) Stop PDMA 0
[61194.074308] mt7622-wmac 18000000.wmac: Message 8 (seq 9) timeout
[61215.958379] mt7622-wmac 18000000.wmac: Message 42 (seq 10) timeout
[61236.438504] mt7622-wmac 18000000.wmac: Message 7 (seq 11) timeout
[61256.918566] mt7622-wmac 18000000.wmac: Message 70 (seq 12) timeout
```

graphine27, 2021-03-22:
```
[11641.256824] US10U
[11641.256824] DigiCert Inc1'0%UDigiCert SHA2 Secure Server CA0�"0
	*�H��

[11647.274122] ieee80211 phy0: N9: (00381271308)EXT_CMD_OPT_BIT_0_ACK
[11647.274122] %	
[11653.282627] ieee80211 phy0: N9: (00381468178)EXT_CMD_OPT_BIT_0_ACK
[11653.282627] �
[11657.529457] ieee80211 phy0: N9: (00381607328)StaRec1 Tag 0x6 
[11657.535311] ieee80211 phy0: N9: (00381607329)StaRec1 Tag 0xd 
[11657.541168] ieee80211 phy0: N9: (00381607331)WTBL# Idx 1 Op 2
[11659.216869] ieee80211 phy0: N9: (00381662615)LMAC HW error !!! (0x80100000) 
[11659.251688] ieee80211 phy0: N9: (00381663756)LMAC HW error !!! (0x  100000) 
[11659.290987] ieee80211 phy0: N9: (00381665045)EXT_CMD_OPT_BIT_0_ACK
[11659.290987] H,
[11661.010914] ieee80211 phy0: N9: (00381721398)LMAC HW error !!! (0x80100000) 
[11661.018069] ieee80211 phy0: N9: (00381721400)SerL1ResetStep1: (A/B) Stop LMAC TxRx and Logic reset LMAC
[11661.027553] ieee80211 phy0: N9: (00381721435)(C) Stop PDMA 0
[11682.817059] mt7622-wmac 18000000.wmac: Message 73 (seq 12) timeout
[11703.297180] mt7622-wmac 18000000.wmac: Message 73 (seq 13) timeout
[11729.921263] mt7622-wmac 18000000.wmac: Message 73 (seq 14) timeout
[11756.545335] mt7622-wmac 18000000.wmac: Message 73 (seq 15) timeout
[11783.169452] mt7622-wmac 18000000.wmac: Message 73 (seq 1) timeout
[11809.793526] mt7622-wmac 18000000.wmac: Message 73 (seq 2) timeout
[11836.417631] mt7622-wmac 18000000.wmac: Message 73 (seq 3) timeout
[11863.041729] mt7622-wmac 18000000.wmac: Message 73 (seq 4) timeout
[11889.665820] mt7622-wmac 18000000.wmac: Message 73 (seq 5) timeout
[11916.289933] mt7622-wmac 18000000.wmac: Message 73 (seq 6) timeout
[11942.914075] mt7622-wmac 18000000.wmac: Message 73 (seq 7) timeout
[11969.538130] mt7622-wmac 18000000.wmac: Message 73 (seq 8) timeout
[11996.162234] mt7622-wmac 18000000.wmac: Message 73 (seq 9) timeout
[12020.738329] mt7622-wmac 18000000.wmac: Message 37 (seq 10) timeout
[12020.744541] wlan0: HW problem - can not stop rx aggregation for f4:f1:5a:0f:09:fd tid 0
[12041.218365] mt7622-wmac 18000000.wmac: Message 37 (seq 11) timeout
[12041.224617] ------------[ cut here ]------------
[12041.229267] WARNING: CPU: 0 PID: 1785 at ___ieee80211_stop_tx_ba_session+0x368/0x3cc [mac80211]
[12041.237953] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet xt_state xt_conntrack pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_chain_nat nf_tables nf_nat_ftp nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack_ftp nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ip_tables cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables ts_kmp ts_fsm ts_bm slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv4 nf_log_common nf_dup_netdev nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat cryptodev msdos vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic
[12041.238105]  authenc sdhci_pltfm sdhci leds_gpio xhci_plat_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug
[12041.338660] CPU: 0 PID: 1785 Comm: hostapd Tainted: G S                5.10.23 #0
[12041.346131] Hardware name: Bananapi BPI-R64 (DT)
[12041.350741] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[12041.356751] pc : ___ieee80211_stop_tx_ba_session+0x368/0x3cc [mac80211]
[12041.363366] lr : ___ieee80211_stop_tx_ba_session+0x210/0x3cc [mac80211]
[12041.369969] sp : ffffffc0114037e0
[12041.373273] x29: ffffffc0114037e0 x28: ffffff8000dd4200 
[12041.378579] x27: ffffffc011403df0 x26: 0000000000000000 
[12041.383883] x25: ffffffc010ad5000 x24: ffffff8002198fc0 
[12041.389188] x23: ffffff8002ec40f0 x22: ffffffc008972a9c 
[12041.394492] x21: 0000000000000003 x20: ffffff8002e07500 
[12041.399797] x19: ffffff8000942000 x18: 0000000000000968 
[12041.405101] x17: 0000000000000000 x16: 000000004384a800 
[12041.410405] x15: ffffffc010a0bfd8 x14: 0000000000001c38 
[12041.415710] x13: 0000000000000968 x12: ffffffc0114032f8 
[12041.421013] x11: ffffffc010a63fd8 x10: 00000000fffff000 
[12041.426318] x9 : ffffffc010a63fd8 x8 : 0000000000000000 
[12041.431621] x7 : 0000000000000000 x6 : 0000000000000002 
[12041.436925] x5 : 0000000000000001 x4 : 0000000000000000 
[12041.442230] x3 : 0000000000000000 x2 : 0000000000000001 
[12041.447534] x1 : 0000000000000002 x0 : 00000000ffffff92 
[12041.452838] Call trace:
[12041.455288]  ___ieee80211_stop_tx_ba_session+0x368/0x3cc [mac80211]
[12041.461556]  ieee80211_sta_tear_down_BA_sessions+0x78/0x12c [mac80211]
[12041.468085]  ieee80211_sta_set_buffered+0x52c/0x6c0 [mac80211]
[12041.473918]  sta_info_destroy_addr_bss+0x38/0x6c [mac80211]
[12041.479492]  ieee80211_csa_finish+0x1668/0x2360 [mac80211]
[12041.484980]  cfg80211_check_station_change+0x10f4/0x36d0 [cfg80211]
[12041.491242]  genl_family_rcv_msg_doit+0xb4/0x110
[12041.495851]  genl_rcv_msg+0xd0/0x1c0
[12041.499418]  netlink_rcv_skb+0x54/0x110
[... 107 lines cut ...]
[12082.497671]  el0_svc+0x10/0x1c
[12082.500717]  el0_sync_handler+0xa8/0xac
[12082.504544]  el0_sync+0x140/0x180
[12082.507850] ---[ end trace a4d03c99c9f13c4c ]---
[... cut ...]
[12409.859687] mt7622-wmac 18000000.wmac: Message 73 (seq 11) timeout
[12436.483757] mt7622-wmac 18000000.wmac: Message 73 (seq 12) timeout
[12463.107848] mt7622-wmac 18000000.wmac: Message 73 (seq 13) timeout
[12489.731923] mt7622-wmac 18000000.wmac: Message 73 (seq 14) timeout
[12516.356013] mt7622-wmac 18000000.wmac: Message 73 (seq 15) timeout
[12542.980140] mt7622-wmac 18000000.wmac: Message 73 (seq 1) timeout
[12569.604195] mt7622-wmac 18000000.wmac: Message 73 (seq 2) timeout
[12596.228297] mt7622-wmac 18000000.wmac: Message 73 (seq 3) timeout
```

nbd168, 2021-03-22:
I think the function should look like this:

```
static void
mt7615_mcu_int_event_trigger(struct mt7615_dev *dev, u8 event)
{
	mt76_wr(dev, MT_MCU_INT_EVENT, event);

	mt7622_trigger_hif_int(dev, true);
	mt7622_trigger_hif_int(dev, false);
}
```

I think the hif interrupt needs to be triggered after MT_MCU_INT_EVENT has been written.

ryderlee1110, 2021-03-22:
right. that's my first version

nbd168, 2021-03-26:
Reset recovery works fine on MT7622 in the latest version. Fixed in b8609066893a399b6aabe4df2e80eed52336fc69

closed 2021-03-26 by nbd168