openwrt/mt76 #511: mt7613 wlan scan cause mcu timeout

ptpt52, 2021-03-25:
chip is mt7613
```
[  660.520319] mt7615e 0000:02:00.0: Message 73 (seq 9) timeout
[  681.000319] mt7615e 0000:02:00.0: Message 73 (seq 10) timeout
[  701.480334] mt7615e 0000:02:00.0: Message 73 (seq 11) timeout
```

I try to `ubus call iwinfo scan {"device": "wlan1"}` 
after Multiple scans
this issue happens

ptpt52, 2021-03-25:
here is how to reproduce it
```
while :; do iwinfo wlan1 scan; echo; uptime; sleep 1; done
```
any help? @nbd168

ptpt52, 2021-03-25:
it is likely that no issue on mt7615 chip
I do the same test one mt7615, no issue

Borromini, 2021-03-25:
@ptpt52 I am seeing this message as well, maybe it's unrelated, but I have two MT7613BE radios here (EAP235-Wall) and it eventually happens on both at some point. It's not a question if, it's a question when. Those timeout messages will fill the log for a while and eventually the radio just dies. I've only seen this when the radios idle - I have pushed hundreds of gigabytes over the radio and that works just fine it seems, it's just when it's idling that at some point it seems to go wrong. I had no way to trigger it but just wait (sometimes it's within a few hours, sometimes it's multiple days though).

I ran your iwinfo command, and this is what `logread` shows (OpenWrt 21.02 with latest mt76 driver from OpenWrt master):

```
Thu Mar 25 23:18:20 2021 authpriv.notice dropbear[6555]: Pubkey auth succeeded for 'root' with key sha1!! <snip> from 10.0.0.5:35098
Thu Mar 25 23:18:21 2021 kern.info kernel: [123931.918046] ieee80211 phy1: N9: (124009.228:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:21 2021 kern.info kernel: [123931.918046] �������;���]�;���]p�b@�r
Thu Mar 25 23:18:27 2021 kern.info kernel: [123937.941969] ieee80211 phy1: N9: (124015.255:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:27 2021 kern.info kernel: [123937.941969] �������;���]�;���]��AЛr
Thu Mar 25 23:18:33 2021 kern.info kernel: [123943.965938] ieee80211 phy1: N9: (124021.282:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:33 2021 kern.info kernel: [123943.965938] �������;���]�;���]��J`�r
Thu Mar 25 23:18:38 2021 kern.info kernel: [123948.667646] ieee80211 phy1: N9: (124025.986:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:38 2021 kern.info kernel: [123948.667646] �������;���]�;���]Q��u
Thu Mar 25 23:18:39 2021 kern.info kernel: [123949.989899] ieee80211 phy1: N9: (124027.310:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:39 2021 kern.info kernel: [123949.989899] �������;���]�;���]�:PP�w
Thu Mar 25 23:18:42 2021 kern.info kernel: [123952.559046] ieee80211 phy1: N9: (124029.880:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:42 2021 kern.info kernel: [123952.559046] �������;���]�;���]�X�
Thu Mar 25 23:18:43 2021 kern.info kernel: [123953.545245] ieee80211 phy1: N9: (124030.867:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:43 2021 kern.info kernel: [123953.545245] �������;���]�;���]�-\�j
Thu Mar 25 23:18:45 2021 kern.info kernel: [123956.016801] ieee80211 phy1: N9: (124033.340:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:45 2021 kern.info kernel: [123956.016801] �������;���]�;���]ІRp�
Thu Mar 25 23:18:47 2021 kern.info kernel: [123957.463012] ieee80211 phy1: N9: (124034.787:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:47 2021 kern.info kernel: [123957.463012] �������;���]�;���] �C �
Thu Mar 25 23:18:48 2021 kern.info kernel: [123958.449243] ieee80211 phy1: N9: (124035.774:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:48 2021 kern.info kernel: [123958.449243] �������;���]�;���]p�O@�
Thu Mar 25 23:18:51 2021 kern.info kernel: [123962.072360] ieee80211 phy1: N9: (124039.399:CMD-E)EXT_CMD_OPT_BIT_0_ACK
Thu Mar 25 23:18:51 2021 kern.info kernel: [123962.072360] �������;���]�;���]�ZYp�@
Thu Mar 25 23:18:52 2021 kern.info kernel: [123962.364951] ieee80211 phy1: N9: Wifi ASSERT @ ../wifi/custom/feature/pm_ap.c:454
Thu Mar 25 23:18:52 2021 kern.info kernel: [123962.364951] ��]�;���]�rWp{�
Thu Mar 25 23:19:13 2021 kern.err kernel: [123984.168757] mt7615e 0000:02:00.0: Message 73 (seq 15) timeout
Thu Mar 25 23:19:34 2021 kern.err kernel: [124004.648606] mt7615e 0000:02:00.0: Message 73 (seq 1) timeout
Thu Mar 25 23:19:54 2021 kern.err kernel: [124025.128456] mt7615e 0000:02:00.0: Message 38 (seq 2) timeout
Thu Mar 25 23:20:15 2021 kern.err kernel: [124045.608303] mt7615e 0000:02:00.0: Message 37 (seq 3) timeout
Thu Mar 25 23:20:35 2021 kern.err kernel: [124066.088147] mt7615e 0000:02:00.0: Message 73 (seq 4) timeout
Thu Mar 25 23:20:56 2021 kern.err kernel: [124086.567997] mt7615e 0000:02:00.0: Message 8 (seq 5) timeout
Thu Mar 25 23:21:16 2021 kern.err kernel: [124107.047845] mt7615e 0000:02:00.0: Message 8 (seq 6) timeout
Thu Mar 25 23:21:37 2021 kern.err kernel: [124127.527696] mt7615e 0000:02:00.0: Message 8 (seq 7) timeout
Thu Mar 25 23:21:57 2021 kern.err kernel: [124148.007541] mt7615e 0000:02:00.0: Message 8 (seq 8) timeout
Thu Mar 25 23:22:18 2021 kern.err kernel: [124168.487391] mt7615e 0000:02:00.0: Message 8 (seq 9) timeout
Thu Mar 25 23:22:38 2021 kern.err kernel: [124188.967236] mt7615e 0000:02:00.0: Message 8 (seq 10) timeout
Thu Mar 25 23:22:59 2021 kern.err kernel: [124209.447088] mt7615e 0000:02:00.0: Message 8 (seq 11) timeout
Thu Mar 25 23:23:19 2021 kern.err kernel: [124229.926931] mt7615e 0000:02:00.0: Message 8 (seq 12) timeout
Thu Mar 25 23:23:40 2021 kern.err kernel: [124250.406779] mt7615e 0000:02:00.0: Message 8 (seq 13) timeout
Thu Mar 25 23:24:00 2021 kern.err kernel: [124270.886634] mt7615e 0000:02:00.0: Message 8 (seq 14) timeout
Thu Mar 25 23:24:21 2021 kern.err kernel: [124291.366479] mt7615e 0000:02:00.0: Message 8 (seq 15) timeout
Thu Mar 25 23:24:41 2021 kern.err kernel: [124311.846333] mt7615e 0000:02:00.0: Message 8 (seq 1) timeout
Thu Mar 25 23:25:02 2021 kern.err kernel: [124332.326180] mt7615e 0000:02:00.0: Message 8 (seq 2) timeout
```

I have pinged nbd about this on IRC a while ago and was asked earlier on to switch on firmware debugging, like this:

`echo 1 > /sys/kernel/debug/ieee80211/phy1/mt76/fw_debug`

I can't say it's the same issue, but at least it's the same error message... I was asked to bisect also, and it looked like with the 2021-01-14 version, it worked fine, but with the 2021-01-27 one it broke. I have attempted multiple bisects but eventually I'd end up with a commit that git marked as bad, only to find out that reverting it didn't help either. So I'm at a loss, but hopefully the above can help.

ptpt52, 2021-03-25:
@Borromini  thanks for you reply
it is likely that it trigger the issue after you run the `iwinfo scan` command?

ryderlee1110, 2021-03-25:
can you try to apply this patch to https://patchwork.kernel.org/project/linux-wireless/patch/fdde0eb8c6dea0e81b5f5ae528085f1ccf53fef3.1616696376.git.ryder.lee@mediatek.com/ see if system ban be recovered back?

Borromini, 2021-03-25:
> it is likely that it trigger the issue after you run the `iwinfo scan` command?

Running your command definitely triggered the timeout message, and the radio died. I ran your command at 23:18, more or less right after I SSH'ed into the device (start of the log excerpt). Log wasn't 

@ryderlee1110 Thanks I will give that a shot.

ryderlee1110, 2021-03-25:
@Borromini  which firmware you're using? please apply https://patchwork.kernel.org/project/linux-wireless/patch/02d52818307a85b66e59ca11e36a9da47e2fe4df.1616657048.git.ryder.lee@mediatek.com/ as well? I fix a mt7615 scan issue. but maybe it's a different one.

Borromini, 2021-03-25:
@ryderlee1110 How can I tell? This is the OpenWrt firmware package: `kmod-mt7663-firmware-ap - 5.4.107+2021-03-15-ca22cc22-4`.

Boot log shows this, I don't see any firmware info on for the MT7615 radio:

```
Thu Mar 25 22:03:56 2021 kern.info kernel: [   29.604167] Loading modules backported from Linux version v5.10.16-0-gde53befa79cf
Thu Mar 25 22:03:56 2021 kern.info kernel: [   29.619268] Backport generated by backports.git v5.10.16-1-0-g21d2a1d2
Thu Mar 25 22:03:56 2021 kern.info kernel: [   29.737818] xt_time: kernel timezone is -0000
Thu Mar 25 22:03:56 2021 kern.info kernel: [   29.811746] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=23
Thu Mar 25 22:03:56 2021 kern.info kernel: [   29.822572] pci 0000:00:00.0: enabling device (0006 -> 0007)
Thu Mar 25 22:03:56 2021 kern.info kernel: [   29.833882] mt7603e 0000:01:00.0: enabling device (0000 -> 0002)
Thu Mar 25 22:03:56 2021 kern.info kernel: [   29.846105] mt7603e 0000:01:00.0: ASIC revision: 76030010
Thu Mar 25 22:03:56 2021 kern.info kernel: [   30.476132] mt7603e 0000:01:00.0: Firmware Version: ap_pcie
Thu Mar 25 22:03:56 2021 kern.info kernel: [   30.487265] mt7603e 0000:01:00.0: Build Time: 20160107100755
Thu Mar 25 22:03:56 2021 kern.info kernel: [   30.522881] mt7603e 0000:01:00.0: firmware init done
Thu Mar 25 22:03:56 2021 kern.debug kernel: [   30.707828] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
Thu Mar 25 22:03:56 2021 kern.info kernel: [   30.720534] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=24
Thu Mar 25 22:03:56 2021 kern.info kernel: [   30.731349] pci 0000:00:01.0: enabling device (0006 -> 0007)
Thu Mar 25 22:03:56 2021 kern.info kernel: [   30.742648] mt7615e 0000:02:00.0: enabling device (0000 -> 0002)
Thu Mar 25 22:03:56 2021 kern.debug kernel: [   30.774070] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'

```

ryderlee1110, 2021-03-25:
Firmware looks right. anyway, we were not deeply involve in mt7613 mt76. It should work but maybe not 100% perfect.

ptpt52, 2021-03-26:
@Borromini  your msg shows the 7603 firmware version, not 7613
@ryderlee1110 

here is my msg:
```
[    2.652532] pci 0000:02:00.0: [14c3:7663] type 00 class 0x000280
[   17.753025] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[   18.531464] mt7615e 0000:02:00.0: N9 Firmware Version: 7663mp1827, Build Time: 20190914043434
```

ptpt52, 2021-03-26:
@ryderlee1110  test all your patch
still issue
and reset recovery not ok

ptpt52, 2021-03-26:
with @ryderlee1110 's patches
@Borromini 
and
I am trying to slow down the `iwinfo scan` frequency
say, scan it in 60s interval
```
while :; do iwinfo wlan1 scan >/dev/null; echo $((i++)); uptime; sleep 60; done
```
and the mcu timeout issue not happens

I did scan in 1 second interval on chip mt7615, no issue happens
```
while :; do iwinfo wlan1 scan >/dev/null; echo $((i++)); uptime; sleep 1; done
```

ryderlee1110, 2021-03-26:
OK. Looks like different issue on mt7613.

ptpt52, 2021-03-26:
@ryderlee1110 
one thing your patch solve is that the `iwinfo scan` never kick sta client, and client could keep stream while no scan doing

ryderlee1110, 2021-03-26:
yes, that's one of issues I fixed for mt7615.

ptpt52, 2021-03-26:
@ryderlee1110  hello, may I ask you something?
I can get chip temperature via
```
cat /sys/kernel/debug/ieee80211/phy1/mt76/temperature 
Temperature: 45
```
on chip mt7615

but cannot get it on chip mt7613, it always return 0:
```
cat /sys/kernel/debug/ieee80211/phy1/mt76/temperature 
Temperature: 0
```

what is the difference between them?

ryderlee1110, 2021-03-26:
@ptpt52 I don't know, it could be command id change etc. Although 7613 reuses 7615 code, I still think there might be some flow changes needs to be modified.

nbd168, 2021-03-26:
Today I fixed reset recovery on MT7622 in a way that might also help MT7613. Please test the latest version

ptpt52, 2021-03-27:
@nbd168  thanks, I would test it next Monday

ptpt52, 2021-03-29:
@nbd168 @ryderlee1110 
very strange, with the latest code (https://github.com/openwrt/mt76/commit/465dda65ee8415169cc39596f913a50b5ba4f5a9), the mcu timeout never happens
I am not sure which commit fix it.

ptpt52, 2021-03-29:
maybe the `0-terminating code ...` commits fix it?

ptpt52, 2021-03-30:
@nbd168  @ryderlee1110  here to report back
the mcu timeout issue still exists
an cannot recovery

ptpt52, 2021-03-30:
it just not esay to triger

ptpt52, 2021-03-30:
the mcu timeout issue happened accidentally, but it is hard for me to reproduce it, maybe it is some sort of hardware issue?

Borromini, 2021-03-30:
Here it ranges from under an hour to multiple days for it to pop up. Happens on both MT7613 devices. I just got a third back (was on loan for porting OpenWrt) and I'll be testing that one as well.

ptpt52, 2021-03-30:
@Borromini should it be related to chip temperature?

ptpt52, 2021-03-31:
I try `echo 1 >fw_debug` and got this random code:
```
[27866.500207] ieee80211 phy1: N9: (27889.441:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27872.131939] ieee80211 phy1: N9: (27895.076:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27872.508168] ieee80211 phy1: N9: (27895.452:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27878.140275] ieee80211 phy1: N9: (27901.088:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27878.516065] ieee80211 phy1: N9: (27901.464:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27884.151745] ieee80211 phy1: N9: (27907.102:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27884.524416] ieee80211 phy1: N9: (27907.475:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27890.159706] ieee80211 phy1: N9: (27913.114:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27890.535863] ieee80211 phy1: N9: (27913.490:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27896.167820] ieee80211 phy1: N9: (27919.125:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27896.544152] ieee80211 phy1: N9: (27919.502:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27902.176016] ieee80211 phy1: N9: (27925.137:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27902.552250] ieee80211 phy1: N9: (27925.513:CMD-E)EXT_CMD_OPT_BIT_0_ACK
[27906.615795] ieee80211 phy1: N9: (27929.579:CMD-E)EXT_CMD_OPT_BIT_0_ACK
?z.N<..a?6.?2#?mY??...???1<????6&w.??.?.???E.?.l)?.???-*.
[28046.811695] h..??G?O\.$???.L.?.??T.. ?N?.T?????.?@_?.?L??._??E?.???.T?\???H<??|.?J?0!;.L^?N???]|S??LU]??SZ-5G..$?H?...
M???2.7.1M???2.7.1[28052.436228] 80211 ??1: N9: (28075.483:CMD-E)EXT CMD OPT BIT 0 ACK
[28052.851760] 80211 ??1: N9: (28075.899:CMD-E)EXT CMD OPT BIT 0 ACK
[28058.444764] 80211 ??1: N9: (28081.495:CMD-E)EXT CMD OPT BIT 0 ACK
[28058.860673] 80211 ??1: N9: (28081.911:CMD-E)EXT CMD OPT BIT 0 ACK
[28064.453809] 80211 ??1: N9: (28087.508:CMD-E)EXT CMD OPT BIT 0 ACK
[28064.872044] 80211 ??1: N9: (28087.926:CMD-E)EXT CMD OPT BIT 0 ACK

```

ptpt52, 2021-03-31:
I try to debug mcu and get this:

```
 448 static void
 449 mt7615_mcu_rx_ext_event(struct mt7615_dev *dev, struct sk_buff *skb)
 450 {
 451         struct mt7615_mcu_rxd *rxd = (struct mt7615_mcu_rxd *)skb->data;
 452 
 453         switch (rxd->ext_eid) {
 454         case MCU_EXT_EVENT_RDD_REPORT:
 455                 mt7615_mcu_rx_radar_detected(dev, skb);
 456                 break;
 457         case MCU_EXT_EVENT_CSA_NOTIFY:
 458                 mt7615_mcu_rx_csa_notify(dev, skb);
 459                 break;
 460         case MCU_EXT_EVENT_FW_LOG_2_HOST:
 461                 mt7615_mcu_rx_log_message(dev, skb);
 462                 break;
 463         default:
 464 
 465                 printk("get ext unhandle eid=%d ext_eid=%d seq=%d\n", rxd->eid, rxd->ext_eid, rxd->seq);
 466                 break;
 467         }
 468 }

```

when iwinfo scan
```
get ext unhandle eid=237 ext_eid=26 seq=0
```

porentak, 2021-03-31:
If it helps, ext_eid=26 is EXT_EVENT_BEACON_LOSS.

ptpt52, 2021-03-31:
what is I disable beacon_offload on driver?
@nbd168 @porentak

ptpt52, 2021-03-31:
I am going to call mt7615_mcu_beacon_loss_event() in ext_eid reach EXT_EVENT_BEACON_LOSS

ptpt52, 2021-03-31:
now I am doing handle the MCU_EXT_EVENT_BEACON_LOSS ext_event

```
@@ -443,7 +480,16 @@ mt7615_mcu_rx_ext_event(struct mt7615_dev *dev, struct sk_buff *skb)
        case MCU_EXT_EVENT_FW_LOG_2_HOST:
                mt7615_mcu_rx_log_message(dev, skb);
                break;
+       case MCU_EXT_EVENT_BEACON_LOSS:
+               //printk("get ext unhandle eid=%d ext_eid=%d seq=%d handle via mt7615_mcu_beacon_loss_event\n", rxd->eid, rxd->ext_eid, rxd->seq);
+               mt7615_mcu_beacon_loss_event(dev, skb);
+               break;
+       case MCU_EXT_EVENT_PS_SYNC:
+               //TODO
+               break;
        default:
+
+               printk("get ext unhandle eid=%d ext_eid=%d seq=%d MCU_EXT_EVENT_FW_LOG_2_HOST=%d\n", rxd->eid, rxd->ext_eid, rxd->seq, MCU_EXT_EVENT_FW_LOG_2_HOST);
                break;
        }
 }
```

ptpt52, 2021-04-01:
I run `killall hostapd`
then it trigger the issue:
```
[ 2410.819856] get ext unhandle eid=237 ext_eid=35 seq=147
[ 2410.825158] get ext unhandle eid=237 ext_eid=35 seq=148
[ 2410.830423] get ext unhandle eid=237 ext_eid=35 seq=149
[ 2410.835712] get ext unhandle eid=237 ext_eid=35 seq=150
[ 2410.841018] get ext unhandle eid=237 ext_eid=35 seq=151
[ 2410.846280] get ext unhandle eid=237 ext_eid=35 seq=152
[ 2410.851572] get ext unhandle eid=237 ext_eid=35 seq=153
[ 2410.856889] get ext unhandle eid=237 ext_eid=35 seq=154
[ 2410.862154] get ext unhandle eid=237 ext_eid=35 seq=155
[ 2410.867441] get ext unhandle eid=237 ext_eid=35 seq=156
[ 2410.872748] get ext unhandle eid=237 ext_eid=35 seq=157
[ 2410.878015] get ext unhandle eid=237 ext_eid=35 seq=158
[ 2410.883332] get ext unhandle eid=237 ext_eid=35 seq=159
[ 2410.888629] get ext unhandle eid=237 ext_eid=35 seq=160
[ 2410.893890] get ext unhandle eid=237 ext_eid=35 seq=161
[ 2410.899179] get ext unhandle eid=237 ext_eid=35 seq=162
[ 2410.904455] get ext unhandle eid=237 ext_eid=35 seq=163
[ 2410.909761] get ext unhandle eid=237 ext_eid=35 seq=164
[ 2410.915046] get ext unhandle eid=237 ext_eid=35 seq=165
[ 2410.920319] get ext unhandle eid=237 ext_eid=35 seq=166
[ 2410.925592] get ext unhandle eid=237 ext_eid=35 seq=167

root@HBD-WAP1200GI:~# [ 2417.695678] mt7615e 0000:02:00.0: Message 39 (seq 9) timeout
[ 2438.175329] mt7615e 0000:02:00.0: Message 39 (seq 10) timeout
[ 2458.655350] mt7615e 0000:02:00.0: Message 39 (seq 11) timeout

root@HBD-WAP1200GI:~# [ 2479.135299] mt7615e 0000:02:00.0: Message 39 (seq 12) timeout
[ 2499.615297] mt7615e 0000:02:00.0: Message 73 (seq 13) timeout

```

ptpt52, 2021-04-01:
@Borromini  use my changes in https://github.com/ptpt52/mt76/commit/36ed7ea7962ca2ed7cfa77de62fb123658bdae6d
Test to see if the mcu timeout issue is fixed

ryderlee1110, 2021-04-01:
Just note that the  mcu_beacon_loss_event format of mt7615 is different compares to connac one. 
struct mt7615_beacon_loss_event {
	u8 bssid[8];
	u8 reason;
	u8 rsv;
} __packed; 

Also, I think we should check reason to update new beacon to firmware if error happens. As for PS_SYN is useless in mt76 you can ignore it.

ptpt52, 2021-04-01:
@ryderlee1110  thanks for the note and I notice there may be some more difference
something like get_temperature from mcu...

ryderlee1110, 2021-04-01:
that's why i don't want to drag every chip in mt7615 :)

ptpt52, 2021-04-01:
@ryderlee1110  should I use the firmware file in `bin/mt7663/rebb/mt7663_patch_e2_hdr.bin` in property driver?

ryderlee1110, 2021-04-01:
I don't know actually ...  I guess it just one of 7663 intermediate version inside MTK, and then someone use it in earlier MT76 bring up. If it works to you in most cases then just use it.

ptpt52, 2021-04-01:
for now I do some hack hard code to handle the mt7613 for test
1. get the temperature: https://github.com/ptpt52/mt76/commit/2198ba5cdf7993ae23d9363bf2c5735db70aa986
2. handle beacon loss event: https://github.com/ptpt52/mt76/commit/9b9e1e8e306e48f1e20a2c5a3d2ee9c44efb3e54

ptpt52, 2021-04-01:
what is `/lib/firmware/mediatek/mt7663_n9_rebb.bin` and where to get it?
@nbd168  @ryderlee1110

ptpt52, 2021-04-01:
each time mcu timeout happens
```
[  969.508658] get ext unhandle eid=237 ext_eid=35 seq=244
[  969.513952] get ext unhandle eid=237 ext_eid=35 seq=245
[  969.519235] get ext unhandle eid=237 ext_eid=35 seq=246
[  969.524559] get ext unhandle eid=237 ext_eid=35 seq=247
```
ext_eid=35
that means `EXT_EVENT_ID_ASSERT_DUMP = 0x23`
something bad happens in the chip?

ptpt52, 2021-04-02:
now I trigger it and get this dump:
```
[ 2271.741418] EXT_EVENT_ID_ASSERT_DUMP:[Wifi ASSERT:]
[ 2271.746445] EXT_EVENT_ID_ASSERT_DUMP:[../wifi/custom/feature/pm_ap.c]
[ 2271.752969] EXT_EVENT_ID_ASSERT_DUMP:[line: #454]
[ 2271.757736] EXT_EVENT_ID_ASSERT_DUMP:[
[ 2271.757736] Asserted file: ++3?]
[ 2271.764771] EXT_EVENT_ID_ASSERT_DUMP:[../wifi/custom/feature/pm_ap.c++]
[ 2271.771472] EXT_EVENT_ID_ASSERT_DUMP:[ (line #454)
[ 2271.771472] 
[ 2271.771472] ++]
[ 2271.779582] EXT_EVENT_ID_ASSERT_DUMP:[EMI Wakup Fail (i=101)++]
[ 2271.785656] EXT_EVENT_ID_ASSERT_DUMP:[<ASSERT> ../wifi/custom/feature/pm_ap.c #454 - , 0x0, 0x0, 0x0, rc=*]
[ 2271.795462] EXT_EVENT_ID_ASSERT_DUMP:[, id=0x2 WIFI, isr=0x0, irq=0x12]
[ 2271.802210] EXT_EVENT_ID_ASSERT_DUMP:[, lr=0x1F0C, sp=0x2009728,{isr_t=0}{asr_t=74733823}]
[ 2271.810523] EXT_EVENT_ID_ASSERT_DUMP:[{shed_trace*]
[ 2271.815526] EXT_EVENT_ID_ASSERT_DUMP:[[idx=2,sched_t=0x2,sched_s=74727314]]
[ 2271.822574] EXT_EVENT_ID_ASSERT_DUMP:[[idx=3,sched_t=0x2,sched_s=74727379]]
[ 2271.829634] EXT_EVENT_ID_ASSERT_DUMP:[[idx=4,sched_t=0x2,sched_s=74727444]]
[ 2271.836661] EXT_EVENT_ID_ASSERT_DUMP:[[idx=5,sched_t=0x2,sched_s=74727510]]
[ 2271.843722] EXT_EVENT_ID_ASSERT_DUMP:[[idx=6,sched_t=0x6,sched_s=74727574]]
[ 2271.850749] EXT_EVENT_ID_ASSERT_DUMP:[[idx=7,sched_t=0x2,sched_s=74728741]]
[ 2271.857785] EXT_EVENT_ID_ASSERT_DUMP:[[idx=8,sched_t=0x6,sched_s=74728806]]
[ 2271.864808] EXT_EVENT_ID_ASSERT_DUMP:[[idx=9,sched_t=0x2,sched_s=74730412]]
[ 2271.871852] EXT_EVENT_ID_ASSERT_DUMP:[[idx=10,sched_t=0x2,sched_s=74730427]]
[ 2271.878955] EXT_EVENT_ID_ASSERT_DUMP:[[idx=11,sched_t=0x6,sched_s=74730541]]
[ 2271.886084] EXT_EVENT_ID_ASSERT_DUMP:[[idx=12,sched_t=0x2,sched_s=74732163]0]
[ 2271.893285] EXT_EVENT_ID_ASSERT_DUMP:[[idx=13,sched_t=0x6,sched_s=74732228]]
[ 2271.900431] EXT_EVENT_ID_ASSERT_DUMP:[[idx=14,sched_t=0x2,sched_s=74733718]]
[ 2271.907534] EXT_EVENT_ID_ASSERT_DUMP:[[idx=0,sched_t=0x2,sched_s=74733783]]
[ 2271.914581] EXT_EVENT_ID_ASSERT_DUMP:[[idx=1,sched_t=0x2,sched_s=74733784]}??]
[ 2271.921908] EXT_EVENT_ID_ASSERT_DUMP:[{irq_trace*F]
[ 2271.926919] EXT_EVENT_ID_ASSERT_DUMP:[[idx=0,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.933628] EXT_EVENT_ID_ASSERT_DUMP:[[idx=1,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.940315] EXT_EVENT_ID_ASSERT_DUMP:[[idx=2,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.947054] EXT_EVENT_ID_ASSERT_DUMP:[[idx=3,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.953766] EXT_EVENT_ID_ASSERT_DUMP:[[idx=4,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.960468] EXT_EVENT_ID_ASSERT_DUMP:[[idx=5,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.967154] EXT_EVENT_ID_ASSERT_DUMP:[[idx=6,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.973884] EXT_EVENT_ID_ASSERT_DUMP:[[idx=7,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.980586] EXT_EVENT_ID_ASSERT_DUMP:[[idx=8,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.987302] EXT_EVENT_ID_ASSERT_DUMP:[[idx=9,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2271.993997] EXT_EVENT_ID_ASSERT_DUMP:[[idx=10,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2272.000782] EXT_EVENT_ID_ASSERT_DUMP:[[idx=11,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2272.007571] EXT_EVENT_ID_ASSERT_DUMP:[[idx=12,irq_t=0xFFFFFFFF,irq_s=0]]
[ 2272.014362] EXT_EVENT_ID_ASSERT_DUMP:[[idx=13,irq_t=0xFFFFFFFF,irq_s=0]..]
[ 2272.021403] EXT_EVENT_ID_ASSERT_DUMP:[[idx=14,irq_t=0xFFFFFFFF,irq_s=0]}?.]
[ 2272.028452] EXT_EVENT_ID_ASSERT_DUMP:[;;;;;;;;;;;;;;;;;;;;;;;;;;;;;++]
[ 2272.035039] EXT_EVENT_ID_ASSERT_DUMP:[; exception type: ASSERT, cnt:1++]
[ 2272.041820] EXT_EVENT_ID_ASSERT_DUMP:[{asser_type=5}]
[ 2272.046944] EXT_EVENT_ID_ASSERT_DUMP:[exp_main: jump from cos_assert++]
M[ 2272.053622] EXT_EVENT_ID_ASSERT_DUMP:[EXP_MAIN_ENTRY_CNT =1++.]
[ 2272.059596] EXT_EVENT_ID_ASSERT_DUMP:[;;[CONNSYS]core dump start ..
[ 2272.059596] ?]
[ 2272.067603] EXT_EVENT_ID_ASSERT_DUMP:[Patch: more log added here -->
[ 2272.067603] ++]
[ 2272.075759] EXT_EVENT_ID_ASSERT_DUMP:[exp_mem_dump_delay_count=1000, exp_mem_dump_delay_count=2000
[ 2272.075759] ++2]
[ 2272.086610] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_2)sched_t=2, sched_start=74727314++ ]
[ 2272.094680] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_3)sched_t=2, sched_start=74727379++]
i[ 2272.102684] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_4)sched_t=2, sched_start=74727444++]
[ 2272.110769] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_5)sched_t=2, sched_start=74727510++.]
n[ 2272.118870] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_6)sched_t=6, sched_start=74727574++]
[ 2272.126922] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_7)sched_t=2, sched_start=74728741++]
[ 2272.134969] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_8)sched_t=6, sched_start=74728806++.]
i[ 2272.143041] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_9)sched_t=2, sched_start=74730412++]
[ 2272.151140] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_10)sched_t=2, sched_start=74730427++]
c[ 2272.159216] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_11)sched_t=6, sched_start=74730541++]
[ 2272.167381] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_12)sched_t=2, sched_start=74732163++]
[ 2272.175463] EXT_EVENT_ID_ASSERT_DUMP:[(sched_info_13)sched_t=6, sched_start=74732228++]
[... 92 lines cut ...]
[ 2272.531995] EXT_EVENT_ID_ASSERT_DUMP:[R.S R18 0x0
[ 2272.531995] 
[ 2272.531995] ++.]
[ 2272.540070] EXT_EVENT_ID_ASSERT_DUMP:[R.S R19 0x0
[ 2272.540070] 
[ 2272.540070] ++]
[ 2272.548078] EXT_EVENT_ID_ASSERT_DUMP:[R.S R20 0x0
.]
[ 2273.011710] EXT_EVENT_ID_ASSERT_DUMP:[[@.]
n[ 2273.015891] EXT_EVENT_ID_ASSERT_DUMP:[[t.]
[ 2273.020090] EXT_EVENT_ID_ASSERT_DUMP:[[?.]
[ 2273.024246] EXT_EVENT_ID_ASSERT_DUMP:[[?.]
i[ 2273.028406] EXT_EVENT_ID_ASSERT_DUMP:[[..]
c[ 2273.032642] EXT_EVENT_ID_ASSERT_DUMP:[[D.]
[ 2273.036879] EXT_EVENT_ID_ASSERT_DUMP:[[x.]
o[ 2273.041070] EXT_EVENT_ID_ASSERT_DUMP:[[?.]
.]
2[ 2273.280412] EXT_EVENT_ID_ASSERT_DUMP:[[@.]
[ 2273.284646] EXT_EVENT_ID_ASSERT_DUMP:[[t.]
.[ 2273.288789] EXT_EVENT_ID_ASSERT_DUMP:[[?.]
7[ 2273.293053] EXT_EVENT_ID_ASSERT_DUMP:[[?.]
[ 2273.297261] EXT_EVENT_ID_ASSERT_DUMP:[[..]
.[ 2273.301490] EXT_EVENT_ID_ASSERT_DUMP:[[D.]

```

ptpt52, 2021-04-02:
@ryderlee1110 @nbd168 @Borromini  any suggestion?

ptpt52, 2021-04-02:
seems good news, it fixed via
```
diff --git a/mt7615/mcu.c b/mt7615/mcu.c
index 20890dc..31e3b99 100644
--- a/mt7615/mcu.c
+++ b/mt7615/mcu.c
@@ -790,6 +788,12 @@ mt7615_mcu_ctrl_pm_state(struct mt7615_dev *dev, int band, int state)
                .pm_state = state ? ENTER_PM_STATE : EXIT_PM_STATE,
                .band_idx = band,
        };
+
+       memset(&req, 0, sizeof(req));
+       req.pm_number = 5;
+       req.pm_state = state ? ENTER_PM_STATE : EXIT_PM_STATE;
+       req.band_idx = band;
+
        printk("mt7615_mcu_ctrl_pm_state state=%d\n", state);
 
        return mt76_mcu_send_msg(&dev->mt76, MCU_EXT_CMD_PM_STATE_CTRL, &req,

```

ptpt52, 2021-04-03:
oops
I got another issue:
```
[ 8131.624960] mt7615e 0000:02:00.0: Message 73 (seq 7) timeout
[ 8152.104957] mt7615e 0000:02:00.0: Message 73 (seq 8) timeout
[ 8172.584919] mt7615e 0000:02:00.0: Message 73 (seq 9) timeout
[ 8193.064954] mt7615e 0000:02:00.0: Message 73 (seq 10) timeout
[ 8213.544952] mt7615e 0000:02:00.0: Message 73 (seq 11) timeout
[ 8234.024950] mt7615e 0000:02:00.0: Message 73 (seq 12) timeout
[ 8254.504963] mt7615e 0000:02:00.0: Message 73 (seq 13) timeout
[ 8274.984973] mt7615e 0000:02:00.0: Message 73 (seq 14) timeout
[ 8295.464945] mt7615e 0000:02:00.0: Message 73 (seq 15) timeout
[ 8315.945054] mt7615e 0000:02:00.0: Message 73 (seq 1) timeout
[ 8336.424973] mt7615e 0000:02:00.0: Message 73 (seq 2) timeout
[ 8356.904969] mt7615e 0000:02:00.0: Message 73 (seq 3) timeout
[ 8377.384969] mt7615e 0000:02:00.0: Message 73 (seq 4) timeout
[ 8397.864973] mt7615e 0000:02:00.0: Message 73 (seq 5) timeout
[ 8418.344987] mt7615e 0000:02:00.0: Message 73 (seq 6) timeout
[ 8438.824956] mt7615e 0000:02:00.0: Message 73 (seq 7) timeout
[ 8459.304989] mt7615e 0000:02:00.0: Message 73 (seq 8) timeout
[ 8479.784983] mt7615e 0000:02:00.0: Message 73 (seq 9) timeout
[ 8500.265021] mt7615e 0000:02:00.0: Message 73 (seq 10) timeout
[ 8520.744979] mt7615e 0000:02:00.0: Message 73 (seq 11) timeout
[ 8541.225027] mt7615e 0000:02:00.0: Message 37 (seq 12) timeout
[ 8541.230838] wlan1: HW problem - can not stop rx aggregation for e4:67:8e:ab:68:5b tid 0
[ 8561.705024] mt7615e 0000:02:00.0: Message 37 (seq 13) timeout
[ 8561.710821] wlan1: HW problem - can not stop rx aggregation for e4:67:8e:ab:68:5b tid 2
[ 8582.185007] mt7615e 0000:02:00.0: Message 37 (seq 14) timeout
[ 8582.190821] wlan1: HW problem - can not stop rx aggregation for e4:67:8e:ab:68:5b tid 3
[ 8602.664992] mt7615e 0000:02:00.0: Message 37 (seq 15) timeout
[ 8602.670788] ------------[ cut here ]------------
[ 8602.675664] WARNING: CPU: 1 PID: 1821 at backports-5.10.16-1/net/mac80211/agg-tx.c:406 ___ieee80211_stop_tx_ba_session+0x2b8/0x2d8 [mac80211]
[ 8602.689149] Modules linked in: xt_connlimit nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_CT nft_redir nft_ct nf_nat nf_t
[ 8602.689329]  ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnec
[ 8602.810029] CPU: 1 PID: 1821 Comm: hostapd Not tainted 5.4.105 #0
[ 8602.816120] Stack : 8042f578 80072e10 80660000 806623c0 806d0000 80662388 806614ec 8642d9fc
[ 8602.824488]         80810000 87d313b4 806aac43 805f58bc 00000001 00000001 8642d9a0 24a5ca47
[ 8602.832829]         00000000 00000000 80850000 00000000 30232035 0000016a 2e352064 30312e34
[ 8602.841169]         00000000 000001ab 00000000 000c5c2d 80000000 806d0000 00000000 86613784
[ 8602.849509]         00000009 86458c20 87ea9070 868fe480 00000000 80302620 00000004 80810004
[ 8602.857852]         ...
[ 8602.860304] Call Trace:
[ 8602.862793] [<8000b61c>] show_stack+0x30/0x100
[ 8602.867275] [<8053e6bc>] dump_stack+0xa4/0xdc
[ 8602.871657] [<8002bcd0>] __warn+0xc0/0x10c
[ 8602.875782] [<8002bd78>] warn_slowpath_fmt+0x5c/0xac
[ 8602.880947] [<86613784>] ___ieee80211_stop_tx_ba_session+0x2b8/0x2d8 [mac80211]
[ 8602.888337] [<86611c98>] ieee80211_sta_tear_down_BA_sessions+0x88/0x160 [mac80211]
[ 8602.895948] [<8660619c>] ieee80211_sta_set_buffered+0x920/0xe50 [mac80211]
[ 8602.903105] ---[ end trace b3aa443694ada836 ]---
[ 8623.144996] mt7615e 0000:02:00.0: Message 37 (seq 1) timeout
[ 8623.150757] ------------[ cut here ]------------
[ 8623.155958] WARNING: CPU: 0 PID: 1821 at backports-5.10.16-1/net/mac80211/agg-tx.c:406 ___ieee80211_stop_tx_ba_session+0x2b8/0x2d8 [mac80211]
[ 8623.168737] Modules linked in: xt_connlimit nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_CT nft_redir nft_ct nf_nat nf_t
[ 8623.169087]  ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnec
[ 8623.289847] CPU: 0 PID: 1821 Comm: hostapd Tainted: G        W         5.4.105 #0
[ 8623.297310] Stack : 8042f578 80072e10 80660000 806623c0 806d0000 80662388 806614ec 8642d9fc
[ 8623.305646]         80810000 87d313b4 806aac43 805f58bc 00000000 00000001 8642d9a0 24a5ca47
[ 8623.313970]         00000000 00000000 80850000 00000000 30232035 0000017f 2e352020 30312e34
[ 8623.322298]         00000000 000001ac 00000000 00046c37 80000000 806d0000 00000000 86613784
[ 8623.330624]         00000009 86458c20 87ea9070 868fe480 00000000 80302620 00000000 80810000
[ 8623.338950]         ...
[ 8623.341388] Call Trace:
[ 8623.343864] [<8000b61c>] show_stack+0x30/0x100
[ 8623.348332] [<8053e6bc>] dump_stack+0xa4/0xdc
[ 8623.352686] [<8002bcd0>] __warn+0xc0/0x10c
[ 8623.356765] [<8002bd78>] warn_slowpath_fmt+0x5c/0xac
[ 8623.361870] [<86613784>] ___ieee80211_stop_tx_ba_session+0x2b8/0x2d8 [mac80211]
[ 8623.369216] [<86611c98>] ieee80211_sta_tear_down_BA_sessions+0x88/0x160 [mac80211]
[ 8623.376806] [<8660619c>] ieee80211_sta_set_buffered+0x920/0xe50 [mac80211]
[ 8623.383948] ---[ end trace b3aa443694ada837 ]---
[ 8643.625010] mt7615e 0000:02:00.0: Message 37 (seq 2) timeout
[ 8664.105047] mt7615e 0000:02:00.0: Message 73 (seq 3) timeout
[ 8684.585069] mt7615e 0000:02:00.0: Message 73 (seq 4) timeout
[ 8705.065234] mt7615e 0000:02:00.0: Message 73 (seq 5) timeout
[ 8725.545030] mt7615e 0000:02:00.0: Message 73 (seq 6) timeout
[ 8746.025047] mt7615e 0000:02:00.0: Message 73 (seq 7) timeout
[ 8766.505002] mt7615e 0000:02:00.0: Message 73 (seq 8) timeout
[ 8786.985012] mt7615e 0000:02:00.0: Message 73 (seq 9) timeout

```

ptpt52, 2021-04-04:
hello, @LorenzoBianconi 
what do you think of this issue? the mcu timeout happens after random hours of time.
the code is the latest.
kernel 5.4 in openwrt master

ptpt52, 2021-04-04:
@ryderlee1110
so I know something about the `mcu timeout`
if `mt7615_mcu_set_fw_ctrl` call, the muc timeout issue happens?

ptpt52, 2021-04-05:
I just hack disable any pm ctrl for mcu, and hope this would help with the mcu timeout issue

```
diff --git a/mt7615/mcu.c b/mt7615/mcu.c
index fcbeb9b8..307d6441 100644
--- a/mt7615/mcu.c
+++ b/mt7615/mcu.c
@@ -821,7 +821,8 @@ mt7615_mcu_ctrl_pm_state(struct mt7615_dev *dev, int band, int state)
        req.pm_state = state ? ENTER_PM_STATE : EXIT_PM_STATE;
        req.band_idx = band;
 
-       printk("%s:%d send MCU_EXT_CMD_PM_STATE_CTRL\n", __func__, __LINE__);
+       printk("%s:%d send MCU_EXT_CMD_PM_STATE_CTRL ignore\n", __func__, __LINE__);
+       return 0;
        return mt76_mcu_send_msg(&dev->mt76, MCU_EXT_CMD_PM_STATE_CTRL, &req,
                                 sizeof(req), true);
 }
```

ptpt52, 2021-04-05:
would the mcu timeout issue similar to this one?
https://github.com/openwrt/mt76/commit/0d5b1a70271503e54cbe173237281bd0c4d8a167
ping @moore-bros

ptpt52, 2021-04-05:
got this fw_debug log:
```
ieee80211 phy1: N9: (6211.799:CMD-E)Unregistered middle function, caller(0x137d02)!!
```

ptpt52, 2021-04-05:
why `mt7615_mcu_add_beacon_offload` call every secs on chip mt7613
but not on mt7615 ?
@ryderlee1110

ptpt52, 2021-04-07:
@ryderlee1110 @nbd168 
hello, ask for some help
you know how to do a reset on the pcie?
I want to do it if I detect the `mcu timeout`

csharper2005, 2021-06-05:
I got this with the latest master:
```
[36120.428956] mt7615e 0000:01:00.0: Message 40000002 (seq 11) timeout
[36146.668734] mt7615e 0000:01:00.0: Message 40000002 (seq 12) timeout
[36172.908453] mt7615e 0000:01:00.0: Message 40000002 (seq 13) timeout
[36199.148153] mt7615e 0000:01:00.0: Message 40000002 (seq 14) timeout
[36225.387877] mt7615e 0000:01:00.0: Message 40000002 (seq 15) timeout
[36251.627608] mt7615e 0000:01:00.0: Message 40000002 (seq 1) timeout
[36277.867321] mt7615e 0000:01:00.0: Message 40000002 (seq 2) timeout
[36304.107041] mt7615e 0000:01:00.0: Message 40000002 (seq 3) timeout
[36330.346776] mt7615e 0000:01:00.0: Message 40000002 (seq 4) timeout
```

csharper2005, 2021-08-24:
Up. Bug is still actual.
Firmware Version | OpenWrt SNAPSHOT r17391-1016790068 / LuCI Master git-21.226.86205-376af36
```
[139272.368392] mt7615e 0000:01:00.0: Message 40000002 (seq 1) timeout
[139298.608159] mt7615e 0000:01:00.0: Message 40000002 (seq 2) timeout
[139324.847934] mt7615e 0000:01:00.0: Message 40000002 (seq 3) timeout
[139351.087745] mt7615e 0000:01:00.0: Message 40000002 (seq 4) timeout
[139377.327479] mt7615e 0000:01:00.0: Message 40000002 (seq 5) timeout
[139403.567250] mt7615e 0000:01:00.0: Message 40000002 (seq 6) timeout
[139429.807026] mt7615e 0000:01:00.0: Message 40000002 (seq 7) timeout
[139456.046798] mt7615e 0000:01:00.0: Message 40000002 (seq 8) timeout
[139482.286572] mt7615e 0000:01:00.0: Message 40000002 (seq 9) timeout
[139508.526347] mt7615e 0000:01:00.0: Message 40000002 (seq 10) timeout
[139534.766119] mt7615e 0000:01:00.0: Message 40000002 (seq 11) timeout
[139561.005907] mt7615e 0000:01:00.0: Message 40000002 (seq 12) timeout
[139582.125712] mt7615e 0000:01:00.0: Message 40000003 (seq 13) timeout
[139582.138400] wlan1: HW problem - can not stop rx aggregation for <censored> tid 0
[139602.605530] mt7615e 0000:01:00.0: Message 40000003 (seq 14) timeout
[139602.618207] wlan1: HW problem - can not stop rx aggregation for <censored> tid 6
[139623.085360] mt7615e 0000:01:00.0: Message 40000003 (seq 15) timeout
[139623.098044] ------------[ cut here ]------------
[139623.107578] WARNING: CPU: 0 PID: 1501 at backports-5.10.42-1/net/mac80211/agg-tx.c:406 ___ieee80211_stop_tx_ba_session+0x2b0/0x2d0 [mac80211]
[139623.133034] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ctr cmac ccm leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[139623.270516] CPU: 0 PID: 1501 Comm: hostapd Not tainted 5.4.142 #0
[139623.282800] Stack : 8d774800 8007c09c 80700000 806fe25c 80760000 806fe224 806fd3b4 8e31ba24
[139623.299590]         808a0000 80690040 8e4e2b08 80748d03 00000000 00000001 8e31b9c8 9037bff3
[139623.316374]         00000000 00000000 808e0000 00000000 30000000 00000164 00000018 312e342e
[139623.333160]         00000000 8e31b8f2 ffffffff 68203a6d 80000000 00000001 00000000 8e7138d4
[139623.349946]         00000009 8e9da680 8e520c20 8e267070 00000008 8038b420 00000000 808a0000
[139623.366730]         ...
[139623.371754] Call Trace:
[139623.376800] [<8000b48c>] show_stack+0x30/0x100
[139623.385821] [<805d7ccc>] dump_stack+0x9c/0xc8
[139623.394662] [<8002b9f8>] __warn+0xcc/0x118
[139623.402973] [<8002baa0>] warn_slowpath_fmt+0x5c/0xac
[139623.413073] [<8e7138d4>] ___ieee80211_stop_tx_ba_session+0x2b0/0x2d0 [mac80211]
[139623.427844] [<8e711e78>] ieee80211_sta_tear_down_BA_sessions+0x88/0x150 [mac80211]
[139623.443123] [<8e705dcc>] ieee80211_sta_eosp+0x54c/0x748 [mac80211]
[139623.455895] ---[ end trace d6c0db60cd2c0260 ]---
[139643.565178] mt7615e 0000:01:00.0: Message 40000003 (seq 1) timeout
[139643.577679] ------------[ cut here ]------------
[139643.587183] WARNING: CPU: 1 PID: 1501 at backports-5.10.42-1/net/mac80211/agg-tx.c:406 ___ieee80211_stop_tx_ba_session+0x2b0/0x2d0 [mac80211]
[139643.612666] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ctr cmac ccm leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[139643.750203] CPU: 1 PID: 1501 Comm: hostapd Tainted: G        W         5.4.142 #0
[139643.765256] Stack : 8d774800 8007c09c 80700000 806fe25c 80760000 806fe224 806fd3b4 8e31ba24
[139643.782048]         808a0000 80690040 8e4e2b08 80748d03 00000001 00000001 8e31b9c8 9037bff3
[139643.798836]         00000000 00000000 808e0000 00000000 30000000 00000178 00000018 20202020
[139643.815625]         00000000 8e31b8f2 ffffffff 20202020 80000000 00000001 00000000 8e7138d4
[139643.832415]         00000009 8f3df800 8e520c20 8e267070 00000008 8038b420 00000004 808a0004
[139643.849203]         ...
[139643.854229] Call Trace:
[139643.859277] [<8000b48c>] show_stack+0x30/0x100
[139643.868299] [<805d7ccc>] dump_stack+0x9c/0xc8
[139643.877145] [<8002b9f8>] __warn+0xcc/0x118
[139643.885461] [<8002baa0>] warn_slowpath_fmt+0x5c/0xac
[139643.895588] [<8e7138d4>] ___ieee80211_stop_tx_ba_session+0x2b0/0x2d0 [mac80211]
[... cut ...]
[142337.931855] mt7615e 0000:01:00.0: Message 40000002 (seq 15) timeout
[142364.171510] mt7615e 0000:01:00.0: Message 40000002 (seq 1) timeout
[142390.411178] mt7615e 0000:01:00.0: Message 40000002 (seq 2) timeout
[142416.650879] mt7615e 0000:01:00.0: Message 40000002 (seq 3) timeout
[142442.890507] mt7615e 0000:01:00.0: Message 40000002 (seq 4) timeout
[142469.130171] mt7615e 0000:01:00.0: Message 40000002 (seq 5) timeout
[142495.369841] mt7615e 0000:01:00.0: Message 40000002 (seq 6) timeout

```

d-me3, 2021-10-30:
I've just experienced this issue this week on a recent build from Tue 26, 2021 (OpenWrt SNAPSHOT r17859-70c12d26ca) on an Archer C6 v3.2 (mt7613).

I thought it was a new problem introduced by the recent build, but reading this thread it seems it exists also in previous builds and it seems that just now it happened to me.

So this bug seems to be still unsolved.

```
(...)
[118768.299525] mt7615e 0000:02:00.0: Message 00000049 (seq 8) timeout
[118788.779195] mt7615e 0000:02:00.0: Message 00000049 (seq 9) timeout
[118815.018699] mt7615e 0000:02:00.0: Message 00000049 (seq 10) timeout
[118841.258256] mt7615e 0000:02:00.0: Message 00000049 (seq 11) timeout
[118867.497741] mt7615e 0000:02:00.0: Message 00000049 (seq 12) timeout
[118893.737335] mt7615e 0000:02:00.0: Message 00000049 (seq 13) timeout
[118919.976890] mt7615e 0000:02:00.0: Message 00000049 (seq 14) timeout
[118946.216353] mt7615e 0000:02:00.0: Message 00000049 (seq 15) timeout
[118972.455966] mt7615e 0000:02:00.0: Message 00000049 (seq 1) timeout
[118998.695429] mt7615e 0000:02:00.0: Message 00000049 (seq 2) timeout
[119024.934961] mt7615e 0000:02:00.0: Message 00000049 (seq 3) timeout
[119051.174495] mt7615e 0000:02:00.0: Message 00000049 (seq 4) timeout
[119077.414001] mt7615e 0000:02:00.0: Message 00000025 (seq 5) timeout
[119077.420282] wlan1: HW problem - can not stop rx aggregation for ee:c8:xx:xx:xx:xx tid 0
[119097.893903] mt7615e 0000:02:00.0: Message 00000025 (seq 6) timeout
[119097.900193] wlan1: HW problem - can not stop rx aggregation for ee:c8:xx:xx:xx:xx tid 1
[119118.373278] mt7615e 0000:02:00.0: Message 00000025 (seq 7) timeout
[119118.379556] ------------[ cut here ]------------
[119118.384546] WARNING: CPU: 1 PID: 1774 at backports-5.15-rc6-1/net/mac80211/agg-tx.c:406 ___ieee80211_stop_tx_ba_session+0x2ac/0x2cc [mac80211]
[119118.397410] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD wireguard pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc poly1305_mips nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libblake2s_generic iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat chacha_mips nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ctr cmac ccm leds_gpio gpio_button_hotplug
[119118.471928] CPU: 1 PID: 1774 Comm: hostapd Not tainted 5.4.155 #0
[119118.478087] Stack : 854cc000 8007eba4 80700000 806fd154 80760000 806fd11c 806fc2e8 868b3a24
[119118.486505]         808a0000 8068fccc 87e5c728 80746d03 00000001 00000001 868b39c8 21110196
[119118.494919]         00000000 00000000 808e0000 00000000 30000000 0000015d 00000018 312e342e
[119118.503331]         00000000 868b38ea ffffffff 68203a6d 80000000 00000001 00000000 86a13728
[119118.511743]         00000009 85491300 86adcc40 86c30070 00000008 8038ada8 00000004 808a0004
[119118.520148]         ...
[119118.522669] Call Trace:
[119118.525218] [<8000b318>] show_stack+0x30/0x100
[119118.529745] [<805d70b4>] dump_stack+0x9c/0xc8
[119118.534187] [<8002e804>] __warn+0xc0/0x10c
[119118.538359] [<8002e8ac>] warn_slowpath_fmt+0x5c/0xac
[119118.543528] [<86a13728>] ___ieee80211_stop_tx_ba_session+0x2ac/0x2cc [mac80211]
[119118.550946] [<86a11c7c>] ieee80211_sta_tear_down_BA_sessions+0x88/0x150 [mac80211]
[119118.558631] [<86a05e84>] ieee80211_sta_block_awake+0x434/0xc28 [mac80211]
[119118.565654] ---[ end trace 5a312c7e005aafc3 ]---
[119138.852953] mt7615e 0000:02:00.0: Message 00000025 (seq 8) timeout
[119138.859222] ------------[ cut here ]------------
[119138.864156] WARNING: CPU: 1 PID: 1774 at backports-5.15-rc6-1/net/mac80211/agg-tx.c:406 ___ieee80211_stop_tx_ba_session+0x2ac/0x2cc [mac80211]
[119138.877014] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD wireguard pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc poly1305_mips nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libblake2s_generic iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat chacha_mips nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ctr cmac ccm leds_gpio gpio_button_hotplug
[119138.951641] CPU: 1 PID: 1774 Comm: hostapd Tainted: G        W         5.4.155 #0
[119138.959190] Stack : 854cc000 8007eba4 80700000 806fd154 80760000 806fd11c 806fc2e8 868b3a24
[119138.967629]         808a0000 8068fccc 87e5c728 80746d03 00000001 00000001 868b39c8 21110196
[119138.976059]         00000000 00000000 808e0000 00000000 30000000 00000171 00000018 20202020
[119138.984488]         00000000 868b38ea ffffffff 20202020 80000000 00000001 00000000 86a13728
[119138.992917]         00000009 86cabd80 86adcc40 86c30070 00000008 8038ada8 01802917 01802957
[... cut ...]
[242325.112697] mt7615e 0000:02:00.0: Message 00000049 (seq 5) timeout
[242351.352141] mt7615e 0000:02:00.0: Message 00000049 (seq 6) timeout
[242377.591633] mt7615e 0000:02:00.0: Message 00000049 (seq 7) timeout
[242403.831154] mt7615e 0000:02:00.0: Message 00000049 (seq 8) timeout
[242430.070651] mt7615e 0000:02:00.0: Message 00000049 (seq 9) timeout
[242456.310308] mt7615e 0000:02:00.0: Message 00000049 (seq 10) timeout
[242482.549627] mt7615e 0000:02:00.0: Message 00000049 (seq 11) timeout
```

nbd168, 2021-11-25:
I pushed a firmware update for 7613 into the 'testing' branch of the mt76 tree. Please test if this improves stability

nwwt, 2022-03-16:
I've been hitting this…
`mt7615e 0000:02:00.0: Message 00000025 (seq 8) timeout`
… error on my T-MB5EU since today's snapshot (5.10.104).

Last week's snapshot (5.10.103) was fine.

ptpt52, 2022-03-16:
> I've been hitting these… `mt7615e 0000:02:00.0: Message 00000025 (seq 8) timeout` … errors on my T-MB5EU since today's snapshot (5.10.104). No wifi, luci fails as well.
> 
> Last week's snapshot (5.10.103) was fine.

T-MB5EU wifi is chip mt7915e, can't be mt7615/mt7613

nwwt, 2022-03-16:
Yes, I've opened a new issue, seeing this one is different. Sorry for the noise here.

[4 more replies]