openwrt/mt76 #512: mt7621: station dump fails with 802.11s

yogo1212, 2021-03-30:
:wave: 

```
$ iw dev mesh24 station dump
failed to parse nested attributes!
failed to parse nested attributes!
$ iw dev mesh5 station dump
failed to parse nested attributes!
failed to parse nested attributes!
```

AP station dumps work.

Environment: openwrt from 
[mid January](https://github.com/openwrt/openwrt/commit/a1a7f3274e0ed27511d45f62ee20281d8d57c7af) (4a90fdf6105e1517897aac1b451b73a54b3d9d0d), [late march](https://github.com/openwrt/openwrt/commit/bc356de2850f14629cb1301be719772fa1212e72) (fbef8bba038f5aff755e383025f0a8b34879aa5c), and [very late march](https://github.com/openwrt/openwrt/commit/f8669c174ee17bbb8d59f901d639f57449dcf751) (b8609066893a399b6aabe4df2e80eed52336fc69) using the built-in radios of a cudy WR2100 (mt7621,).

The configuration always being:
```
config wifi-iface 'mesh5'                        
        option ifname 'mesh5'                    
        option network 'mesh5' 
        option encryption 'psk2+ccmp'              
        option mesh_id 'nom'                       
        option bssid '02:BA:CC:11:55:90'
        option key ',,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,'        
        option mode 'mesh'                         
        option mesh_fwding '0'          
        option mesh_ttl '1'
        option device 'radio5'
```
Alongside an AP per radio.

In the version from January (4a90fdf6105e1517897aac1b451b73a54b3d9d0d), it would work on at least one radio from time to time. Just `wifi` and reboots until it worked seemingly random pattern.

yogo1212, 2021-04-14:
@karlpip found this [possibly related commit](https://github.com/openwrt/mt76/commit/3ef5f3444a94e450f16260a85c2ade3237e813bc)

it is included in the mt76 package for openwrt. but there is no noise in the station dump.
```
Survey data from mesh24_0
        frequency:                      2437 MHz [in use]
        channel active time:            51009580 ms
        channel busy time:              2816918 ms
        channel BSS receive time:       264 ms
        channel transmit time:          585 ms
```

if the noise isn't present there but the alleged fix is in this repo, there might be more discrepancy...

ptpt52, 2021-04-22:
same issue here on my 7613/7915

ryderlee1110, 2021-04-27:
https://git.kernel.org/pub/scm/linux/kernel/git/jberg/iw.git/commit/?id=0ba98b9041e4e0f41ee72fa0fce9c43f8c6f2db3