openwrt/mt76 #513: mt7615e cannot be clean remove

ptpt52, 2021-04-01:
I do `rmmod mt7615e` and then I try to insmod, it get errors
```
rmmod mt7615e
modprobe mt7615e

[ 9348.129223] mt7615e 0000:02:00.0: Message 16 (seq 1) timeout
[ 9348.136372] mt7615e 0000:02:00.0: Failed to get patch semaphore
[ 9348.146331] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[ 9368.609775] mt7615e 0000:02:00.0: Message 16 (seq 2) timeout
[ 9368.615603] mt7615e 0000:02:00.0: Failed to get patch semaphore
[ 9368.621647] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[ 9389.089205] mt7615e 0000:02:00.0: Message 16 (seq 3) timeout
[ 9389.094936] mt7615e 0000:02:00.0: Failed to get patch semaphore
[ 9389.101001] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[ 9409.569246] mt7615e 0000:02:00.0: Message 16 (seq 4) timeout
[ 9409.576273] mt7615e 0000:02:00.0: Failed to get patch semaphore
[ 9409.584138] mt7615e 0000:02:00.0: failed to load mediatek/mt7663pr2h_rebb.bin
[ 9430.049234] mt7615e 0000:02:00.0: Message 16 (seq 5) timeout
[ 9430.055072] mt7615e 0000:02:00.0: Failed to get patch semaphore
[ 9430.061200] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin

```

ryderlee1110, 2021-04-02:
```
root@OpenWrt:/# rmmod mt7615e
root@OpenWrt:/# modprobe mt7615e
[  500.586130] mt7615e 0001:01:00.0: assign IRQ: got 141
[  500.591307] mt7615e 0001:01:00.0: enabling bus mastering
[  500.599570] ieee80211 phy3: Selected rate control algorithm 'minstrel_ht'
[  500.624329] ieee80211 phy4: Selected rate control algorithm 'minstrel_ht'
[  500.633222] mt7615e 0001:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
root@OpenWrt:/# [  500.647162] mt7615e 0001:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
[  500.660838] mt7622-wmac 18000000.wmac: HW/SW Version: 0x8a108a10, Build Time: 20190801210006a
[  500.660838]
[  500.683415] mt7622-wmac 18000000.wmac: N9 Firmware Version: 2.0, Build Time: 20200131180931
```
Do you have a mt7663 device on platform?

ptpt52, 2021-04-02:
@ryderlee1110  actually it is mt7613.

ptpt52, 2021-04-02:
ok, I know now, it is just issue with mt7613

ptpt52, 2021-04-02:
so the issue is somewhat related to `mt7615_mcu_exit()`, where it does the wrong thing?
@ryderlee1110

ryderlee1110, 2021-04-02:
Don't know. I just have 7615 and 7622, and both work well.

ryderlee1110, 2021-04-02:
A related question - I didn't pay much attention on mt76x3 work but did you try setting prefer_offload_fw = false first?

ptpt52, 2021-04-02:
> A related question - I didn't pay much attention on mt76x3 work but did you try setting prefer_offload_fw = false first for all your 7613 activities?

for now I am using the `mt7663-firmware-ap`, say `mt7663_n9_rebb.bin + mt7663pr2h_rebb.bin`

and switch to `mt7663-firmware-sta`, still all issues persist

1. cannot rmmod and insmod
2. ssid remain existing after disable

ryderlee1110, 2021-04-02:
what if you replace the firmware with the one you get from proprietary SDK?

ptpt52, 2021-04-02:
I did it before, using the `mt7663/bin/mt7663/rebb/mt7663_patch_e2_hdr.bin` to replace `mt7663pr2h_rebb.bin`
and all issue persist.

BTW, which one is the potential replacement of mt7663_n9_rebb.bin?

ptpt52, 2021-04-02:
> regarding ssid problem, so it already exists for a long time, right?

sure.

>BTW, which one is the potential replacement of mt7663_n9_rebb.bin?

I get these bins:
mt7663/bin/mt7663/rebb/MT7663_EEPROM.bin
mt7663/bin/mt7663/rebb/mt7663_patch_e1_hdr.bin
mt7663/bin/mt7663/rebb/mt7663_patch_e2_hdr.bin
mt7663/bin/mt7663/rebb/WIFI_RAM_CODE_MT7663.bin
mt7663/bin/mt7663/rebb/WIFI_RAM_CODE_MT7663_E2.bin

ptpt52, 2021-04-02:
and more strange things would happen

setup `ssid1` `ssid2`, channel is auto
and then disable `ssid1`, and rename `ssid2` to `ssid1`

then scan one phone, 2 `ssid1` would be found on channel 60 and channel 64

ptpt52, 2021-04-03:
this issue is somewhat like that:
```
bss beacon start and can never stop (set beacon offload disable cannot stop it)
mcu start run and can never stop (remove kmod cannot stop it)
```

it may be something issue in the chip of mt7613?

ptpt52, 2021-04-03:
@ryderlee1110 
I get some point,

when disable beacon in `mt7615_mcu_add_beacon_offload`
the call `ieee80211_beacon_get_template` return NULL, so the beacon disable command never reach to mcu

here is my changes to fix:

```
diff --git a/mt7615/mcu.c b/mt7615/mcu.c
index e3a7d73..53a67b4 100644
--- a/mt7615/mcu.c
+++ b/mt7615/mcu.c
@@ -736,33 +736,40 @@ mt7615_mcu_add_beacon_offload(struct mt7615_dev *dev,
 	req.band_idx = mvif->mt76.band_idx;
 
 	skb = ieee80211_beacon_get_template(hw, vif, &offs);
-	if (!skb)
-		return -EINVAL;
-
-	if (skb->len > 512 - MT_TXD_SIZE) {
-		dev_err(dev->mt76.dev, "Bcn size limit exceed\n");
-		dev_kfree_skb(skb);
-		return -EINVAL;
-	}
-
-	if (mvif->mt76.band_idx) {
-		info = IEEE80211_SKB_CB(skb);
-		info->hw_queue |= MT_TX_HW_QUEUE_EXT_PHY;
-	}
+	if (!skb) {
+		printk("mt7615_mcu_add_beacon_offload call ieee80211_beacon_get_template return NULL");
+		if (enable)
+			return -EINVAL;
+		req.pkt_len = 0;
+		req.tim_ie_pos = 0;
+		req.csa_ie_pos = 0;
+		req.csa_cnt = 0;
+	} else {
+		if (skb->len > 512 - MT_TXD_SIZE) {
+			dev_err(dev->mt76.dev, "Bcn size limit exceed\n");
+			dev_kfree_skb(skb);
+			return -EINVAL;
+		}
 
-	mt7615_mac_write_txwi(dev, (__le32 *)(req.pkt), skb, wcid, NULL,
-			      0, NULL, true);
-	memcpy(req.pkt + MT_TXD_SIZE, skb->data, skb->len);
-	req.pkt_len = cpu_to_le16(MT_TXD_SIZE + skb->len);
-	req.tim_ie_pos = cpu_to_le16(MT_TXD_SIZE + offs.tim_offset);
-	if (offs.cntdwn_counter_offs[0]) {
-		u16 csa_offs;
+		if (mvif->mt76.band_idx) {
+			info = IEEE80211_SKB_CB(skb);
+			info->hw_queue |= MT_TX_HW_QUEUE_EXT_PHY;
+		}
 
-		csa_offs = MT_TXD_SIZE + offs.cntdwn_counter_offs[0] - 4;
-		req.csa_ie_pos = cpu_to_le16(csa_offs);
-		req.csa_cnt = skb->data[offs.cntdwn_counter_offs[0]];
+		mt7615_mac_write_txwi(dev, (__le32 *)(req.pkt), skb, wcid, NULL,
+				0, NULL, true);
+		memcpy(req.pkt + MT_TXD_SIZE, skb->data, skb->len);
+		req.pkt_len = cpu_to_le16(MT_TXD_SIZE + skb->len);
+		req.tim_ie_pos = cpu_to_le16(MT_TXD_SIZE + offs.tim_offset);
+		if (offs.cntdwn_counter_offs[0]) {
+			u16 csa_offs;
+
+			csa_offs = MT_TXD_SIZE + offs.cntdwn_counter_offs[0] - 4;
+			req.csa_ie_pos = cpu_to_le16(csa_offs);
+			req.csa_cnt = skb->data[offs.cntdwn_counter_offs[0]];
+		}
+		dev_kfree_skb(skb);
 	}
-	dev_kfree_skb(skb);
 

```

ptpt52, 2021-04-03:
test ok
no need to change mt7615_remove_interface
beacon would be disable in `mt7615_bss_info_changed`

```
ifconfig wlan1-1 down
beacon disable as expected
```

ryderlee1110, 2021-04-03:
ah. nice catch. mt7615_mcu_add_bss is used by mt7615/mt7622/mt7915 to disable BSS state (stop beaconing) so we don't run into this.

ptpt52, 2021-04-03:
> ah. nice catch. mt7615_mcu_add_bss is used by mt7615/mt7622/mt7915 to disable BSS state (stop beaconing) so we don't run into this. btw next time you can use "7613" as the subject.
> 
> I think we can just [ryderlee1110/wireless@bbd0990](https://github.com/ryderlee1110/wireless/commit/bbd09905c1bdf17712ba78220925e4f0a0c79a9d)

true
but I have to do something more to make sure `req` initialized, or it just cause mcu timeout on my chip 7613
```
diff --git a/mt7615/mcu.c b/mt7615/mcu.c
index 3c664b79..badcc3e4 100644
--- a/mt7615/mcu.c
+++ b/mt7615/mcu.c
@@ -716,6 +716,7 @@ mt7615_mcu_add_beacon_offload(struct mt7615_dev *dev,
        struct mt76_wcid *wcid = &dev->mt76.global_wcid;
        struct ieee80211_mutable_offsets offs;
        struct ieee80211_tx_info *info;
+       struct sk_buff *skb;
        struct req {
                u8 omac_idx;
                u8 enable;
@@ -731,13 +732,16 @@ mt7615_mcu_add_beacon_offload(struct mt7615_dev *dev,
                /* bss color change */
                u8 bcc_cnt;
                __le16 bcc_ie_pos;
-       } __packed req = {
-               .omac_idx = mvif->mt76.omac_idx,
-               .enable = enable,
-               .wlan_idx = wcid->idx,
-               .band_idx = mvif->mt76.band_idx,
-       };
-       struct sk_buff *skb;
+       } __packed req;
+
+       memset(&req, 0, sizeof(req));
+       req.omac_idx = mvif->mt76.omac_idx;
+       req.enable = enable;
+       req.wlan_idx = wcid->idx;
+       req.band_idx = mvif->mt76.band_idx;
+
+       if (!enable)
+               goto out;
 
        skb = ieee80211_beacon_get_template(hw, vif, &offs);
        if (!skb)
@@ -768,6 +772,7 @@ mt7615_mcu_add_beacon_offload(struct mt7615_dev *dev,
        }
        dev_kfree_skb(skb);
 
+out:
        return mt76_mcu_send_msg(&dev->mt76, MCU_EXT_CMD_BCN_OFFLOAD, &req,
                                 sizeof(req), true);
 }
```

ptpt52, 2021-04-03:
another very strange issue, I try to dump_stack() in `mt7615_mcu_add_bss()`
but get all wrong stack for no reason!!
```
[   64.405989] mt7615_bss_info_changed: call mt7615_mcu_add_bss_info
[   64.412117] ------------[ cut here ]------------
[   64.416843] WARNING: CPU: 3 PID: 1822 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2021-04-03-5282b867/mt7615/mcu.c:932 mt7615_mcu_exit+0x1fc/0x7bc [mt7615_common]
[   64.432161] Modules linked in: xt_connlimit nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_CT nft_redir nft_ct nf_nat nf_t
[   64.432446]  ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnec
[   64.553248] CPU: 3 PID: 1822 Comm: hostapd Not tainted 5.4.105 #0
[   64.559328] Stack : 864ab9e8 80072e10 80660000 806623c0 806d0000 80662388 806614ec 864ab7b4
[   64.567675]         80810000 87cb23d4 806aac43 805f58bc 00000003 00000001 864ab758 9562ee79
[   64.576017]         00000000 00000000 80850000 00000000 30232035 00000126 2e352064 30312e34
[   64.584352]         00000000 00000003 00000000 00087120 80000000 806d0000 00000000 866a5530
[   64.592686]         00000009 866b0000 00000000 86443978 00000000 80302620 0000000c 8081000c
[   64.601017]         ...
[   64.603465] Call Trace:
[   64.605935] [<8000b61c>] show_stack+0x30/0x100
[   64.610394] [<8053e6bc>] dump_stack+0xa4/0xdc
[   64.614758] [<8002bcd0>] __warn+0xc0/0x10c
[   64.618847] [<8002bd78>] warn_slowpath_fmt+0x5c/0xac
[   64.623834] [<866a5530>] mt7615_mcu_exit+0x1fc/0x7bc [mt7615_common]
```

I dump these stack via this changes:
```
diff --git a/mt7615/main.c b/mt7615/main.c
index d275ea31..2e49df0b 100644
--- a/mt7615/main.c
+++ b/mt7615/main.c
@@ -554,6 +554,7 @@ static void mt7615_bss_info_changed(struct ieee80211_hw *hw,
        }
 
        if (changed & BSS_CHANGED_BEACON_ENABLED && info->enable_beacon) {
+               printk("%s: call mt7615_mcu_add_bss_info\n", __func__);
                mt7615_mcu_add_bss_info(phy, vif, NULL, true);
                mt7615_mcu_sta_add(phy, vif, NULL, true);
 
diff --git a/mt7615/mcu.c b/mt7615/mcu.c
index 9fea9ab2..b1612d93 100644
--- a/mt7615/mcu.c
+++ b/mt7615/mcu.c
@@ -929,6 +929,8 @@ mt7615_mcu_add_bss(struct mt7615_phy *phy, struct ieee80211_vif *vif,
        struct mt7615_dev *dev = phy->dev;
        struct sk_buff *skb;
 
+       WARN_ON(1);
+
        if (mvif->mt76.omac_idx >= REPEATER_BSSID_START)
                mt7615_mcu_muar_config(dev, vif, true, enable);
 

```

you see it, this is total wrong stack.
I don't know what's going on, or maybe something stack overflow happens?

ptpt52, 2021-04-03:
another question when I review on the code https://github.com/ryderlee1110/wireless/blob/b2e60a964f244309a430e4262cdca1e6c7cad687/drivers/net/wireless/mediatek/mt76/mt7615/mcu.c#L1191

```
	struct {
		struct req_hdr {
			u8 bss_idx;
			u8 pad[3];
		} __packed hdr;
		struct bcn_content_tlv {
			__le16 tag;
			__le16 len;
			__le16 tim_ie_pos;
			__le16 csa_ie_pos;
			__le16 bcc_ie_pos;
			/* 0: enable beacon offload
			 * 1: disable beacon offload
			 * 2: update probe respond offload
			 */
			u8 enable;
			/* 0: legacy format (TXD + payload)
			 * 1: only cap field IE
			 */
			u8 type;
			__le16 pkt_len;
			u8 pkt[512];
		} __packed beacon_tlv;
	} req = {
		.hdr = {
			.bss_idx = mvif->mt76.idx,
		},
		.beacon_tlv = {
			.tag = cpu_to_le16(UNI_BSS_INFO_BCN_CONTENT),
			.len = cpu_to_le16(sizeof(struct bcn_content_tlv)),
			.enable = enable,
		},
	};
```

beacon_tlv.enable = 0 or 1 or 2
0: enable beacon offload
1: disable beacon offload
2: update probe respond offload

so if enable is true, what the value set to beacon_tlv.enable is 1
this is correct as expected?

ryderlee1110, 2021-04-03:
I don't know 7663 actually. one thing weird is why do you need to memest every time to make sure req initialized?

ptpt52, 2021-04-04:
> I guess it's typo. Can you apply my patch to test it with prefer_offload_fw = true? One more thing weird is why do you have to memest to make req initialized? The members that are not explicitly initialized will be zero initialized as part of the designated initializer. You don't even need that 0 there. What's your complier?

what I do is trying to figure out the cause of mcu timeout issue the initialize I do maybe not necessary but I just make it a sure.
for now, the mcu timeout issue, it happens in rarely but persist.

ptpt52, 2021-04-04:
Issues on mt7613:
1. mcu timeout
2. kmod mt7615e remove and insmod would not work again
3. dump_stack() in mt76 output incorrect call stacks

ptpt52, 2021-04-04:
prefer_offload_fw = true means using these firmware bins: `mt7663pr2h.bin mt7663_n9_v3.bin`
this is intended for sta client usage

my unit is intended to use as AP
so I use these bins: `mt7663_n9_rebb.bin   mt7663pr2h_rebb.bin`

ptpt52, 2021-04-16:
@nbd168  I know the point:

1. mt7615_load_patch() for the first time it load the `mediatek/mt7663pr2h_rebb.bin`
```
[   33.462722] mt7615e 0000:02:00.0: enabling device (0000 -> 0002)
[   35.240726] mt7615e 0000:02:00.0: mediatek/mt7663pr2h.bin not found, switching to mediatek/mt7663pr2h_rebb.bin
[   35.515725] mt7615e 0000:02:00.0: HW/SW Version: 0x65322d31, Build Time: 19091404338b809
[   35.913911] mt7615e 0000:02:00.0: N9 Firmware Version: 7663mp1827, Build Time: 20190914043434
[   35.922535] mt7615e 0000:02:00.0: Region number: 0x3
[   35.927525] mt7615e 0000:02:00.0: Parsing tailer Region: 0
[   35.948607] mt7615e 0000:02:00.0: Region 0, override_addr = 0x00112c00
[   35.955182] mt7615e 0000:02:00.0: Parsing tailer Region: 1
[   35.964528] mt7615e 0000:02:00.0: Parsing tailer Region: 2
[   35.971615] mt7615e 0000:02:00.0: override_addr = 0x00112c00, option = 3
```
2. the second time it mt7615_load_patch() return success without trying to load `mt7663pr2h_rebb.bin
`
actually the loaded patch is `mediatek/mt7663pr2h_rebb.bin`, but it assume it as `mediatek/mt7663pr2h.bin`
then `mt7663_load_n9()` trying to load the wrong n9_firmware so it failed.
```
1836 int __mt7663_load_firmware(struct mt7615_dev *dev)
1837 {
1838         const char *n9_firmware;
1839         int ret;
1840 
1841         ret = mt76_get_field(dev, MT_CONN_ON_MISC, MT_TOP_MISC2_FW_N9_RDY);
1842         if (ret) {
1843                 dev_dbg(dev->mt76.dev, "Firmware is already download\n");
1844                 return -EIO;
1845         }
1846 
1847         ret = mt7663_load_rom_patch(dev, &n9_firmware);
1848         if (ret)
1849                 return ret;
1850 
1851         ret = mt7663_load_n9(dev, n9_firmware);
1852         if (ret)
1853                 return ret;
```

nbd168, 2021-04-16:
How about this one? http://nbd.name/mt7663-fw-load.patch

ptpt52, 2021-04-16:
@nbd168  with this patch, it works now.
kmod remove and re-insmod ok.

nbd168, 2021-04-16:
Great. I'll push out the change and send it upstream then. Thanks a lot for your help in fixing these issues.

ptpt52, 2021-04-16:
thanks for you patches and big help!

Now the only big issue left is memory leak/airtime leak issue.

closed 2021-04-16 by ptpt52