openwrt/mt76 #515: mcu timeout on mt7628

ptpt52, 2021-04-06:
in recent code
random happens

```
[936121.645135] mt76_wmac 10300000.wmac: MCU message 8 (seq 4) timed out
[936125.997103] mt76_wmac 10300000.wmac: MCU message 8 (seq 5) timed out
[936129.069084] mt76_wmac 10300000.wmac: MCU message 8 (seq 6) timed out
[936132.141059] mt76_wmac 10300000.wmac: MCU message 8 (seq 7) timed out
[936135.213038] mt76_wmac 10300000.wmac: MCU message 8 (seq 8) timed out
[936138.285014] mt76_wmac 10300000.wmac: MCU message 8 (seq 9) timed out
[936141.356995] mt76_wmac 10300000.wmac: MCU message 8 (seq 10) timed out
[936144.428978] mt76_wmac 10300000.wmac: MCU message 8 (seq 11) timed out
[936147.500947] mt76_wmac 10300000.wmac: MCU message 8 (seq 12) timed out
[936150.572926] mt76_wmac 10300000.wmac: MCU message 8 (seq 13) timed out
[936153.644901] mt76_wmac 10300000.wmac: MCU message 8 (seq 14) timed out
[936156.716880] mt76_wmac 10300000.wmac: MCU message 8 (seq 15) timed out
[936159.788857] mt76_wmac 10300000.wmac: MCU message 8 (seq 1) timed out
[936162.860832] mt76_wmac 10300000.wmac: MCU message 8 (seq 2) timed out
[936165.932807] mt76_wmac 10300000.wmac: MCU message 8 (seq 3) timed out
[936170.284781] mt76_wmac 10300000.wmac: MCU message 8 (seq 4) timed out
[936173.356758] mt76_wmac 10300000.wmac: MCU message 8 (seq 5) timed out
[936176.428752] mt76_wmac 10300000.wmac: MCU message 8 (seq 6) timed out
[936179.500710] mt76_wmac 10300000.wmac: MCU message 8 (seq 7) timed out
[936182.572687] mt76_wmac 10300000.wmac: MCU message 8 (seq 8) timed out
[936185.644667] mt76_wmac 10300000.wmac: MCU message 8 (seq 9) timed out
[936188.716645] mt76_wmac 10300000.wmac: MCU message 8 (seq 10) timed out
[936191.788621] mt76_wmac 10300000.wmac: MCU message 8 (seq 11) timed out
[936194.860600] mt76_wmac 10300000.wmac: MCU message 8 (seq 12) timed out
[936197.932574] mt76_wmac 10300000.wmac: MCU message 8 (seq 13) timed out
[936201.004555] mt76_wmac 10300000.wmac: MCU message 8 (seq 14) timed out
[936204.076533] mt76_wmac 10300000.wmac: MCU message 8 (seq 15) timed out
[936207.148506] mt76_wmac 10300000.wmac: MCU message 8 (seq 1) timed out
[936210.220485] mt76_wmac 10300000.wmac: MCU message 8 (seq 2) timed out
```

sanitariu, 2021-08-10:
Tested latest git and after heavy torrent traffic i have got the same. Disabling / Enabling wifi does not help. Only solution is to reboot the router. Nothing i the logs except that message.

SvenRoederer, 2021-11-16:
also seen here on a TPLink wr841-v13 running OpenWrt-21.02.1

~~~
[210358.885075] device wlan0-dhcp-2 left promiscuous mode
[210358.890531] br-dhcp: port 3(wlan0-dhcp-2) entered disabled state
[210389.105644] mt76_wmac 10300000.wmac: MCU message 08 (seq 12) timed out
[210392.177650] mt76_wmac 10300000.wmac: MCU message ef (seq 13) timed out
[210472.920226] mt76_wmac 10300000.wmac: ASIC revision: 76280001
[210472.994918] mt76_wmac 10300000.wmac: Firmware Version: 20151201
[210473.001100] mt76_wmac 10300000.wmac: Build Time: 20151201183641
[210473.007217] mt76_wmac 10300000.wmac: Firmware already running...
[210473.013422] mt76_wmac 10300000.wmac: firmware init done
[210476.144743] mt76_wmac 10300000.wmac: MCU message 21 (seq 1) timed out
[210476.156398] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
~~~

seq 12, 13 are when unloading the mt7603e module, `message 21 (seq 1)` is when loading the module again.

Djfe, 2024-02-09:
@ptpt52 is this still an issue?

beege, 2024-02-25:
I'm having this issue as well. Happens under load. Running  OpenWrt 22.03.5, r20134-5f15225c1e. I'll grab a later version and see if it's still happening.

beege, 2024-02-26:
Just happened again for me running a later version. Running a snapshot from 2 days ago: OpenWrt SNAPSHOT, r25318-3e8d2f2439

beege, 2024-02-27:
OK I'll give 23.05.2 a whirl and see what happens. Usually takes a few days for the error to occur in my experience so I'll check back in a week or so if it doesn't happen.

beege, 2024-03-06:
Haven't seen it in a week now. Looks solved to me.

Djfe, 2024-03-06:
great, please close it then, you can still reopen it if you experience it again 👍

closed 2024-03-06 by ptpt52