openwrt/mt76 #517: mt7915 low TX speeds from client

tkso1997, 2021-04-07:
I have an Linksys e8450 with mt7915 up and running. The AC performance is in general very good. If I am very close to the router I can receive  up to 650 Mbits/sec and send up to 600 Mbits/sec. As soon as I move a bit further away the sending rate goes dramatically down. About 5 meters away RX on my MacBook Pro (AC2x2) is still 450 Mbits/sec, while TX goes down to 170 Mbits/sec.
Is there anything I can tweak? With my old R3P (mt7615) the TX performance was always better than RX, so I am wondering whether its the mt7915 driver/hardware limitations or anything in the settings I can optimize?

ryderlee1110, 2021-04-08:
cat /sys/kernel/debug/ieee80211/phyX/mt76/txpower_sku and paste the output here.

tkso1997, 2021-04-08:
```
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy1/mt76/txpower_sku 

Band 0
       CCK: 34 34 34 34
      OFDM: 34 34 34 34 34 34 34 34
      HT20: 34 34 34 34 34 34 34 34
      HT40: 34 34 34 34 34 34 34 34
     VHT20: 34 34 34 34 34 34 34 34 34 34
     VHT40: 34 34 34 34 34 34 34 34 34 34
     VHT80: 34 34 34 34 34 34 34 34 34 34
    VHT160: 34 34 34 34 34 34 34 34 34 34
      RU26: 34 34 34 34 34 34 34 34 34 34 32 32
      RU52: 34 34 34 34 34 34 34 34 34 34 32 32
     RU106: 34 34 34 34 34 34 34 34 34 34 32 32
RU242/SU20: 34 34 34 34 34 34 34 34 34 34 32 32
RU484/SU40: 34 34 34 34 34 34 34 34 34 34 32 32
RU996/SU80: 34 34 34 34 34 34 34 34 34 34 32 32
RU2x996/SU160: 34 34 34 34 34 34 34 34 34 34 32 32
```

ryderlee1110, 2021-04-09:
hm. txpower looks normal. I just tested rate vs range seems ok.

tkso1997, 2021-04-10:
Okay, I will do some further testing.
Overall performance is very good!

DragonBluep, 2022-09-16:
The latest version still has some problems in 5 GHz AX mode.

AP: MT7621A + MT7915D + 256 MiB RAM (mt76 driver version 2022/09/15)
STA: Windows 10 + MT7921 (driver version 3.03.00.0615 15.07.2022)

The iperf3 server runs on mt7621, default ap mode on latest OpenWrt. Not sure if the problem is related to the performance of the mt7621, iperf3 takes a lot of resources. There is almost no difference between 2.4G and 5G.
<details>
<summary>iperf3 test</summary>

```
5 GHz (AP <- STA):
PS D:\Applications\iperf-3.11-win64> ./iperf3 -c 192.168.1.1 -P 4
Connecting to host 192.168.1.1, port 5201
[ ID] Interval           Transfer     Bitrate
[SUM]   0.00-10.00  sec   562 MBytes   471 Mbits/sec                  sender
[SUM]   0.00-10.01  sec   557 MBytes   466 Mbits/sec                  receiver

5 GHz (AP -> STA):
PS D:\Applications\iperf-3.11-win64> ./iperf3 -c 192.168.1.1 -P 4 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[ ID] Interval           Transfer     Bitrate         Retr
[SUM]   0.00-10.09  sec   383 MBytes   318 Mbits/sec    0             sender
[SUM]   0.00-10.00  sec   379 MBytes   318 Mbits/sec                  receiver

2.4 GHz (AP <- STA):
PS D:\Applications\iperf-3.11-win64> ./iperf3 -c 192.168.1.1 -P 4
Connecting to host 192.168.1.1, port 5201
[ ID] Interval           Transfer     Bitrate
[SUM]   0.00-10.01  sec   542 MBytes   455 Mbits/sec                  sender
[SUM]   0.00-10.05  sec   541 MBytes   452 Mbits/sec                  receiver

2.4 GHz (AP -> STA):
PS D:\Applications\iperf-3.11-win64> ./iperf3 -c 192.168.1.1 -P 4 -R
Connecting to host 192.168.1.1, port 5201
Reverse mode, remote host 192.168.1.1 is sending
[ ID] Interval           Transfer     Bitrate         Retr
[SUM]   0.00-10.02  sec   390 MBytes   326 Mbits/sec    0             sender
[SUM]   0.00-10.00  sec   389 MBytes   326 Mbits/sec                  receiver
```
</details>

rany2, 2023-06-08:
Just a hint but on SNAPSHOT, restarting the router a couple of times causes it to work again. Not precise numbers but if you reboot it the TX ends up being normal (i.e. 800Mbit/s+ in HE80) about 1 in 10 times.

Anteus, 2023-06-09:
Sounds like this problem https://github.com/openwrt/mt76/pull/768
https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466

rany2, 2023-06-09:
@Anteus I have that patch applied.

Anteus, 2023-06-11:
This commit mentioned in the mt76 thread on openwrt forum sounds promising, might be worth to try and backport? https://lore.kernel.org/linux-wireless/20230604120651.48d262b6b42d.Ia15532657c17535c28ec0c5df263b65f0f80663c@changeid/

Could try the full removal commit of 160MHz including AX and see if that makes a difference?

@rany2 what device are you testing on? I'm on rt3200/e8450

rany2, 2023-06-11:
@Anteus I have already tried reverting the 160Mhz, it's unrelated. I'll backport that commit to the mac80211 stack

rany2, 2023-06-11:
@Anteus No dice, same issue. https://github.com/rany2/openwrt/commit/e24d1c3ab24e93f9aac154c617a315402f15e11d