openwrt/mt76 #521: quit OPENWRT felix, you're terrible

i3roly, 2021-04-10:
or quit life, felix.

waiting for benchies to top my build, felix.

wasted two years trying to challenge LEGACY RALINK.

how many shares did you get felix, to try and topple the giant?

QUIT, FELIX. EVERYONE AT THE OPENWRT TEAM HAS SUFFERED BECAUSE OF YOU.

WIRELESS-NEXT HAS SUFFERED, BECAUSE YOU, FELIX.

QUIT.

i3roly, 2021-04-10:
were you born a serpent @nbd168?

the shame is that the entire project must go through you. look at the below so people can see how much @brainslayer has to deal with. 

you are a serpent, felix. he still has to kiss your rings, because obviously you are the one who undeservedly leads all of the opensource wireless projects. and you have done nothing to deserve it, irq_spin_{un}lock_{save,restore} "king".

QUIT, SERPENT.

```
From: Gagan Sidhu <broly@mac.com>
Subject: Re: some bugs in jansson
Date: February 1, 2018 at 9:57:22 AM MST
To: Sebastian Gottschall <s.gottschall@dd-wrt.com>

picture log attached (better than text log). i want us to always have clear lines of communication.

i’m pretty sure this is felix fietkau. the person’s IP resolves to germany and his name was ‘fefelix’ at the time

IP was 80.70.189.230

you sure it’s not felix, man?

this guy seemed to know a lot, and i certainly did not see “nbd” or felix on at the time of this conversation. wouldn’t he make sure not to use his nbd vhost when having a conversation like this?

On Feb 1, 2018, at 9:39 AM, Sebastian Gottschall <s.gottschall@dd-wrt.com> wrote:

Am 01.02.2018 um 17:05 schrieb Gagan Sidhu:
I'm dropping a fat one gimme half and I will. He pretty much suggested there was a court proceeding against you for the webgui and that you do things the wrong way.
there was never a legal action against me. i believe that the felix you mentioned about is not felix fietkau (nbd)
this gui exists for 10 years now and is written from scratch just with look and feel with the old linksys routers
for me it sounds like you talked with imre kaloz and he is a total asshole
```

[image]
[image]
[image]
[image]
[image]
```

I will get it for you ASAP. it's nothing like my statements ("loser" "fag" etc lol) but moreso a denigration of your approach.

On Feb 1, 2018, at 9:02 AM, Sebastian Gottschall <s.gottschall@dd-wrt.com> wrote:

Am 01.02.2018 um 16:45 schrieb Gagan Sidhu:
lol how is Felix working for you when I've got irc logs of him bad mouthing you?
Felix is a loser.
show me
I wouldn't be surprised if the nasty story behind the formation of lede (kong's words) was all Felix.
not just felix, but the cause is deeper. there was a evil guy behind the old openwrt team which caused the split
No one can save mr "no big deal" from the embarrassment he has managed to artfully elude for a decade (and you know it!)

Lol "don't put shit on him" I love you man hahahahaha

On Feb 1, 2018, at 8:35 AM, Sebastian Gottschall <s.gottschall@dd-wrt.com> wrote:

Am 01.02.2018 um 06:28 schrieb Gagan Sidhu:
fuckign bricked my 882 because of uboot! all i did was type “spi erase all” and i just expected it to clean all of the partitions, turned out it wiped it clean and i couldn’t even get a serial console afterwards.

got another one and while it cost a pretty penny (that’s ignoring how fucking annoying this thing is to install serial pins for. it is seriously the worst router ever for it. i guess i can understand why hehe), it seems setting rootfs
to the partition address (0x429000 in my case) won’t always work.
dont know. mediatek was always crap. but never had troubles with them. except that the wifi chipsets arent good
the early successes lead me to believe i could just wipe my spi and try again, hoping that any corruption on the partitions would then be addressed

googling around about the issue suggests it effects many lede distros (lol), and that maybe this is something i should take some time to develop a full solution for.

really upset about bricking one dir882, but as you know i feel sort of indebted to those guys (they were the fuel for my years of youth), so i bought another one anyways (and almost fucked this one up installing the pins, fuck! lol).

god damnit BS we need this shit working so felix is holding a “will suck dick for food” sign ;)
ahm felix fietkau? he is working in our company too. dont put shit on him
[... 339 lines cut ...]
Mit freundlichen Grüssen / Regards

Sebastian Gottschall / CTO

NewMedia-NET GmbH - DD-WRT
Firmensitz:  Stubenwaldallee 21a, 64625 Bensheim
Registergericht: Amtsgericht Darmstadt, HRB 25473
Geschäftsführer: Peter Steinhäuser, Christian Scheele
http://www.dd-wrt.com
email: s.gottschall@dd-wrt.com
Tel.: +496251-582650 / Fax: +496251-5826565

-- 
Mit freundlichen Grüssen / Regards

Sebastian Gottschall / CTO

NewMedia-NET GmbH - DD-WRT
Firmensitz:  Stubenwaldallee 21a, 64625 Bensheim
Registergericht: Amtsgericht Darmstadt, HRB 25473
Geschäftsführer: Peter Steinhäuser, Christian Scheele
http://www.dd-wrt.com
email: s.gottschall@dd-wrt.com
Tel.: +496251-582650 / Fax: +496251-5826565
```

closed 2021-04-10 by dangowrt

hahahalmao, 2021-04-10:
test 
Thanks,
Gagan

Begin forwarded message:

From: Gagan Sidhu <broly@mac.com>
Subject: Re: some bugs in jansson
Date: February 1, 2018 at 9:57:22 AM MST
To: Sebastian Gottschall <s.gottschall@dd-wrt.com>

picture log attached (better than text log). i want us to always have clear lines of communication.

i’m pretty sure this is felix fietkau. the person’s IP resolves to germany and his name was ‘fefelix’ at the time

IP was 80.70.189.230

you sure it’s not felix, man?

this guy seemed to know a lot, and i certainly did not see “nbd” or felix on at the time of this conversation. wouldn’t he make sure not to use his nbd vhost when having a conversation like this?

On Feb 1, 2018, at 9:39 AM, Sebastian Gottschall <s.gottschall@dd-wrt.com> wrote:

Am 01.02.2018 um 17:05 schrieb Gagan Sidhu:
I'm dropping a fat one gimme half and I will. He pretty much suggested there was a court proceeding against you for the webgui and that you do things the wrong way.
there was never a legal action against me. i believe that the felix you mentioned about is not felix fietkau (nbd)
this gui exists for 10 years now and is written from scratch just with look and feel with the old linksys routers
for me it sounds like you talked with imre kaloz and he is a total asshole

Screen Shot 2018-02-01 at 9.52.17 AM.pngScreen Shot 2018-02-01 at 9.52.30 AM.pngScreen Shot 2018-02-01 at 9.52.41 AM.pngScreen Shot 2018-02-01 at 9.52.49 AM.pngScreen Shot 2018-02-01 at 9.53.01 AM.png

I will get it for you ASAP. it's nothing like my statements ("loser" "fag" etc lol) but moreso a denigration of your approach.

On Feb 1, 2018, at 9:02 AM, Sebastian Gottschall <s.gottschall@dd-wrt.com> wrote:

Am 01.02.2018 um 16:45 schrieb Gagan Sidhu:
lol how is Felix working for you when I've got irc logs of him bad mouthing you?
Felix is a loser.
show me
I wouldn't be surprised if the nasty story behind the formation of lede (kong's words) was all Felix.
not just felix, but the cause is deeper. there was a evil guy behind the old openwrt team which caused the split
No one can save mr "no big deal" from the embarrassment he has managed to artfully elude for a decade (and you know it!)

Lol "don't put shit on him" I love you man hahahahaha

On Feb 1, 2018, at 8:35 AM, Sebastian Gottschall <s.gottschall@dd-wrt.com> wrote:

Am 01.02.2018 um 06:28 schrieb Gagan Sidhu:
fuckign bricked my 882 because of uboot! all i did was type “spi erase all” and i just expected it to clean all of the partitions, turned out it wiped it clean and i couldn’t even get a serial console afterwards.

got another one and while it cost a pretty penny (that’s ignoring how fucking annoying this thing is to install serial pins for. it is seriously the worst router ever for it. i guess i can understand why hehe), it seems setting rootfs
to the partition address (0x429000 in my case) won’t always work.
dont know. mediatek was always crap. but never had troubles with them. except that the wifi chipsets arent good
the early successes lead me to believe i could just wipe my spi and try again, hoping that any corruption on the partitions would then be addressed

googling around about the issue suggests it effects many lede distros (lol), and that maybe this is something i should take some time to develop a full solution for.

really upset about bricking one dir882, but as you know i feel sort of indebted to those guys (they were the fuel for my years of youth), so i bought another one anyways (and almost fucked this one up installing the pins, fuck! lol).

god damnit BS we need this shit working so felix is holding a “will suck dick for food” sign ;)
ahm felix fietkau? he is working in our company too. dont put shit on him

On Jan 30, 2018, at 5:19 PM, Sebastian Gottschall <s.gottschall@dd-wrt.com> wrote:

Am 30.01.2018 um 22:24 schrieb Gagan Sidhu:
sorry one last question about the squashfs approach (i had this question in september too, but it wasn’t relevant enough at the time since i didn’t have a working kernel).

since i’m a u-boot neophyte, i guess i should ask: why is there a difference in magic bytes when flashing dd-wrt for the first time (from stock) and “webflash” (dd-wrt to dd-wrt)?
because dd-wrt uses a firmware format and dlink, asus, netgear all are using a individual different format and uboot is not the only bootloader on this planet.
dd-wrt was started where uboot did not exist on all the current platforms back in 2005
[... 334 lines cut ...]
NewMedia-NET GmbH - DD-WRT
Firmensitz:  Stubenwaldallee 21a, 64625 Bensheim
Registergericht: Amtsgericht Darmstadt, HRB 25473
Geschäftsführer: Peter Steinhäuser, Christian Scheele
http://www.dd-wrt.com
email: s.gottschall@dd-wrt.com
Tel.: +496251-582650 / Fax: +496251-5826565

-- 
Mit freundlichen Grüssen / Regards

Sebastian Gottschall / CTO

NewMedia-NET GmbH - DD-WRT
Firmensitz:  Stubenwaldallee 21a, 64625 Bensheim
Registergericht: Amtsgericht Darmstadt, HRB 25473
Geschäftsführer: Peter Steinhäuser, Christian Scheele
http://www.dd-wrt.com
email: s.gottschall@dd-wrt.com
Tel.: +496251-582650 / Fax: +496251-5826565
[image]
[image]
[image]
[image]
[image]