openwrt/mt76 #522: 802.11s broken in latest version

amaumene, 2021-04-11:
Hi,

I'm running OpenWrt based on version 024c81adb8 between 2 Xiaomi Redmi AC2100 without any issue. They both have this configuration:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option cell_density '0'
	option country 'SG'
	option htmode 'HT20'
	option channel '1'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'plop'
	option key 'my_very_secure_password'
	option encryption 'psk2+ccmp'
	option ieee80211r '1'
	option mobility_domain '1b23'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'VHT80'
	option cell_density '0'
	option country 'SG'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'plop'
	option key 'my_very_secure_password'
	option encryption 'psk2+ccmp'
	option ieee80211r '1'
	option mobility_domain '1a23'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'

config wifi-iface 'wifinet2'
	option device 'radio1'
	option mode 'mesh'
	option encryption 'sae'
	option mesh_id 'mesh_plop'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
	option key 'my_very_secure_key'
	option network 'lan'
```

I've tried upgrading to OpenWrt version e887049fbb but with this version 802.11s isn't working anymore. I can see this stack trace:
```
Fri Apr  9 23:05:19 2021 daemon.notice wpa_supplicant[1757]: wlan1: CTRL-EVENT-CONNECTED - Connection to 00:00:00:00:00:00 completed [id=0 id_str=]
Fri Apr  9 23:05:19 2021 daemon.notice wpa_supplicant[1757]: wlan1: MESH-GROUP-STARTED ssid="mesh_plop" id=0
Fri Apr  9 23:05:19 2021 daemon.notice wpa_supplicant[1757]: wlan1: new peer notification for 28:d1:27:11:6b:eb
Fri Apr  9 23:05:19 2021 daemon.notice wpa_supplicant[1757]: wlan1: mesh plink with 28:d1:27:11:6b:eb established
Fri Apr  9 23:05:19 2021 daemon.notice wpa_supplicant[1757]: wlan1: MESH-PEER-CONNECTED 28:d1:27:11:6b:eb
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.792988] ------------[ cut here ]------------
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.797667] WARNING: CPU: 0 PID: 3651 at net/core/flow_dissector.c:958 __skb_flow_dissect+0x2d4/0x16c4
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.806974] Modules linked in: iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG tcp_bbr poly1305_mips nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libblake2s_generic iptable_mangle iptable_filter ip_tables compat chacha_mips nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel kpp leds_gpio gpio_button_hotplug
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.871536] CPU: 0 PID: 3651 Comm: kworker/u9:2 Not tainted 5.4.109 #0
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.878079] Workqueue: napi_workq napi_workfn
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.882433] Stack : 806792e4 860eda3c 80700000 80740000 8717c900 8068b164 80425d30 00000009
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.890760]         872c119c 861ce018 00000000 8007d904 00000000 00000001 860ed9f8 9b3ca898
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.899085]         00000000 00000000 00000000 00000000 716b726f 00000142 6f775f69 6e666b72
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.907411]         00000000 00000001 00000000 000d65ff 00000000 80760000 00000000 80425d30
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.915734]         00000009 872c119c 861ce018 00000000 00000000 8035c7a8 00000000 808a0000
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.924057]         ...
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.926496] Call Trace:
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.928950] [<8000b64c>] show_stack+0x30/0x100
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.933379] [<805ca160>] dump_stack+0xa4/0xdc
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.937733] [<8002bfe8>] __warn+0xc0/0x10c
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.941812] [<8002c090>] warn_slowpath_fmt+0x5c/0xac
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.946762] [<80425d30>] __skb_flow_dissect+0x2d4/0x16c4
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.952056] [<804273d0>] __skb_get_hash+0x7c/0x258
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.956968] [<86932e94>] ieee80211_unreserve_tid+0x720/0xbcc [mac80211]
Fri Apr  9 23:05:20 2021 kern.warn kernel: [   36.963807] ---[ end trace 4e6a3e8bca6b0164 ]---
```

It looks like both router can see each other (output from the 2nd router):
```
root@route3:~# iw dev wlan1 mpath dump
DEST ADDR         NEXT HOP          IFACE	SN	METRIC	QLEN	EXPTIME		DTIM	DRET	FLAGS	HOP_COUNT	PATH_CHANGE
28:d1:27:11:6b:eb 28:d1:27:11:6b:eb wlan1	93840	1490	0	3796	100	0	0x15	1	1
```
(route3 is the router connecting to route2, and route2 has internet access, there is no route1).

This is also not working anymore:
```
root@route3:~# iw dev wlan1 station dump
failed to parse nested attributes!
```

The MAC address of route2 is properly learn by route3.
```
root@route3:~# arp -d
IP address       HW type     Flags       HW address            Mask     Device
192.168.1.2      0x1         0x0         28:d1:27:11:6b:e9     *        br-lan
```

You can see how I'm generating my OpenWrt images using the image builder [here](https://github.com/amaumene/xiaomi_redmi-router-ac2100_openwrt_mesh)

Please let me know if you need more logs.

amaumene, 2021-04-11:
Hi,

Since I build my images with a GitHub action, I have some kind of history, so I went ahead and did a bit of investigation:
* OpenWrt version acbea54bc2 is the first getting stack trace but 802.11s is still working
* OpenWrt version 1cd098784e is the latest working
* OpenWrt version d7c8ca4d0b doesn’t work anymore

What I can conclude is:
* There is a change between 024c81adb8 and acbea54bc2 that is creating the stack trace, but the stack trace doesn't prevent 802.11s to work properly.
* There is a change between 1cd098784e and d7c8ca4d0b that is breaking 802.11s (at least with my hardware).

I hope that can help, I generate images every day but sometimes image builder is broken in snapshot which explains why some times there is some missing daily images and I can't narrow it down more precisely.

amaumene, 2021-04-11:
Hi,

I suspect the update of mt76 [here](https://github.com/openwrt/openwrt/commit/12b5f898f9222d35b80058fee5bd2ae935226070) brought a change in the driver that broke 802.11s for my hardware.

I've attached the full output of logread before and after this update of mt76:
* [full_log.d7c8ca4d0b.txt (after)](https://github.com/openwrt/mt76/files/6291335/full_log.d7c8ca4d0b.txt)
* [full_log.1cd098784e.txt (before)](https://github.com/openwrt/mt76/files/6291334/full_log.1cd098784e.txt)

amaumene, 2021-04-11:
Hi,

I did a bit more testing and found out that this [patch](https://github.com/openwrt/mt76/commit/8e3f5bfe74f6634232a8d032d8dcd2f94f781b73) is the culprit. I built from scratch an image based on OpenWrt dc61c32fdc and as expected 802.11s didn't work. I then built the same image but with a patch to revert 8e3f5bfe74f6634232a8d032d8dcd2f94f781b73 and 802.11s is again working fine.

I will try this build for a few days and will report back. I wonder if this is specific to my hardware or if anyone running 802.11s on mt7615e with a recent snapshot is having the same issue.

ptpt52, 2021-04-11:
@ryderlee1110  ping

nbd168, 2021-04-11:
I've fixed this in latest master. Please test, and I'll update the mt76 package in OpenWrt afterwards

nbd168, 2021-04-11:
6b293c411d22b2bbcc5231be47905f51e1898039 is the fix

amaumene, 2021-04-11:
Hi,

Thanks for your quick answer. I've done a quick test and so far it looks good. I will need to do more extensive testing tomorrow especially with regards to roaming between both APs where I had previous issues. But this issue can be marked as resolved for me, as 802.11s is working again.

Thanks for your help.

nbd168, 2021-04-11:
Thanks for testing

closed 2021-04-11 by nbd168

amaumene, 2021-04-14:
Hi @nbd168

I have a new issue with the latest snapshot. When I run only 1 AP everything works fine, but when I run 2 APs I have a lot of disconnections:
```
Wed Apr 14 09:35:55 2021 daemon.info hostapd: wlan0: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:35:58 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:36:03 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:36:11 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:36:18 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:36:24 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:36:32 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:36:38 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:36:46 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:36:52 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:36:58 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:37:03 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:37:09 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:37:53 2021 daemon.info hostapd: wlan1-1: STA f0:18:98:26:a3:d7 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:38:01 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:38:05 2021 daemon.info hostapd: wlan1-1: STA f0:18:98:26:a3:d7 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 09:38:08 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

I've narrowed it down to either one of those 2 commits:
* https://github.com/openwrt/openwrt/commit/dfdb28c24aa60cf057ec45cbc11ab48ba2655f53
* https://github.com/openwrt/openwrt/commit/7d8e14e44f76e18d1696565569ec50ccdce963b3

I don't see how to open issues on OpenWrt and since you committed both I thought it would be easier to reach out to you here. I'm sorry if that's not the right way of reporting issues for hostapd.