openwrt/mt76 #524: MT7615 read pcie data

zhangsd20, 2021-04-12:
The device I tested is U7621-06 + mt7615N pcie card.
But it seems to work abnormally .

dmesg:
[ 12.837712] mt7615e 0000:02:00.0: Invalid MAC address, using random address 9e:1b:ba:4e:73:8a
[ 12.860242] mt7615e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a

ryderlee1110, 2021-04-12:
https://github.com/openwrt/mt76/issues/478

From your previous report, are you sure the calibration data all set?

zhangsd20, 2021-04-13:
I am using U7615N-L3 pcie card, it should read calibration data from eeprom or efuse, not from flash.
 (I bought U7621 and U7615N from different manufacturers)

 From the results of my test, we can see that 7615 can only read calibration data from the flash (factory partition).

 Can 7615 read calibration data from pcie card?

#478

ryderlee1110, 2021-04-13:
Sure. Did you specify "mediatek,mtd-eeprom" in dts? Please check the returned value of mt76_get_of_eeprom() or mt76_eeprom_init() in mt7615_eeprom_load().

zhangsd20, 2021-04-14:
Hi,
         At the beginning ，not specify "mediatek,mtd-eeprom" in dts.
        log: [ 12.837712] mt7615e 0000:02:00.0: Invalid MAC address, using random address

        

        mt76_eeprom_init in mt7615_eeprom_load return 0.

zhangsd20, 2021-04-14:
Hi,

        The maximum connection rate of mt7615 on openwrt 19.07 is only 1300 Mbit/s, and VHT-NSS shows 3.
         rx bitrate:     1300.0 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 3 
         Could it be that 3 channels are working?

ryderlee1110, 2021-04-14:
Are you sure the data of efuse is correct? Try hexdump to check it.

zhangsd20, 2021-04-15:
Hi,
================================================

   log:
       [   15.797583] mt7615e 0000:01:00.0: enabling device (0140 -> 0142)
       [   15.798144] AAAAA ret = 0      /*ret is mt7615_eeprom_load return */
       [   15.804012] mt7615e 0000:01:00.0: Invalid MAC address, using random address ee:5a:e9:64:ae:e9
       [   15.829430] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
       [   16.147194] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
       [   16.174696] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
===================================================

 hexdump -C  /sys/kernel/debug/ieee80211/phyX/mt76/eeprom

root@OpenWrt:/#  hexdump -C  /sys/kernel/debug/ieee80211/phy0/mt76/eeprom
00000000  00 00 20 00 00 00 00 00  00 00 15 76 c3 14 00 80  |.. ........v....|
00000010  02 00 c3 14 15 76 03 22  ff ff 23 04 0d 02 8f 02  |.....v."..#.....|
00000020  00 80 0a 20 00 00 00 00  00 00 00 00 00 00 00 00  |... ............|
00000030  20 08 00 00 44 00 00 60  01 01 00 00 00 00 00 00  | ...D..`........|
00000040  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000050  00 00 00 b3 40 b5 40 c4  00 00 00 00 c0 c3 00 00  ***@***.***@.........|
00000060  00 00 c0 c4 00 00 00 00  40 c4 00 00 00 00 40 c5  ***@***.***@.|
00000070  00 00 00 c0 c4 00 00 00  c0 c3 00 00 00 40 c3 00  ***@***.***|
00000080  00 00 40 c2 00 00 00 40  c2 00 00 00 40 c2 00 00  ***@***.***@....@...|
00000090  00 c0 c2 00 00 00 40 c3  00 00 00 40 c3 00 00 00  ***@***.***@....|
000000a0  40 c3 00 00 00 c0 c2 00  00 00 40 c2 00 00 00 c0  ***@***.***@.....|
000000b0  c1 00 00 00 40 c1 00 00  00 c0 c0 00 00 00 00 00  ***@***.***|
000000c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000000f0  00 08 00 00 c1 00 00 86  00 00 00 00 00 00 00 86  |................|
00000100  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000110  7d 02 be 0f 95 66 bd 56  c5 00 00 00 00 00 00 00  |}....f.V........|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000140  40 c1 00 00 00 c0 c1 00  00 00 40 c1 00 00 00 40  ***@***.***@....@|
00000150  c1 00 00 00 40 c1 00 00  00 c0 c0 00 00 00 c0 c0  ***@***.***|
00000160  00 00 00 c0 c0 00 00 00  c0 c5 00 00 00 40 c5 00  ***@***.***|
00000170  00 00 40 c4 00 00 00 c0  c3 00 00 00 40 c3 00 00  ***@***.***@...|
00000180  00 c0 c2 00 00 00 c0 c2  00 00 00 c0 c2 00 00 00  |................|
00000190  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001b0  00 00 00 00 00 85 85 86  00 00 00 00 00 00 00 00  |................|
000001c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000003a0  00 00 00 00 00 00 00 00  00 00 00 00 07 03 f1 03  |................|
000003b0  f1 03 f1 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000003c0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00004da0  00 00 00 00 00 00 00 00                           |........|
00004da8

namidairo, 2021-04-15:
Hmm, the mac address check is only for 00:00:00:00:00:00 or multicast bit being set... 

There is a perfectly placed 00:00:00:00:00:00 right in the offset where the mac address would normally be, and the 15,76 further in...

Maybe mt76_efuse_init has loaded it all from 0x0 instead of 0xA? I don't see any explicit check for the model like the eeprom init does.

Solution would be to call mt7615_check_eeprom in the efuse init + iterate through about 0xF bytes to try get a valid header for calibration?

zhangsd20, 2021-04-25:
Hi，

   mt7615 supports regular 20/40/80/160M bandwidth,does it support 5/10M bandwidth?

source code

===============

  staticc int  mt7615_mcu_uni_add_bss (......)
{
        switch (chandef->width) {
         ...

        case NL80211_CHAN_WIDTH_5:
                rlm_req.rlm.bw = CMD_CBW_5MHZ;
                break;
        case NL80211_CHAN_WIDTH_10:
                rlm_req.rlm.bw = CMD_CBW_10MHZ;
                break;
      ...

        }

}

I did not find a way to set it in the code. （In ath9k, “chanbw” can be set to set 5/10M bandwidth.）

ryderlee1110, 2021-04-26:
hardware supports 5M/10M, but mt76 doesn't add it yet. Is your issue resolved?

zhangsd20, 2021-04-26:
Thaank you very much！

I am not sure if there is a problem with the mt7615 card I bought.

I looked at the opt data of the two cards, there are only minor differences,

but it does not seem to be the mac address.

If I want to support 5M/10M, does the relevant firmware also need to be added accordingly?

If I only modify the mt75 source code, I can try it.

Do I need to try to modify it? please tell me.  Thanks!

ryderlee1110, 2021-04-26:
You just need to handle 5M/10M path in driver.

zhangsd20, 2021-04-28:
I added a chanbw file to mt7615, and I used the instrument to check that the actual bandwidth is indeed 5 /10M. But the test throughput of the two mt7615 is very low, even 0

zhangsd20, 2021-04-28:
The connection rate is always displayed Bit Rate: 6.5 MBit/s

vogelfreiheit, 2022-05-05:
> You just need to handle 5M/10M path in driver. I think both tickets can be closed since I don't see any driver issues.

I opened a specific ticket about this:
https://github.com/openwrt/mt76/issues/667

GregorioBrandani, 2025-06-03:
I also stumbled into this thread looking for an answer. I'm about to hard code 10MHz into the MT7925 driver (I know this thread was originally about the MT7615) and will report back. More generally, it would be awesome if someone can generate a list of all the MT chips/drivers and whether or not they currently support 5/10Mhz channel widths.