openwrt/mt76 #525: mt7603e max tx power is only 8 dBm on HiWiFi HC5962

LGA1150, 2021-04-12:
It's fine on MI Router 3G's mt7603e, so I think the driver misinterpreted the calibration data.

[Content of `factory` partition](https://github.com/openwrt/mt76/files/6294283/mtd2.zip)
Please take a look.

Colani1200, 2021-04-15:
Can you post a dmesg?

LGA1150, 2021-04-26:
dmesg seems normal

```
[   17.185880] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=23
[   17.191408] pci 0000:00:00.0: enabling device (0006 -> 0007)
[   17.197143] mt7603e 0000:01:00.0: enabling device (0000 -> 0002)
[   17.203426] mt7603e 0000:01:00.0: ASIC revision: 76030010
[   17.829314] mt7603e 0000:01:00.0: Firmware Version: ap_pcie
[   17.834920] mt7603e 0000:01:00.0: Build Time: 20160107100755
[   17.867991] mt7603e 0000:01:00.0: firmware init done
[   18.047889] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
```

Colani1200, 2021-04-26:
That's is just an excerpt, the interesting parts are missing. Can you post the full dmesg?
Or if you don't want to, check if you find entries like this in it:
`Bad eraseblock`
and
`generated random MAC address`

LGA1150, 2021-04-26:
Not at all

asiaon123, 2021-08-07:
I have the same issue

asiaon123, 2021-08-07:
[image]

asiaon123, 2021-11-24:
> That's is just an excerpt, the interesting parts are missing. Can you post the full dmesg? Or if you don't want to, check if you find entries like this in it: `Bad eraseblock` and `generated random MAC address`

Does this means it has broken eeprom?

asiaon123, 2021-11-26:
[image]

And this is my eeprom

asiaon123, 2021-11-26:
@nbd168 Hi, can you help fix out this problem?

[image]

This is my eeprom

dfateyev, 2022-11-29:
I have the same issue on the latest master, with kernel-5.15.
Interesting, that the 22.03-SNAPSHOT with kernel-5.10 doesn't produce this error.