openwrt/mt76 #528: 802.11s and AP on mt7615e de-authenticate clients in loop

amaumene, 2021-04-14:
Hi,

Follow up on my previous [issue](https://github.com/openwrt/mt76/issues/522). While the 802.11s link of mt7615e is stable I cannot use it as AP at the same time. Running OpenWrt d4f2c2914a with this configuration:
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option cell_density '0'
	option htmode 'HT20'
	option channel '1'
	option country 'SG'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'plop'
	option key 'my_very_secure_password'
	option encryption 'psk2'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'VHT80'
	option cell_density '0'
	option channel '36'
	option country 'SG'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'plop'
	option key 's my_very_secure_password'
	option encryption 'psk2'

config wifi-iface 'wifinet2'
	option device 'radio1'
	option mode 'mesh'
	option encryption 'sae'
	option mesh_id 'mesh_plop'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
	option key 'my_very_secure_key'
	option network 'lan'
```

If I connect to radio1 on any AP I will see my client connecting/disconnecting in loop. And on the router itself I can see:
```
Wed Apr 14 11:59:29 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: authenticated
Wed Apr 14 11:59:29 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: associated (aid 1)
Wed Apr 14 11:59:29 2021 daemon.notice hostapd: wlan1-1: AP-STA-CONNECTED 1e:04:26:56:b9:92
Wed Apr 14 11:59:29 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 RADIUS: starting accounting session 16E7D7CD6C1944C4
Wed Apr 14 11:59:29 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 WPA: pairwise key handshake completed (RSN)
Wed Apr 14 11:59:30 2021 daemon.notice hostapd: wlan1-1: AP-STA-DISCONNECTED 1e:04:26:56:b9:92
Wed Apr 14 11:59:30 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: disassociated
Wed Apr 14 11:59:31 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 11:59:32 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: authenticated
Wed Apr 14 11:59:32 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: associated (aid 1)
Wed Apr 14 11:59:32 2021 daemon.notice hostapd: wlan1-1: AP-STA-CONNECTED 1e:04:26:56:b9:92
Wed Apr 14 11:59:32 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 RADIUS: starting accounting session B5F439640ABCF72A
Wed Apr 14 11:59:32 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 WPA: pairwise key handshake completed (RSN)
Wed Apr 14 11:59:34 2021 daemon.notice hostapd: wlan1-1: AP-STA-DISCONNECTED 1e:04:26:56:b9:92
Wed Apr 14 11:59:34 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: disassociated
Wed Apr 14 11:59:35 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 11:59:37 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: authenticated
Wed Apr 14 11:59:37 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: associated (aid 1)
Wed Apr 14 11:59:37 2021 daemon.notice hostapd: wlan1-1: AP-STA-CONNECTED 1e:04:26:56:b9:92
Wed Apr 14 11:59:37 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 RADIUS: starting accounting session A9C7D9C076E2881C
Wed Apr 14 11:59:37 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 WPA: pairwise key handshake completed (RSN)
Wed Apr 14 11:59:39 2021 daemon.notice hostapd: wlan1-1: AP-STA-DISCONNECTED 1e:04:26:56:b9:92
Wed Apr 14 11:59:39 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: disassociated
Wed Apr 14 11:59:40 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 12:00:05 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: authenticated
Wed Apr 14 12:00:05 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: associated (aid 1)
Wed Apr 14 12:00:05 2021 daemon.notice hostapd: wlan1-1: AP-STA-CONNECTED 1e:04:26:56:b9:92
Wed Apr 14 12:00:05 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 RADIUS: starting accounting session 071C68756BE096FF
Wed Apr 14 12:00:05 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 WPA: pairwise key handshake completed (RSN)
Wed Apr 14 12:00:07 2021 daemon.notice hostapd: wlan1-1: AP-STA-DISCONNECTED 1e:04:26:56:b9:92
Wed Apr 14 12:00:07 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: disassociated
Wed Apr 14 12:00:08 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 12:00:10 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: authenticated
Wed Apr 14 12:00:10 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: associated (aid 1)
Wed Apr 14 12:00:10 2021 daemon.notice hostapd: wlan1-1: AP-STA-CONNECTED 1e:04:26:56:b9:92
Wed Apr 14 12:00:10 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 RADIUS: starting accounting session 9748F719F04C9888
Wed Apr 14 12:00:10 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 WPA: pairwise key handshake completed (RSN)
Wed Apr 14 12:00:12 2021 daemon.notice hostapd: wlan1-1: AP-STA-DISCONNECTED 1e:04:26:56:b9:92
Wed Apr 14 12:00:12 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: disassociated
Wed Apr 14 12:00:13 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Apr 14 12:00:18 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: authenticated
Wed Apr 14 12:00:18 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: associated (aid 1)
Wed Apr 14 12:00:18 2021 daemon.notice hostapd: wlan1-1: AP-STA-CONNECTED 1e:04:26:56:b9:92
Wed Apr 14 12:00:18 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 RADIUS: starting accounting session 8916434659BD0CDB
Wed Apr 14 12:00:18 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 WPA: pairwise key handshake completed (RSN)
Wed Apr 14 12:00:19 2021 daemon.notice hostapd: wlan1-1: AP-STA-DISCONNECTED 1e:04:26:56:b9:92
Wed Apr 14 12:00:19 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: disassociated
Wed Apr 14 12:00:20 2021 daemon.info hostapd: wlan1-1: STA 1e:04:26:56:b9:92 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

If I then set radio1 to ```option disabled '1'``` my clients stay connect on mt7603e and it's nice and stable. If I run only 1 AP clients can connect on mt7615e. As mentioned in https://github.com/openwrt/mt76/issues/522 this configuration was running stable with OpenWrt 024c81adb8. I will try to narrow it down to a specific commit.

amaumene, 2021-04-16:
Hi,

I've reverted 00628061b54692210480cb30a9933c0b52eab6d6 out of 42a2dddb706b47d6a6c36698aeb594bd0d20f92b and my mt7615e can be used for 802.1s and STA at the same time. I need to do a bit of testing but I don't experience the death in loop anymore.

ryderlee1110, 2021-04-16:
I don't see how this causes deauth issue. Just a capability change for dbdc second phy

amaumene, 2021-04-19:
I have no idea either but since it was working previously and I'm not familiar with the code base I realised the only thing I could do was to revert patches related to mt7615 one by one, and without this one it works again.

Sorry if I cannot provide an explanation.

ryderlee1110, 2021-04-19:
Are you using DBDC?

amaumene, 2021-04-20:
Hi,

Yes it seems similar. It also seems similar to https://github.com/openwrt/mt76/issues/474 and https://github.com/openwrt/mt76/issues/483 except setting:
```
	option phy 'phyX'
	option serialize '1'
	option country 'country_code'
```
doesn't fix it (I've just tried).

ryderlee1110, 2021-04-20:
I didn't get the point. Those tickets you posted were just interface order issue.

ryderlee1110, 2021-04-20:
Are you using dbdc?

amaumene, 2021-04-20:
Hi,

I was just saying that the symptoms in these issues were the same. I didn't enable anything related to DBDC manually. FWIW:
```
root@route2:~# cat /sys/kernel/debug/ieee80211/phy1/mt76/dbdc
0
```

amaumene, 2021-04-21:
Hi,

I tried to revert the mentioned commit but it didn't fix this issue.

ryderlee1110, 2021-04-21:
Not sure which one. But this looks like mac80211 issue to me.

amaumene, 2021-04-21:
But if it were mac80211 I should face this issue with both mt7615 and mt7603 correct? and I face it only with mt7615.

ryderlee1110, 2021-04-21:
I don't know at the moment. The above ticket showed that this also happened on mt7603/mt7612. But if you're just using the 7615N then my changes should be harmless.

ptpt52, 2021-04-22:
I try 80211s, seems not work
```
iw dev wlan1 station dump
failed to parse nested attributes!
```

ptpt52, 2021-04-22:
```
AP1(7915)---AP2(7613)---internet
```

ptpt52, 2021-04-22:
should I use the `STA offload firmware` on mt7613?

ryderlee1110, 2021-04-22:
@ptpt52 7915 works, so let's don't get 7613 involved here.

ptpt52, 2021-04-22:
on 7915, 2.4G
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option phy 'phy0'
	option htmode 'HT20'
	option country 'CN'
	option txpower '20'
	option cell_density '0'
	option channel '6'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option phy 'phy1'
	option htmode 'VHT80'
	option country 'CN'
	option cell_density '0'
	option txpower '19'
	option channel '52'

config wifi-iface 'wifinet1'
	option device 'radio0'
	option mode 'mesh'
	option encryption 'none'
	option mesh_id 'x-wrt-mesh-id'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
	option network 'lan'

```

on 7603, 2.4G
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option country 'CN'
	option txpower '20'
	option htmode 'HT40'
	option noscan '1'
	option beacon_int '100'
	option rts '2347'
	option cell_density '0'
	option channel '6'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'
	option country 'CN'
	option cell_density '0'
	option txpower '19'
	option beacon_int '100'
	option rts '2347'
	option channel '52'

config wifi-iface 'wifinet1'
	option device 'radio0'
	option mode 'mesh'
	option encryption 'none'
	option mesh_id 'x-wrt-mesh-id'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
	option network 'lan'
```

I have to fix channel on 6, to get them connected,
but network ping fail

amaumene, 2021-04-22:
> I try 80211s, seems not work
> 
> ```
> iw dev wlan1 station dump
> failed to parse nested attributes!
> ```

Hi,

I don't think station dump ever worked for me. But 802.11s works fine:
```
root@route3:~# iwinfo wlan1 assoclist
28:D1:27:11:6B:EB  -68 dBm / -91 dBm (SNR 23)  50 ms ago
	RX: 780.0 MBit/s, VHT-MCS 5, 80MHz, VHT-NSS 3    686524 Pkts.
	TX: 780.0 MBit/s, VHT-MCS 4, 80MHz, VHT-NSS 4    229017 Pkts.
	expected throughput: 439.9 MBit/s

root@route3:~# iw dev wlan1 mpath dump
DEST ADDR         NEXT HOP          IFACE	SN	METRIC	QLEN	EXPTIME		DTIM	DRET	FLAGS	HOP_COUNT	PATH_CHANGE
28:d1:27:11:6b:eb 28:d1:27:11:6b:eb wlan1	36651	19	0	590	100	0	0x15	1	1
```

802.11s was broken until https://github.com/openwrt/mt76/issues/522 but the issue now is that if I have mesh and ap working on mt7615 at the same time it infinite loop on the DEAUTH message showed above and my clients keep connecting/disconnecting. It seems that reverting a patch I mentioned earlier about DBDC it works fine again, but it's probably a side effect of the revert that is fixing the issue.

ptpt52, 2021-04-22:
I drop mt7915, and use `(AP 7603) -- (AP 7603)`
mesh connected and works, ping ok
bug I got some log:
```
[  124.759015] ------------[ cut here ]------------
[  124.763817] WARNING: CPU: 2 PID: 204 at net/core/flow_dissector.c:958 __skb_flow_dissect+0x2d4/0x16c8
[  124.773167] Modules linked in: xt_connlimit nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_CT nft_redir nft_ct nf_nat nft
[  124.773420]  ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnc
[  124.894269] CPU: 2 PID: 204 Comm: kworker/u9:0 Not tainted 5.4.113 #0
[  124.900747] Workqueue: napi_workq napi_workfn
[  124.905119] Stack : 805e49bc 87e67a3c 80660000 806a0000 87ece380 805f5934 803cb218 00000009
[  124.913469]         87f6f19c 85627818 00000000 80072fdc 00000002 00000001 87e679f8 8de802f8
[  124.921803]         00000000 00000000 00000000 00000000 716b726f 00000112 6f775f69 6e666b72
[  124.930133]         00000000 00000006 00000000 000dbe8b 00000000 806d0000 00000000 803cb218
[  124.938463]         00000009 87f6f19c 85627818 00000000 00000000 803027a4 00000008 80810008
[  124.946793]         ...
[  124.949232] Call Trace:
[  124.951699] [<8000b61c>] show_stack+0x30/0x100
[  124.956161] [<8053e9c8>] dump_stack+0xa4/0xdc
[  124.960520] [<8002bcd0>] __warn+0xc0/0x10c
[  124.964610] [<8002bd78>] warn_slowpath_fmt+0x5c/0xac
[  124.969574] [<803cb218>] __skb_flow_dissect+0x2d4/0x16c8
[  124.974870] [<803cc8bc>] __skb_get_hash+0x7c/0x258
[  124.979814] [<86532df4>] ieee80211_unreserve_tid+0x720/0xbcc [mac80211]
[  124.986666] ---[ end trace 7bbf0d9a083d481d ]---
```

ptpt52, 2021-04-22:
mesh speed is very low on mt7603:
```
iperf3 -c 192.168.18.1  -R
Connecting to host 192.168.18.1, port 5201
Reverse mode, remote host 192.168.18.1 is sending
[  5] local 192.168.18.170 port 58570 connected to 192.168.18.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  43.8 KBytes   358 Kbits/sec                  
[  5]   1.00-2.00   sec   102 KBytes   834 Kbits/sec                  
[  5]   2.00-3.00   sec  77.8 KBytes   638 Kbits/sec                  
[  5]   3.00-4.00   sec  73.5 KBytes   602 Kbits/sec                  
[  5]   4.00-5.00   sec  67.9 KBytes   556 Kbits/sec                  
[  5]   5.00-6.00   sec  84.8 KBytes   695 Kbits/sec                  
[  5]   6.00-7.00   sec  63.6 KBytes   521 Kbits/sec                  
[  5]   7.00-8.00   sec  43.8 KBytes   359 Kbits/sec                  
[  5]   8.00-9.00   sec  46.7 KBytes   382 Kbits/sec   
```

ptpt52, 2021-04-22:
Then I try the 
```
client---(AP ssid+mesh)---(AP mesh)---internet
```
connection ok
no `de-authenticate clients in loop` issue
chip is mt7603

ryderlee1110, 2021-04-22:
@ptpt52 did you test 7915 or 7615 only?

ptpt52, 2021-04-22:
I have no more than one unit of 7615/7915, I will try 7615 + 7612 latter

ptpt52, 2021-04-22:
Tested `(AP 7615e)--(AP 7612e)`
connection ok
no de-authenticate clients in loop issue

speed 100Mbps

amaumene, 2021-04-26:
I found a workaround, if I let the router boot, then I run ```wifi down; wifi up``` clients can connect fine and roam between both AP on 5Ghz (with mesh and ap mode). This has been verified with OpenWrt f4473baf6e and no patches.

ryderlee1110, 2021-04-27:
station dump fixup - https://git.kernel.org/pub/scm/linux/kernel/git/jberg/iw.git/commit/?id=0ba98b9041e4e0f41ee72fa0fce9c43f8c6f2db3

widewing, 2021-10-17:
I have exactly same issue, on same device (Redmi ac2100), on release 21.02.0. as @amaumene mentioned, simply `wifi down;wifi up` can workaround it.

amaumene, 2021-10-18:
Hi,

Same for me with 21.02.0. I don't need this workaround anymore with the latest snapshot fwiw.

widewing, 2021-10-18:
good to know! waiting for next release

Suxsem, 2021-11-28:
Same issue with 21.02.1, any news?

mvarchdev, 2023-09-18:
Some news? Incoutered this issue on:

Xiaomi Redmi Router AC2100
--
Architecture | MediaTek MT7621 ver:1 eco:3
Target Platform | ramips/mt7621
Firmware Version | OpenWrt 22.03.5 r20134-5f15225c1e / LuCI openwrt-22.03 branch git-23.093.57104-ce20b4a
Kernel Version | 5.10.176

Djfe, 2024-02-09:
@amaumene shall this be closed? :)

amaumene, 2024-02-09:
Definitely. I don't own this router and don't use mesh anymore ;)

closed 2024-02-09 by amaumene