openwrt/mt76 #530: Low transmission power on MT7610EN

ThatMishakov, 2021-04-17:
Hello. I have been trying to get my router (Archer D2, very similar to Archer vr200v, but with newer SoC) with ARX368 SoC and MT710EN 5Ghz wifi supported by openwrt, however I have discovered that after seemengly initializing and reporting to be working normally, the radio would not let me set the transmission power higher that 10-11dbm and my phone would now discover the network. What could be the issue? It seems to be very similar to https://github.com/openwrt/mt76/issues/216, but that was closed.

[iw phy0 info output](https://pastebin.com/FN3ciQ9M)

```
root@OpenWrt:/# cat /sys/kernel/debug/ieee80211/phy0/mt76/txpower
Target power: 13
     Delta:  0  0
```
```
root@OpenWrt:/# cat /sys/kernel/debug/ieee80211/phy0/mt76/rate_txpower
       CCK: -1 -1 -1 -1
      OFDM:  6  6  6  6  3  3 -1 -1
      STBC: -1 -1 -1 -1 -1 -1 -1 -1 -1 -1
        HT:  6  6  6  6  3  3 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1
       VHT:  6  6  6  6  3  3 -1 -1 -2 -2

```
[radio partition.zip](https://github.com/openwrt/mt76/files/6329088/radio.partition.zip)

sanitariu, 2021-05-31:
I have Archer C20 version 5. Again MT7610EN and tx-power is 25mw. Changing countries does not help.
Seems like its reading eeprom from the wrong place. Also 5G is very unstable and disconnects each 5-10 minutes.

GrzybDev, 2021-06-18:
Same issue on RE200 v3 and v4

ArtemiisFowl, 2021-10-13:
Same Problem on the vr200v. Seems like tplink did something different to other vendors with 6710e