openwrt/mt76 #531: MT7615DN max tx power issue

MeIsReallyBa, 2021-04-17:
Phicomm K2P，MT7615DN+external pa/lna, max tx power is only 9dbm.
JCG Y2,MT7615DN+internal pa/lna ,max tx power could reach 23dbm with the same openwrt firmware.
[image]
K2P eeprom:
[k2peeprom.zip](https://github.com/openwrt/mt76/files/6329300/k2peeprom.zip)

kar200, 2021-04-20:
I have the same issue with DIR-853-A3 - DBDC (not on openwrt official yet)

I get 20 on the 2g and only 8 on 5g. Someone suggested to try a different country but after trying a couple it did not change.

````
wlan0     ESSID: "Openwrt2G"
          Access Point: F4:8C:EB:A8:F0:66
          Mode: Master  Channel: 3 (2.422 GHz)
          Center Channel 1: 3 2: unknown
          Tx-Power: 20 dBm  Link Quality: 47/70
          Signal: -63 dBm  Noise: -91 dBm
          Bit Rate: 35.7 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgnac
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "Openwrt5G"
          Access Point: F4:8C:EB:A8:F0:66
          Mode: Master  Channel: 40 (5.200 GHz)
          Center Channel 1: 42 2: unknown
          Tx-Power: 8 dBm  Link Quality: 54/70
          Signal: -56 dBm  Noise: -86 dBm
          Bit Rate: 487.5 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7615 7615:14C3 [MediaTek MT7615E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
````

ptpt52, 2021-05-07:
Whether the actual working power is normal, please use the mobile phone Wi-Fi scanner to scan

ptpt52, 2021-05-07:
eeprom offset 0x73 byte is `0x00`, this is the target tx power limit byte

ptpt52, 2021-05-07:
if you want to hack:
```
diff --git a/mt7615/init.c b/mt7615/init.c
index 3f8c84c4..59458bae 100644
--- a/mt7615/init.c
+++ b/mt7615/init.c
@@ -277,6 +277,9 @@ void mt7615_init_txpower(struct mt7615_dev *dev,
                        index = mt7615_eeprom_get_target_power_index(dev, chan, j);
                        if (index < 0)
                                continue;
+                       if (eep[index] == 0) {
+                               eep[index] = 0x14; //20dB
+                       }
 
                        target_power = max(target_power, eep[index]);
                }
```

MeIsReallyBa, 2021-06-17:
> if you want to hack:
> 
> ```
> diff --git a/mt7615/init.c b/mt7615/init.c
> index 3f8c84c4..a6d0b04a 100644
> --- a/mt7615/init.c
> +++ b/mt7615/init.c
> @@ -277,6 +277,9 @@ void mt7615_init_txpower(struct mt7615_dev *dev,
>                         index = mt7615_eeprom_get_target_power_index(dev, chan, j);
>                         if (index < 0)
>                                 continue;
> +                       if (eep[index] == 0 && mt7615_ext_pa_enabled(dev, chan->band)) {
> +                               eep[index] = 0x14; //20dB
> +                       }
>  
>                         target_power = max(target_power, eep[index]);
>                 }
> ```

it works.thanks.

closed 2021-06-17 by MeIsReallyBa

badrr19, 2024-03-20:
> it works.thanks.

can you tell me how to do it  :) Iam new in this ....

badrr19, 2024-03-21:
> how
can you tell me where to do it ? using ssh?