openwrt/mt76 #532: 21.02 branch, seeing crashes and high Inerrupt errors

JFtico, 2021-04-19:
Made a build on 4/15 from the 21.02 branch to test on a ZBT-WE3526 (MT7621 / 7612 / 7603 )

Many nice improvements over the 19.07.x releases, such as much stronger signals, less latency on both bands.

However, still seeing recurrences of 7603 crashes, and rapidly increasing /proc/interrupt errors.

Here are the crash logs for two instances:

JFtico, 2021-04-22:
After three days, still seeing crashes in 7603 driver

`[242220.609306] ------------[ cut here ]------------
[242220.614065] WARNING: CPU: 1 PID: 1487 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2021-04-11-bf45b30d/mt7603/mac.c:206 mt7603_filter_tx+0x16c/0x174 [mt7603e]
[242220.628927] Modules linked in: pppoe ppp_async pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY slhc sch_cake nf_reject_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conncount iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat act_connmark nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci
[242220.629224]  xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[242220.723604] CPU: 1 PID: 1487 Comm: hostapd Not tainted 5.4.111 #0
[242220.729754] Stack : 00000001 8007d738 806c0000 806beb24 80720000 806beaec 806bdc40 8ec059cc
[242220.738166]         80860000 8fe99a28 80708d03 80651b4c 00000001 00000001 8ec05970 10ec5cf9
[242220.746574]         00000000 00000000 808a0000 00000000 30000000 0000015d 00000018 312e342e
[242220.754983]         00000000 00002f13 00000000 68203a6d 80000000 80720000 00000000 8ee2452c
[242220.763392]         00000009 00000008 00000008 8ee30000 00000008 8034e8dc 00000004 80860004
[242220.771801]         ...
[242220.774321] Call Trace:
[242220.776869] [<8000b64c>] show_stack+0x30/0x100
[242220.781398] [<805950b8>] dump_stack+0xa4/0xdc
[242220.785839] [<8002bfe8>] __warn+0xc0/0x10c
[242220.790008] [<8002c090>] warn_slowpath_fmt+0x5c/0xac
[242220.795058] [<8ee2452c>] mt7603_filter_tx+0x16c/0x174 [mt7603e]
[242220.801048] [<8ee219d4>] mt7603_sta_remove+0x94/0xfc [mt7603e]
[242220.806975] [<8ee12d44>] __mt76_sta_remove+0x78/0xec [mt76]
[242220.812624] [<8ee12e90>] mt76_sta_state+0xd8/0x1e4 [mt76]
[242220.818233] [<8ef09ee4>] sta_set_sinfo+0xfac/0x102c [mac80211]
[242220.824398] ---[ end trace f8497b66a9028e1b ]---
`

Mushoz, 2021-05-17:
I am on the 21.02 RC1 version with a mt7621 device, MT7603EN and MT7612EN WiFi chips, with similar stack traces. Posted below for reference. First a mac80211 crash:

```
[922688.827778] ------------[ cut here ]------------
[922688.837246] WARNING: CPU: 2 PID: 20 at backports-5.10.16-1/net/mac80211/tx.c:4611 0x864b2470 [mac80211@9be18d50+0x7d2e0]
[922688.859148] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard pppox ppp_generic nf_nat_pptp nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY ts_kmp ts_fsm ts_bm slhc sch_cake poly1305_mips nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libblake2s_generic iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat chacha_mips br_netfilter sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit
[922688.859343]  act_mirred ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel kpp leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[922689.135708] CPU: 2 PID: 20 Comm: ksoftirqd/2 Tainted: G    B             5.4.111 #0
[922689.151125] Stack : 00000000 80840000 00000000 8007d6e0 00000000 00000000 00000000 00000000
[922689.167945]         00000000 00000000 00000000 00000000 00000000 00000001 87c91b40 5fc2ce17
[922689.184762]         87c91bd8 00000000 00000000 00000000 00000038 805e1804 00000008 42202020
[922689.201579]         00000000 0000b353 00000000 20202020 00000000 87c91b20 00000000 864b2470
[922689.218397]         00000009 00000000 86f584a0 8515fc30 00000018 80359e2c 00000008 80810008
[922689.235219]         ...
[922689.240271] Call Trace:
[922689.240293] [<8007d6e0>] 0x8007d6e0
[922689.252439] [<805e1804>] 0x805e1804
[922689.259594] [<864b2470>] 0x864b2470 [mac80211@9be18d50+0x7d2e0]
[922689.271612] [<80359e2c>] 0x80359e2c
[922689.278740] [<8000b05c>] 0x8000b05c
[922689.285849] [<8000b064>] 0x8000b064
[922689.292961] [<805c6f9c>] 0x805c6f9c
[922689.300082] [<8007a19c>] 0x8007a19c
[922689.307201] [<8002bfe8>] 0x8002bfe8
[922689.314365] [<864b2470>] 0x864b2470 [mac80211@9be18d50+0x7d2e0]
[922689.326336] [<8002c090>] 0x8002c090
[922689.333521] [<864b2470>] 0x864b2470 [mac80211@9be18d50+0x7d2e0]
[922689.345488] [<805e1824>] 0x805e1824
[922689.352641] [<864b24c0>] 0x864b24c0 [mac80211@9be18d50+0x7d2e0]
[922689.364634] [<864b7360>] 0x864b7360 [mac80211@9be18d50+0x7d2e0]
[922689.376626] [<80089dd4>] 0x80089dd4
[922689.383763] [<80010670>] 0x80010670
[922689.390900] [<864b73c0>] 0x864b73c0 [mac80211@9be18d50+0x7d2e0]
[922689.402864] [<800544a4>] 0x800544a4
[922689.410065] [<865781d0>] 0x865781d0 [mt76x02_lib@75106930+0x9dc0]
[922689.422377] [<8005510c>] 0x8005510c
[922689.429529] [<864bd810>] 0x864bd810 [mac80211@9be18d50+0x7d2e0]
[922689.441524] [<8657321c>] 0x8657321c [mt76x02_lib@75106930+0x9dc0]
[922689.453842] [<86578190>] 0x86578190 [mt76x02_lib@75106930+0x9dc0]
[922689.466225] [<864bd90c>] 0x864bd90c [mac80211@9be18d50+0x7d2e0]
[922689.478230] [<86575360>] 0x86575360 [mt76x02_lib@75106930+0x9dc0]
[922689.490588] [<86574e4c>] 0x86574e4c [mt76x02_lib@75106930+0x9dc0]
[922689.502907] [<80030220>] 0x80030220
[922689.510108] [<800301a8>] 0x800301a8
[922689.517232] [<805e7d1c>] 0x805e7d1c
[922689.524339] [<8004fed8>] 0x8004fed8
[922689.531447] [<8004fed8>] 0x8004fed8
[922689.538579] [<80030478>] 0x80030478
[922689.545706] [<805e2d88>] 0x805e2d88
[922689.552814] [<80050060>] 0x80050060
[922689.559925] [<8004b1cc>] 0x8004b1cc
[922689.567030] [<805e319c>] 0x805e319c
[922689.574152] [<8004b6dc>] 0x8004b6dc
[922689.581259] [<8004b59c>] 0x8004b59c
[922689.588395] [<8004b59c>] 0x8004b59c
[922689.595543] [<8004b59c>] 0x8004b59c
[922689.602683] [<80006718>] 0x80006718
[922689.609808] 
[922689.613709] ---[ end trace 52e7a6fe65a762b8 ]---
```

And secondly, a mt7603e crash:

```
[1089896.185299] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[1089896.196699] mt76x2e 0000:01:00.0: Build: 1
[1089896.205268] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[1089896.231996] mt76x2e 0000:01:00.0: Firmware running!
[1089896.248160] ieee80211 phy1: Hardware restart was requested
[1094976.568100] ------------[ cut here ]------------
[... 38 lines cut ...]
[1094977.255108] [<80496e54>] 0x80496e54
[1094977.262393] [<80497674>] 0x80497674
[1094977.269712] [<80172194>] 0x80172194
[1094977.277002] [<80497e60>] 0x80497e60
[1094977.284286] [<80496d70>] 0x80496d70
[1094977.291593] [<80172194>] 0x80172194
[1094977.298878] [<8005dcf8>] 0x8005dcf8
[1094977.306202] [<80497140>] 0x80497140
[1094977.313512] [<80178920>] 0x80178920
[... cut ...]
[1094977.335425] [<80097da4>] 0x80097da4
[1094977.342756] [<80407a48>] 0x80407a48
[1094977.350046] [<805e6cac>] 0x805e6cac
[1094977.357324] [<801d03c8>] 0x801d03c8
[1094977.364635] [<800975fc>] 0x800975fc
[1094977.371961] [<805e6d44>] 0x805e6d44
[1094977.379273] [<80408c68>] 0x80408c68
[1094977.386546] [<8007037c>] 0x8007037c
[1094977.393825] [<8040612c>] 0x8040612c
[1094977.401117] [<80179cc0>] 0x80179cc0
[1094977.408464] [<80014558>] 0x80014558
[1094977.415748] 
[1094977.419767] ---[ end trace 52e7a6fe65a762b9 ]---
```

easyteacher, 2021-09-15:
Do you set htmode to Legacy/None?