openwrt/mt76 #534: mt7915 dbdc, AP + STA hostapd crash loop

ptpt52, 2021-04-25:
```

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option htmode 'HT20'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option disabled '0'
	option country 'CN'
	option channel 'auto'
	option txpower '20'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option htmode 'VHT80'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
	option country 'CN'
	option channel 'auto'
	option cell_density '0'
	option txpower '19'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'X-WRT_6859'
	option encryption 'psk2'
	option skip_inactivity_poll '1'
	option wpa_group_rekey '0'
	option wpa_pair_rekey '0'
	option wpa_master_rekey '0'
	option disassoc_low_ack '0'
	option key '88888888'

config wifi-iface 'wifinet1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'X-WRT_6859'
	option encryption 'psk2'
	option skip_inactivity_poll '1'
	option wpa_group_rekey '0'
	option wpa_pair_rekey '0'
	option wpa_master_rekey '0'
	option disassoc_low_ack '0'
	option key '88888888'

config wifi-iface 'wifinet2'
	option device 'radio1'
	option mode 'sta'
	option network 'wwan'
	option ssid 'X-WRT_5G'
	option encryption 'psk2'
	option key '88888888'

```

dmesg:
```
dmesg | grep wpad
[  174.196131] epc = 5566836d in wpad[55598000+104000]
[  174.201036] ra  = 556684c3 in wpad[55598000+104000]
[  190.797398] epc = 5570736d in wpad[55637000+104000]
[  190.802390] ra  = 557074c3 in wpad[55637000+104000]
[  207.426912] epc = 556dd36d in wpad[5560d000+104000]
[  207.431852] ra  = 556dd4c3 in wpad[5560d000+104000]
[  224.096612] epc = 556e336d in wpad[55613000+104000]
[  224.101642] ra  = 556e34c3 in wpad[55613000+104000]
[  384.187815] epc = 5565736d in wpad[55587000+104000]
[  384.192771] ra  = 556574c3 in wpad[55587000+104000]
[  455.833616] epc = 5566b36d in wpad[5559b000+104000]
[  455.838552] ra  = 5566b4c3 in wpad[5559b000+104000]
[  472.459355] epc = 5568536d in wpad[555b5000+104000]
[  472.464327] ra  = 556854c3 in wpad[555b5000+104000]
[  489.068988] epc = 5565536d in wpad[55585000+104000]
[  489.074016] ra  = 556554c3 in wpad[55585000+104000]
[  544.637079] epc = 5567a36d in wpad[555aa000+104000]
[  544.642052] ra  = 5567a4c3 in wpad[555aa000+104000]
```

ptpt52, 2021-04-25:
maybe somewhat similar to https://github.com/openwrt/mt76/issues/528

ptpt52, 2021-04-25:
gdb debug info:
```
gdb attach 18390
GNU gdb (GDB) 10.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "mipsel-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
attach: No such file or directory.
Attaching to process 18390
Reading symbols from /usr/sbin/wpad...
Reading symbols from /lib/ld-musl-mipsel-sf.so.1...
(No debugging symbols found in /lib/ld-musl-mipsel-sf.so.1)

warning: Unable to find dynamic linker breakpoint function.
GDB will be unable to debug shared library initializers
and track explicitly loaded dynamic code.
0x77d7fb48 in ?? (
warning: GDB can't find the start of the function at 0x77d7fb48.

    GDB is unable to find the start of the function at 0x77d7fb48
and thus can't determine the size of that function's stack frame.
This means that GDB may be unable to access that stack frame, or
the frames below it.
    This problem is most likely caused by an invalid program counter or
stack pointer.
    However, if you think GDB should simply search farther back
from 0x77d7fb48 for code which looks like the beginning of a
function, you can increase the range of the search using the `set
heuristic-fence-post' command.
)
(gdb) c
Continuing.

Program received signal SIGSEGV, Segmentation fault.
0x5565b36d in hostapd_reload_config ()
(gdb) bt
#0  0x5565b36d in hostapd_reload_config ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

ptpt52, 2021-04-25:
crash here:
```
hostapd-2020-06-08-5a8b3662/src/ap/hostapd.c

 238 int hostapd_reload_config(struct hostapd_iface *iface, int reconf)
............
 316         for (i = 0; i < iface->num_hw_features; i++) {
 317                 struct hostapd_hw_modes *mode = &iface->hw_features[i];
 318                 if (mode->mode == iface->conf->hw_mode) {
 319                         iface->current_mode = mode;
 320                         break;
 321                 }
 322         }
.......
```
`mode == NULL`

ptpt52, 2021-04-25:
the same case on mt7613 (AP + STA) no issue

nbd168, 2021-11-25:
Fixed in the latest version

closed 2021-11-25 by nbd168

Suxsem, 2021-11-28:
@nbd168 is it already fixed in 21.02.1?