openwrt/mt76 #535: EAP - wifi not connecting / keys exchange failure - local deauth request after exactly 10 seconds

farfade, 2021-05-02:
Hello,

Using 19.07.7 on a router newifi3 (newifi D2 - mt76 wifi driver), and EAP-TTLS, I'm facing the following with Windows clients (intel wifi chip with latest intel driver from january 2021)

```
    Sun May 2 11:50:47 2021 daemon.info hostapd: wlan1: STA 34:f3:9a:72:e7:55 IEEE 802.11: authenticated
    Sun May 2 11:50:47 2021 daemon.info hostapd: wlan1: STA 34:f3:9a:72:e7:55 IEEE 802.11: associated (aid 1)
    Sun May 2 11:50:47 2021 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 34:f3:9a:72:e7:55
    Sun May 2 11:50:47 2021 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
    Sun May 2 11:50:47 2021 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED 34:f3:9a:72:e7:55 off
    Sun May 2 11:50:47 2021 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-STARTED 34:f3:9a:72:e7:55
    Sun May 2 11:50:47 2021 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
    Sun May 2 11:50:47 2021 daemon.notice hostapd: wlan1: CTRL-EVENT-EAP-SUCCESS2 34:f3:9a:72:e7:55
    Sun May 2 11:50:57 2021 daemon.info hostapd: wlan1: STA 34:f3:9a:72:e7:55 IEEE 802.11: deauthenticated due to local deauth request
```

The windows 10 client speaks about a failure in dynamic key exchanges
The radius servers correctly sends an Accept:Accept message and MPPE Keys.

Whereas

    The same windows 10 client connects flawlessly on a second AP (a DD-WRT router serving the same SSID, with the same RADIUS backend and client account).
    Android 10 client can connect flawlessly
    raspbian buster client can connect if and only if I use the driver wext (the driver nl80211 cannot connect, it speaks about a failure in dynamic keys exchange too)

Isn't there a bug in the driver when sending the dynamic keys to the client, after having received it from the radius server ?

Thank you for reading !

Cheers

blogic, 2021-05-02:
what does your uci for the wifi-iface look like ? EAP is driver agnostic and handled solely by hostapd.

farfade, 2021-05-02:
Thank you @blogic for you interest.

Here is my uci

`config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option auth_secret '--'
        option auth_server '192.168.2.4'
        option ieee80211w '1'
        option ssid 'RM-5G'
        option encryption 'wpa2+ccmp'
        option wpa_disable_eapol_key_retries '1'

config wifi-iface 'wifinet2'
        option network 'lan'
        option ssid 'infra5G-2'
        option encryption 'psk2+ccmp'
        option device 'radio1'
        option mode 'ap'
        option key '--'
        option wpa_disable_eapol_key_retries '1'

config wifi-iface 'wifinet3'
        option network 'GUEST'
        option encryption 'psk2+ccmp'
        option device 'radio1'
        option mode 'ap'
        option isolate '1'
        option key '--'
        option ssid 'Nous-5G'`