openwrt/mt76 #539: Xiaomi r3p mt7615N low tx power 2ghz 5ghz

legale, 2021-05-12:
Yeasterday i've compiled firmware for target Xiaomi mi r3p from openwrt master branch. 

`Linux version 5.4.117 (ru@localhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16708-3c3c2c4f37)) #0 SMP Mon May 10 17:52:14 2021`
```
root@OpenWrt:/etc# iw list | grep -i mhz
                        Supported Channel Width: 160 MHz, 80+80 MHz
                        short GI (80 MHz)
                        short GI (160/80+80 MHz)
                        * 5180 MHz [36] (6.0 dBm)
                        * 5200 MHz [40] (6.0 dBm)
                        * 5220 MHz [44] (6.0 dBm)
                        * 5240 MHz [48] (6.0 dBm)
                        * 5260 MHz [52] (6.0 dBm)
                        * 5280 MHz [56] (6.0 dBm)
                        * 5300 MHz [60] (6.0 dBm)
                        * 5320 MHz [64] (6.0 dBm)
                        * 5500 MHz [100] (6.0 dBm)
                        * 5520 MHz [104] (6.0 dBm)
                        * 5540 MHz [108] (6.0 dBm)
                        * 5560 MHz [112] (6.0 dBm)
                        * 5580 MHz [116] (6.0 dBm)
                        * 5600 MHz [120] (6.0 dBm)
                        * 5620 MHz [124] (6.0 dBm)
                        * 5640 MHz [128] (6.0 dBm)
                        * 5660 MHz [132] (6.0 dBm)
                        * 5680 MHz [136] (6.0 dBm)
                        * 5700 MHz [140] (6.0 dBm)
                        * 5720 MHz [144] (6.0 dBm)
                        * 5745 MHz [149] (6.0 dBm)
                        * 5765 MHz [153] (6.0 dBm)
                        * 5785 MHz [157] (6.0 dBm)
                        * 5805 MHz [161] (6.0 dBm)
                        * 5825 MHz [165] (6.0 dBm)
                        * 5845 MHz [169] (disabled)
                        * 5865 MHz [173] (disabled)
                   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }
                 * short GI for 40 MHz
                        * 2412 MHz [1] (6.0 dBm)
                        * 2417 MHz [2] (6.0 dBm)
                        * 2422 MHz [3] (6.0 dBm)
                        * 2427 MHz [4] (6.0 dBm)
                        * 2432 MHz [5] (6.0 dBm)
                        * 2437 MHz [6] (6.0 dBm)
                        * 2442 MHz [7] (6.0 dBm)
                        * 2447 MHz [8] (6.0 dBm)
                        * 2452 MHz [9] (6.0 dBm)
                        * 2457 MHz [10] (6.0 dBm)
                        * 2462 MHz [11] (6.0 dBm)
                        * 2467 MHz [12] (6.0 dBm)
                        * 2472 MHz [13] (6.0 dBm)
                        * 2484 MHz [14] (disabled)
                   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }
                 * short GI for 40 MHz
```

Wireless configuration
```
root@OpenWrt:/etc# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option cell_density '0'
        option htmode 'HT40'
        option channel 'auto'
        option country 'PA'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'tobe20'
        option encryption 'sae-mixed'
        option key '88888888'
        option ieee80211w '1'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11a'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option channel 'auto'
        option cell_density '0'
        option htmode 'VHT40'
        option country 'PA'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'tobe50'
        option encryption 'sae-mixed'
        option key '88888888'
        option ieee80211w '1'
        
```
Max tx power is only 6dBm.
[eeprom.zip](https://github.com/openwrt/mt76/files/6465060/eeprom.zip)

ptpt52, 2021-05-13:
I think there is no problem with mt76 and no problem with eeprom. Can you try the firmware of x-wrt and give feedback?

https://downloads.x-wrt.com/rom/

legale, 2021-05-13:
> I think there is no problem with mt76 and no problem with eeprom. Can you try the firmware of x-wrt and give feedback?
> 
> https://downloads.x-wrt.com/rom/
> 

I will.

mouth4war, 2022-08-28:
Was this ever fixed?