openwrt/mt76 #544: 使用transmission下载pt时，导致page-fault

siwind, 2021-05-30:
使用transmission下载pt任务， 网络下载流量大点(下载速度超过2M以上),  transmission就会报错退出：

'''
Sun May 30 19:39:10 2021 user.notice transmission: Starting with 256538000 virt mem
Sun May 30 19:40:07 2021 kern.info kernel: [75901.470157] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 00000000
Sun May 30 19:40:07 2021 kern.info kernel: [75901.479315] epc = 00407747 in transmission-daemon[400000+3f000]
Sun May 30 19:40:07 2021 kern.info kernel: [75901.485253] ra  = 0040773f in transmission-daemon[400000+3f000]
Sun May 30 19:40:07 2021 daemon.info procd: Instance transmission::instance1 s in a crash loop 7 crashes, 58 seconds since last crash
'''

硬件: Newfi-D2,  mt7603/mt7612
软件： openwrt-19.07.7

'''
root@OpenWrt:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.7'
DISTRIB_REVISION='r11306-c4a6851c72'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.7 r11306-c4a6851c72'
DISTRIB_TAINTS='no-all'

root@OpenWrt:~# opkg list-installed | grep mt76
kmod-mt76-core - 4.14.221+2021-02-15-5c768dec-1
kmod-mt7603 - 4.14.221+2021-02-15-5c768dec-1
kmod-mt76x02-common - 4.14.221+2021-02-15-5c768dec-1
kmod-mt76x2 - 4.14.221+2021-02-15-5c768dec-1
kmod-mt76x2-common - 4.14.221+2021-02-15-5c768dec-1
'''

=====================
感谢驱动程序作者的辛苦工作~

siwind, 2021-05-30:
root@OpenWrt:~# opkg list-installed | grep tran
getrandom - 2019-06-16-4df34a4d-3
luci-app-transmission - git-21.044.30835-34e0d65-1
luci-i18n-transmission-zh-cn - git-21.044.30835-34e0d65-1
transmission-daemon-mbedtls - 3.00-4
transmission-web - 3.00-4

namidairo, 2021-05-30:
This is not the correct place to lodge an issue about the transmission package.

https://github.com/openwrt/packages/ would be a more appropriate place to lodge an issue.

siwind, 2021-05-30:
The problem disappear  with Openwrt master branch and kernel 5.4.123.

'
root@OpenWrt:~# opkg list-installed | grep mt76
kmod-mt76-core - 5.4.123+2021-05-15-9d736545-4
kmod-mt7603 - 5.4.123+2021-05-15-9d736545-4
kmod-mt76x02-common - 5.4.123+2021-05-15-9d736545-4
kmod-mt76x2 - 5.4.123+2021-05-15-9d736545-4
kmod-mt76x2-common - 5.4.123+2021-05-15-9d736545-4
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.123 #0 SMP Sun May 30 11:49:31 2021 mips GNU/Linux
root@OpenWrt:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r17175-bc337bb3d3'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r17175-bc337bb3d3'
DISTRIB_TAINTS='no-all'

root@OpenWrt:~# opkg list-installed | grep tran
getrandom - 2020-10-25-9ef88681-2
luci-app-transmission - git-21.115.57293-52297b1
luci-i18n-transmission-zh-cn - git-21.136.59095-94d35bb
transmission-daemon - 3.00-11
transmission-web - 3.00-11
'

Thanks.

closed 2021-06-06 by nbd168