openwrt/mt76 #545: mt7915d never work on latest code

ptpt52, 2021-06-03:
openwrt/master
mt76/master

/etc/config/wireless
```
config wifi-device 'radio0'
        option type 'mac80211'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option disabled '1'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

```

this config is not correct? it is DBDC and works before.

ptpt52, 2021-06-03:
get log on boot:
```
[   30.042189] ------------[ cut here ]------------
[   30.047309] WARNING: CPU: 2 PID: 4039 at backports-5.10.34-1/net/mac80211/main.c:118 ieee80211_hw_config+0x34c/0x390 [mac80211]
[   30.058851] control:0.44032 MHz width:0 center: 2412.000/0 MHz
[   30.058869] Modules linked in: pppoe ppp_async wireguard pptp pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xp
[   30.065056]  nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sit
[   30.151643]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ic
[   30.283223] CPU: 2 PID: 4039 Comm: iw Not tainted 5.4.123 #0
[   30.288889] Stack : 804784d0 80073288 806b0000 806b4070 80720000 806b4038 806b2fb0 8b8b1a44
[   30.297248]         80860000 8e2a5994 806fec63 80643f18 00000002 00000001 8b8b19e8 ebc95499
[   30.305580]         00000000 00000000 808a0000 00000000 00000030 00000139 746f4e20 69617420
[   30.313912]         00000000 00000001 00000000 00045257 80000000 80720000 00000000 8d801df4
[   30.322244]         00000009 8ea9b6c0 8b8b1c74 8ea66000 00000003 8033cb7c 00000008 80860008
[   30.330576]         ...
[   30.333014] Call Trace:
[   30.335482] [<8000b61c>] show_stack+0x30/0x100
[   30.339941] [<8058a680>] dump_stack+0xa4/0xdc
[   30.344300] [<8002be20>] __warn+0xc0/0x10c
[   30.348385] [<8002bef8>] warn_slowpath_fmt+0x8c/0xac
[   30.353483] [<8d801df4>] ieee80211_hw_config+0x34c/0x390 [mac80211]
[   30.359819] [<8d822704>] ieee80211_nan_func_match+0xafc/0x4448 [mac80211]
[   30.366744] ---[ end trace 06bff96597ba4686 ]---
```

crash log if run command `iwinfo`
```
root@X-WRT:~# iwinfo
wlan0     ESSID: "X-WRT_6859"
          Access Point: E4:67:8E:29:68:5B
          Mode: Master  Channel: unknown (2.407 GHz)
          Center Channel 1: unknown 2: unknown
          Tx-Pow[  296.956682] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 00000000, ra == 8df0a304
er: 3 dBm  Link [  296.968380] Oops[#1]:
[  296.971931] CPU: 0 PID: 6924 Comm: iwinfo Tainted: G        W         5.4.123 #0
[  296.979287] $ 0   : 00000000 00000001 00000000 00000000
[  296.984491] $ 4   : 8de01ec0 83006000 00000001 00000688
[  296.989693] $ 8   : 00000689 8ebb9240 00000003 807225e0
[  296.994895] $12   : 00000000 00000402 8071db30 00000056
[  297.000096] $16   : 83006000 00000000 8df0ece5 00000000
[  297.005298] $20   : 00000000 8de039a0 8de01ec0 8ea4b080
[  297.010499] $24   : 00000000 8dfe1940                  
[  297.015700] $28   : 8cb8e000 8cb8fa38 8df0ecef 8df0a304
[  297.020902] Hi    : 00000000
[  297.023762] Lo    : 0000000e
[  297.026631] epc   : 00000000 0x0
[  297.029867] ra    : 8df0a304 mt7915_mcu_get_rx_rate+0x500/0x928 [mt7915e]
[  297.036617] Status: 11008403 KERNEL EXL IE 
[  297.040781] Cause : 50800008 (ExcCode 02)
[  297.044765] BadVA : 00000000
[  297.047627] PrId  : 0001992f (MIPS 1004Kc)
[  297.051696] Modules linked in: pppoe ppp_async wireguard pptp pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xp
[  297.051877]  nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sit
[  297.138180]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ic
[  297.269302] Process iwinfo (pid: 6924, threadinfo=551a3e78, task=282eabbe, tls=77e53ec8)
[  297.277350] Stack : 804784d0 8dfabe3c 00000001 00000002 83006000 8df10000 8de01ec0 8de01ec0
[  297.285670]         8cb8fb00 8ebb9180 8de01ec0 8de020a0 00000000 8cb8fee8 804784d0 8df0c690
[  297.293989]         00000000 807225e0 00000000 8cb8fafc 8de01ec0 8dfe3cc8 8b8cde40 8b8cde40
[  297.302308]         8cb8fafc 8cb8faf8 00000000 8cb8fb00 8ebb9180 8dfe3fa8 00000000 00000000
[  297.310627]         00000000 00000000 8ec2f000 00000000 00000000 8b8cde40 8ebb9180 8ec2e800
[  297.318947]         ...
[  297.321380] Call Trace:
[  297.321382] 
[  297.325298] [<804784d0>] netlink_sendmsg+0x0/0x3d8
[  297.330134] [<8dfabe3c>] nl80211_put_sta_rate+0x1450/0x15cc [cfg80211]
[  297.336769] [<804784d0>] netlink_sendmsg+0x0/0x3d8
[  297.341543] [<8df0c690>] mt7915_update_channel+0x20/0x50 [mt7915e]
[  297.347740] [<8dfe3cc8>] mt76_update_survey+0x2c/0xd0 [mt76]
[  297.353427] [<8dfe3fa8>] mt76_get_survey+0x5c/0x4d4 [mt76]
[  297.358951] [<8dfa71e0>] nl80211_prepare_wdev_dump+0x2e4/0xcac [cfg80211]
[... 7 lines cut ...]
[  297.401755] [<8047a410>] genl_rcv_msg+0x0/0x494
[  297.406280] [<80478cdc>] netlink_rcv_skb+0xac/0x158
[  297.411175] [<8047949c>] genl_rcv+0x30/0x48
[  297.415369] [<804783ec>] netlink_unicast+0x1b0/0x294
[  297.420345] [<8013c430>] __check_object_size+0x170/0x1d4
[  297.425711] [<8047868c>] netlink_sendmsg+0x1bc/0x3d8
[  297.430684] [<80142b2c>] rw_copy_check_uvector+0x50/0x1f8
[  297.436066] [<80402abc>] simple_copy_to_iter+0x0/0xa0
[  297.441151] [<803ee390>] ____sys_sendmsg+0xc0/0x264
[  297.446046] [<803eeb9c>] copy_msghdr_from_user+0xcc/0x160
[  297.451446] [<803fc208>] skb_queue_purge+0x24/0x3c
[... cut ...]
[  297.465517] [<801936fc>] fsnotify_destroy_marks+0x18/0x120
[  297.471009] [<80162b7c>] __destroy_inode+0x28/0x16c
[  297.475960] [<8014381c>] __fput+0x180/0x2a0
[  297.480154] [<805a74e4>] _cond_resched+0x20/0x60
[  297.484813] [<803efec8>] __sys_sendmsg+0x4c/0x94
[  297.489409] [<80143ecc>] fput_many+0x74/0xcc
[  297.493693] [<80014558>] syscall_common+0x34/0x58
[  297.498376] 
[  297.499854] Code: (Bad address in epc)
[  297.503587] 
Quality: unknown[  297.505467] ---[ end trace 06bff96597ba4687 ]---
```

ptpt52, 2021-06-03:
installed ipks:
```
opkg list | grep mt
kmod-mt76-connac - 5.4.123+2021-06-02-20210602-4
kmod-mt76-core - 5.4.123+2021-06-02-20210602-4
kmod-mt7615-common - 5.4.123+2021-06-02-20210602-4
kmod-mt7615-firmware - 5.4.123+2021-06-02-20210602-4
kmod-mt7915e - 5.4.123+2021-06-02-20210602-4
```

ryderlee1110, 2021-06-04:
does it work after dropping https://github.com/openwrt/mt76/commit/d9dd7635b0551839d2dc5544855c1ebb5205c800 ? I don't have dbdc devices but iwinfo works on my 7915a

ptpt52, 2021-06-05:
> can you try if this fix iwinfo for 7915 dbdc https://pastebin.com/cAwpxsaB?

no help, crash still:
```
root@X-WRT:~# iwinfo 
wlan0     ESSID: "X-WRT_6859"
          Access Point: E4:67:8E:29:68:5B
          Mode: Master  Channel: unknown (2.407 GHz)
          Center Channel 1: unknown 2: unknown
          Tx-Pow[   81.688193] CPU 3 Unable to handle kernel paging request at virtual address 000010c0, epc == 000010c0, ra == 8dfea054
er: 3 dBm  Link [   81.699816] Oops[#1]:
Quality: unknown[   81.703426] CPU: 3 PID: 6652 Comm: iwinfo Not tainted 5.4.123 #0
/70
[   81.710770] $ 0   : 00000000 00000001 000010c0 8b078560
[   81.716386] $ 4   : 8e021ec0 0002482c 8b0bfb00 8b0bfb00
[   81.721588] $ 8   : 8eefc800 8eefc800 00000000 00000002
[   81.726790] $12   : fffffffd 00000402 8071dff0 00000040
[   81.731993] $16   : 8e021ec0 8e021ec0 00020000 8ff98f00
[   81.737196] $20   : 8e0239a0 8e0220a0 8e021ec0 8b0bfee8
[   81.742399] $24   : 00000000 8d803f4c                  
[   81.747602] $28   : 8b0be000 8b0bfa28 804786a8 8dfea054
[   81.752807] Hi    : 00000000
[   81.755668] Lo    : 0000000d
[   81.758536] epc   : 000010c0 0x10c0
[   81.762036] ra    : 8dfea054 mt7915_mcu_get_rx_rate+0x4a0/0x9ac [mt7915e]
[   81.768787] Status: 11007c03 KERNEL EXL IE 
[   81.772955] Cause : 50800008 (ExcCode 02)
[   81.776939] BadVA : 000010c0
[   81.779800] PrId  : 0001992f (MIPS 1004Kc)
[   81.783871] Modules linked in: pppoe ppp_async wireguard pptp pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xp
[   81.784051]  nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sit
[   81.870378]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ic
[   82.001511] Process iwinfo (pid: 6652, threadinfo=bf00a2b2, task=5abc6ab1, tls=77e7fec8)
[   82.009557] Stack : fffffffe 00000002 8fefe488 8b0bfee8 804786a8 8dfabe3c 00000001 00000002
[   82.017878]         00000001 805a10a4 8e021ec0 8e021ec0 8b0bfb00 8ff98f00 8e021ec0 8e0220a0
[   82.026201]         00000000 8b0bfee8 804786a8 8dfec4cc 00000000 00000000 8d998640 8b0bfafc
[   82.034522]         8e021ec0 8d803cc8 8d998640 8d998640 8b0bfafc 8b0bfaf8 00000000 8b0bfb00
[   82.042845]         8ff98f00 8d803fa8 00000000 00000000 00000000 00000000 8eeff800 00000000
[   82.051167]         ...
[   82.053604] Call Trace:
[   82.053606] 
[   82.057523] [<804786a8>] netlink_sendmsg+0x0/0x3d8
[   82.062365] [<8dfabe3c>] nl80211_put_sta_rate+0x1450/0x15cc [cfg80211]
[   82.068863] [<805a10a4>] xas_alloc+0xc4/0x104
[   82.073315] [<804786a8>] netlink_sendmsg+0x0/0x3d8
[   82.078093] [<8dfec4cc>] mt7915_update_channel+0x18/0x8c [mt7915e]
[   82.084293] [<8d803cc8>] mt76_update_survey+0x2c/0xd0 [mt76]
[   82.089955] [<8d803fa8>] mt76_get_survey+0x5c/0x4d4 [mt76]
[   82.095461] [<8dfa71e0>] nl80211_prepare_wdev_dump+0x2e4/0xcac [cfg80211]
[   82.102248] [<803f9f34>] __alloc_skb+0x98/0x180

```

ptpt52, 2021-06-05:
wait, I retry it

ptpt52, 2021-06-05:
@ryderlee1110 
revert or using patch in https://pastebin.com/cAwpxsaB
no help, still crash

ptpt52, 2021-06-05:
```
[   28.665245] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   30.130396] ------------[ cut here ]------------
[   30.135563] WARNING: CPU: 0 PID: 4049 at backports-5.10.34-1/net/mac80211/main.c:118 ieee80211_hw_config+0x34c/0x390 [mac80211]
[   30.147157] control:0.35416 MHz width:0 center: 2412.000/0 MHz
[   30.147174] Modules linked in: pppoe ppp_async wireguard pptp pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xp
[   30.153402]  nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sit
[   30.240082]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ic
[   30.371635] CPU: 0 PID: 4049 Comm: iw Not tainted 5.4.123 #0
[   30.377298] Stack : 804786a8 80073288 806b0000 806b5190 80720000 806b5158 806b40d0 8b87ba44
[   30.385660]         80860000 8fd193b4 806fec63 80644f60 00000000 00000001 8b87b9e8 0f03a340
[   30.394000]         00000000 00000000 808a0000 00000000 00000030 00000138 746f4e20 69617420
[   30.402333]         00000000 00000001 00000000 0005abb3 80000000 80720000 00000000 8d801df4
[   30.410664]         00000009 8ff50a80 8b87bc74 8ecbc000 00000003 8033cb7c 00000000 80860000
[   30.418995]         ...
[   30.421434] Call Trace:
[   30.423901] [<8000b61c>] show_stack+0x30/0x100
[   30.428362] [<8058a858>] dump_stack+0xa4/0xdc
[   30.432721] [<8002be20>] __warn+0xc0/0x10c
[   30.436807] [<8002bef8>] warn_slowpath_fmt+0x8c/0xac
[   30.441912] [<8d801df4>] ieee80211_hw_config+0x34c/0x390 [mac80211]
[   30.448232] [<8d822704>] ieee80211_nan_func_match+0xafc/0x4448 [mac80211]
[   30.455201] ---[ end trace a66436d61ea4c13d ]---
root@X-WRT:~# iwinfo 
wlan0     ESSID: "X-WRT_6859"
          Access Point: E4:67:8E:29:68:5B
          Mode: Master  Channel: unknown (2.407 GHz)
          Center Channel 1: unknown 2: unknown
          Tx-Pow[   74.867657] CPU 2 Unable to handle kernel paging request at virtual address 00000000, epc == 00000001, ra == 8df8c53c
er: 3 dBm  Link [   74.879286] Oops[#1]:
Quality: unknown[   74.882899] CPU: 2 PID: 5957 Comm: iwinfo Tainted: G        W         5.4.123 #0
/70
[   74.891620] $ 0   : 00000000 00000001 00000001 00000000
[   74.897238] $ 4   : 8de69ec0 83006000 00000001 000007c0
[   74.902441] $ 8   : 000007c1 8d8ab240 8cbc9c74 8cbc9ee8
[   74.907646] $12   : 00000000 00000402 8071dff0 00000056
[   74.912847] $16   : 83006000 00000000 8de6b9a0 00000000
[   74.918049] $20   : 8df8ecb5 00000000 00000000 8de69ec0
[   74.923252] $24   : 00000000 8def1940                  
[   74.928455] $28   : 8cbc8000 8cbc9a48 8df8ecbf 8df8c53c
[   74.933659] Hi    : 00000000
[   74.936520] Lo    : 00000002
[   74.939388] epc   : 00000001 0x1
[   74.942630] ra    : 8df8c53c mt7915_update_channel+0xc8/0x218 [mt7915e]
[   74.949208] Status: 11007c03 KERNEL EXL IE 
[   74.953375] Cause : 50800008 (ExcCode 02)
[   74.957360] BadVA : 00000000
[   74.960221] PrId  : 0001992f (MIPS 1004Kc)
[   74.964293] Modules linked in: pppoe ppp_async wireguard pptp pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xp
[   74.964475]  nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sit
[   75.050785]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ic
[   75.181918] Process iwinfo (pid: 5957, threadinfo=3f5d620d, task=05742f03, tls=77e41ec8)
[   75.189964] Stack : 00000001 fffffff1 8cbc9a78 806fc3b0 81025464 8ffea480 83006000 8df90000
[   75.198284]         8de69ec0 8de69ec0 8cbc9b00 8d8ab000 8de69ec0 8de6a0a0 00000000 8cbc9ee8
[   75.206606]         804786a8 8def3ccc 8ffeae40 8ffeae40 8cbc9afc 8cbc9af8 00000000 8de69ec0
[   75.214930]         8cbc9b00 8def3f90 00000000 00000000 00000000 00000000 8ea24000 00000000
[   75.223254]         00000000 8ffeae40 8d8ab000 8ea25000 fffffffe 8df271e0 00000000 00000000
[   75.231576]         ...
[   75.234010] Call Trace:
[   75.234012] 
[   75.238091] [<804786a8>] netlink_sendmsg+0x0/0x3d8
[   75.242882] [<8def3ccc>] mt76_update_survey+0x30/0xc0 [mt76]
[   75.248572] [<8def3f90>] mt76_get_survey+0x5c/0x4d4 [mt76]
[   75.254145] [<8df271e0>] nl80211_prepare_wdev_dump+0x2e4/0xcac [cfg80211]
[   75.260999] [<803f9f34>] __alloc_skb+0x98/0x180

```

ryderlee1110, 2021-06-05:
does it work after reverting https://github.com/openwrt/mt76/commit/d9dd7635b0551839d2dc5544855c1ebb5205c800#diff-29503a65d2a468b0ddbcdb23c60145f560036a0261e39a4bbee3b0b6cb9b523e ?

ptpt52, 2021-06-05:
> does it work after reverting [d9dd763#diff-29503a65d2a468b0ddbcdb23c60145f560036a0261e39a4bbee3b0b6cb9b523e](https://github.com/openwrt/mt76/commit/d9dd7635b0551839d2dc5544855c1ebb5205c800#diff-29503a65d2a468b0ddbcdb23c60145f560036a0261e39a4bbee3b0b6cb9b523e) ?

no help

lister-wrt, 2021-06-06:
Same issue reported in the forum (https://forum.openwrt.org/t/issues-with-recent-mt76/98311)

nbd168, 2021-06-06:
Fixed in the latest version

closed 2021-06-06 by nbd168