openwrt/mt76 #547: Wifi unstable and wan connection drops on Xiaomi MI router 4c on SoC MediaTek MT7628AN running on OpenWrt 21.02.0-rc2 stable release

router4c, 2021-06-03:
system log attached below:

wifi disconnect
Thu Jun 3 19:55:06 2021 kern.info kernel: [21337.734846] rt3050-esw 10110000.esw: link changed 0x06

after restarting wifi radio
Thu Jun 3 20:00:24 2021 daemon.notice hostapd: Remove interface 'wlan0'
Thu Jun 3 20:00:24 2021 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Thu Jun 3 20:00:24 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 26:0f:41:a2:14:99
Thu Jun 3 20:00:24 2021 daemon.notice hostapd: wlan0: AP-DISABLED
Thu Jun 3 20:00:24 2021 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Thu Jun 3 20:00:24 2021 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Thu Jun 3 20:00:24 2021 kern.info kernel: [21655.465888] device wlan0 left promiscuous mode
Thu Jun 3 20:00:24 2021 kern.info kernel: [21655.470650] br-lan: port 2(wlan0) entered disabled state
Thu Jun 3 20:00:27 2021 kern.err kernel: [21658.906057] mt76_wmac 10300000.wmac: MCU message 08 (seq 7) timed out
Thu Jun 3 20:00:29 2021 daemon.notice netifd: Network device 'wlan0' link is down
Thu Jun 3 20:00:31 2021 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Thu Jun 3 20:00:34 2021 kern.err kernel: [21665.817985] mt76_wmac 10300000.wmac: MCU message 08 (seq 8) timed out
Thu Jun 3 20:00:36 2021 kern.info kernel: [21667.884073] br-lan: port 2(wlan0) entered blocking state
Thu Jun 3 20:00:36 2021 kern.info kernel: [21667.889549] br-lan: port 2(wlan0) entered disabled state
Thu Jun 3 20:00:36 2021 kern.info kernel: [21667.895378] device wlan0 entered promiscuous mode
Thu Jun 3 20:00:36 2021 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Thu Jun 3 20:00:36 2021 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Thu Jun 3 20:00:39 2021 kern.err kernel: [21670.937919] mt76_wmac 10300000.wmac: MCU message 08 (seq 9) timed out
Thu Jun 3 20:00:44 2021 kern.err kernel: [21676.057867] mt76_wmac 10300000.wmac: MCU message 08 (seq 10) timed out
Thu Jun 3 20:00:49 2021 kern.err kernel: [21681.177788] mt76_wmac 10300000.wmac: MCU message 08 (seq 11) timed out
Thu Jun 3 20:00:55 2021 kern.err kernel: [21686.297726] mt76_wmac 10300000.wmac: MCU message 08 (seq 12) timed out
Thu Jun 3 20:01:00 2021 kern.err kernel: [21691.417664] mt76_wmac 10300000.wmac: MCU message 08 (seq 13) timed out
Thu Jun 3 20:01:05 2021 kern.err kernel: [21696.537602] mt76_wmac 10300000.wmac: MCU message 08 (seq 14) timed out
Thu Jun 3 20:01:10 2021 kern.err kernel: [21701.657538] mt76_wmac 10300000.wmac: MCU message 08 (seq 15) timed out
Thu Jun 3 20:01:15 2021 kern.err kernel: [21706.777479] mt76_wmac 10300000.wmac: MCU message 08 (seq 1) timed out
Thu Jun 3 20:01:20 2021 kern.err kernel: [21711.897424] mt76_wmac 10300000.wmac: MCU message 08 (seq 2) timed out
Thu Jun 3 20:01:25 2021 kern.err kernel: [21717.017355] mt76_wmac 10300000.wmac: MCU message 08 (seq 3) timed out
Thu Jun 3 20:01:30 2021 kern.err kernel: [21722.137291] mt76_wmac 10300000.wmac: MCU message 08 (seq 4) timed out
Thu Jun 3 20:01:36 2021 kern.err kernel: [21727.257229] mt76_wmac 10300000.wmac: MCU message 08 (seq 5) timed out
Thu Jun 3 20:01:41 2021 kern.err kernel: [21732.377165] mt76_wmac 10300000.wmac: MCU message 08 (seq 6) timed out
Thu Jun 3 20:01:46 2021 kern.err kernel: [21737.497107] mt76_wmac 10300000.wmac: MCU message 08 (seq 7) timed out
Thu Jun 3 20:01:51 2021 kern.err kernel: [21742.873039] mt76_wmac 10300000.wmac: MCU message 08 (seq 8) timed out
Thu Jun 3 20:01:53 2021 kern.info kernel: [21744.922760] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Thu Jun 3 20:01:53 2021 kern.info kernel: [21744.929532] br-lan: port 2(wlan0) entered blocking state
Thu Jun 3 20:01:53 2021 kern.info kernel: [21744.934979] br-lan: port 2(wlan0) entered forwarding state
Thu Jun 3 20:01:53 2021 daemon.notice netifd: Network device 'wlan0' link is up
Thu Jun 3 20:01:53 2021 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Thu Jun 3 20:01:53 2021 daemon.notice hostapd: wlan0: AP-ENABLED

wan connection drop
Thu Jun 3 20:08:38 2021 kern.err kernel: [22150.164143] mt76_wmac 10300000.wmac: MCU message 08 (seq 9) timed out
Thu Jun 3 20:08:43 2021 kern.err kernel: [22154.516085] mt76_wmac 10300000.wmac: MCU message 08 (seq 10) timed out
Thu Jun 3 20:08:49 2021 kern.err kernel: [22160.660009] mt76_wmac 10300000.wmac: MCU message 08 (seq 11) timed out
Thu Jun 3 20:08:54 2021 kern.err kernel: [22165.779948] mt76_wmac 10300000.wmac: MCU message 08 (seq 12) timed out
Thu Jun 3 20:08:59 2021 kern.err kernel: [22170.899888] mt76_wmac 10300000.wmac: MCU message 08 (seq 13) timed out
Thu Jun 3 20:09:04 2021 kern.err kernel: [22176.019826] mt76_wmac 10300000.wmac: MCU message 08 (seq 14) timed out
Thu Jun 3 20:09:09 2021 kern.err kernel: [22181.139765] mt76_wmac 10300000.wmac: MCU message 08 (seq 15) timed out
Thu Jun 3 20:09:15 2021 kern.err kernel: [22186.259707] mt76_wmac 10300000.wmac: MCU message 08 (seq 1) timed out
Thu Jun 3 20:09:20 2021 kern.err kernel: [22191.379642] mt76_wmac 10300000.wmac: MCU message 08 (seq 2) timed out
Thu Jun 3 20:09:25 2021 kern.err kernel: [22196.499585] mt76_wmac 10300000.wmac: MCU message 08 (seq 3) timed out
Thu Jun 3 20:09:30 2021 kern.err kernel: [22201.619520] mt76_wmac 10300000.wmac: MCU message 08 (seq 4) timed out
Thu Jun 3 20:09:35 2021 kern.err kernel: [22206.739463] mt76_wmac 10300000.wmac: MCU message 08 (seq 5) timed out
Thu Jun 3 20:09:40 2021 kern.err kernel: [22211.859397] mt76_wmac 10300000.wmac: MCU message 08 (seq 6) timed out
Thu Jun 3 20:09:45 2021 kern.err kernel: [22216.979335] mt76_wmac 10300000.wmac: MCU message 08 (seq 7) timed out
Thu Jun 3 20:09:50 2021 kern.err kernel: [22222.099276] mt76_wmac 10300000.wmac: MCU message 08 (seq 8) timed out
Thu Jun 3 20:09:55 2021 kern.err kernel: [22227.219215] mt76_wmac 10300000.wmac: MCU message 08 (seq 9) timed out
Thu Jun 3 20:10:01 2021 kern.err kernel: [22232.339153] mt76_wmac 10300000.wmac: MCU message 08 (seq 10) timed out
Thu Jun 3 20:10:06 2021 kern.err kernel: [22237.459096] mt76_wmac 10300000.wmac: MCU message 08 (seq 11) timed out
Thu Jun 3 20:10:11 2021 kern.err kernel: [22242.579030] mt76_wmac 10300000.wmac: MCU message 08 (seq 12) timed out
Thu Jun 3 20:10:16 2021 kern.err kernel: [22247.698970] mt76_wmac 10300000.wmac: MCU message 08 (seq 13) timed out
Thu Jun 3 20:10:21 2021 kern.err kernel: [22252.818917] mt76_wmac 10300000.wmac: MCU message 08 (seq 14) timed out
Thu Jun 3 20:10:26 2021 kern.err kernel: [22257.938869] mt76_wmac 10300000.wmac: MCU message 08 (seq 15) timed out
Thu Jun 3 20:10:31 2021 kern.err kernel: [22263.058788] mt76_wmac 10300000.wmac: MCU message 08 (seq 1) timed out
Thu Jun 3 20:10:36 2021 kern.err kernel: [22268.178725] mt76_wmac 10300000.wmac: MCU message 08 (seq 2) timed out
Thu Jun 3 20:10:42 2021 kern.err kernel: [22273.298665] mt76_wmac 10300000.wmac: MCU message 08 (seq 3) timed out
Thu Jun 3 20:10:47 2021 kern.err kernel: [22278.418620] mt76_wmac 10300000.wmac: MCU message 08 (seq 4) timed out
Thu Jun 3 20:10:52 2021 kern.err kernel: [22283.538544] mt76_wmac 10300000.wmac: MCU message 08 (seq 5) timed out
Thu Jun 3 20:10:57 2021 kern.err kernel: [22288.658776] mt76_wmac 10300000.wmac: MCU message 08 (seq 6) timed out
Thu Jun 3 20:28:51 2021 kern.info kernel: [23362.924721] rt3050-esw 10110000.esw: link changed 0x04

Wifi disconnects randomly and gets restored after restarting the router.

I think there is bug in kernel driver or implementation of the driver not done properly with openwrt stable release.
thank you

router4c, 2021-06-18:
After getting below system logs, I disabled 'Disassociate On Low Acknowledgement' as suggested in Wireless FAQ in openwrt page
`Fri Jun 11 11:34:39 2021 daemon.info hostapd: wlan0: STA 26:0f:41:a2:14:99 IEEE 802.11: disconnected due to excessive missing ACKs
Fri Jun 11 11:34:39 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 26:0f:41:a2:14:99
Fri Jun 11 11:35:09 2021 daemon.info hostapd: wlan0: STA 26:0f:41:a2:14:99 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)`
But still the problem exists, only logs slightly changed
`Fri Jun 11 12:24:19 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 26:0f:41:a2:14:99
Fri Jun 11 12:24:19 2021 daemon.info hostapd: wlan0: STA 26:0f:41:a2:14:99 IEEE 802.11: disassociated due to inactivity
Fri Jun 11 12:24:20 2021 daemon.info hostapd: wlan0: STA f26:0f:41:a2:14:99 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)`
Then I tried resetting and reflashing the OpenWrt 21.02.0-rc2 firmware, I get no problem for 24 hours after that same thing starts happening again, system log attached
`Fri Jun 11 17:35:19 2021 kern.notice kernel: [    0.000000] Linux version 5.4.119 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16122-c2139eef27)) #0 Fri May 28 14:04:04 2021
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Board has DDR2
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Analog PMU set to hw control
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Digital PMU set to hw control
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7628AN ver:1 eco:2
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] MIPS: machine is Xiaomi Mi Router 4C
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Fri Jun 11 17:35:19 2021 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Fri Jun 11 17:35:19 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Zone ranges:
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Early memory node ranges
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Fri Jun 11 17:35:19 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Fri Jun 11 17:35:19 2021 kern.debug kernel: [    0.000000]   Normal zone: 144 pages used for memmap
Fri Jun 11 17:35:19 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Fri Jun 11 17:35:19 2021 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Fri Jun 11 17:35:19 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Fri Jun 11 17:35:19 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
Fri Jun 11 17:35:19 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=0007e009
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=0007e009
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] Memory: 57508K/65536K available (5058K kernel code, 205K rwdata, 632K rodata, 1200K init, 205K bss, 8028K reserved, 0K cma-reserved)
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] NR_IRQS: 256
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] intc: using register map from devicetree
Fri Jun 11 17:35:19 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from 0x805c4a28 with crng_init=0
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] CPU Clock: 575MHz
Fri Jun 11 17:35:19 2021 kern.crit kernel: [    0.000000] timer_probe: no matching timers found
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6647862422 ns
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.000010] sched_clock: 32 bits at 287MHz, resolution 3ns, wraps every 7469508094ns
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.007561] Calibrating delay loop... 382.46 BogoMIPS (lpj=764928)
Fri Jun 11 17:35:19 2021 kern.info kernel: [    0.041415] pid_max: default: 32768 minimum: 301
[... cut ...]
Sat Jun 12 01:56:03 2021 kern.info kernel: [ 6046.930855] br-lan: port 2(wlan0) entered blocking state
Sat Jun 12 01:56:03 2021 kern.info kernel: [ 6046.936302] br-lan: port 2(wlan0) entered forwarding state
Sat Jun 12 01:56:03 2021 daemon.notice netifd: Network device 'wlan0' link is up
Sat Jun 12 01:56:03 2021 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Sat Jun 12 01:56:03 2021 daemon.notice hostapd: wlan0: AP-ENABLED
Sat Jun 12 01:58:15 2021 daemon.err uhttpd[1316]: luci: accepted login on / for root from 192.168.1.135`
thank you

1Light, 2022-02-07:
I have also faced same problem but with only on one of the client Device which has BROADCAST chip (Macbook air 2015 running linux).

**System Log from laptop:**

Feb 07 17:44:14  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-DISCONNECTED bssid=<BSSID> reason=0
Feb 07 17:44:14  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Feb 07 17:44:14  NetworkManager[392]: <info>  [1644236054.1569] device (wlp3s0): supplicant interface state: completed -> disconnected
Feb 07 17:44:14  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:15  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:16  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:17  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:18  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:19  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:20  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:21  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:22  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:23  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:24  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:25  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:26  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:27  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:28  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:29  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:29  NetworkManager[392]: <warn>  [1644236069.5744] device (wlp3s0): link timed out.
Feb 07 17:44:29  NetworkManager[392]: <info>  [1644236069.5747] device (wlp3s0): state change: activated -> failed (reason 'ssid-not-found', sys-iface-state: 'managed')
Feb 07 17:44:29  NetworkManager[392]: <info>  [1644236069.5758] manager: NetworkManager state is now DISCONNECTED
Feb 07 17:44:29  avahi-daemon[389]: Interface wlp3s0.IPv4 no longer relevant for mDNS.
Feb 07 17:44:29  avahi-daemon[389]: Leaving mDNS multicast group on interface wlp3s0.IPv4 with address <local ip>.
Feb 07 17:44:29  NetworkManager[392]: <info>  [1644236069.5788] device (wlp3s0): set-hw-addr: set MAC address to <MAC> (scanning)
Feb 07 17:44:29  avahi-daemon[389]: Withdrawing address record for <local ip> on wlp3s0.
Feb 07 17:44:29  avahi-daemon[389]: Joining mDNS multicast group on interface wlp3s0.IPv4 with address <local ip>.
Feb 07 17:44:29  avahi-daemon[389]: New relevant interface wlp3s0.IPv4 for mDNS.
Feb 07 17:44:29  avahi-daemon[389]: Registering new address record for <local ip> on wlp3s0.IPv4.
Feb 07 17:44:29  wpa_supplicant[883]: wlp3s0: CTRL-EVENT-SCAN-FAILED ret=-22 retry=1
Feb 07 17:44:29  NetworkManager[392]: <warn>  [1644236069.5840] device (wlp3s0): Activation: failed for connection '<SSID>'
Feb 07 17:44:29  gnome-shell[1281]: An active wireless connection, in infrastructure mode, involves no access point?
Feb 07 17:44:29  NetworkManager[392]: <info>  [1644236069.5882] device (wlp3s0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Feb 07 17:44:29  dbus-daemon[391]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.5' (uid=0 pid=392 comm="/usr/bin/NetworkManager --no-daemon ")
Feb 07 17:44:29  NetworkManager[392]: <info>  [1644236069.6004] dhcp4 (wlp3s0): canceled DHCP transaction
Feb 07 17:44:29  NetworkManager[392]: <info>  [1644236069.6004] dhcp4 (wlp3s0): state changed bound -> terminated
Feb 07 17:44:29  systemd[1]: Starting Network Manager Script Dispatcher Service...
Feb 07 17:44:29  avahi-daemon[389]: Withdrawing address record for <local ip> on wlp3s0.
Feb 07 17:44:29  avahi-daemon[389]: Leaving mDNS multicast group on interface wlp3s0.IPv4 with address <local ip>.
Feb 07 17:44:29  avahi-daemon[389]: Interface wlp3s0.IPv4 no longer relevant for mDNS.
Feb 07 17:44:29  gnome-shell[1281]: An active wireless connection, in infrastructure mode, involves no access point?
Feb 07 17:44:29  gnome-shell[1281]: An active wireless connection, in infrastructure mode, involves no access point?
Feb 07 17:44:29  gnome-shell[1281]: An active wireless connection, in infrastructure mode, involves no access point?
Feb 07 17:44:29  dbus-daemon[391]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Feb 07 17:44:29  systemd[1]: Started Network Manager Script Dispatcher Service.
Feb 07 17:44:29  audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Feb 07 17:44:29  kernel: audit: type=1130 audit(1644236069.626:167): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=NetworkManager-dispatcher comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Feb 07 17:44:34  NetworkManager[392]: <info>  [1644236074.9912] policy: auto-activating connection '<SSID>' (5a827cab-7ad3-470a-acc9-3632a71cb99f)
[... cut ...]
Feb 07 17:44:34  NetworkManager[392]: <info>  [1644236074.9965] Config: added 'scan_ssid' value '1'
Feb 07 17:44:34  NetworkManager[392]: <info>  [1644236074.9965] Config: added 'bgscan' value 'simple:30:-70:86400'
Feb 07 17:44:34  NetworkManager[392]: <info>  [1644236074.9965] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256'
Feb 07 17:44:34  NetworkManager[392]: <info>  [1644236074.9965] Config: added 'psk' value '<hidden>'
Feb 07 17:44:35  NetworkManager[392]: <info>  [1644236075.0085] device (wlp3s0): supplicant interface state: disconnected -> inactive