openwrt/mt76 #548: mt7921e: probe of 0000:02:00.0 failed with error -5

Roman-, 2021-06-12:
Running Ubuntu 21.04 on my ASUS ROG Zephyrus G14. Kernel version: 5.12.10.
In the network settings, it says **No Wi-Fi Adapter Found**.

# rfkill list all
```
0: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```

# lshw -C network
```
*-network UNCLAIMED       
       description: Network controller
       product: MEDIATEK Corp.
       vendor: MEDIATEK Corp.
       physical id: 0
       bus info: pci@0000:02:00.0
       version: 00
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm cap_list
       configuration: latency=0
       resources: iomemory:fe0-fdf iomemory:fe0-fdf iomemory:fe0-fdf memory:fe20300000-fe203fffff memory:fe20400000-fe20403fff memory:fe20404000-fe20404fff
```
# /var/log/syslog
```
Jun 12 19:13:29 rog-laptop kernel: [    3.190487] Error: Driver 'asus' is already registered, aborting...
Jun 12 19:13:29 rog-laptop kernel: [    3.190685] usbcore: registered new interface driver btusb
Jun 12 19:13:29 rog-laptop kernel: [    3.202090] mt7921e 0000:02:00.0: enabling device (0000 -> 0002)
Jun 12 19:13:29 rog-laptop kernel: [    3.202805] mt7921e 0000:02:00.0: disabling ASPM  L1
Jun 12 19:13:29 rog-laptop kernel: [    3.203087] mt7921e 0000:02:00.0: ASIC revision: 79610010
...
Jun 12 19:13:29 rog-laptop kernel: [    3.205066] mt7921e: probe of 0000:02:00.0 failed with error -5
Jun 12 19:13:29 rog-laptop kernel: [    3.206692] AMD-Vi: AMD IOMMUv2 driver by Joerg Roedel <jroedel@suse.de>
Jun 12 19:13:29 rog-laptop kernel: [    3.209335] kvm: Nested Virtualization enabled
Jun 12 19:13:29 rog-laptop kernel: [    3.209358] SVM: kvm: Nested Paging enabled
...
```
I have Windows 10 insalled on this PC as well, and Wi-Fi works fine there.
Fast boot is disabled. Secure boot disabled.

Any suggestions?

Roman-, 2021-06-12:
# Solved by powering off the laptop (instead of rebooting it)

Following [this](https://askubuntu.com/a/998450/597252) suggestion, I powered off the laptop (instead of rebooting it) and unplugged the power cable for a few minutes. Then the Wi-Fi adapter magically appeared.
If this is a hardware issue that can't be solved with the drivers, please close my issue.

Ciochirca, 2021-06-13:
Hello Roman,

I am fairly new to Ubuntu and Linux in general. I have the same problem on an ASUS TUF F15 and I took the same steps as you have by updating the kernel. For me, sometimes the Wi-Fi works, other times it does not. Apparently, as I found [here](https://forums.linuxmint.com/viewtopic.php?t=349575) there are no drivers yet available for our Wi-Fi adapter. I have disabled secure boot and fast boot as said there. Hopefully there will be a driver release soon. As I understand, sometimes if you turn off your laptop and then turn it back on without having a the charger plugged in, the adapter may work. You may also find useful information [here](https://askubuntu.com/questions/984303/no-wifi-adapter-found-dual-boot-windows-10-and-ubuntu-17-10/998450#998450). I hope this helps you. If you find better solutions, please contact me and let me know as I am in the same position.

Best wishes,
Teodor-Avram

Ciochirca, 2021-06-13:
BTW, is your trackpad working?

shizonic, 2021-06-18:
@Ciochirca My trackpad stops working from time to time...

Ciochirca, 2021-06-18:
@shizonic have you tried the instructions [here](https://wiki.ubuntu.com/DebuggingTouchpadDetection)? My issue is that apparently there is a bug in the Linux kernel, but for some reason, I can not report the problem. When I enter the command `ubuntu-bug linux` I get a pop up with the title "Problem in linux-image-5.12.10-051210-generic" containing the description "The problem can not be reported: This report is about a package that is not installed.". I have read on a couple of forums that it might be an xor problem, but I can not say for sure. I hope you will resolve your problem from the instructions provided by their team.

objelf, 2021-06-21:
> # Solved by powering off the laptop (instead of rebooting it)
> Following [this](https://askubuntu.com/a/998450/597252) suggestion, I powered off the laptop (instead of rebooting it) and unplugged the [...]

HI Roman, 

Could you help cherry-pick the patch "mt76: mt7921: reset wfsys during hw probe"  [1] onto your testing kernel source,
recompile the module and retry again ? 

[1] https://git.kernel.org/pub/scm/linux/kernel/git/kvalo/wireless-drivers-next.git/commit/drivers/net/wireless/mediatek/mt76/mt7921?id=01f7da40917923bf9d8fd8d5c9a6ed646004e47c

go2jimmys, 2021-07-03:
Hello I also experience the same problem with the same mt7921 on asus tuf f15. Everytime I reboot the wifi firmware fails to build and load and the wifi icon is missing from the top right corner. It works fine after a shutdown and then boot.

I tried blacklisting the mt7921 modules from modprobe on boot and run the modprobe command manually after boot. I get no errors on command line but again the module seems to initialize normally after a shutdown and does not work after a reboot (failed with error -5 on dmesg).

I will try to build the module using the patch mentioned by objelf and report back (probably tomorrow) 

`# modinfo mt7921e
filename:       /lib/modules/5.12.13-300.fc34.x86_64/kernel/drivers/net/wireless/mediatek/mt76/mt7921/mt7921e.ko.xz
license:        Dual BSD/GPL
author:         Lorenzo Bianconi <lorenzo@kernel.org>
author:         Sean Wang <sean.wang@mediatek.com>
firmware:       mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
firmware:       mediatek/WIFI_RAM_CODE_MT7961_1.bin
alias:          pci:v000014C3d00007961sv*sd*bc*sc*i*
depends:        mt76,mt76-connac-lib,mac80211,cfg80211
retpoline:      Y
intree:         Y
name:           mt7921e
vermagic:       5.12.13-300.fc34.x86_64 SMP mod_unload`

On my setup I have windows 10 dual boot on a separate EFI partition with fast boot disabled and secure boot disabled.

gasbi, 2021-07-15:
Hello, I am experiencing the same issue here with the mt7921e on a ROG G15 GA503QS, and I am on Fedora 34 with the custom kernel from asus-ctl (https://asus-linux.org). Setup with a dual boot with Windows 10.

Also, I never got the Bluetooth to work properly; does it work for any of you?

Let me know if I can do anything to help fix this issue, or if you find a solution!

```
modinfo mt7921e
filename:       /lib/modules/5.12.15-300.rog.fc34.x86_64/kernel/drivers/net/wireless/mediatek/mt76/mt7921/mt7921e.ko.xz
license:        Dual BSD/GPL
author:         Lorenzo Bianconi <lorenzo@kernel.org>
author:         Sean Wang <sean.wang@mediatek.com>
firmware:       mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
firmware:       mediatek/WIFI_RAM_CODE_MT7961_1.bin
alias:          pci:v000014C3d00007961sv*sd*bc*sc*i*
depends:        mt76,mt76-connac-lib,mac80211,cfg80211
retpoline:      Y
intree:         Y
name:           mt7921e
vermagic:       5.12.15-300.rog.fc34.x86_64 SMP mod_unload 
sig_id:         PKCS#7
signer:         Fedora kernel signing key
```

Andrey1994, 2021-07-17:
I have ASUS ROG Zephyrus G15, tried fedora 34 with dual boot with kernels 5.11 and 5.12, the same issue for both.(@Ciochirca touchpad didnt work for 5.11 for me, after update it was ok)

Moved to ubuntu 21, updated kernel to 5.13.2. Without power cable wifi works, if I connect a cable and reboot it doesnt.
Bluetooth doesnt work at all for all kernels, distributions, etc

lovesh, 2021-07-21:
> Hello Roman,
>
> I am fairly new to Ubuntu and Linux in general. I have the same problem on an ASUS TUF F15 and I took the same steps as you have by updating the kernel. For me, sometimes the Wi-Fi [...]

@Ciochirca @shizonic I have TUF F15 and cant get touchpad working. How did you guys resolve this?

krviolent, 2021-07-24:
I had got same issue on my Asus F15 as you, Roman, so powering off helped (Ubuntu 20.04 kernel 5.12.5).
But mainly I have issue with Bluetooth adapter, it doesn't work..

> hciconfig hci0
hci0: Type: Primary Bus: USB
BD Address: 00:00:00:00:00:00 ACL MTU: 0:0 SCO MTU: 0:0
DOWN
RX bytes:0 acl:0 sco:0 events:0 errors:0
TX bytes:3 acl:0 sco:0 commands:1 errors:0

Have anyone fased such problem?

divine-comedian, 2021-07-29:
I have the same issue on my ASUS A15 
```
$ inxi -nz
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet 
  driver: r8169 
  IF: enp2s0 state: down mac: <filter> 
  Device-2: MEDIATEK driver: N/A 
  IF-ID-1: usb0 state: unknown speed: -1 duplex: half mac: <filter> 
```
--- 
```
lspci -v 
03:00.0 Network controller: MEDIATEK Corp. Device 7961
	Subsystem: AzureWave Device 4680
	Flags: fast devsel, IRQ 91, IOMMU group 12
	Memory at fe20300000 (64-bit, prefetchable) [size=1M]
	Memory at fe20400000 (64-bit, prefetchable) [size=16K]
	Memory at fe20404000 (64-bit, prefetchable) [size=4K]
	Capabilities: <access denied>
	Kernel modules: mt7921e
```
The Wifi showed up on install of 21.04, I rebooted my computer and now no wi-fi

Gesrua, 2021-08-04:
I have the same issue on ASUS ROG Zephyrus M16. Changing my wifi adapter from MT7921 to AX200 doesn't help.

```
Power cable plugged. Boot Windows10 -> Shutdown Windows10 -> Boot Arch Linux -> No WiFi adapter
Power cable unplugged. Boot Windows10 -> Shutdown Windows10 -> Boot Arch Linux -> The WiFi adapter works well -> Plug the cable, reboot Arch Linux -> The WiFi adapter works well
```

Logs when No WiFi adapter. Line 1121
```
zephyrus kernel: Intel(R) Wireless WiFi driver for Linux	
zephyrus kernel: iwlwifi 0000:30:00.0: enabling device (0000 -> 0002)	
zephyrus kernel: i915 0000:00:02.0: [drm] VT-d active for gfx access	
zephyrus kernel: checking generic (4000000000 fa0000) vs hw (622c000000 1000000)	
zephyrus kernel: checking generic (4000000000 fa0000) vs hw (4000000000 10000000)	
zephyrus kernel: fb0: switching to inteldrmfb from EFI VGA	
zephyrus kernel: i915 0000:00:02.0: vgaarb: deactivate vga console	
zephyrus kernel: AVX2 version of gcm_enc/dec engaged.	
zephyrus kernel: AES CTR mode by8 optimization enabled	
zephyrus systemd[1]: Finished Monitoring of LVM2 mirrors, snapshots etc. using dmeventd or progress polling.	
zephyrus audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=lvm2-monitor comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'	
zephyrus systemd[1]: Reached target Preparation for Local File Systems.	
zephyrus kernel: ACPI: battery: new extension: ASUS Battery Extension	
zephyrus kernel: iwlwifi 0000:30:00.0: api flags index 2 larger than supported by driver	
zephyrus kernel: iwlwifi 0000:30:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37	
zephyrus kernel: iwlwifi 0000:30:00.0: loaded firmware version 63.c04f3485.0 cc-a0-63.ucode op_mode iwlmvm
```

Logs when the WiFi adapter works. Line 1111
```
zephyrus kernel: Intel(R) Wireless WiFi driver for Linux
zephyrus kernel: iwlwifi 0000:30:00.0: enabling device (0000 -> 0002)
zephyrus kernel: Creating 1 MTD partitions on "0000:00:1f.5":
zephyrus kernel: 0x000000000000-0x000001000000 : "BIOS"
zephyrus kernel: i915 0000:00:02.0: [drm] VT-d active for gfx access
zephyrus kernel: checking generic (4000000000 fa0000) vs hw (622c000000 1000000)
zephyrus kernel: checking generic (4000000000 fa0000) vs hw (4000000000 10000000)
zephyrus kernel: fb0: switching to inteldrmfb from EFI VGA
zephyrus kernel: i915 0000:00:02.0: vgaarb: deactivate vga console
zephyrus kernel: input: Asus WMI hotkeys as /devices/platform/asus-nb-wmi/input/input11
zephyrus kernel: usb 3-4: new low-speed USB device number 3 using xhci_hcd
zephyrus systemd[1]: Finished Monitoring of LVM2 mirrors, snapshots etc. using dmeventd or progress polling.
zephyrus audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=lvm2-monitor comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
zephyrus systemd[1]: Reached target Preparation for Local File Systems.
zephyrus kernel: iwlwifi 0000:30:00.0: api flags index 2 larger than supported by driver
zephyrus kernel: iwlwifi 0000:30:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37
zephyrus kernel: iwlwifi 0000:30:00.0: loaded firmware version 63.c04f3485.0 cc-a0-63.ucode op_mode iwlmvm
zephyrus kernel: i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
zephyrus kernel: i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/tgl_dmc_ver2_08.bin (v2.8)
```

[work.txt](https://github.com/openwrt/mt76/files/6931153/work.txt)
[notwork.txt](https://github.com/openwrt/mt76/files/6931155/notwork.txt)

lawRathod, 2021-09-18:
Experience very similar problem with arch on kernel 5.14.5. The kernel panics on reboots and when poweron. A solution that seems to be working is a hard shutdown and then boot.

Couldn't dump the panic logs in a text file. The logs in the image are fairly readable

[image]

9prady9, 2021-11-09:
I am currently facing identical issue with Wifi adapter on ASUS ROG G15: reboot fails to bring up wifi adapter on Arch Linux(kernel version 5.14). @divine-comedian I see you have G15, have you faced any ethernet related issues ? or Anyone with latest G15 models who has ethernet port, have you faced ethernet issues ?

Thank you

divine-comedian, 2021-11-09:
> I am currently facing identical issue with Wifi adapter on ASUS ROG G15: reboot fails to bring up wifi adapter on Arch Linux(kernel version 5.14). @divine-comedian I see you have G15, have you faced [...]

I have actually a TUF506QM but my issues are with wi-fi, bluetooth and powering back from sleep/hibernate. wired connections have been my only saviour often.

9prady9, 2021-11-09:
> I have actually a TUF506QM but my issues are with wi-fi, bluetooth and powering back from sleep/hibernate. wired connections have been my only saviour often.

Thanks for the prompt reply. Shutdown-start trick works for my wifi. I haven't had any luck with bluetooth though. My mouse pad's right click doesn't work. Those are my issues as far as I know at the moment. I believe my ethernet issue should be resolved once [this commit](https://github.com/torvalds/linux/commit/72f898ca0ab85fde6facf78b14d9f67a4a7b32d1#diff-9d61beebd4949ecbf8877624e7f25547715934c98cf968bed6e3a6d47d797a7f) is released in stable kernel - already tagged to 5.15.

go2jimmys, 2021-11-13:
So sorry, I didn't find the time to test and I installed windows on it. Now I installed ubuntu 21.10 on dual boot with UEFI and wifi works as intended. My kernel version is the following:
```$ uname -r
5.13.0-21-generic```

Bluetooth still doesn't work.

sharp2448, 2021-11-30:
@9prady9 I have also ASUS ROG G15 and yes I also have the issues with ethernet and wifi during arch(5.14.15-arch1-1) installation. 
Btw trick with power off and not have plugged in power cable works

9prady9, 2021-11-30:
> @9prady9 I have also ASUS ROG G15 and yes I also have the issues with ethernet and wifi during arch(5.14.15-arch1-1) installation. Btw trick with power off and not have plugged in power cable works

Glad to know that wifi trick works for you as well. Ethernet one should go away with 5.15 kernel as that added the newer card that is present in G15 to r8169 module https://github.com/torvalds/linux/commit/72f898ca0ab85fde6facf78b14d9f67a4a7b32d1#diff-9d61beebd4949ecbf8877624e7f25547715934c98cf968bed6e3a6d47d797a7f there is new fix to the same module in 5.16 soon to be released

FireBurn, 2021-12-02:
The issues with suspend and resume have been fixed in 5.16-rc1+ on, there's also a small patch that solves them on older kernels too 

I raised a bug about the coldboot wifi issue in the kernel bugzilla but there hasn't been much interaction with it

https://bugzilla.kernel.org/show_bug.cgi?id=214557

divine-comedian, 2021-12-02:
> The issues with suspend and resume have been fixed in 5.16-rc1+ on, there's also a small patch that solves them on older kernels too
>
> I raised a bug about the coldboot wifi issue in the kernel [...]

HUGE that's awesome 

I understand this isn't explicitly related to the topic but since we're discussing ASUS ubuntu issues - I updated to ubuntu 21.10 and the only working kernel I can seem to use is 5.12.16 - kernel 5.13 (default for Ub 21.10) leads to a blank/frozen terminal screen on load. kernel 5.15+ blank terminal screen as well. smells like a graphics driver issue, not sure if anyone has found the magic kernel past 5.12.16 to get this bad boy to load.

FireBurn, 2021-12-03:
> 

> HUGE that's awesome
>
> I understand this isn't explicitly related to the topic but since we're discussing ASUS ubuntu issues - I updated to ubuntu 21.10 and the only working kernel I can seem to use [...]

That's very weird, is that the AMD+AMD system or AMD+NVIDIA? If it's the former try and capture the dmesg and you can report the bug at https://gitlab.freedesktop.org/drm/amd/-/issues, if it's nVidia have you made sure you've got a matching nvidia module for that kernel?

divine-comedian, 2021-12-03:
@FireBurn 

Using the ubuntu drivers tool I updated the proprietary nvidia drivers from 470 -> 495 and now it works on the default kernel for impish indri (5.13)  but no further - (5.15+ still breaks).

FireBurn, 2021-12-03:
Probably best to raise a bug on the ubuntu bug tracker, give them your dmesg too, and just check that the kernel you've installed has a matching nvidia entry in /lib/modules/{kernel version}, at a guess I'd think that the nvidia proprietary module hasn't been generated for the matching version

pythys, 2021-12-09:
I face the same issue on artix-linux. It doesn't seem to be distro-specific since ubuntu was mentioned in this discussion

ljk5403, 2021-12-29:
Same issue on Arch Linux, with my Thinkbook 14 G3 ACL.

The Wifi adapter can not function after reboot, but when I put the laptop to sleep and reopen it, it works!
A shutdown and then boot also work.

Yey007, 2022-01-09:
The mt7921e is still not working for me on Pop!_OS 20.10 (kernel 5.15) on the ASUS M16.

chiefdinkan, 2022-01-11:
I face the same issue on fedora 35(kernal 5.15) on the asus tuf f15 laptop

pythys, 2022-01-11:
Can we know if this is a kernel issue, driver issue, combo? The reason I ask is to know whether we should continue reporting here or move the discussion elsewhere?

wjes, 2022-01-11:
Had the same issue on a Thinkpad L14 with Arch and `linux-lts 5.10.90-1`. Installed `linux 5.15.13` and now it works.

vlad2005, 2022-01-15:
I have an Asus Vivobook pro 15, and mediatek WiFi ( mt7921e driver).
With kernel 5.15.13 WiFi and Bluetooth work perfect but after suspend WiFi not working. Also if I try to shutdown, not work and I need to press power button for few seconds to force shutdown.
I tried to install version 5.16 but failed to start with it.
I have dual boot with win10, I try it with fast boot disabled and enabled, no difference.
Only solution for moment is to disable suspend and hibernate.

ljk5403, 2022-01-18:
> Same issue on Arch Linux, with my Thinkbook 14 G3 ACL. (also has mt7921e)
> 
> The Wifi adapter can not function after reboot, but when I put the laptop to sleep and reopen it, it works! A shutdown and then boot also work.

Now with kernel 5.16.1-zen1-1-zen, a sleep would lead to a crash, and `modprobe -r mt7921e` would not work.

```
Jan 18 09:36:42 JKLiTB sddm[483]: Loading theme configuration from "/usr/share/sddm/themes/breeze>
Jan 18 09:36:42 JKLiTB sddm[483]: Greeter starting...
Jan 18 09:36:42 JKLiTB sddm[483]: Signal received: SIGTERM
Jan 18 09:36:42 JKLiTB sddm[483]: Socket server stopping...
Jan 18 09:36:42 JKLiTB sddm[483]: Socket server stopped.
Jan 18 09:36:42 JKLiTB sddm[483]: Display server stopping...
Jan 18 09:36:42 JKLiTB sddm-helper[4106]: [PAM] Starting...
Jan 18 09:36:42 JKLiTB audit[4106]: USER_AUTH pid=4106 uid=0 auid=4294967295 ses=4294967295 msg='>
Jan 18 09:36:42 JKLiTB audit[4106]: USER_ACCT pid=4106 uid=0 auid=4294967295 ses=4294967295 msg='>
Jan 18 09:36:42 JKLiTB sddm-helper[4106]: [PAM] Authenticating...
Jan 18 09:36:42 JKLiTB sddm-helper[4106]: [PAM] returning.
Jan 18 09:36:43 JKLiTB sddm[483]: Display server stopped.
Jan 18 09:36:43 JKLiTB sddm[483]: Running display stop script  "/usr/share/sddm/scripts/Xstop"
Jan 18 09:36:43 JKLiTB sddm[483]: QProcess: Destroyed while process ("/usr/lib/sddm/sddm-helper")>
Jan 18 09:36:43 JKLiTB systemd[1]: sddm.service: Deactivated successfully.
Jan 18 09:36:43 JKLiTB systemd[1]: Stopped Simple Desktop Display Manager.
Jan 18 09:36:43 JKLiTB audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='uni>
Jan 18 09:36:42 JKLiTB sddm[483]: Loading theme configuration from "/usr/share/sddm/themes/breeze/theme.conf"
Jan 18 09:36:42 JKLiTB sddm[483]: Greeter starting...
Jan 18 09:36:42 JKLiTB sddm[483]: Signal received: SIGTERM
Jan 18 09:36:42 JKLiTB sddm[483]: Socket server stopping...
Jan 18 09:36:42 JKLiTB sddm[483]: Socket server stopped.
Jan 18 09:36:42 JKLiTB sddm[483]: Display server stopping...
Jan 18 09:36:42 JKLiTB sddm-helper[4106]: [PAM] Starting...
Jan 18 09:36:42 JKLiTB audit[4106]: USER_AUTH pid=4106 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:authentication grantors=pam_permit acct="sddm" exe="/usr/lib/sddm/sddm-helper" hostname=? add>
Jan 18 09:36:42 JKLiTB audit[4106]: USER_ACCT pid=4106 uid=0 auid=4294967295 ses=4294967295 msg='op=PAM:accounting grantors=pam_permit acct="sddm" exe="/usr/lib/sddm/sddm-helper" hostname=? addr=? >
Jan 18 09:36:42 JKLiTB sddm-helper[4106]: [PAM] Authenticating...
Jan 18 09:36:42 JKLiTB sddm-helper[4106]: [PAM] returning.
Jan 18 09:36:43 JKLiTB sddm[483]: Display server stopped.
Jan 18 09:36:43 JKLiTB sddm[483]: Running display stop script  "/usr/share/sddm/scripts/Xstop"
Jan 18 09:36:43 JKLiTB sddm[483]: QProcess: Destroyed while process ("/usr/lib/sddm/sddm-helper") is still running.
Jan 18 09:36:43 JKLiTB systemd[1]: sddm.service: Deactivated successfully.
Jan 18 09:36:43 JKLiTB systemd[1]: Stopped Simple Desktop Display Manager.
Jan 18 09:36:43 JKLiTB audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=sddm comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
Jan 18 09:36:43 JKLiTB systemd[1]: sddm.service: Consumed 23.655s CPU time.
Jan 18 09:36:43 JKLiTB kernel: mt7921e 0000:05:00.0: Message 80000010 (seq 4) timeout
Jan 18 09:36:43 JKLiTB kernel: mt7921e 0000:05:00.0: Failed to get patch semaphore
Jan 18 09:36:43 JKLiTB kernel: mt7921e 0000:05:00.0: chip reset failed
Jan 18 09:36:46 JKLiTB wpa_supplicant[934]: Could not set interface wlan0 flags (UP): Connection timed out
Jan 18 09:36:46 JKLiTB kernel: mt7921e 0000:05:00.0: Message 00000046 (seq 5) timeout
Jan 18 09:36:46 JKLiTB kernel: mt7921e 0000:05:00.0: chip reset
Jan 18 09:36:46 JKLiTB wpa_supplicant[934]: WEXT: Could not set interface 'wlan0' UP
Jan 18 09:36:46 JKLiTB wpa_supplicant[934]: wlan0: Failed to initialize driver interface
Jan 18 09:36:46 JKLiTB wpa_supplicant[934]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Jan 18 09:36:52 JKLiTB wpa_supplicant[934]: Could not set interface wlan0 flags (UP): Connection timed out
Jan 18 09:36:52 JKLiTB kernel: mt7921e 0000:05:00.0: Message 00000046 (seq 6) timeout
Jan 18 09:36:55 JKLiTB kernel: mt7921e 0000:05:00.0: Message 80000010 (seq 7) timeout
Jan 18 09:36:55 JKLiTB kernel: mt7921e 0000:05:00.0: Failed to get patch semaphore
```

ljk5403, 2022-01-18:
A simple but dirty way to fix: put this file in `/usr/lib/systemd/system-sleep/`, name it as `mt7921e_fix` or whatever.

```bash
#!/bin/sh

# manually unload and load mt7921e to avoid problem after sleep

case $1 in
    pre)  modprobe -r mt7921e  ;;
    post) modprobe mt7921e  ;;
esac
```

This file automatically unload the mt7961e before sleep and load it back after waking up.

vlad2005, 2022-01-18:
Tested with 5.16.1 and still not work wifi after suspend.
`A simple but dirty way to fix: put this file in /usr/lib/systemd/system-sleep/, name it as mt7921e_fix or whatever.`
I know about this solution but in my case not work (kernel 5.15.13 or 14 I don't remember exactly). 
What version of the kernel works best for you with the above solution?

ljk5403, 2022-01-18:
> Tested with 5.16.1 and still not work wifi after suspend. `A simple but dirty way to fix: put this file in /usr/lib/systemd/system-sleep/, name it as mt7921e_fix or whatever.` I know about this [...]

I mainly use linux-zen 5.16.1.zen1-1, and I did a test with linux 5.16.1.arch1-1, both works.
By the way, I am using Arch Linux and only sleep works, I have fatal problem with hibernate.

Veirt, 2022-01-18:
I have a MSI Modern 14 B5m laptop and running a dual boot (Windows and Arch). 
I managed to make my Wi-Fi worked by following this [answer](https://askubuntu.com/a/1380495/1152688) before. 

But recently after I updated my kernel to `5.16.arch1-1`, it keeps making the kernel to panic (my `journalctl -b -1` logged `mt7921e 0000:01:00.0: Timeout for driver own`). I'm currently running the linux-lts kernel version `5.15.15-1`, although sometimes, it randomly stopped working. But I could fix it by updating my laptop firmware on Windows.

vlad2005, 2022-01-18:
I use Ubuntu 20.04, and for me, removing kernel module not work.
I test first before any suspend (fresh start). Remove kernel module, then load again, everything work perfect.
Then I remove kernel, go to suspend and back, But when i try to load kernel module nothing happens.

tuwuna, 2022-01-20:
# After 6 months they finally fixed it

Kernel patch: https://patchwork.kernel.org/project/linux-wireless/patch/70e27cbc652cbdb78277b9c691a3a5ba02653afb.1641540175.git.objelf@gmail.com/

FireBurn, 2022-01-20:
The kernel refuses to boot with this applied

[image]

tuwuna, 2022-01-20:
@FireBurn I added a warning

It seems like things are finally moving quite a bit for the mediatek driver: https://patchwork.kernel.org/project/linux-mediatek/list/?submitter=169671. It should be fixed in the next release cycle.

For ubuntu folks, the fix is included in the staging firmware https://bugs.launchpad.net/ubuntu/+source/linux-firmware/+bug/1958286

vlad2005, 2022-01-20:
How to test this firmware, or when will be available for non geeks?
I test until now kernels from here: https://kernel.ubuntu.com/~kernel-ppa/mainline/
but it's a bit to much to download source code, apply patch, compile, ......

HelloEdit, 2022-01-29:
Some users (myself included) are seeing their system stability fall apart with the latest kernel 5.16.3 update on arch https://bbs.archlinux.org/viewtopic.php?pid=2019011

vlad2005, 2022-01-30:
For me (Ubuntu 20.04) the latest kernel versions don't even boot (5.16.3, 5.16.4, 5.15.16-18). I even updated from 20.04 to 21.10, but without success.

9prady9, 2022-01-30:
I have `5.16.3-arch1-1` now on Arch Linux with that update what used to work stopped working completely now

aelzenaar, 2022-01-30:
On Arch too with an Asus TUF506LH, 5.16.3 is totally broken for me (it panics 50% of the time on boot due to the mt7921e driver and the other 50% it boots but the network stack seems to be completely broken), I tried the lts kernel (5.15.17) and it is also not working (now it boots almost every time but mt7921e is still broken)!

I burned a new live disc yesterday and when booting from there I could connect to wifi using iwd; rebooting back into my permanent install then messes up the wlan chip state, and if I boot back into the live disc iwd no longer works. To get it to work again I have to reset the firmware (unplug adapter & hold power button for 60sec, as in https://forums.linuxmint.com/viewtopic.php?t=349575).

In any case here is another relevant bug report not linked above. https://lore.kernel.org/all/YfY+C9hiX2V7LnT6@kroah.com/t/

rnestler, 2022-02-01:
> On Arch too with an Asus TUF506LH, 5.16.3 is totally broken for me (it panics 50% of the time on boot due to the mt7921e driver and the other 50% it boots but the network stack seems to be completely broken),

@aelzenaar I'm on arch as well and `5.16.3` and `5.16.4` were completely broken for me on the a Lenovo P14s. The mt7921e WiFi didn't work and the amdgpu caused issues as well, including troubles entering standby.

> I tried the lts kernel (5.15.17) and it is also not working (now it boots almost every time but mt7921e is still broken)!

`5.15.18-1-lts` works for me, it boots without problems and wifi and amdgpu seem to work without problems so far.

aelzenaar, 2022-02-01:
@rnestler 

I am now on 5.15.18 and still having wifi problems, it now works on boot but after a random period of time it just stops -  for instance if I run `ping 8.8.8.8` it displays no output for 2min or so before failing with "destination host unreadable", and if I try to use the plasma network manager to disconnect and reconnect from the wifi it hangs and the fan spins up to full. After this, poweroff hangs with a stop job waiting for dhcpcd and NetworkManager and on reboot I need to reset the firmware again before wifi will start working again. No problem with the GPU so far as I can see though, but I don't use standby mode that often.

benat12, 2022-02-03:
> I have an Asus Vivobook pro 15, with mediatek WiFi ( mt7921e driver). With kernel 5.15.13, WiFi and Bluetooth work perfect but after suspend WiFi not working. Bluetooth continue to work as expected after suspend. Also if I try to shutdown, not work and I need to press power button for few seconds to force shutdown. I tried to install version 5.16 but failed to start with it. I have dual boot with win10, I try it with fast boot disabled and enabled, no difference. Only solution for moment is to disable suspend and hibernate.

Hi! I actually have the same problem on the same computer. I've tried different solutions and used newer kernels, but as I am just starting with Ubuntu many of the fixes in this issue are beyond my understanding. If you've (or any other) found a way to solving this, please write back as I'm currently forced to be plugged in via ethernet or switching off and on the computer every time I want it to suspend.

aelzenaar, 2022-02-03:
@benat12  Try downgrading to 5.14, that should fix it.

benat12, 2022-02-04:
@aelzenaar Thanks for the help, but either I'm doing something wrong as I install the kernels (I'm using ubuntu mainline) or the problem arises from another place. I've recently seen that not only does the wifi not work, but when I suspend the device it slows down heavily, I can't do any sudo commands and generaly the only option is a complete shut off by pressing the power button, as the power off option in ubuntu needs a long time to shut things off.

aelzenaar, 2022-02-04:
@benat12  As you see above this is a known problem with this hardware and recent kernels, if the problem persists after downgrading I suggest you open a post on e.g. the Ubuntu forums and provide your dmesg output & other logs there.

JuanuSt, 2022-02-05:
System Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: ASUS TUF Gaming A17 FA706QM_FA706QM
        
Kernel 
Linux  5.12.15-051215-generic #202107080949-Ubuntu

Error
dmesg: mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin failed with error -2

Solution
Download 3 files from https://anduin.linuxfromscratch.org/sources/linux-firmware/mediatek/
and copy to firmware directory

cd /lib/firmware/mediatek/
wget https://anduin.linuxfromscratch.org/sources/linux-firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin
wget https://anduin.linuxfromscratch.org/sources/linux-firmware/mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
wget https://anduin.linuxfromscratch.org/sources/linux-firmware/mediatek/WIFI_RAM_CODE_MT7961_1.bin

One year ago I resolved the wifi problem with the steps described above. This dirty hack works fine with wifi (still you have to shutdown/start to make the wifi works) but not with bluetooth.

It looks like is loaded, but not:

[   10.387820] Bluetooth: Core ver 2.22
[   10.387854] NET: Registered protocol family 31
[   10.387855] Bluetooth: HCI device and connection manager initialized
[   10.387860] Bluetooth: HCI socket layer initialized
[   10.387863] Bluetooth: L2CAP socket layer initialized
[   10.387867] Bluetooth: SCO socket layer initialized
[   10.400462] loop6: detected capacity change from 0 to 113680
[   10.400581] loop7: detected capacity change from 0 to 333552
...
[   10.449316] mt7921e 0000:03:00.0: enabling device (0000 -> 0002)
[   10.449812] mt7921e 0000:03:00.0: disabling ASPM  L1
[   10.449882] mt7921e 0000:03:00.0: ASIC revision: 79610010
[   10.459479] MCE: In-kernel MCE decoding enabled.
[   10.459965] mt7921e 0000:03:00.0: HW/SW Version: 0x8a108a10, Build Time: 20211014150838a

EnlightenedBacon, 2022-02-05:
Just adding that Kernel 5.16.5 still has the same Kernel Panic issue unfortunately.

vlad2005, 2022-02-05:
I tested the kernel version 5.15.20. The problem remained the same. Wifi does not work if I suspend my laptop.
Otherwise, wifi and bluetooth work properly.

pythys, 2022-02-06:
I did not do any testing yet, but it seems some [fix](https://github.com/torvalds/linux/commit/1d1c950faa81e1c287c9e14f307f845b190eb578) has been made to 5.16. I don't know if we'll see the light at the end of the tunnel but it might be worth testing for anyone with free time.

chabad360, 2022-02-08:
On Arch (5.16.7) on an ASUS TUF F15 (`FX506HM`/`TUF506HM`), the WiFi works, but if I reboot I get a kernel panic (haven't tested sleep yet), however, I'm not sure it's always consistent (I just got this laptop yesterday).

9prady9, 2022-02-08:
> On Arch (5.16.7) on an ASUS TUF F15 (`FX506HM`/`TUF506HM`), the WiFi works, but if I reboot I get a kernel panic (haven't tested sleep yet), however, I'm not sure it's always consistent (I just got this laptop yesterday).

Just updated to 5.16.7 , I can confirm similar behavior. It doesn't work on reboot.

EnlightenedBacon, 2022-02-08:
I have this problem too on an ASUS Zephyrus S17. I don't know if this helps at all, but I suspect it isn't just an issue with the MediaTek wifi adaptor, but may be related to that adaptor and something with the BIOS in ASUS (and maybe MSI) laptops.

I swapped out my MT7921 for an Intel AX200 and while it worked perfectly while the computer was on in any BIOS version, occasionally it just wouldn't appear on boot and then need a hard reset to appear again.

Both wifi adaptors worked perfectly with Windows 10, so not a hardware issue. Perhaps the reason the bug is so obscure is because it's an interaction with some other specific device, rather than the adaptor on its own that's the problem?

aelzenaar, 2022-02-08:
Definitely not a hardware issue, I had no trouble with the mt7921 driver before updating my kernel to the 5.16 series in January.

[134 more replies]