openwrt/mt76 #549: MT7628NN: Wi-Fi clients randomly unable to connect

anyuta1166, 2021-06-13:
I have TP-Link TL-WR842N v5 (MediaTek MT7628NN) router running OpenWrt 21.02.0-rc2.
My Wi-Fi setup is a simple WPA2-PSK + CCMP access point. I've faced the issue with Wi-Fi. Sometimes some clients cannot connect to AP. More precisely speaking, they connect and then disconnect after a few seconds. System log shows successful connection and then "hostapd: deauthenticated due to local deauth request" exactly 10 seconds later.
When this happens, all other currently connected clients can disconnect and reconnect successfully. Only one client cannot connect.
This happens to different clients every time. For example, yesterday my notebook couldn't connect, today my phone cannot connect, but the notebook connects successfully and so on.
Restarting Wi-Fi doesn't help. Rebooting the router does help for some time. After a reboot all clients can successfully connect but the issue repeats some time later.
This happens only with this MediaTek based router. I have another 3 routers with Qualcomm Atheros Wi-Fi hardware in a different location and these routers never had this issue. That's why I think that this may be mt76 issue.

anyuta1166, 2021-06-13:
Initially I thought that there may be a limit on the number of client connections, but after disconnecting another client the affected client still stuck in authenticated -> deauthenticated loop.

anyuta1166, 2021-06-13:
I can also confirm that this is not related to the poor signal strength. Moving the affected device to be near to the router didn't help too. It is still being connected and disconnected after 10 seconds.

anvnhcm, 2021-06-14:
[image]

Hi all, 
Image show that, wifi was connected but mobile can not access the internet. But we can use LAN to access the internet.

It happens some Samsung model.

Wifi signal is lower than Stock firmware about 10 dB. Padavan firmware get the highest power signal.

Regards,

anyuta1166, 2021-06-24:
Encountered a crash today on OpenWrt SNAPSHOT r16962-a603e82dd3:
```
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.752053] ------------[ cut here ]------------
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.756805] WARNING: CPU: 0 PID: 8778 at target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2021-06-06-22b69033/mt7603/mac.c:206 mt7603_filter_tx+0x16c/0x174 [mt7603e]
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.771820] Modules linked in: xt_connlimit qcserial pppoe ppp_async option nf_conncount l2tp_ppp ipw iptable_nat cdc_mbim xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_FLOWOFFLOAD xt_CT usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nf_nat_ftp nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_netlink nf_conntrack_ftp nf_conntrack mt7603e mt76 mac80211 kalmia ipt_REJECT huawei_cdc_ncm cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem xt_u32 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_hashlimit xt_esp xt_ecn xt_dscp xt_comment xt_cgroup xt_bpf xt_addrtype xt_TPROXY xt_TEE xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY usbserial usbnet ts_kmp ts_fsm ts_bm slhc nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_dup_ipv6 nf_dup_ipv4 nf_defrag_ipv6
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.772038]  nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ipt_rpfilter ipt_ah ipt_ECN ipheth ip6t_rpfilter ip_tables hso crc_ccitt compat cdc_wdm cdc_acm ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip_gre gre sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel tunnel4 ip_tunnel tun sha1_generic ecb arc4 leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common mii aead cryptomgr crypto_null crypto_hash
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.937560] CPU: 0 PID: 8778 Comm: kworker/u3:2 Not tainted 5.4.124 #0
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.944190] Workqueue: napi_workq napi_workfn
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.948615] Stack : 804ce668 833e44ac 00000009 00000005 00000005 80058800 83e7e040 00000000
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.957099]         00000017 82c8243c 804bf630 82179bf4 80530000 00000001 82179bc8 7a184408
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.965573]         00000000 00000000 8054f280 00000000 716b726f 000000f7 6f775f69 6e666b72
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.974061]         00000000 00000231 00000000 000e683e 00000000 00000009 00000000 833e44ac
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.982548]         00000009 00000005 00000005 833f0000 00000000 80276b94 00000000 80690000
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.991031]         ...
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.993507] Call Trace:
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11294.996001] [<80009c60>] show_stack+0x30/0x100
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.000523] [<80023a78>] __warn+0xc0/0x10c
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.004674] [<80023b20>] warn_slowpath_fmt+0x5c/0xac
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.009743] [<833e44ac>] mt7603_filter_tx+0x16c/0x174 [mt7603e]
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.015760] [<833e4644>] mt7603_wtbl_set_ps+0x12c/0x134 [mt7603e]
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.021972] [<833e1a1c>] mt7603_sta_ps+0x38/0x428 [mt7603e]
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.027662] [<82c547e4>] mt76_rx_poll_complete+0x2e0/0x5b0 [mt76]
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.033857] [<82c51530>] mt76_dma_rx_poll+0x3cc/0xeec [mt76]
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.039621] [<802f3e4c>] __napi_poll+0x3c/0x10c
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.044213] [<802f3f5c>] napi_workfn+0x40/0x90
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.048750] [<8003a40c>] process_one_work+0x220/0x480
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.053878] [<8003a7d4>] worker_thread+0x168/0x5c8
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.058753] [<8003f3d8>] kthread+0x138/0x140
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.063083] [<80005158>] ret_from_kernel_thread+0x14/0x1c
Thu Jun 24 13:31:47 2021 kern.warn kernel: [11295.068570] ---[ end trace bf1af95c53611112 ]---
```

valga, 2021-11-16:
@anyuta1166 Does downgrade to 19.07.2 solve your issue?

sanitariu, 2021-11-16:
I am at revision: r18070-49f615022c. Still have the same problem. It happens on heavy wifi traffic. Wifi just stops. Disconnects all clients, then disappears. After approx. 1 minute it comes back. There is nothing in the log file.

valga, 2021-11-16:
@sanitariu I had the same issue with 21.02.1 and solved it by downgrading to 19.07.2.

More details [here](https://github.com/openwrt/mt76/issues/569#issuecomment-970072115).