openwrt/mt76 #550: Two MT7612U working in AP mode at the same time cause a crash

guoxiaoqiao, 2021-06-16:
[ 5485.697438] skbuff: skb_under_panic: text:ecd4927a len:375 put:4 head:8c2ecd5c data:4972c483 tail:0xe9fc0070 end:0xe9fc0f00 dev:wlan1
[ 5485.697530] Internal error: Oops - undefined instruction: 0 [#1] SMP ARM
[ 5485.708434] Modules linked in: pppoe ppp_async iptable_nat ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt76_usb mt76 mac80211 ipt_REJECT cfg80211 cdc_ncm xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG usbnet slhc r8152 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon exfat crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 vfat fat ntfs autofs4 nls_utf8 nls_iso8859_1 nls_cp437 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 mii crc32c_generic
[ 5485.768759] CPU: 0 PID: 470 Comm: mt76-tx phy1 Not tainted 5.4.124 #0
[ 5485.790987] Hardware name: Generic DT based system
[ 5485.797505] PC is at skb_panic+0x5c/0x60
[ 5485.802181] LR is at skb_panic+0x5c/0x60
[ 5485.806262] pc : [<c07301a0>]    lr : [<c07301a0>]    psr: 60000013
[ 5485.810168] sp : ee245d70  ip : 00000000  fp : 00000002
[ 5485.816158] r10: edfe9ee0  r9 : edf9e944  r8 : 000000f3
[ 5485.821366] r7 : 00000000  r6 : bf0e0d90  r5 : eb84a9c0  r4 : e9fc0070
[ 5485.826577] r3 : ef6b5650  r2 : 2eb77000  r1 : ef6b029c  r0 : 00000079
[ 5485.833177] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[ 5485.839687] Control: 10c5387d  Table: ad36806a  DAC: 00000051
[ 5485.846892] Process mt76-tx phy1 (pid: 470, stack limit = 0xcd2e0aab)
[ 5485.852622] Stack: (0xee245d70 to 0xee246000)
[ 5485.859045] 5d60:                                     00000004 e9fc0000 e9fbfffe e9fc0070
[ 5485.863391] 5d80: e9fc0f00 ef336000 edfe9ee0 c07301e4 05080174 bf0e0d68 edf9e940 edfe9ee0
[ 5485.871554] 5da0: bf0e0d90 bf38d714 ee47c748 ee245db4 ef0652c8 00000000 00000000 00000000
[ 5485.879710] 5dc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5485.887872] 5de0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5485.896029] 5e00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5485.904191] 5e20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5485.912348] 5e40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5485.920512] 5e60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5485.928667] 5e80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5485.936831] 5ea0: 00000000 00000000 00000000 00000000 00000000 eb84a9c0 00000000 00000000
[ 5485.944988] 5ec0: 00000012 ee47c748 edf9e940 bf38d69c edfe9ee0 edf9e944 edfe9ee0 bf380198
[ 5485.953150] 5ee0: ee47c748 ef1ae300 ee47c850 ee4790a4 edf9e940 eb84a9c0 ee47c850 bf380ca0
[ 5485.961309] 5f00: ee47c748 ee245f2f ef6b6a00 00000001 edfe9ee0 edfe9ee0 00000002 edfeaee0
[ 5485.969469] 5f20: 00000002 00000000 c03247b4 00b77000 60000013 00000002 edfea110 00000002
[ 5485.977627] 5f40: edfe9ee0 ee244000 bf37b454 ee17dd1c ee0854dc bf380ed4 00000000 00000001
[ 5485.985789] 5f60: edfeb118 edfeb110 ee244000 bf37b4cc ee0854c0 ee29b980 00000000 ee244000
[ 5485.993946] 5f80: edfeb110 c033e3e0 0000000d ee29b980 c033e294 00000000 00000000 00000000
[ 5486.002109] 5fa0: 00000000 00000000 00000000 c03010e8 00000000 00000000 00000000 00000000
[ 5486.010264] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 5486.018428] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[ 5486.026593] [<c07301a0>] (skb_panic) from [<c07301e4>] (__skb_checksum+0x0/0x338)
[ 5486.034747] [<c07301e4>] (__skb_checksum) from [<00000000>] (0x0)
[ 5486.042210] Code: e34cc09a 11a0c00e e58dc014 ebf0d2c1 (e7f001f2) 
[ 5486.048279] ---[ end trace 1ac96b6ac65266f7 ]---
[ 5486.054354] Kernel panic - not syncing: Fatal exception in interrupt
[ 5486.059057] CPU1: stopping
[ 5486.065381] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D           5.4.124 #0
[ 5486.067898] Hardware name: Generic DT based system
[ 5486.075459] [<c030e144>] (unwind_backtrace) from [<c030ab60>] (show_stack+0x10/0x14)
[ 5486.080062] [<c030ab60>] (show_stack) from [<c0876640>] (dump_stack+0x8c/0xa0)
[ 5486.087960] [<c0876640>] (dump_stack) from [<c030d418>] (handle_IPI+0x168/0x19c)
[ 5486.094989] [<c030d418>] (handle_IPI) from [<c05bd2fc>] (gic_handle_irq+0x8c/0x90)
[ 5486.102543] [<c05bd2fc>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[ 5486.109914] Exception stack(0xef095f60 to 0xef095fa8)
[ 5486.117469] 5f60: 00000000 0801c194 ef6c6234 c0313a20 ef094000 00000001 c0c04e68 c0c04ea8
[ 5486.122508] 5f80: 00000000 c0b3ede8 00000000 ef095fb8 2eb86000 ef095fb0 c030808c c0308090
[ 5486.130659] 5fa0: 60000013 ffffffff
[ 5486.138823] [<c0301a8c>] (__irq_svc) from [<c0308090>] (arch_cpu_idle+0x38/0x3c)
[ 5486.142127] [<c0308090>] (arch_cpu_idle) from [<c034b464>] (do_idle+0x21c/0x270)
[ 5486.149760] [<c034b464>] (do_idle) from [<c034b748>] (cpu_startup_entry+0x18/0x1c)
[ 5486.157136] [<c034b748>] (cpu_startup_entry) from [<803024ac>] (0x803024ac)
[ 5486.164514] CPU2: stopping
[ 5486.171370] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G      D           5.4.124 #0
[ 5486.174146] Hardware name: Generic DT based system
[ 5486.181706] [<c030e144>] (unwind_backtrace) from [<c030ab60>] (show_stack+0x10/0x14)
[ 5486.186310] [<c030ab60>] (show_stack) from [<c0876640>] (dump_stack+0x8c/0xa0)
[ 5486.194205] [<c0876640>] (dump_stack) from [<c030d418>] (handle_IPI+0x168/0x19c)
[ 5486.201236] [<c030d418>] (handle_IPI) from [<c05bd2fc>] (gic_handle_irq+0x8c/0x90)
[ 5486.208788] [<c05bd2fc>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[ 5486.216161] Exception stack(0xef097f60 to 0xef097fa8)
[ 5486.223716] 7f60: 00000000 067dc254 ef6d5234 c0313a20 ef096000 00000002 c0c04e68 c0c04ea8
[ 5486.228755] 7f80: 00000000 c0b3ede8 00000000 ef097fb8 2eb95000 ef097fb0 c030808c c0308090
[ 5486.236910] 7fa0: 60000013 ffffffff
[ 5486.245069] [<c0301a8c>] (__irq_svc) from [<c0308090>] (arch_cpu_idle+0x38/0x3c)
[ 5486.248372] [<c0308090>] (arch_cpu_idle) from [<c034b464>] (do_idle+0x21c/0x270)
[ 5486.256010] [<c034b464>] (do_idle) from [<c034b748>] (cpu_startup_entry+0x18/0x1c)
[ 5486.263391] [<c034b748>] (cpu_startup_entry) from [<803024ac>] (0x803024ac)
[ 5486.270765] CPU3: stopping
[ 5486.277619] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G      D           5.4.124 #0
[ 5486.280396] Hardware name: Generic DT based system
[ 5486.287958] [<c030e144>] (unwind_backtrace) from [<c030ab60>] (show_stack+0x10/0x14)
[ 5486.292557] [<c030ab60>] (show_stack) from [<c0876640>] (dump_stack+0x8c/0xa0)
[ 5486.300458] [<c0876640>] (dump_stack) from [<c030d418>] (handle_IPI+0x168/0x19c)
[ 5486.307486] [<c030d418>] (handle_IPI) from [<c05bd2fc>] (gic_handle_irq+0x8c/0x90)
[ 5486.315036] [<c05bd2fc>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[ 5486.322411] Exception stack(0xef099f60 to 0xef099fa8)
[ 5486.329969] 9f60: 00000000 06cd0618 ef6e4234 c0313a20 ef098000 00000003 c0c04e68 c0c04ea8
[ 5486.335005] 9f80: 00000000 c0b3ede8 00000000 ef099fb8 2eba4000 ef099fb0 c030808c c0308090
[ 5486.343159] 9fa0: 60000013 ffffffff
[ 5486.351318] [<c0301a8c>] (__irq_svc) from [<c0308090>] (arch_cpu_idle+0x38/0x3c)
[ 5486.354627] [<c0308090>] (arch_cpu_idle) from [<c034b464>] (do_idle+0x21c/0x270)
[ 5486.362260] [<c034b464>] (do_idle) from [<c034b748>] (cpu_startup_entry+0x18/0x1c)
[ 5486.369637] [<c034b748>] (cpu_startup_entry) from [<803024ac>] (0x803024ac)
[ 5486.377021] Rebooting in 3 seconds..

CPU：Qualcomm IPQ4019
OpenWrt Version: 21.02 RC2
Kernel Version: 5.4.124

guoxiaoqiao, 2021-06-16:
Only connect to one of the MT7612U APs, run stable, if connect to two MT7612U APs at the same time, and download data, it will crash soon

guoxiaoqiao, 2021-06-23:
root@OpenWrt:/# [ 2969.801595] skbuff: skb_under_panic: text:79286df7 len:372 put:4 head:fe3b9854 data:bcfe866f tail:0xebd24070 end:0xebd24f00 dev:wlan2
[ 2969.801690] Internal error: Oops - undefined instruction: 0 [#1] SMP ARM
[ 2969.812585] Modules linked in: pppoe ppp_async iptable_nat ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt76_usb mt76 mac80211 ipt_REJECT cfg80211 cdc_ncm xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG usbnet slhc r8152 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon exfat crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 vfat fat ntfs autofs4 nls_utf8 nls_iso8859_1 nls_cp437 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 mii crc32c_generic
[ 2969.872907] CPU: 1 PID: 521 Comm: mt76-tx phy2 Not tainted 5.4.124 #0
[ 2969.895135] Hardware name: Generic DT based system
[ 2969.901649] PC is at skb_panic+0x70/0x74
[ 2969.906330] LR is at __irq_work_queue_local+0x40/0x7c
[ 2969.910410] pc : [<c07d6274>]    lr : [<c03f89d4>]    psr: 60000013
[ 2969.915361] sp : ed92dcc8  ip : ed92dbf0  fp : ed92dcec
[ 2969.921435] r10: 0000015a  r9 : ecef0748  r8 : ebd24030
[ 2969.926642] r7 : ebd24016  r6 : edb99ee0  r5 : ed2b66c0  r4 : ed67c000
[ 2969.931851] r3 : ef6b8650  r2 : 2eb4a000  r1 : ef6b329c  r0 : 00000079
[ 2969.938452] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[ 2969.944959] Control: 10c5387d  Table: ade2006a  DAC: 00000051
[ 2969.952164] Process mt76-tx phy2 (pid: 521, stack limit = 0xe8031aa8)
[ 2969.957895] Stack: (0xed92dcc8 to 0xed92e000)
[ 2969.964319] dcc0:                   00000004 ebd24000 ebd23ffe ebd24070 ebd24f00 ed67c000
[ 2969.968665] dce0: ed92dd04 ed92dcf0 c07d62c8 c07d6210 05080170 ed2b66c0 ed92dd1c ed92dd08
[ 2969.976825] dd00: bf209ef4 c07d6284 ecef0850 ed92dd6c ed92dd5c ed92dd20 bf20a088 bf209ebc
[ 2969.984988] dd20: ecef0748 0000015a 40c00050 00000000 edd10d00 edce9040 edb99ee0 bf209f1c
[ 2969.993146] dd40: 00000035 00000000 ed2b66c0 edb99ee0 ed92dea4 ed92dd60 bf42d730 bf209f28
[ 2970.001307] dd60: ecef0748 ed92dd6c ee389000 00000000 00000000 00000000 00000000 00000000
[ 2970.009463] dd80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2970.017628] dda0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2970.025785] ddc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2970.033942] dde0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2970.042104] de00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2970.050261] de20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2970.058424] de40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2970.066582] de60: 00000000 00000000 00000000 ed2b66c0 00000000 00000000 ed92de78 ecef0748
[ 2970.074743] de80: 00000012 edce9040 bf42d6a8 edb9aee0 ecef40a4 edb99ee0 ed92decc ed92dea8
[ 2970.082907] dea0: bf424cf0 bf42d6b4 ecef0748 ecef0850 edce9040 ed2b66c0 ecef0850 edce9044
[ 2970.091066] dec0: ed92df2c ed92ded0 bf4256b0 bf424cb8 ecef0748 ed92deff ed92defc 00000000
[ 2970.099224] dee0: 00000002 c03eaa38 edb99ee0 edb99ee0 00000002 ed92df00 c095195c 00329864
[ 2970.107383] df00: ed92df2c 00000002 edb9a110 00000002 edb99ee0 00000001 edbaac9c edb9b110
[ 2970.115539] df20: ed92df4c ed92df30 bf425b00 bf425520 00000000 edb9b118 ed92c000 edb9b110
[ 2970.123705] df40: ed92df74 ed92df50 bf41f6c4 bf425acc edbaac80 edce5c00 00000000 ed92c000
[ 2970.131859] df60: ede8fb34 edbaac9c ed92dfac ed92df78 c03461e8 bf41f648 ffffe000 bf41f63c
[ 2970.140021] df80: ed92dfac edce5c00 c034607c 00000000 00000000 00000000 00000000 00000000
[ 2970.148178] dfa0: 00000000 ed92dfb0 c03010b4 c0346088 00000000 00000000 00000000 00000000
[ 2970.156337] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2970.164498] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[ 2970.172647] Backtrace: 
[ 2970.180813] [<c07d6204>] (skb_panic) from [<c07d62c8>] (__skb_checksum+0x0/0x34c)
[ 2970.183086] [<c07d6278>] (skb_push) from [<bf209ef4>] (mt76x02u_skb_dma_info+0x44/0x6c [mt76x02_usb])
[ 2970.190714]  r5:ed2b66c0 r4:05080170
[ 2970.199917] [<bf209eb0>] (mt76x02u_skb_dma_info [mt76x02_usb]) from [<bf20a088>] (mt76x02u_tx_prepare_skb+0x16c/0x10e4 [mt76x02_usb])
[ 2970.203563]  r5:ed92dd6c r4:ecef0850
[ 2970.215457] [<bf209f1c>] (mt76x02u_tx_prepare_skb [mt76x02_usb]) from [<bf42d730>] (mt76u_tx_queue_skb+0x88/0x1a8 [mt76_usb])
[ 2970.219141]  r10:edb99ee0 r9:ed2b66c0 r8:00000000 r7:00000035 r6:bf209f1c r5:edb99ee0
[ 2970.230294]  r4:edce9040
[ 2970.238124] [<bf42d6a8>] (mt76u_tx_queue_skb [mt76_usb]) from [<bf424cf0>] (__mt76_set_tx_blocked+0xc8/0x148 [mt76])
[ 2970.240718]  r10:edb99ee0 r9:ecef40a4 r8:edb9aee0 r7:bf42d6a8 r6:edce9040 r5:00000012
[ 2970.251207]  r4:ecef0748
[ 2970.258956] [<bf424cac>] (__mt76_set_tx_blocked [mt76]) from [<bf4256b0>] (mt76_tx_status_check+0x208/0x44c [mt76])
[ 2970.261548]  r7:edce9044 r6:ecef0850 r5:ed2b66c0 r4:edce9040
[... cut ...]
[ 2970.833246] [<c0355b94>] (do_idle) from [<c03560f0>] (cpu_startup_entry+0x28/0x2c)
[ 2970.840626]  r10:00000000 r9:410fc075 r8:8020406a r7:c0c5e308 r6:10c0387d r5:00000002
[ 2970.847995]  r4:0000008a
[ 2970.855894] [<c03560c8>] (cpu_startup_entry) from [<c030f29c>] (secondary_start_kernel+0x15c/0x164)
[ 2970.858511] [<c030f140>] (secondary_start_kernel) from [<8030280c>] (0x8030280c)
[ 2970.867269]  r5:00000051 r4:af12006a
[ 2970.874913] Rebooting in 3 seconds..

I found only one adapter can cause cash, this crash log inluce trace info

LorenzoBianconi, 2021-06-24:
@guoxiaoqiao do you mean two AP vif on the same adapter or 2 due different phy?
what do you mean with "I found only one adapter can cause cash"?
I have just tested using mt76x0u with 2 AP vifs on the same phy and it is working fine

guoxiaoqiao, 2021-06-29:
Sorry, I can’t reproduce this crash now, I’m not sure if it’s my hardware problem

closed 2021-11-25 by nbd168

esp0xdeadbeef, 2026-11-15:
This one is likely the same root cause as #711: `skb_under_panic` in the mt76 USB tx path, i.e. an skb handed down without enough headroom, and `skb_push()` walking below `skb->head`.

For mt7925 that is `mt7925_usb_sdio_tx_prepare_skb()` missing the `skb_cow_head()` that its mt7921 sibling has. If you can still reproduce, the fix in #711 applies the same pattern:

```c
err = skb_cow_head(skb, MT_SDIO_TXD_SIZE + MT_SDIO_HDR_SIZE);
if (err)
	return err;
```

Patch for mt7925 (adjust the constants for mt7612 if the layout differs):
https://github.com/esp0xdeadbeef/nixos/blob/cca1932f/patches/mt7925-tx-headroom.patch

esp0xdeadbeef, 2026-11-15:
Correction: the function lives in `mt7925/mac.c`, not `mt7925/usb.c`.

Current mainline, for reference:
- `mt7921/mac.c:781` in `mt7921_usb_sdio_tx_prepare_skb()` has `skb_cow_head(skb, MT_SDIO_TXD_SIZE + MT_SDIO_HDR_SIZE)` before the push
- `mt7925/mac.c:1645` in `mt7925_usb_sdio_tx_prepare_skb()` does not