openwrt/mt76 #551: mt76-tx phy1 Tainted

siwind, 2021-06-21:
My mt7621 router ran into ph1 tained status after a long period of  download data.
Following is the log:

```
#dmesg
...
[106420.982549] ------------[ cut here ]------------
[106420.987432] WARNING: CPU: 1 PID: 996 at backports-5.10.42-1/net/mac80211/airtime.c:457 ieee80211_ocb_leave+0x564/0x644 [mac80211]
[106420.999460] Modules linked in: pppoe ppp_async wireguard pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc poly1305_mips nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_netlink nf_conncount macvlan libcurve25519_generic libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat chacha_mips ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet
[106420.999645]  ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_udp_tunnel udp_tunnel tun vfat fat ntfs autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ctr cmac ccm uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 exfat(C) usbcore nls_base usb_common crc32c_generic
[106421.157244] CPU: 1 PID: 996 Comm: mt76-tx phy1 Tainted: G         C        5.4.124 #0
[106421.165137] Stack : 00000000 8007b218 806e0000 806e53cc 80750000 806e5394 806e42fc 8dcedc94
[106421.173561]         80880000 8fdf03a4 8072ed63 80676d68 00000001 00000001 8dcedc38 3d2e2d7b
[106421.181988]         00000000 00000000 808c0000 00000000 30000000 00000193 00000018 20202047
[106421.190401]         00000000 000014ae 00000000 20432020 00000000 80750000 00000000 8eb57cb0
[106421.198816]         00000009 00000014 00000004 00000002 00000008 80374fc4 00000004 80880004
[106421.207233]         ...
[106421.209758] Call Trace:
[106421.212316] [<8000b64c>] show_stack+0x30/0x100
[106421.216843] [<805b96d0>] dump_stack+0xa4/0xdc
[106421.221286] [<8002c18c>] __warn+0xc0/0x10c
[106421.225453] [<8002c234>] warn_slowpath_fmt+0x5c/0xac
[106421.230658] [<8eb57cb0>] ieee80211_ocb_leave+0x564/0x644 [mac80211]
[106421.237152] ---[ end trace 9f60eee69308e792 ]---
[109117.870999] mt7530 mdio-bus:1f lan3: Link is Down
[109117.876626] br-lan: port 3(lan3) entered disabled state
[109132.430215] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control rx/tx
[109132.438002] br-lan: port 3(lan3) entered blocking state
[109132.443352] br-lan: port 3(lan3) entered forwarding state
[109133.469824] mt7530 mdio-bus:1f lan3: Link is Down
[109133.474852] br-lan: port 3(lan3) entered disabled state
[109143.869967] mt7530 mdio-bus:1f lan3: Link is Up - 10Mbps/Full - flow control rx/tx
[109143.877692] br-lan: port 3(lan3) entered blocking state
[109143.883034] br-lan: port 3(lan3) entered forwarding state
[109144.909844] mt7530 mdio-bus:1f lan3: Link is Down
[109144.914896] br-lan: port 3(lan3) entered disabled state
[109151.150015] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[109151.157636] br-lan: port 3(lan3) entered blocking state
[109151.162996] br-lan: port 3(lan3) entered forwarding state
......
```

And hardware and software:
```
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.124 #0 SMP Wed Jun 16 12:14:33 2021 mips GNU/Linux
root@OpenWrt:~# opkg list-installed | grep mt76
kmod-mt76-core - 5.4.124+2021-06-06-22b69033-4
kmod-mt7603 - 5.4.124+2021-06-06-22b69033-4
kmod-mt76x02-common - 5.4.124+2021-06-06-22b69033-4
kmod-mt76x2 - 5.4.124+2021-06-06-22b69033-4
kmod-mt76x2-common - 5.4.124+2021-06-06-22b69033-4
root@OpenWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10
```

Router is  newifi-d2 with openwrt master branch.

nekos, 2021-07-08:
i have the same issue with cudy wr1300 (openwrt master branch), wifi hang for seconds then recover by himself.

root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.128 #0 SMP Mon Jun 28 15:24:15 2021 mips GNU/Linux

[83713.702321] ------------[ cut here ]------------
[83713.707120] WARNING: CPU: 3 PID: 1023 at backports-5.10.42-1/net/mac80211/airtime.c:457 ieee80211_ocb_leave+0x564/0x644 [mac80211]
[83713.718882] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc sch_cake nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac ghash_generic
[83713.719170]  gf128mul gcm ctr cmac ccm leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[83713.824279] CPU: 3 PID: 1023 Comm: mt76-tx phy1 Not tainted 5.4.128 #0
[83713.830781] Stack : 00000000 8007ddb0 80740000 8073bc38 807a0000 8073bc00 8073ab68 8601dc94
[83713.839108]         808e0000 87eb03a8 80788d63 806c9c50 00000003 00000001 8601dc38 ce0a5a21
[83713.847431]         00000000 00000000 80920000 00000000 00000030 00000153 746e6961 35206465
[83713.855752]         00000000 00001045 00000000 000c93d7 00000000 807a0000 00000000 86757cb0
[83713.864071]         00000009 00000014 00000004 00000002 00000001 8038e07c 0000000c 808e000c
[83713.872396]         ...
[83713.874836] Call Trace:
[83713.877299] [<8000b64c>] show_stack+0x30/0x100
[83713.881743] [<80605f88>] dump_stack+0xa4/0xdc
[83713.886101] [<8002c130>] __warn+0xc0/0x10c
[83713.890188] [<8002c1d8>] warn_slowpath_fmt+0x5c/0xac
[83713.895283] [<86757cb0>] ieee80211_ocb_leave+0x564/0x644 [mac80211]
[83713.902142] ---[ end trace bd0eb20e33d61eca ]---

kofec, 2021-08-10:
I see the same:

`2021-08-09T09:21:49.370842+00:00 Xiaomi3G kernel: [228635.161866] ------------[ cut here ]------------
2021-08-09T09:21:49.370872+00:00 Xiaomi3G kernel: [228635.174743] WARNING: CPU: 1 PID: 899 at backports-5.10.42-1/net/mac80211/airtime.c:457 ieee80211_ocb_leave+0x564/0x644 [mac80211]
2021-08-09T09:21:49.518149+00:00 Xiaomi3G kernel: [228635.261613] Stack : 00000000 8007de7c 80650000 8064e090 806b0000 8064e058 8064d0d4 8e4b7c94
2021-08-09T09:21:49.518259+00:00 Xiaomi3G kernel: [228635.270044]         807f0000 8febf428 806966a3 805e69d4 00000001 00000001 8e4b7c38 e3365d63
2021-08-09T09:21:49.518457+00:00 Xiaomi3G kernel: [228635.278473]         00000000 00000000 80830000 00000000 30000000 0000016d 00000018 35206465
2021-08-09T09:21:49.518497+00:00 Xiaomi3G kernel: [228635.286918]         00000000 00002c6f 00000000 746d203a 00000000 806b0000 00000000 8e657080
2021-08-09T09:21:49.518523+00:00 Xiaomi3G kernel: [228635.295381]         00000009 000000e9 00000004 00000000 00000008 80350774 00000004 807f0004
2021-08-09T09:21:49.518711+00:00 Xiaomi3G kernel: [228635.303814]         ...
2021-08-09T09:21:49.518743+00:00 Xiaomi3G kernel: [228635.306348] Call Trace:
2021-08-09T09:21:49.518828+00:00 Xiaomi3G kernel: [228635.308906] [<8000b64c>] show_stack+0x30/0x100
2021-08-09T09:21:49.518856+00:00 Xiaomi3G kernel: [228635.313467] [<8052fd34>] dump_stack+0xa4/0xdc
2021-08-09T09:21:49.518889+00:00 Xiaomi3G kernel: [228635.317944] [<8002c140>] __warn+0xc0/0x10c
2021-08-09T09:21:49.518911+00:00 Xiaomi3G kernel: [228635.322121] [<8002c1e8>] warn_slowpath_fmt+0x5c/0xac
2021-08-09T09:21:49.518934+00:00 Xiaomi3G kernel: [228635.327324] [<8e657080>] ieee80211_ocb_leave+0x564/0x644 [mac80211]
2021-08-09T09:21:49.519044+00:00 Xiaomi3G kernel: [228635.334129] ---[ end trace f1dbb4c173992ad1 ]---
`

siwind, 2021-09-05:
The problem appear again with  HWNAT(Hardware offload enable):

```
[21915.846073] mt7530 mdio-bus:1f lan3: Link is Down
[21915.851292] br-lan: port 3(lan3) entered disabled state
[21917.926273] mt7530 mdio-bus:1f lan3: Link is Up - 10Mbps/Full - flow control off
[21917.933726] br-lan: port 3(lan3) entered blocking state
[21917.939000] br-lan: port 3(lan3) entered forwarding state
[21950.165197] mt7530 mdio-bus:1f lan3: Link is Down
[21950.170060] br-lan: port 3(lan3) entered disabled state
[22862.093904] ------------[ cut here ]------------
[22862.098745] WARNING: CPU: 2 PID: 1078 at backports-5.10.42-1/net/mac80211/airtime.c:456 ieee80211_ocb_leave+0x23c/0x38c [mac80211]
[22862.110532] Modules linked in: pppoe ppp_async wireguard pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc poly1305_mips nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323
[22862.110714]  nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat chacha_mips asn1_decoder ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsd msdos ip6_udp_tunnel udp_tunnel rpcsec_gss_krb5 auth_rpcgss oid_registry tun vfat fat ntfs lockd sunrpc grace autofs4 nls_utf8 nls_iso8859_1 nls_cp936 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac
[22862.196592]  ghash_generic gf128mul gcm ecb des_generic libdes cts ctr cmac ccm cbc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 exfat(C) usbcore nls_base usb_common crc32c_generic
[22862.307205] CPU: 2 PID: 1078 Comm: mt76-tx phy1 Tainted: G         C        5.4.142 #0
[22862.315095] Stack : 00000000 80079a04 80700000 806ffb0c 80760000 806ffad4 806fec64 8ee47c94
[22862.323422]         808a0000 80690c04 8ffab0a4 8074ad63 00000002 00000001 8ee47c38 ec83f854
[22862.331743]         00000000 00000000 808e0000 00000000 00000030 0000017c 20202020 35202020
[22862.340065]         00000000 8ee47b61 ffffffff ffffffff 00000000 00000001 00000000 8e458f14
[22862.348392]         00000009 00000014 00000004 00000002 00000001 803845d8 00000008 808a0008
[22862.356714]         ...
[22862.359147] Call Trace:
[22862.361611] [<8000b48c>] show_stack+0x30/0x100
[22862.366061] [<805d8300>] dump_stack+0x9c/0xc8
[22862.370415] [<8002ba54>] __warn+0xcc/0x118
[22862.374501] [<8002bafc>] warn_slowpath_fmt+0x5c/0xac
[22862.379600] [<8e458f14>] ieee80211_ocb_leave+0x23c/0x38c [mac80211]
[22862.385985] ---[ end trace fcb393c5990da00f ]---
```

The driver is  mt76-07-15 as following:
```
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.142 #0 SMP Tue Aug 31 14:15:59 2021 mips GNU/Linux
root@OpenWrt:~# opkg list | grep mt76
kmod-mt76-core - 5.4.142+2021-07-15-bbebea7d-4
kmod-mt7603 - 5.4.142+2021-07-15-bbebea7d-4
kmod-mt76x02-common - 5.4.142+2021-07-15-bbebea7d-4
kmod-mt76x2 - 5.4.142+2021-07-15-bbebea7d-4
kmod-mt76x2-common - 5.4.142+2021-07-15-bbebea7d-4
```

easyteacher, 2021-09-06:
2.4GHz WiFi disappears after these messages.

> [272540.999849] ------------[ cut here ]------------
> [272541.004626] WARNING: CPU: 3 PID: 1068 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2021-07-15-bbebea7d/mt7603/mac.c:206 mt7603_filter_tx+0x16c/0x174 [mt7603e]
> [272541.019741] Modules linked in: pppoe ppp_async wireguard pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 lz4 libchacha20poly1305 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211 cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota2 xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_ipp2p xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm tcp_bbr slhc sch_cake poly1305_mips nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_flow_table nf_conntrack_netlink nf_conncount macvlan lz4_decompress lz4_compress
> [272541.020662]  libcurve25519_generic libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 crc_ccitt compat_xtables compat chacha_mips fuse sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_csum libcrc32c act_connmark sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4
> [272541.108931]  ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel tun nls_utf8 nls_cp950 nls_cp936 zram zsmalloc sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 mbcache jbd2 usbcore nls_base usb_common mii crc32c_generic crc32_generic
> [272541.248108] CPU: 3 PID: 1068 Comm: napi/phy0-7 Not tainted 5.10.52 #0
> [272541.254651] Stack : 80756cb0 839a0000 00000001 80081c5c 80860000 80757c10 00000000 00000000
> [272541.263142]         82f79c64 809b0000 81c261c8 807ecde7 80725f68 00000001 82f79c08 ff4e4248
> [272541.271605]         00000000 00000000 80725f68 82f79aa8 ffffefff 808567e4 00000000 00000000
> [272541.280061]         00000000 000034f7 00000000 0003c92c 00000000 80860000 00000000 80720000
> [272541.288515]         00000009 00000005 00000005 839a0000 00000000 803ef7dc 0000000c 809b000c
> [272541.296970]         ...
> [272541.299508] Call Trace:
> [272541.302070] [<80008108>] show_stack+0x30/0x100
> [272541.306604] [<80372bac>] dump_stack+0xa4/0xe0
> [272541.311061] [<8002d884>] __warn+0xc0/0xe8
> [272541.315151] [<8002d908>] warn_slowpath_fmt+0x5c/0xac
> [272541.320237] [<839944ec>] mt7603_filter_tx+0x16c/0x174 [mt7603e]
> [272541.326244] [<83994688>] mt7603_wtbl_set_ps+0x130/0x138 [mt7603e]
> [272541.332420] [<83991a34>] mt7603_sta_ps+0x38/0x430 [mt7603e]
> [272541.338098] [<839b4744>] mt76_rx_poll_complete+0x2cc/0x59c [mt76]
> [272541.344280] [<839b15b8>] mt76_dma_rx_poll+0x3c4/0xd7c [mt76]
> [272541.350056] [<804d7f8c>] __napi_poll+0x3c/0x154
> [272541.354669] [<804d8164>] napi_threaded_poll+0xc0/0x128
> [272541.359888] [<8004d478>] kthread+0x13c/0x144
> [272541.364239] [<80003038>] ret_from_kernel_thread+0x14/0x1c
> [272541.369709] 
> [272541.371442] ---[ end trace 77f95f2224e36093 ]---
> [284968.476930] sched: RT throttling activated
> [434259.443402] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
> [434259.448982] mt76x2e 0000:01:00.0: Build: 1
> [434259.453533] mt76x2e 0000:01:00.0: Build Time: 201507311614____
> [434259.482091] mt76x2e 0000:01:00.0: Firmware running!
> [434259.492362] ieee80211 phy1: Hardware restart was requested
> [700144.654471] ------------[ cut here ]------------
> [700144.659652] WARNING: CPU: 1 PID: 1097 at backports-5.10.42-1/net/mac80211/airtime.c:457 ieee80211_ocb_leave+0x55c/0x63c [mac80211]
> [700144.671922] Modules linked in: pppoe ppp_async wireguard pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 lz4 libchacha20poly1305 libblake2s ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211 cdc_ether xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota2 xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_ipp2p xt_hl xt_helper xt_hashlimit xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TPROXY xt_TCP [... cut ...]
> [700144.944279]         00000009 00000014 00000004 00000002 00000000 803ef7dc 00000004 809b0004
> [700144.952790]         ...
> [700144.955345] Call Trace:
> [700144.957908] [<80008108>] show_stack+0x30/0x100
> [700144.962496] [<80372bac>] dump_stack+0xa4/0xe0
> [700144.967001] [<8002d884>] __warn+0xc0/0xe8
> [700144.971101] [<8002d908>] warn_slowpath_fmt+0x5c/0xac
> [700144.976323] [<83959380>] ieee80211_ocb_leave+0x55c/0x63c [mac80211]
> [700144.982788] 
> [700144.984724] ---[ end trace 77f95f2224e36094 ]---

siwind, 2021-09-21:
After a heavy download,   this appear again:

```
[56494.639616] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[56494.647138] br-lan: port 3(lan3) entered blocking state
[56494.652429] br-lan: port 3(lan3) entered forwarding state
[56510.238989] mt7530 mdio-bus:1f lan3: Link is Down
[56510.243825] br-lan: port 3(lan3) entered disabled state
[56513.359297] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[56513.366829] br-lan: port 3(lan3) entered blocking state
[56513.372112] br-lan: port 3(lan3) entered forwarding state
[63805.490250] ------------[ cut here ]------------
[63805.495046] WARNING: CPU: 1 PID: 1078 at backports-5.10.42-1/net/mac80211/airtime.c:456 ieee80211_ocb_leave+0x23c/0x38c [mac80211]
[63805.506788] Modules linked in: pppoe ppp_async wireguard pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc poly1305_mips nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323
[63805.506997]  nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libblake2s_generic iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat chacha_mips asn1_decoder ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsd msdos ip6_udp_tunnel udp_tunnel rpcsec_gss_krb5 auth_rpcgss oid_registry tun vfat fat ntfs lockd sunrpc grace autofs4 nls_utf8 nls_iso8859_1 nls_cp936 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac
[63805.592764]  ghash_generic gf128mul gcm ecb des_generic libdes cts ctr cmac ccm cbc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 exfat(C) usbcore nls_base usb_common crc32c_generic
[63805.702824] CPU: 1 PID: 1078 Comm: mt76-tx phy1 Tainted: G         C        5.4.142 #0
[63805.710733] Stack : 00000000 80079a04 80700000 806ffb0c 80760000 806ffad4 806fec64 8e311c94
[63805.719070]         808a0000 80690c04 8fe90ee4 8074ad63 00000001 00000001 8e311c38 904a82e7
[63805.727405]         00000000 00000000 808e0000 00000000 00000030 00000174 20202020 35202020
[63805.735737]         00000000 8e311b61 ffffffff ffffffff 00000000 00000001 00000000 8e458f14
[63805.744072]         00000009 00000014 00000004 00000002 00000001 803845d8 00000004 808a0004
[63805.752400]         ...
[63805.754843] Call Trace:
[63805.757310] [<8000b48c>] show_stack+0x30/0x100
[63805.761770] [<805d8300>] dump_stack+0x9c/0xc8
[63805.766136] [<8002ba54>] __warn+0xcc/0x118
[63805.770228] [<8002bafc>] warn_slowpath_fmt+0x5c/0xac
[63805.775336] [<8e458f14>] ieee80211_ocb_leave+0x23c/0x38c [mac80211]
[63805.781742] ---[ end trace 984476f7a9352e72 ]---
[69601.470488] mt7530 mdio-bus:1f lan3: Link is Down
[69601.475606] br-lan: port 3(lan3) entered disabled state
[69603.550515] mt7530 mdio-bus:1f lan3: Link is Up - 10Mbps/Full - flow control off
[69603.558030] br-lan: port 3(lan3) entered blocking state
[69603.563349] br-lan: port 3(lan3) entered forwarding state
[70540.569339] mt7530 mdio-bus:1f lan3: Link is Down
[70540.574703] br-lan: port 3(lan3) entered disabled state
[70542.649485] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
[70542.657050] br-lan: port 3(lan3) entered blocking state
[70542.662369] br-lan: port 3(lan3) entered forwarding state
[70562.408783] mt7530 mdio-bus:1f lan3: Link is Down
[70562.413623] br-lan: port 3(lan3) entered disabled state
[70565.528973] mt7530 mdio-bus:1f lan3: Link is Up - 10Mbps/Full - flow control off
[70565.536410] br-lan: port 3(lan3) entered blocking state
[70565.541701] br-lan: port 3(lan3) entered forwarding state
[71524.387371] mt7530 mdio-bus:1f lan3: Link is Down
[71524.392924] br-lan: port 3(lan3) entered disabled state
[96841.594804] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off
[96841.602326] br-lan: port 3(lan3) entered blocking state
[96841.607589] br-lan: port 3(lan3) entered forwarding state
[96862.394137] mt7530 mdio-bus:1f lan3: Link is Down
[96862.399258] br-lan: port 3(lan3) entered disabled state
[96865.514278] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control rx/tx
```

LearZhou, 2021-10-14:
Similar issue with my D-Link DIR-860L B1, running OpenWrt SNAPSHOT, r17750-213ce1d837:
```
[66043.471224] ------------[ cut here ]------------
[66043.480655] WARNING: CPU: 2 PID: 956 at backports-5.10.68-1/net/mac80211/airtime.c:456 ieee80211_ocb_leave+0x240/0x38c [mac80211]
[66043.503881] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ctr cmac ccm leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[66043.642134] CPU: 2 PID: 956 Comm: mt76-tx phy1 Not tainted 5.4.152 #0
[66043.654947] Stack : 00000000 8007bba8 80750000 8074929c 807b0000 80749264 80748430 86751c94
[66043.671571]         808f0000 806d850c 86b85808 80796d63 00000002 00000001 86751c38 e096fe20
[66043.688194]         00000000 00000000 80930000 00000000 30232032 0000013b 65746e69 2e352064
[66043.704818]         00000000 86751b59 ffffffff ffffffff 00000000 00000001 00000000 86659578
[66043.721439]         00000009 00000014 00000004 00000002 00000000 80396744 00000008 808f0008
[66043.738062]         ...
[66043.742919] Call Trace:
[66043.747803] [<8000b318>] show_stack+0x30/0x100
[66043.756652] [<80619ff0>] dump_stack+0x9c/0xc8
[66043.765330] [<8002b808>] __warn+0xc0/0x10c
[66043.773467] [<8002b8b0>] warn_slowpath_fmt+0x5c/0xac
[66043.783476] [<86659578>] ieee80211_ocb_leave+0x240/0x38c [mac80211]
[66043.796772] ---[ end trace 543c041620db15b9 ]---
```

laigor, 2021-10-15:
cat /proc/cpuinfo 
system type		: MediaTek MT7621 ver:1 eco:3
machine			: Xiaomi Mi Router 4A Gigabit Edition
OPENWRT_RELEASE="OpenWrt 21.02-SNAPSHOT r0-e7c5e08"
```[112886.348085] ------------[ cut here ]------------
[112886.352991] WARNING: CPU: 2 PID: 882 at backports-5.10.68-1/net/mac80211/airtime.c:457 ieee80211_ocb_leave+0x564/0x644 [mac80211]
[112886.364751] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mac80211 libchacha20poly1305 libblake2s iptable_raw iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake poly1305_mips nf_reject_ipv4 nf_log_ipv4 nf_log_common nf_defrag_ipv4 libcurve25519_generic libblake2s_generic hwmon crypto_hw_eip93 crc_ccitt compat chacha_mips sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev ifb
[112886.364967]  udp_tunnel tun sha256_generic libsha256 sha1_generic md5 des_generic libdes authenc leds_gpio gpio_button_hotplug
[112886.465143] CPU: 2 PID: 882 Comm: mt76-tx phy1 Tainted: G        W         5.4.150 #0
[112886.473033] Stack : 00000000 8007dea0 80660000 8065a9a8 806c0000 8065a970 806599ec 86e8fc94
[112886.481451]         807f0000 87f27428 806a26a3 805f2400 00000002 00000001 86e8fc38 0289a392
[112886.489872]         00000000 00000000 80830000 00000000 30000000 00000153 00000018 20202047
[112886.498301]         00000000 000015f0 00000000 20205720 00000000 806c0000 00000000 86957060
[112886.506720]         00000009 00000014 00000004 00000002 00000008 803503f0 00000008 807f0008
[112886.515135]         ...
[112886.517667] Call Trace:
[112886.520218] [<8000b64c>] show_stack+0x30/0x100
[112886.524762] [<8053b5d0>] dump_stack+0xa4/0xdc
[112886.529215] [<8002c140>] __warn+0xc0/0x10c
[112886.533384] [<8002c1e8>] warn_slowpath_fmt+0x5c/0xac
[112886.538566] [<86957060>] ieee80211_ocb_leave+0x564/0x644 [mac80211]
[112886.545580] ---[ end trace d783939086ce889a ]---
```

siwind, 2021-10-16:
Problem again with latest version:

```
[30094.405412] ------------[ cut here ]------------
[30094.414874] WARNING: CPU: 2 PID: 1027 at backports-5.10.68-1/net/mac80211/airtime.c:456 ieee80211_ocb_leave+0x240/0x38c [mac80211]
[30094.438496] Modules linked in: pppoe ppp_async wireguard pptp pppox ppp_mppe ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_state xt_socket xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_FLOWOFFLOAD usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc poly1305_mips nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_netlink macvlan libcurve25519_generic libblake2s_generic iptable_nat iptable_mangle iptable_filter ip_tables crc_ccitt compat chacha_mips xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark
[30094.438673]  ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig msdos ip_gre gre ip6_udp_tunnel udp_tunnel ip_tunnel tun vfat fat ntfs autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ecb ctr cmac ccm arc4 usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd uhci_hcd ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat(C) btrfs xor raid6_pq libcrc32c usbcore nls_base usb_common crc32c_generic
[30094.750711] CPU: 2 PID: 1027 Comm: mt76-tx phy1 Tainted: G         C        5.4.152 #0
[30094.766486] Stack : 00000000 8007eba4 80700000 807070cc 80770000 80707094 80706260 8deb9c94
[30094.783133]         808b0000 80699764 8ebcab08 80752d03 00000002 00000001 8deb9c38 e102afac
[30094.799770]         00000000 00000000 808f0000 00000000 00000030 00000194 20202020 35202020
[30094.816394]         00000000 8deb9b59 ffffffff ffffffff 00000000 00000001 00000000 8e059578
[30094.833025]         00000009 00000014 00000004 00000002 00000001 803934fc 00000008 808b0008
[30094.849658]         ...
[30094.854530] Call Trace:
[30094.859420] [<8000b318>] show_stack+0x30/0x100
[30094.868317] [<805e00c8>] dump_stack+0x9c/0xc8
[30094.877009] [<8002e804>] __warn+0xc0/0x10c
[30094.885154] [<8002e8ac>] warn_slowpath_fmt+0x5c/0xac
[30094.895198] [<8e059578>] ieee80211_ocb_leave+0x240/0x38c [mac80211]
[30094.908168] ---[ end trace bd7f73c72428fe1f ]---
```

The latest  openwrt is:

```
root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.152 #0 SMP Sat Oct 16 02:31:59 2021 mips GNU/Linux
root@OpenWrt:~# opkg list | grep mt76
kmod-mt76-core - 5.4.152+2021-10-12-83598c2e-4
kmod-mt7603 - 5.4.152+2021-10-12-83598c2e-4
kmod-mt76x02-common - 5.4.152+2021-10-12-83598c2e-4
kmod-mt76x2 - 5.4.152+2021-10-12-83598c2e-4
kmod-mt76x2-common - 5.4.152+2021-10-12-83598c2e-4
root@OpenWrt:~# 
```

freed00m, 2022-01-27:
Is someone working on this? Or should we start editing mt76 devices on the wiki as not supported for v20 and v21?

candiesdoodle, 2022-01-28:
Has anyone tried with snapshot with kernel v 5.10 or higher? read on the forum that it solved some issues related to speed, not necessarily related to this, but maybe worth a try?

freed00m, 2022-01-28:
I am testing it now.

```
Model	Asus RT-AC51U
Architecture	MediaTek MT7620A ver:2 eco:6
Target Platform	ramips/mt7620
Firmware Version	OpenWrt SNAPSHOT r18647-17135ae091 / LuCI Master git-22.025.79016-22e2bfb
Kernel Version	5.10.92
```

Will see what happens, running it as dumbAP and turned of some services like led and firewall as they are not needed :)

FusionDragon, 2022-02-07:
I'm running a weird setup where i have the router connected to another router via wifi, but 1 antenna is connected to the 5ghz (6220>main) and the 2.4ghz to devices on my side of the house. all in all it works great except when i get that error and 1 of 2 things happen, the radio crashes and reboots or the whole router crashes and restarts. i noticed this mostly happens during the daytime when there is a lot of activity. but at night it's usually perfectly fine.

FusionDragon, 2022-02-07:
I also noticed this interesting line, even though the tx power doesn't actually go up to 30 for me because the max is 25 for 2.4g and 23 for 5g.

> Mon Feb  7 04:33:10 2022 kern.debug kernel: [ 4772.971485] INTERNET: Limiting TX power to 30 (30 - 0)

freed00m, 2022-02-07:
@FusionDragon - this is reported I think already in some other issue.

gitnewcomer, 2022-02-13:
> I am testing it now.
> 
> ```
> Model	Asus RT-AC51U
> Architecture	MediaTek MT7620A ver:2 eco:6
> Target Platform	ramips/mt7620
> Firmware Version	OpenWrt SNAPSHOT r18647-17135ae091 / LuCI Master git-22.025.79016-22e2bfb
> Kernel Version	5.10.92
> ```
> 
> Will see what happens, running it as dumbAP and turned of some services like led and firewall as they are not needed :)
> 
> EDIT: after 2days no drop and phy1 not tainted yet.. Looking good so far. My router reboots daily(my cronjob) and switching channels are on auto.
> 
> EDIT2: as far I see, there is no tainted crashes, however I still see drops of 5Ghz most likely due #633
> 
> EDIT3: drops were solved by lowering the antena power due to previously mentioned bug limiting my max power, now the limit cannot be increased without flashing the ROM chip until upstream kernel fix is merged.

Hi @freed00m,

I have 3 Xiaomi Router R3 and I would like to test the changes you've made.
I am using the X-WRT firmware (fork from OpenWRT) and hat the same problems described here: WiFi dropping randomly.
I know that the Xiaomi Router R3 isn't directly supported by OpenWRT because of the NAND, but I will try to use this script I've found (I don't know why this isn't already on OpenWRT main source code): https://github.com/XFY9326/Xiaomi-R3-OpenWrt-Stable
Could you please share the steps you used to compile the OpenWRT snapshot with newer kernel?
Thanks! :)

freed00m, 2022-02-13:
@gitnewcomer I used the snapshot, exact version mentioned in my post.

FusionDragon, 2022-02-13:
I remember reading somewhere that the issue can also occur with devices using wmm power save, I haven't gotten to test this but one way I've been able to test the drops is watching videos from YouTube or prime video. It usually happens then and either the radio will crash or the whole router will crash and restart.

FusionDragon, 2022-02-13:
> @FusionDragon - this lower power is reported/discussed I think already in some other issue [#227 (comment)](https://github.com/openwrt/mt76/issues/227#issuecomment-944611998).

I read this but idk if it's the same issue, I don't think my power is low, I actually believe the power is inaccurate and higher than what's listed. I think that the radio is broadcasting too hard if that's possible and maybe that's what's causing it to crash.

gitnewcomer, 2022-02-13:
> I read this but idk if it's the same issue, I don't think my power is low, I actually believe the power is inaccurate and higher than what's listed. I think that the radio is broadcasting too hard [...]

This is what I have also seen on my Xiaomi Router R3: if I start demanding to much bandwith, like streaming Full HD or 4K videos on Youtube, the WiFi will crash or the router will restart...
I tought it could be a heating problem and added a bigger heatsink on both chipsets, but that didn't help. Then I tried to lower the transmitting power down, but this also didn't help. I switched to PandoraBox 19.01 firmware that uses the MTK binary for WiFi and I'm without drop outs for 5 days. But I would love to change to the newer OpenWRT and not rely on the MTK Binary... :)

FusionDragon, 2022-02-13:
> > > @FusionDragon - this lower power is reported/discussed I think already in some other issue [#227 (comment)](https://github.com/openwrt/mt76/issues/227#issuecomment-944611998).
> > 
> > 
> > I read this but idk if it's the same issue, I don't think my power is low, I actually believe the power is inaccurate and higher than what's listed. I think that the radio is broadcasting too hard if that's possible and maybe that's what's causing it to crash.
> 
> This is what I have also seen on my Xiaomi Router R3: if I start demanding to much bandwidth, like streaming Full HD or 4K videos on Youtube, the WiFi will crash or the router will restart... I tought it could be a heating problem and added a bigger heatsink on both chipsets, but that didn't help. Then I tried to lower the transmitting power down, but this also didn't help. I switched to PandoraBox 19.01 firmware that uses the MTK binary for WiFi and I'm without drop outs for 5 days. But I would love to change to the newer OpenWRT and not rely on the MTK Binary... :)

I did it a little different, I thought it might be a collision issue because of country code and how high the dbm was so instead I switched to a higher band on 5g and lower band on 2.4g and ended up getting less crashes overall.

khanjui, 2023-03-29:
```
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.147190] ------------[ cut here ]------------
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.151854] WARNING: CPU: 2 PID: 687 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2023-03-01-c32d6d84/mt7603/mac.c:208 mt7603_filter_tx+0x178/0x180 [mt7603e]
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.166590] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ifb sha256_generic libsha256
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.167395]  seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug crc32c_generic
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.263313] CPU: 2 PID: 687 Comm: napi/phy0-7 Not tainted 5.15.104 #0
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.269751] Stack : 00000000 80085bdc 00000000 00000004 00000000 00000000 82e7fc54 80a90000
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.278143]         808d0000 807f4df8 81ce5518 808d0ee3 00000000 00000001 82e7fc00 81471900
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.286533]         00000000 00000000 807f4df8 82e7faa0 ffffefff 00000000 ffffffea 00000000
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.294916]         82e7faac 0000018b 808d6b90 ffffffff 807f4df8 00000001 00000000 830b4708
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.303301]         00000009 00000000 00003a98 80000000 00000018 80431124 00000008 80a90008
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.311685]         ...
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.314133] Call Trace:
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.316566] [<8000812c>] show_stack+0x28/0xf0
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.320952] [<803a3448>] dump_stack_lvl+0x60/0x80
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.325672] [<8002f6fc>] __warn+0x9c/0x124
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.329787] [<8002f7e0>] warn_slowpath_fmt+0x5c/0xac
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.334764] [<830b4708>] mt7603_filter_tx+0x178/0x180 [mt7603e]
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.340705] [<830b489c>] mt7603_wtbl_set_ps+0x130/0x138 [mt7603e]
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.346811] [<830b1af4>] mt7603_sta_ps+0x38/0x440 [mt7603e]
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.352398] [<82e65998>] mt76_rx_poll_complete+0x2e4/0x658 [mt76]
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.358535] [<82e623f4>] mt76_dma_rx_poll+0x274/0x4e0 [mt76]
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.364215] [<80545200>] __napi_poll+0x70/0x1f8
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.368766] [<805454d8>] napi_threaded_poll+0x150/0x19c
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.373989] [<8005301c>] kthread+0x140/0x164
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.378268] [<80003058>] ret_from_kernel_thread+0x14/0x1c
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.383670]
Tue Mar 28 23:39:22 2023 kern.warn kernel: [44354.385320] ---[ end trace 5d93ce66b789cf73 ]---

Still happens with latest snapshot on Xiaomi Mi 4 A Gigabit Ediiton
```