openwrt/mt76 #553: Unreliable start of multiple associations on MT7610E

mlichvar, 2021-06-25:
I'm observing this on a C20 v5 running openwrt-21.02-rc3. I don't see it on a different hardware (e.g. MT76x2E) using the same configuration, so I assume this is an issue specific to MT7610E.

The 5GHz radio is configured as two stations connecting to the same AP in order to bridge two different VLANs (I'd like to avoid the GRETAP approach). It takes long time (minutes or hours) for the two associations to fully establish. After that happens, the connections seem stable. It looks like there is something in the initial phase that breaks the other association. The number of retries is random. When I change the configuration to use only one connection, it starts quickly.

In the log I see many messages like this (timestamps are wrong due to unsynchronized clock):

````
Sun Jun 20 16:03:11 2021 kern.info kernel: [ 1047.063927] wlan1: send auth to xxx (try 1/3)
Sun Jun 20 16:03:13 2021 kern.info kernel: [ 1048.498988] wlan1: send auth to xxx (try 2/3)
Sun Jun 20 16:03:13 2021 kern.info kernel: [ 1048.606968] wlan1: send auth to xxx (try 3/3)
````

and on the AP I see messages like this:

````
Fri Jun 25 10:45:23 2021 daemon.notice hostapd: wlan1-1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Fri Jun 25 10:45:24 2021 daemon.notice hostapd: wlan1-1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Fri Jun 25 10:46:12 2021 daemon.info hostapd: wlan1-1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jun 25 10:47:13 2021 daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Fri Jun 25 10:47:14 2021 daemon.notice hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: did not acknowledge authentication response
Fri Jun 25 10:49:36 2021 daemon.info hostapd: wlan1-1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Fri Jun 25 10:49:36 2021 daemon.info hostapd: wlan1-1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Fri Jun 25 10:49:36 2021 daemon.info hostapd: wlan1-1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 2)
````

mlichvar, 2021-07-06:
It seems the issue is on the receiving side of the station. In a packet capture I see that it rarely sends any acknowledgements and the AP is repeating everything many times until it suddenly starts working.

Also, it seems to change with the distance from the AP. When right next to it, even a single association has troubles to start.

I tried some modifications in the driver. Decreasing the gain value calculated in `mt76x0_phy_set_chan_bbp_params()` by 10 or 20 seems to completely fix the issue for me. Both associations start on first attempt. The original value of the gain is 84. Is that correct in this case?

Here is the data from EEPROM:
````
00000000  10 76 00 02 XX XX XX XX  XX XX 10 76 c3 14 00 00
00000010  00 00 10 76 c3 14 00 00  00 00 00 00 00 00 00 00
00000020  ff ff ff 37 ef ff 30 3e  3c 9b ff ff ff ff ff ff
00000030  ff ff ff ff 11 05 0a 10  ff ff 76 01 ff ff ff ff
00000040  ff ff 11 16 00 09 00 00  00 0c 00 00 00 0b 00 00
00000050  ff 80 00 00 00 00 00 00  00 00 00 00 00 00 00 00
00000060  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff
00000070  ff ff ff ff ff 00 00 ff  14 13 13 12 11 11 10 10
00000080  10 10 10 10 10 10 10 10  10 10 10 10 10 10 10 10
00000090  10 10 19 19 19 19 19 19  19 19 19 ff ff ff ff ff
000000a0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff
*
000000d0  ff fc 24 c1 40 74 95 00  00 00 00 00 64 9b 00 00
000000e0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
000000f0  80 80 80 80 f1 fb fd 00  0e 10 7f 7f 7f 7f 80 80
00000100  80 80 ee fb fd 03 0d 14  7f 7f 7f 7f 80 fc ff ff
00000110  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff
00000120  08 06 02 00 06 04 02 00  07 07 04 00 00 00 ff ff
00000130  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff
````

newmem12, 2021-09-21:
`Decreasing the gain value calculated in mt76x0_phy_set_chan_bbp_params() by 10 or 20 seems to completely fix the issue for me`
Can you give more specific instructions on how to do it?
Thanks!