openwrt/mt76 #554: mt7921e: Freeze on resume from suspend

Nocrex, 2021-06-27:
My laptop (Acer Nitro 5) suspends normally, but when resuming locks up completely. The last entry in the logs is `kernel: PM: suspend entry (deep)`.

I did some testing and narrowed the problem down to the wifi driver. The laptop does not freeze if the driver hasn't been loaded since the last reboot and kernel panics if it was loaded previously (blinking caps lock). Having the driver loaded results in the freeze.

I know that #290 is very similar, but it deals with a different device and also seems to be resolved already.
Distro is Manjaro 5.12.9

Nocrex, 2021-07-06:
After adding `initcall_debug` to my kernel parameters, these messages are shown when it freezes:
[image]

because it mentioned async, I tried disabling async pm, but it still froze:
[image]

Darkasf, 2021-07-07:
I'm having the same issue; joe biden

Nocrex, 2021-07-14:
After upgrading to kernel 5.13 the issue seems to be resolved.

closed 2021-07-14 by Nocrex

Diegovsky, 2022-12-29:
I still get the same problem on Acer Aspire 5. Should I open a new issue?