openwrt/mt76 #555: Low TX Power mt7615e

SonicFM, 2021-06-30:
Hello,
I am currently using https://downloads.openwrt.org/snapshots/targets/mediatek/mt7623/kmods/5.4.116-1-d23a908e04ea1380c22d98d8a83dafc9/ kmods for my mt7615e card.
Unfortunately I am only getting 6dbm max transmit power from this card.

Searching google has led me to several posts stating that this could be a problem with the eeprom of the card.

Is this problem already being addressed or is there even a fix yet?
If so, what files do I need to download and install in openwrt or where do I enter what code to compile the needed packages myself?

Thanks a lot in advance!

Borromini, 2021-07-06:
If it's the EEPROM then it's not a driver but a filesystem problem, so to speak. You'll need to check your flash for the correct location, if you have NAND and bad blocks that might be the issue. E.g.:

https://forum.openwrt.org/t/netgear-r6850-very-low-transmit-power-on-5ghz/90984

This isn't an mt76 bug, it doesn't belong here.

SonicFM, 2021-07-06:
Thanks for the clarification and the link. I will investigate this further.
Sorry if this was wrong here, I just didn't know how to help myself, since i am not a dev or experienced with such kind of problems.

closed 2021-07-06 by SonicFM