openwrt/mt76 #556: MT7915 Firmware crashes in Monitor mode when bandwidth is set to 80 MHz and above

wii1989, 2021-07-12:
While experimeting with MT7915 , i found this weird behavior where the firmware crashes exactly when the module is in monitor mode and is set to 80 or 160MHz bandwidth.

The following is the error message. This seems to be coming from the firmware when looking into the driver. This message spams the complete dmesg, the only way to recover is to reload the firmware or reboot the device.

```
[169420.512879] ieee80211 phy0: WM: (3966.051244:82:CMD-S)ProtectCtrl!! Protection Index = 1, DBDC = 0
[169420.819249] ieee80211 phy0: WM: (3966.391180:83:SER-E)BN0 TRB prt corruption!!
[169420.826610] ieee80211 phy0: WM: (3966.391241:84:SER-W)[SER]SerL3RxAbort done code:0
[169421.039684] ieee80211 phy0: WM: (3966.611547:85:SER-E)BN0 TRB prt corruption!!
[169421.047059] ieee80211 phy0: WM: (3966.611608:86:SER-W)[SER]SerL3RxAbort done code:0
[169421.200015] ieee80211 phy0: WM: (3966.771856:87:SER-E)BN0 TRB prt corruption!!
[169421.207382] ieee80211 phy0: WM: (3966.771917:88:SER-W)[SER]SerL3RxAbort done code:0
[169421.360358] ieee80211 phy0: WM: (3966.932165:89:SER-E)BN0 TRB prt corruption!!
[169421.367716] ieee80211 phy0: WM: (3966.932226:90:SER-W)[SER]SerL3RxAbort done code:0
[169421.550740] ieee80211 phy0: WM: (3967.122503:91:SER-E)BN0 TRB prt corruption!!
[...]
```
I tried firmware files obtained from openwrt backports and from MT76 project in GitHub from the following link.

https://github.com/openwrt/mt76/tree/master/firmware

The build dates on backports was from 20200819112407 and the ones from MT76 project is 20201105222323

I ran the test on monitor mode simulating a high traffic environment using iperf. The firmware crashes within a few seconds after i start the test. 

What is the cause for this behavior? is it due to the firmware or the driver is not ready yet for monitor mode yet?

ryderlee1110, 2021-07-15:
Can you observe if the received packets larger than 8K?

wii1989, 2021-07-15:
No, the packets were not larger than 8k, sticking to 80MHz bandwidth.

wii1989, 2021-07-15:
If you are talking about frames, there are A-MPDU frames that are larger than 8k but  wireshark shows de-aggregated MPDU's. Could this be an issue?

ryderlee1110, 2021-07-15:
I mean MPDU > 8K not A-MPDU.

wii1989, 2021-07-16:
> I mean MPDU > 8K not A-MPDU.

No, the MPDU size is not greater than 8k. I cross verified this once again.

wii1989, 2021-07-20:
@ryderlee1110 Do you need any further information regarding the crash?

ryderlee1110, 2021-07-21:
Drop mt76_rmw_field(dev, MT_DMA_DCR0(band), MT_DMA_DCR0_RXD_G5_EN, enabled);  to see if it works

wii1989, 2021-07-28:
@ryderlee1110 I removed the the following line from the mt7915 driver as you mentioned, but the issue still exists. The firmware crashes the same way.
`mt76_rmw_field(dev, MT_DMA_DCR0(band), MT_DMA_DCR0_RXD_G5_EN, enabled);`

Is there any other option in the driver that i can try ? To investigate the cause and fix the firmware crash or  is there a new firmware that i could try?

wii1989, 2021-08-05:
@ryderlee1110 Im still stuck with this issue, can you please let me know if there are other ways to test these modules?

wii1989, 2021-08-24:
I have now tried this on a standard PC Mainboard running the latest Ubuntu disto 21.04. I tested this with the kernel v5.13.8 and v5.14 but the firmware still crashes in 80MHz when simulating a high load with iPerf3.

@ryderlee1110 Please advice, what i could try next.

oyla12, 2021-09-05:
Hey @wii1989, while re-looking for this Problem by googleing I've seen you made already a ticket. See #570, seem to be the same behavior. Do you have any progress with this issue? In the meanwhile my feeling is, that this bug comes directly from the card and is not a bug in this firmware part. This would be very fatal bug of the card itself and maybe the manufactor should definitely solve this problem?

@ryderlee1110 do you have further ideas how to avoid this problem or additional thoughts?

Have both a lovely start in the new week :)

nishaero, 2021-10-08:
> Hi,
> 
> I'm going to send a patch to set proper max rx len of some hw modules. You can try below two lines to see if it works:
> 
> mt76_rmw_field(dev, MT_DMA_DCR0(band), MT_DMA_DCR0_MAX_RX_LEN, 0x680); mt76_rmw_field(dev, MT_MDP_DCR1, MT_MDP_DCR1_MAX_RX_LEN, 0x400);

Hi, thank you for your suggestion.
Which function should i add this? in 'mac_init' of init.c??

Your help is greatly appreciated.

wii1989, 2021-10-08:
> Hi,
> 
> I'm going to send a patch to set proper max rx len of some hw modules. You can try below two lines to see if it works:
> 
> mt76_rmw_field(dev, MT_DMA_DCR0(band), MT_DMA_DCR0_MAX_RX_LEN, 0x680); mt76_rmw_field(dev, MT_MDP_DCR1, MT_MDP_DCR1_MAX_RX_LEN, 0x400);

Hi, thank you for your suggestion.
Which function should i add this? in 'mac_init' of init.c??

Your help is greatly appreciated.

csyuanc, 2021-10-08:
> Hi, thank you for your suggestion. Which function should i add this? in 'mac_init' of init.c??
>
> Your help is greatly appreciated.

These two registers are set in mt7915/init.c, please replace those values with new values.

closed 2021-11-25 by nbd168

oyla12, 2021-12-06:
Moin

I use this ticket here, as my ticket #570 seems to be just an duplication and more communication was going on here. Now I had time to continiue some test and to rebuild / expand the necessary setup. Sorry for the delay.

### tl;dr
This bug is not fixed and is still existing as before with some small changes. So we should reopen that ticket.

### Setup
For the testing setup I used an AX Router and two boards. One board ist generating the load, the other has the MT7915e in monitoring mode. I created some load using iperf. So it's the classical setup for monitoring load tests from my point of view.

I used the OpenWRT Master `7a496e4b4` from Nov 22 17:23:03 having the latest MT updates Integrated (2 commits before `24bb494b6`).

By the way for other people: I noticed that the setting the debug mode changed, there are two files now see below.

I used the following commands in the given order to bring the card in monitoring mode. There are also log messages.

```sh
root@OpenWrt:/# echo '1' > /sys/kernel/debug/ieee80211/phy0/mt76/fw_debug_wa
root@OpenWrt:/# echo '1' > /sys/kernel/debug/ieee80211/phy0/mt76/fw_debug_wm
root@OpenWrt:/# iw wlan0 set monitor none
root@OpenWrt:/# iw wlan0 set channel 36 80MHz
root@OpenWrt:/#
[ 83.043861] ieee80211 phy0: WM: ( 101.754151:45:BF-E)StaRec is Null
[ 83.346197] ieee80211 phy0: WM: ( 102.056458:46:BF-E)StaRec is Null
[ 85.004163] ieee80211 phy0: WM: ( 103.714020:47:BF-E)StaRec is Null
[ 85.348360] ieee80211 phy0: WM: ( 104.058136:48:BF-E)StaRec is Null
[ 87.043861] ieee80211 phy0: WM: ( 105.753266:49:BF-E)StaRec is Null
```

After that I started the load by using:

Client:
```sh
iperf -c 192.168.xxx.yyy -t 1000
```
Server:
```sh
iperf -s
```

I used `tcpdump -i wlan0 -w /tmp/crash.pcap` to get the packages. Before load testing I was (always) able to see a continuous flow of beacon frames.
The card is not getting any frame after about 1 second high load. I also tested some rates with the `-m` Flag, but anyway having more then 5-10 Mbit, the card is playing dead.

Also it seems that I have to restart the full system to get it working again.

Using 80MHz I get the following log messages:

```log
[ 4178.279381] ieee80211 phy0: WM: (4192.128327:77:BF-E)StaRec is Null
[ 4178.383733] ieee80211 phy0: WM: (4192.232636:78:BF-E)StaRec is Null
[ 4178.488646] ieee80211 phy0: WM: (4192.337525:79:BF-E)StaRec is Null
[ 4178.593760] ieee80211 phy0: WM: (4192.442597:80:BF-E)StaRec is Null
[ 4178.818510] ieee80211 phy0: WM: (4192.667298:81:SER-E)BN0 TRB prt corruption!!
[ 4178.826200] ieee80211 phy0: WM: (4192.667359:82:SER-W)[SER]SerL3RxAbort done code:0
[ 4178.918793] ieee80211 phy0: WM: (4192.767579:83:SER-E)BN0 TRB prt corruption!!
[ 4178.927075] ieee80211 phy0: WM: (4192.767640:84:SER-W)[SER]SerL3RxAbort done code:0
```

There is a small difference to my first small tests. The `BN0 TRB prt corruption!!` is not triggered anytime, so the message is not reliable reproducable. But the praxis behavior (no captured frames) has not changed.

Using 160MHz the card is doing mostly the same. I just get some other messages by switching to monitor mode. Also I was never able to get the `BN0 TRB prt corruption!!` message.
The card is dead after a very small load again.

```
root@OpenWrt:/# iw wlan0 set monitor none
root@OpenWrt:/# iw wlan0 set channel 36 160MHz
[ 447.893029] ieee80211 phy0: WM: ( 470.776154:45:CAL-S)halPhyCalReloadFdFi: Reload SX0-WF0/1 RXFD/FI !
[ 447.902325] ieee80211 phy0: WM: ( 470.776185:46:CAL-S)G2 Sx0 Ant0 Freq5210000 Bw3
<STARTING LOAD HERE>
[ 780.875611] device wlan0 left promiscuous mode
```

### So some facts in summay

- I'm quite sure this problem still existis as you see.
- There is a new `SerL3RxAbort` message
- There is a new `ieee80211 phy0: WM: ( 101.754151:45:BF-E)StaRec is Null` message
- There is this `device wlan0 left promiscuous mode` message I havent noticed before.
- The `BN0 TRB prt corruption!!` is not an safe indicator anymore in my case

### Questions
It is a bit time intensive to redopt the setup always, so I have some questions :)

- Did I really have the correct updates? There was no commit reference here as orientation.
- How do I know, which firmware blobs are used? Is there a single point of truth for this files, or is the OpenWRT Repository the first point they may appear? For a possible next test I like to make sure to have an correct commit and source base.
- Do you have some hints about the log messages in debug mode, some seems to get directly from the hardware/firmware.
- Any ideas how to get the 80MHz workable? 160MHz is maybe to exotic these days, so this would be ok, but 80MHz is common used.

Thanks for your work, I'm willing to made tests again and would keep the test setup warm now :)

Greetings from Germany

oyla12, 2021-12-20:
Moin 

As it is two weeks later I would like assure that my post is at least seen, probably without having an instant solution now for the problem. This ticket is still not reopened and I can not see any kind of a small communicative feedback as ACK. Furthermore I have the working test setup to be able to make some new test if someone has an interesting hint for me.

Lovely x-mas days with your favorites :)