openwrt/mt76 #557: Wireless is hardly broken with latest mt76 update

ivanich, 2021-07-14:
Commit https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=99a22d48f2003910f0968369b61b9131a8f53d80 broke wifi, can't connect to 2.4 or 5ghz networks, reverting this commit fixes issue. Device gl.inet beryl mt-1300

ryderlee1110, 2021-07-15:
which chip?

ivanich, 2021-07-15:
Openwrt wiki says it's a MT7615D

kar200, 2021-07-15:
Same here . MT7615DN (DBDC). Wlan 5ghz showing enabled and on in Luci and from console but nothing shows up from the phone (or computer).
The 2GHZ cards does show but I cannot connect - Tried in WPA+PSK and even with no encryption it does not work.

nbd168, 2021-07-15:
I tested the latest version on my own device with MT7615D, but could not reproduce the issue.
Can you try to use git bisect on the mt76 git tree to figure out which commit broke it?

dibdot, 2021-07-15:
Same problem here with a Beryl MT-1300 and TP-RE650 - both APs went down as soon as I install latest MT76 update.

TWijit, 2021-07-15:
Same here on MT7615E. Revert this commit fix my problem.
https://github.com/TWijit/mt76/commit/ca64a36908b7a1b9d4a0cdc3c64dd5434010e302

ryderlee1110, 2021-07-15:
@TWijit can you help to change '6' back to '8' in mt76_rates to see if this issue solved?

TWijit, 2021-07-15:
Change '6' to '8' in mt76_rates also works for me.

nbd168, 2021-07-15:
Please try the latest mt76.git. I just pushed a revert + fix for related mt7915/mt7921 changes

TWijit, 2021-07-15:
I have no issue with the latest commit.

dibdot, 2021-07-15:
@nbd168 re-tested the latest m76 git with your revert/fix successfully on the above mentioned TP-RE650, AP comes up as before and STA connections are working fine as well. Many thanks!

nbd168, 2021-07-15:
Fix pushed to master, thanks for testing and reporting your findings

ivanich, 2021-07-15:
Fixed, thanks

closed 2021-07-15 by ivanich