openwrt/mt76 #560: mt7603/mac.c:206 mt7603_filter_tx+0x16c/0x174 [mt7603e]

953537867, 2021-07-27:
For me only these warnings every 3/4 days, connection is stable with ~ 10 devices
I'm NOT using option disassoc_low_ack='0'

[105622.833568] ------------[ cut here ]------------ [105622.838329] WARNING: CPU: 2 PID: 1487 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2021-04-11-bf45b30d/mt7603/mac.c:206 mt7603_filter_tx+0x16c/0x174 [mt7603e] [105622.853172] Modules linked in: pppoe ppp_async pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY slhc sch_cake nf_reject_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conncount iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat act_connmark nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci [105622.853386] xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common [105622.947785] CPU: 2 PID: 1487 Comm: hostapd Not tainted 5.4.111 #0 [105622.953936] Stack : 00000001 8007d738 806c0000 806beb24 80720000 806beaec 806bdc40 8ee639cc [105622.962359] 80860000 8fe5d268 80708d03 80651b4c 00000002 00000001 8ee63970 e3a64ee8 [105622.970779] 00000000 00000000 808a0000 00000000 30000000 0000015f 00000018 312e342e [105622.979197] 00000000 00001487 00000000 68203a6d 80000000 80720000 00000000 8eeb452c [105622.987615] 00000009 00000009 00000009 8eec0000 00000008 8034e8dc 00000008 80860008 [105622.996032] ... [105622.998557] Call Trace: [105623.001110] [<8000b64c>] show_stack+0x30/0x100 [105623.005643] [<805950b8>] dump_stack+0xa4/0xdc [105623.010091] [<8002bfe8>] __warn+0xc0/0x10c [105623.014263] [<8002c090>] warn_slowpath_fmt+0x5c/0xac [105623.019317] [<8eeb452c>] mt7603_filter_tx+0x16c/0x174 [mt7603e] [105623.025314] [<8eeb19d4>] mt7603_sta_remove+0x94/0xfc [mt7603e] [105623.031242] [<8eea2d44>] __mt76_sta_remove+0x78/0xec [mt76] [105623.036897] [<8eea2e90>] mt76_sta_state+0xd8/0x1e4 [mt76] [105623.042533] [<8e809ee4>] sta_set_sinfo+0xfac/0x102c [mac80211] [105623.048727] ---[ end trace 91b91b401c16c82b ]---

Related syslog entries for the abve:

Fri Apr 16 22:43:04 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 4c:24:98:88:f0:0d Fri Apr 16 22:43:05 2021 daemon.info hostapd: wlan0: STA 4c:24:98:88:f0:0d IEEE 802.11: authenticated Fri Apr 16 22:43:05 2021 daemon.info hostapd: wlan0: STA 4c:24:98:88:f0:0d IEEE 802.11: associated (aid 1) Fri Apr 16 22:43:05 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 4c:24:98:88:f0:0d Fri Apr 16 22:43:05 2021 daemon.info hostapd: wlan0: STA 4c:24:98:88:f0:0d WPA: pairwise key handshake completed (RSN) Fri Apr 16 22:43:33 2021 daemon.info hostapd: wlan0: STA d4:a3:3d:6e:98:da IEEE 802.11: authenticated Fri Apr 16 22:43:33 2021 daemon.info hostapd: wlan0: STA d4:a3:3d:6e:98:da IEEE 802.11: associated (aid 10) Fri Apr 16 22:43:33 2021 daemon.notice hostapd: wlan1: Prune association for d4:a3:3d:6e:98:da Fri Apr 16 22:43:33 2021 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED d4:a3:3d:6e:98:da Fri Apr 16 22:43:33 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED d4:a3:3d:6e:98:da Fri Apr 16 22:43:33 2021 daemon.info hostapd: wlan0: STA d4:a3:3d:6e:98:da WPA: pairwise key handshake completed (RSN) Fri Apr 16 22:44:03 2021 daemon.info hostapd: wlan1: STA d4:a3:3d:6e:98:da IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE) Fri Apr 16 22:57:17 2021 daemon.info hostapd: wlan1: STA d4:a3:3d:6e:98:da IEEE 802.11: authenticated Fri Apr 16 22:57:17 2021 daemon.info hostapd: wlan1: STA d4:a3:3d:6e:98:da IEEE 802.11: associated (aid 2) Fri Apr 16 22:57:17 2021 daemon.notice hostapd: wlan0: Prune association for d4:a3:3d:6e:98:da Fri Apr 16 22:57:17 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED d4:a3:3d:6e:98:da Fri Apr 16 22:57:17 2021 daemon.notice hostapd: wlan1: AP-STA-CONNECTED d4:a3:3d:6e:98:da Fri Apr 16 22:57:17 2021 daemon.info hostapd: wlan1: STA d4:a3:3d:6e:98:da WPA: pairwise key handshake completed (RSN) Fri Apr 16 22:57:19 2021 kern.warn kernel: [105622.833568] ------------[ cut here ]------------

easyteacher, 2021-09-15:
Do you set htmode to Legacy/None?

krumelmonster, 2022-11-03:
```
[5532307.299579] ------------[ cut here ]------------
[5532307.304426] WARNING: CPU: 3 PID: 1560 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2021-06-06-22b69033/mt7603/mac.c:206 0x86f6454c [mt7603e@34e0ed6
e+0x95a0]
[5532307.319184] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_RE
DIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76
03e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt
_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc sch_cake poly1305_mips nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_def
rag_ipv4 libcurve25519_generic libblake2s_generic iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat chacha_mips sch_tbf sch_ingress
sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact nf_log_ipv6 nf_log_common ip6table
_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables
[5532307.319465]  nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel kpp leds_gpio gpio_button_hotplug
[5532307.415599] CPU: 3 PID: 1560 Comm: hostapd Not tainted 5.4.143 #0
[5532307.421836] Stack : 00000000 80850000 00000001 8007dec4 00000000 00000000 00000000 00000000
[5532307.430333]         00000000 00000000 00000000 00000000 00000000 00000001 86913968 a8180e7f
[5532307.438831]         86913a00 00000000 00000000 00000000 00000038 805eb4e4 2e342e35 20333431
[5532307.447332]         00000000 00043339 00000000 0006576f 80000000 86913948 00000000 86f6454c
[5532307.455826]         00000009 00000007 00000007 86f70000 00000002 8035c59c 0000000c 8082000c
[5532307.464321]         ...
[5532307.466927] Call Trace:
[5532307.466939] [<8007dec4>] 0x8007dec4
[5532307.473222] [<805eb4e4>] 0x805eb4e4
[5532307.476892] [<86f6454c>] 0x86f6454c [mt7603e@34e0ed6e+0x95a0]
[5532307.482791] [<8035c59c>] 0x8035c59c
[5532307.486443] [<8000b05c>] 0x8000b05c
[5532307.490087] [<8000b064>] 0x8000b064
[5532307.493732] [<805d0e1c>] 0x805d0e1c
[5532307.497376] [<8007a96c>] 0x8007a96c
[5532307.501028] [<8002c140>] 0x8002c140
[5532307.504680] [<86f6454c>] 0x86f6454c [mt7603e@34e0ed6e+0x95a0]
[5532307.510573] [<8002c1e8>] 0x8002c1e8
[5532307.514230] [<86f6454c>] 0x86f6454c [mt7603e@34e0ed6e+0x95a0]
[5532307.520155] [<86f619d4>] 0x86f619d4 [mt7603e@34e0ed6e+0x95a0]
[5532307.526058] [<86f52db0>] 0x86f52db0 [mt76@4b68a40d+0x96c0]
[5532307.531714] [<86f52efc>] 0x86f52efc [mt76@4b68a40d+0x96c0]
[5532307.537354] [<86f87fdc>] 0x86f87fdc [mac80211@1028d069+0x7d9d0]
[5532307.543435] [<86f89f24>] 0x86f89f24 [mac80211@1028d069+0x7d9d0]
[5532307.549499] [<80070ab4>] 0x80070ab4
[5532307.553151] [<80431964>] 0x80431964
[5532307.556806] [<86f89fd4>] 0x86f89fd4 [mac80211@1028d069+0x7d9d0]
[5532307.562887] [<86f8a084>] 0x86f8a084 [mac80211@1028d069+0x7d9d0]
[5532307.568958] [<86f1d3b0>] 0x86f1d3b0 [cfg80211@ba90ac77+0x43740]
[5532307.575032] [<86f14160>] 0x86f14160 [cfg80211@ba90ac77+0x43740]
[5532307.581106] [<804a1e10>] 0x804a1e10
[5532307.584788] [<804a1be0>] 0x804a1be0
[5532307.588436] [<8049fb98>] 0x8049fb98
[5532307.592079] [<804a03a4>] 0x804a03a4
[5532307.595731] [<804a0b90>] 0x804a0b90
[5532307.599379] [<8049fab4>] 0x8049fab4
[5532307.603027] [<8017382c>] 0x8017382c
[5532307.606687] [<8049fe84>] 0x8049fe84
[5532307.610337] [<80179fc0>] 0x80179fc0
[5532307.613996] [<8040f68c>] 0x8040f68c
[5532307.617654] [<8040fe8c>] 0x8040fe8c
[5532307.621319] [<8040ff9c>] 0x8040ff9c
[5532307.624996] [<804111bc>] 0x804111bc
[5532307.628639] [<80070ab4>] 0x80070ab4
[5532307.632282] [<8040e680>] 0x8040e680
[5532307.635938] [<80014558>] 0x80014558
[5532307.639587]
[5532307.641646] ---[ end trace 998bf9da49552ec2 ]---
```

This leads to a few seconds of lost connection but devices stay associated to wifi I believe. Only seen this with iOS clients by now.

OpenWrt 21.02.0, r16279-5cc0535800

seeing this on multiple devices, all Xiaomi 4A Gigabit, happens about daily.

disassoc_low_ack on default (so 1?)

I'll upgrade to 21.03.1 next week

krumelmonster, 2022-11-03:
Should I try change the htmode setting?

lukasz1992, 2022-11-04:
DMA busy. Disabling ht will move you 20 years back to wifi3 technology