openwrt/mt76 #561: with mt7615 some clients have low download speed on 5ghz

mcaptur, 2021-07-27:
I have two types of mt76 routers mir3g and mi4a gigabit (mt7612) and redmi ac2100 and dlink dir878 (mt7615). Certain clients e.g. mitv stick have very slow download ~3mbps when connected to mt7615 devices, but good speed > 300mbps when connected to mt7612 devices.. I tried same frequencies same channgel width (80 mhz) etc etc

Any info i can send to help debug?

TWijit, 2021-07-28:
I also have this problem. There appear to be high tx retry rate only on single spatial stream device. Upload to working fine though.

JulioMC81, 2021-07-28:
Same problem here. Low tx speed with some clients (I think with VHT-NSS 1 devices, VHT-NSS 2 seems to work OK).

Router: Xiaomi MI Router AC2100.
OpenWRT version: OpenWrt 21.02-SNAPSHOT r16249-55d9c020a1 / LuCI openwrt-21.02 branch git-21.207.43342-132c72c.

csszep, 2021-08-03:
Same problem here after upgrading 21.02-RC1 to 21.02-RC4. 
21.02-RC1 seems ok.

LaKardo, 2021-08-13:
Same problem here. Asus RT-AC65P (two mt7615N). Snapshot (r17318-eaa9c94c75) and 21.02.0-rc4 (r16256-2d5ee43dc6).
Very slow download ~ 2-4 mbps on both (2.4g and 5g) but upload still ok.

elcovv, 2021-10-02:
Found the same behaviour here with one of my Cudy WR2100s (7615e) running 21.02.0. With a second router, same type, config and release, but different channel I had good performance. The problem remained on the same router after swapping the channels. It only happens with a VHT-NSS 1 device, with a VHT-NSS 2 it worked fine.

elcovv, 2021-10-13:
Same problem on Cudy WR2100 with VHT-NSS 1 device (no problems with VHT-NSS 2) running openwrt 21.02.0 or today's snapshot (r17750-213ce1d837). Performance is good on a custom snapshot version provided by Cudy.

With 21.02.0 and snapshot the VHT-MCS rates are constantly changing. [More details](https://pastebin.com/mJRbhe5n)

elcovv, 2021-11-01:
Tested the latest snapshot (`OpenWrt SNAPSHOT r17926-a44e4aaef9`) on Cudy WR2100 running version `5.4.155+2021-10-23.1-f6bde7ba-4` of kmod-mt7615e.

With the following /etc/config/wireless (kept it as basic as possible):

```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'HT20'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Test'
        option encryption 'psk2'
        option key '<key>'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
        option htmode 'VHT80'
        option cell_density '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Test_5G'
        option encryption 'psk2'
        option key '<key>'
```

Performance is still unstable for NSS-1 clients. The problemas are mainly with download (AP->client) but also the other  way:

```
Connecting to host <ip>, port 5201
Reverse mode, remote host <ip> is sending
[ 4] local <ip> port 37884 connected to <ip> port 5201
[ ID] Interval Transfer Bandwidth
[ 4] 0.00-1.01 sec 9.90 KBytes 80.6 Kbits/sec
[ 4] 1.01-2.00 sec 0.00 Bytes 0.00 bits/sec
[ 4] 2.00-3.00 sec 1.04 MBytes 8.79 Mbits/sec
[ 4] 3.00-4.00 sec 669 KBytes 5.47 Mbits/sec
[ 4] 4.00-5.00 sec 45.2 KBytes 370 Kbits/sec
[ 4] 5.00-6.00 sec 0.00 Bytes 0.00 bits/sec
[ 4] 6.00-7.00 sec 1.41 KBytes 11.6 Kbits/sec
[ 4] 7.00-8.00 sec 0.00 Bytes 0.00 bits/sec
[ 4] 8.00-9.01 sec 0.00 Bytes 0.00 bits/sec
[ 4] 9.01-10.00 sec 0.00 Bytes 0.00 bits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval Transfer Bandwidth Retr
[ 4] 0.00-10.00 sec 1.84 MBytes 1.54 Mbits/sec 46 sender
[ 4] 0.00-10.00 sec 1.75 MBytes 1.47 Mbits/sec receiver

iperf Done.

Connecting to host ineke, port 5201
[ 4] local <ip> port 37918 connected to <ip> port 5201
[ ID] Interval Transfer Bandwidth
[ 4] 0.00-1.00 sec 12.9 MBytes 108 Mbits/sec
[ 4] 1.00-2.00 sec 11.1 MBytes 92.8 Mbits/sec
[ 4] 2.00-3.01 sec 10.6 MBytes 88.5 Mbits/sec
[ 4] 3.01-4.00 sec 9.57 MBytes 80.8 Mbits/sec
[ 4] 4.00-5.00 sec 11.2 MBytes 93.8 Mbits/sec
[ 4] 5.00-6.00 sec 11.3 MBytes 94.9 Mbits/sec
[ 4] 6.00-7.00 sec 11.1 MBytes 92.8 Mbits/sec
[ 4] 7.00-8.00 sec 11.1 MBytes 93.1 Mbits/sec
[ 4] 8.00-9.00 sec 10.3 MBytes 86.7 Mbits/sec
[ 4] 9.00-10.00 sec 11.2 MBytes 93.8 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval Transfer Bandwidth
[ 4] 0.00-10.00 sec 110 MBytes 92.5 Mbits/sec sender
[ 4] 0.00-10.00 sec 109 MBytes 91.2 Mbits/sec receiver

Connecting to host 192.168.178.251, port 5201
Reverse mode, remote host 192.168.178.251 is sending
[ 4] local 192.168.178.75 port 59808 connected to 192.168.178.251 port 5201
[ ID] Interval Transfer Bandwidth
[ 4] 0.00-1.00 sec 148 KBytes 1.22 Mbits/sec
[ 4] 1.00-2.00 sec 1.41 MBytes 11.8 Mbits/sec
[ 4] 2.00-3.00 sec 216 KBytes 1.77 Mbits/sec
[ 4] 3.00-4.00 sec 339 KBytes 2.79 Mbits/sec
[ 4] 4.00-5.00 sec 9.42 MBytes 79.0 Mbits/sec
[ 4] 5.00-6.00 sec 19.8 MBytes 166 Mbits/sec
[ 4] 6.00-7.00 sec 15.0 MBytes 126 Mbits/sec
[ 4] 7.00-8.00 sec 5.50 MBytes 46.1 Mbits/sec
[ 4] 8.00-9.00 sec 7.02 MBytes 59.0 Mbits/sec
[ 4] 9.00-10.00 sec 6.52 MBytes 54.5 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval Transfer Bandwidth Retr
[ 4] 0.00-10.00 sec 66.2 MBytes 55.5 Mbits/sec 876 sender
[ 4] 0.00-10.00 sec 65.4 MBytes 54.8 Mbits/sec receiver

iperf Done.

Connecting to host 192.168.178.251, port 5201
[ 4] local 192.168.178.75 port 59862 connected to <ip> port 5201
[ ID] Interval Transfer Bandwidth
[ 4] 0.00-1.01 sec 4.95 MBytes 41.3 Mbits/sec
[ 4] 1.01-2.00 sec 1.25 MBytes 10.5 Mbits/sec
[ 4] 2.00-3.00 sec 3.75 MBytes 31.5 Mbits/sec
[ 4] 3.00-4.00 sec 6.25 MBytes 52.5 Mbits/sec
[ 4] 4.00-5.00 sec 1.25 MBytes 10.5 Mbits/sec
[ 4] 5.00-6.00 sec 1.25 MBytes 10.5 Mbits/sec
[ 4] 6.00-7.00 sec 1.25 MBytes 10.5 Mbits/sec
[ 4] 7.00-8.00 sec 1.25 MBytes 10.5 Mbits/sec
[ 4] 8.00-9.00 sec 0.00 Bytes 0.00 bits/sec
[ 4] 9.00-10.00 sec 1.25 MBytes 10.5 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval Transfer Bandwidth
[ 4] 0.00-10.00 sec 22.4 MBytes 18.8 Mbits/sec sender
[ 4] 0.00-10.00 sec 19.3 MBytes 16.2 Mbits/sec receiver

iperf Done.
```

A restart of wifi or a reboot doesn't help. My NSS-2 client still works fine.

@nbd168: Is there any way I can provide more detailed information, maybe enable more detailed logging or traces?

TWijit, 2021-11-02:
Change channel bandwidth to 160 MHz improves download performance and stability on NSS-1 clients for me.

-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.1.34, port 42550
[  5] local 192.168.1.20 port 5201 connected to 192.168.1.34 port 42552
[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-1.00   sec  23.5 MBytes   197 Mbits/sec
[  5]   1.00-2.00   sec  27.5 MBytes   231 Mbits/sec
[  5]   2.00-3.00   sec  27.5 MBytes   231 Mbits/sec
[  5]   3.00-4.00   sec  21.2 MBytes   178 Mbits/sec
[  5]   4.00-5.00   sec  26.9 MBytes   225 Mbits/sec
[  5]   5.00-6.00   sec  24.6 MBytes   207 Mbits/sec
[  5]   6.00-7.00   sec  24.8 MBytes   208 Mbits/sec
[  5]   7.00-8.00   sec  27.1 MBytes   227 Mbits/sec
[  5]   8.00-9.00   sec  29.0 MBytes   244 Mbits/sec
[  5]   9.00-10.00  sec  27.7 MBytes   233 Mbits/sec
[  5]  10.00-10.12  sec  3.46 MBytes   237 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-10.12  sec  0.00 Bytes  0.00 bits/sec                  sender
[  5]   0.00-10.12  sec   263 MBytes   218 Mbits/sec                  receiver
-----------------------------------------------------------

-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.1.109, port 49952
[  5] local 192.168.1.20 port 5201 connected to 192.168.1.109 port 49954
[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-1.00   sec  24.5 MBytes   206 Mbits/sec
[  5]   1.00-2.00   sec  26.0 MBytes   218 Mbits/sec
[  5]   2.00-3.00   sec  25.5 MBytes   214 Mbits/sec
[  5]   3.00-4.00   sec  25.4 MBytes   213 Mbits/sec
[  5]   4.00-5.00   sec  25.5 MBytes   214 Mbits/sec
[  5]   5.00-6.00   sec  25.6 MBytes   215 Mbits/sec
[  5]   6.00-7.00   sec  25.2 MBytes   212 Mbits/sec
[  5]   7.00-8.00   sec  25.3 MBytes   212 Mbits/sec
[  5]   8.00-9.00   sec  25.5 MBytes   214 Mbits/sec
[  5]   9.00-10.00  sec  25.4 MBytes   213 Mbits/sec
[  5]  10.00-10.00  sec  52.3 KBytes   111 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-10.00  sec  0.00 Bytes  0.00 bits/sec                  sender
[  5]   0.00-10.00  sec   254 MBytes   213 Mbits/sec                  receiver
-----------------------------------------------------------

elcovv, 2021-11-02:
I've tried 160MHz and it increased performance, but not stability.

The performance depends on the location and direction where I place the mobile. Placing at the right location there is stable, high performance, up to 200Mbps at 80MHz. Just moving it a centimeter can cause performance to drop below 1Mbps or 0.

elcovv, 2021-11-02:
@ryderlee1110 Could it be that the unreliable download speed is caused by incorrect/empty eeprom data as was the cause in this [issue](https://github.com/openwrt/mt76/issues/376#issuecomment-639184227)?

ryderlee1110, 2021-11-02:
Could be.

elcovv, 2021-11-14:
I tried a number of things with regard to the eeprom:

- There is eeprom data (7615, mac address, etc.) on the factory partition as specified in the Cudy WR2100 dts file (`mediatek,mtd-eeprom = <&factory 0x8000>;`), so the eeprom is not 'empty.'
- I found an mt7615e.eeprom.bin file in the /etc/wireless/mt7615e directory of the stock firmware, which uses the Mediatek 5.0.4.0 driver. Writing that eeprom to the factory partition didn't resolve the issue.
- The 5Ghz wifi MAC address is different running the stock firmware and so far I was not able to find that MAC address anywhere on the stock firmware image. So I was wondering if the stock firmware could be using efuse instead of flash or a file. In an attempt to make the mt7615 driver use the efuse data, I build a snapshot of which I removed the 7615 eeprom line from the dts file. The driver complaint about a missing mac address, but for the rest it worked. The issue however was still there. 

- Is it correct that the efuse data can be accessed via /sys/kernel/debug/ieee80211/phy1/mt76/otp?
- Would it be useful if I send the different eeproms (from factory partition, from stock image and the otp one)
- Is there anything else I can troubleshoot regarding the eeprom?

elcovv, 2021-11-14:
@mcaptur, @TWijit @JulioMC81 @csszep @LaKardo Is the download rate in your situations also depending on the location or position of the client? For me it is very consistent as long as I don't move the router or the antennas: there are places where performance is fine, and others where it is bad, it's just a matter of a few centimeters. When I move the router a bit, the locations change.

The Cudy WR2100 has 4 external antennas, of which 2 are also used for the 2.4GHz signal. I found components on the board that seem to mix the antenna signal. Do you know what the antenna setup is for your routers? Maybe we can find similarities here.

TWijit, 2021-11-15:
> @mcaptur, @TWijit @JulioMC81 @csszep @LaKardo Is the download rate in your situations also depending on the location or position of the client? For me it is very consistent as long as I don't move [...]

It was once. Right now I get same performance in those place.
Redmi AC2100 has 6 external antennas.

LaKardo, 2021-11-18:
> HI, @elcovv

I have an Asus RT-AC65P, MT7621AT chip + two mt7615n, as indicated on the openwrt website. But deviwiki lists mt7615s for n + mt7615b for ac. As far as I know, it has 4 antennas: 3 external and 1 internal. Configuration mu-mimo 3x3. The stability and speed are poor. Even without moving the phone, for example, the speed can fluctuate from about 1-2 to ~ 50-60 Mbits / sec in the same place. On the stock firmware, the router immediately outputs ~ 97 Mbits / sec in both directions, and this is very stable throughout the entire speed test. Tariff speed from the provider is 100 Mbits / sec.

I don't know what else I can check or do to help fix this problem.

elcovv, 2021-11-18:
> > HI, @elcovv
> 
> I have an Asus RT-AC65P, MT7621AT chip + two mt7615n, as indicated on the openwrt website. But deviwiki lists mt7615s for n + mt7615b for ac. Perhaps this is the same chip or with minimal differences, which I hope are not the reason for such poor performance. As far as I know, this router has 4 antennas: 3 external and 1 internal. Configuration mu-mimo 3x3. The stability and speed are poor. Even without moving the phone, for example, the speed can fluctuate from about 1-2 to ~ 50-60 Mbits / sec in the same place. On the stock firmware, the router immediately outputs ~ 97 Mbits / sec in both directions, and this is very stable throughout the entire speed test. Tariff speed from the provider is 100 Mbits / sec.
> 
> I want to test the X-wrt firmware from @ptpt52, he made some changes for the mt7621, hopefully it has some effect on stability or performance.
> 
> I don't know what else I can check or do to help fix this problem.

[Wikidevi](https://wikidevi.wi-cat.ru/ASUS_RT-AC65P) says that the ASUS RT-AC65P has 3 antennas, you might want to check [this ](https://github.com/openwrt/mt76/issues/329) issue.

Wikidevi also lists additional chips: 5GHz RF Front End Module;Qorvo;QPF4519;Q, QPF4519, 5JUF;4; Maybe the mt76 driver needs to be adapted/configured for this...

Thanks for the X-WRT tip, I'm going to check it out!

elcovv, 2021-11-18:
@LaKardo What I've understood from [wikidevi ](https://wikidevi.wi-cat.ru/ASUS_RT-AC65P) is that there are 7615n chips in this router, one for n and the other for ac. You might want to check the board to see if there is really a fourth internal antenna, if not, [this ](https://github.com/openwrt/mt76/issues/329) issue might be of interest for you.

Also wikidevi mentions additional chips: `5GHz RF Front End Module;Qorvo;QPF4519;Q, QPF4519, 5JUF;4;` I wonder if these additional chips require some specific configuration or adaption of the driver.

elcovv, 2021-11-29:
> I want to test the X-wrt firmware from @ptpt52, he made some changes for the mt7621, hopefully it has some effect on stability or performance.

Hi @LaKardo, I've build and x-wrt firmware for the Cudy-WR2100, but it didn't make any difference.

elcovv, 2021-11-29:
> Could be.

Hi @ryderlee1110, I managed to get root access to the stock firmware of a Cudy WR2100, which is using `MT7615E AP Driver version-5.0.4.0`. The command `iwpriv rai0 e2p` gave the exact same output as the eeprom in /dev/mtd2 at address 0x8000 that the mt76 driver is using in OpenWrt. So it seems that the right eeprom data is used.

Do you have suggestions on where to look for next? Is there any usefull output from the stock firmware that I could send?

LaKardo, 2021-11-29:
Hi @elcovv, yes, I know, already checked. The problem is clearly in wifi, because lan works as stable as on the stock firmware.  Unfortunately, I have not checked the latest mt76 driver yet. In theory, I have no problems with eeprom, txpower is fine, mac address too. The range of wifi is excellent. There were no network disconnections either. But the stability and speed are very poor.

mcaptur, 2021-12-31:
> Hi @elcovv, yes, I know, already checked. The problem is clearly in wifi, because lan works as stable as on the stock firmware. Unfortunately, I have not checked the latest mt76 driver yet. In theory, I have no problems with eeprom, txpower is fine, mac address too. The range of wifi is excellent. There were no network disconnections either. But the stability and speed are very poor.
> 
> Update: The above problems in my case are most related to download. The upload is good enough.

adding         option rx_stbc '1' or '2' to wireless config seems to improve situation on download.. can you try?

mcaptur, 2022-01-01:
> adding option rx_stbc '1' or '2' to wireless config seems to improve situation on download.. can you try?

Actually still seem to have issue.. is there anything i can capture or submit to help troubleshoot? mt7603 works fine.. mt7615 has extremely slow performance with some clients (both 2.4 and 5ghz) I can repeat issue at will with a mitv stick

LaKardo, 2022-01-05:
> adding option rx_stbc '1' or '2' to wireless config seems to improve situation on download.. can you try?

Hi, I just tried it. The result has hardly changed. Unless in one case it affected the upload. 
For comparison, I will attach screenshots of speed tests on the stock firmware and openwrt.

### Stock openwrt

**Wifi 2.4 Ghz**
[image]

**Wifi 5 Ghz**
[image]

### Openwrt with rx_stbc = 1

**Wifi 2.4 Ghz**
[image]

**Wifi 5 Ghz**
[image]

### Openwrt with rx_stbc = 2

**Wifi 2.4 Ghz**
[image]

**Wifi 5 Ghz**
[image]

### Results on stock asus rom

**Wifi 2.4 Ghz**
[image]

**Wifi 5 Ghz**
[image]

elcovv, 2022-01-06:
One thing I noticed with the stock firmware of the Cudy WR2100 is that MCS8 and MCS9 are not used. The higest MCS rate used is MCS7. Does anyone know if there is a way in Openwrt to exclude these rates?

paulfertser, 2022-02-02:
I suggest you try ``iw dev wlan0 set bitrates ht-mcs-2.4 1 2 3 4 5 6 7`` or some variants of that.

elcovv, 2022-03-01:
Hi Paul, I found the arguments to exclude MCS8 and 9 for NSS-1 stations: `iw dev wlan1 set bitrates vht-mcs-5 1:0-7` 

Unfortunately it didn't make any difference in the throughput or stability...

mmt84, 2022-09-25:
I have 3 different phones and one tablet that have great download speeds. But download speed with my laptop which has a realtek rtl8821ce chipset is terrible. It seems to be a NSS-1 client. Tx rate on my laptop is 390mbit (max rate on my laptop) and seems to be constant. Rx rate on the other hand seems to be changing a lot, going between mcs0 and mcs9 every second. Tried different (config and iw) combinations with no success. Rolled back to stock for time being.

The problem is most definitely with the mt7615 driver. Having this problem on Tp-link re650 v1 and v2.

Not sure if it helps but I don't have any problems with a Xiaomi 4a gigabit router using a 7612 chipset.

wraypa1, 2022-10-30:
I have the exact same issues, I have read pretty much every thread I can find online. There are some mentions of a fix but I can't say I quite follow. Has anyone made any progress. I am on a Redmi ac2100.

Cheers

mmt84, 2023-04-15:
Not sure if it's relevant but I'm not seeing any bug reports for routers/access points with mt7622/mt7615 combo. All the mentioned devices here seem to have mt7621/mt7615 combo.

lukasz1992, 2023-04-16:
@mmt84 Assuming you're right, I wonder if the problem could be with mt76 does not use threaded NAPI for all official releases (uses in master). mt7621 is slower than mt7622 - maybe that makes a difference.

Anyone has the problem should check snapshot and check if issue still persists.

mmt84, 2023-04-16:
I have tried the snapshot from a few days ago, also checking and trying snapshots on a regular basis. The problem still persists. One thing I noticed is mt7615.h header file contains code for mt7622 but not for mt7621.

shaman79, 2023-07-19:
I have exactly the same symptoms. TPLINK RE305v1 (MT7612EN, MediaTek MT7628AN).

cliobrando, 2024-05-08:
Redmi AX6000 suffers the same problem, using OpenWrt SNAPSHOT r26144-219018185e

Lu1sr0550, 2024-07-13:
Hi. Any news? I have the same problem with Redmi AC2100 OpenWrt SNAPSHOT r26917-d92c42f469
Wifi 2.4 sync 150 and 100-30 real speed , it depends on the room in the house.
Wifi 5G nearest room Sync 433 and 40-5 real speed, chuwi ubookx( VHT-NSS 1)
Wifi 5G farthest roomwith poco f5 pro  about 100( VHT-NSS 2).

rokezu, 2024-08-15:
Having the same problem with a Netgear R6850 and OpenWrt 23.05.4, horrible download speeds, expected upload speeds.

Have tried several 80MHz channels and some fiddling, not luck yet.  Using factory default image gives me expected speeds.

renatoyamane, 2024-08-29:
> Same problem here after upgrading 21.02-RC1 to 21.02-RC4 (OpenWrt 21.02.0-rc4 r16256-2d5ee43dc6) 21.02-RC1 seems ok.

This user said exactly when the problem started, so it's a good starting point to investigate.

rokezu, 2024-08-30:
I found a thread in openwrt.org which helped me while this is fixed in a future release: 

https://forum.openwrt.org/t/solved-openwrt-on-mt7621-mt7615n-devices-with-5ghz-problems/107392/288

Summarizing: just lower the TX power little by little until acceptable speeds are achieved. Out of 23 dBm (199 mW), I'm using it at 18 dBm (63 mW), and getting 85MB/s from a Netgear R6850 to a Intel AX200 wifi adapter.  I also get around 50-60MB/s in older Intel wifi adapters, (7260, 9560 I think).  Those speeds are acceptable for me.  Also, the AX200 replaces a Ralink one that was behaving erratically (1MB/s to top 20MB/s, but mostly around 5MB/s).  I would suggest to do an upgrade if possible.

Svyat-admin, 2025-01-10:
Beeline Turbo+ shows very low speed on 5 GHz, results are random for speed in tests near router. Changing channels and width, country, power doing nothing. Speed returns to normal after reset or two.

ivanusevanton, 2025-02-12:
I had a similar problem on Keenetic Viva (KN-1910). The patch to fix the incorrect power, as well as the power change did not solve my problem. As a temporary solution, periodically restarting the 5G interface helped, but then the problem returned.

This patch fixed everything: https://github.com/rany2/openwrt/blob/42894a8f58cda08de1e1f789285f23101c41ba47/package/kernel/mt76/patches/9000-wifi-mt76-mt7615-disable-precal.patch

Now clients with nss-1 are running at about 260/210 MBps. Before that, most of the time, they didn't work.

PS. What should I do to ensure that this patch is now applied when building openwrt from the official firmware selector? It will probably need to be upgraded, I don't have enough such knowledge.

KiralyCraft, 2025-04-26:
I have a Linksys EA7300 V2 running `OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.083.69138-0a0ce2a`. I reduced power from 29 dbM to 18 and now my 5 GHz AP is at least usable. The client I was having trouble with is NSS-2, so I didn't get to try @ivanusevanton's patch.

I see that there is no "patches" folder in the official repo, so upstreaming them would probably mean forking this repository, applying them on the raw source files, and then submitting a PR.

ivanusevanton, 2025-04-26:
@KiralyCraft The patch I mentioned above also affects clients with NSS-2. I did not specify this in this thread, because it is not critical for me. But I tested the speed for clients with NSS-2 and without a patch, the speed is about 80-100 Mbps, with a patch - 350-400 mbps. Things are really bad with NSS-1 clients. The speed without a patch is up to 1 Mbps.

KiralyCraft, 2025-04-26:
My NSS-2 client is a Meta Quest 2, and streaming is indeed most stable with at most 80 mbps, so you may be right. Unfortunately I'm a bit low on time with reflashing and compiling, but it does align with the symptoms. 

Do you happen to know whether the other patches are beneficial for this particular issue (or stability otherwise?)

ivanusevanton, 2025-04-26:
@KiralyCraft I've tried different patches. At the moment, this is the only patch that has solved my problem. It works on version 24.10.1. If we talk about optimizing wifi, I changed the DTIM to 1 and the Beacon Interval to 50. These settings are optimal for me. The improvement is especially noticeable when using VoWiFi.

navhawk80, 2025-04-26:
> What should I do to ensure that this patch is now applied when building openwrt from the official firmware selector? It will probably need to be upgraded, I don't have enough such knowledge.

Did you find the answer to your question? How did you apply the patch in the end?

ivanusevanton, 2025-04-26:
> Did you find the answer to your question? How did you apply the patch in the end?

I did not find an answer. At the moment, I'm compiling the firmware on a PC.

baracudaz, 2026-10-11:
Another data point: **D-Link DIR-853 R1** (MT7621AT + MT7615DN DBDC, EU) on **OpenWrt 25.12.5** (r33051, kernel 6.12.94, mt76 `2026.03.19~39c960c3`).

Same symptom as above: 5 GHz download (AP→client) is far below upload on all clients (MacBook Pro BCM4388 2SS, iPhone), even at -48 dBm. AP `tx retries` climbs 15–40k per 8 s test and the AP→client rate drops to VHT-MCS 2–3, while client→AP is fine. Channel (36 vs 100), width (40 MHz is worse) and macOS AWDL make no difference.

**Lowering txpower fixes most of it.** iperf3 AP→Mac, router as server, VHT80, Mbit/s, 3 runs each:

| txpower | close (-48 dBm) | far room (-70 dBm) |
|---|---|---|
| 23 dBm | 41–94 | 51–99 |
| 20 dBm | 68–152 | – |
| 17 dBm | 78–254 | 99–172 |
| 14 dBm | 182–348 | 113–182 |

**EEPROM notes** (factory partition vs. the stock D-Link firmware, which uses MediaTek's `mt_wifi.ko` with its own `/etc_ro/wlan/MT7615_EEPROM.bin` instead of the flash partition):

- `0x037` (`NIC_CONF_1`+1) = `0x20` → `TSSI_5G` off → `mt7615_ext_pa_enabled()` is true for 5 GHz, so `mt7615_eeprom_get_target_power_index()` returns `MT_EE_EXT_PA_5G_TARGET_POWER` (`0x0f3`). That's the same path as #1129. Here `0x0f3` = `0x28` in factory (stock file: `0x00`). On the DIR-853 A3 in #1129 it's `0x04` and power gets capped at 9 dBm; on this R1 it seems to overdrive the PA instead.
- `0x052` (`MT_EE_CALDATA_FLASH`) = `0x02` in factory (stock file `0x13`), so pre-calibration is loaded from flash, which is what `9000-wifi-mt76-mt7615-disable-precal.patch` clears.
- The per-group 5 GHz target bytes at `0x070`/`0x098` are identical in byte 0 (`0x17`), but bytes 3–4 of each 5-byte group are `00 00` in factory and populated in the stock file; `0x142`/`0x16a` (chains 2/3) differ more.

Running at 14 dBm for now. I plan to test the precal patch next and will report back.

baracudaz, 2026-10-11:
Follow-up to my DIR-853 R1 report above, with two updates.

**1. Correction: clean re-test.** My earlier Mac numbers were taken while a Wi-Fi scanner app (WiFi Explorer) was running, which makes the client leave the channel periodically. Re-tested with it closed. Same router (25.12.5, mt76 `39c960c3`), Mac in a far room at -67…-71 dBm, iperf3 to a wired host, VHT80, 3 runs each, Mbit/s:

| txpower | AP→client | AP→client ×4 | client→AP | AP→client rate at end |
|---|---|---|---|---|
| 14 dBm | 181 / 198 / 324 | 210 / 163 / 286 | 126 / 168 / 148 | VHT-MCS 9 |
| 23 dBm | 121 / 118 / 85 | 124 / 89 / 108 | 141 / 116 / 135 | VHT-MCS 3 |

So the earlier conclusion holds without the scanner: at 23 dBm AP→client collapses to MCS 3; at 14 dBm it reaches MCS 9 and download roughly doubles. Client→AP is unaffected.

**2. The precal patch doesn't help on this board.** Instead of rebuilding mt76 I cleared `MT_EE_CALDATA_FLASH` (EEPROM `0x052`) in the factory partition, `0x02` → `0x00`. That's equivalent to `9000-wifi-mt76-mt7615-disable-precal.patch`: in `39c960c3` the byte is only read by `mt7615_mcu_apply_rx_dcoc()` / `mt7615_mcu_apply_tx_dpd()`, `cal_free_data` doesn't touch it, and there's no checksum. After reboot the driver's EEPROM in debugfs shows `00`. On this board only `RX_CAL` was set, so this only disables the RX DCOC precal; TX DPD was already off.

| | AP→client | AP→client ×4 | client→AP |
|---|---|---|---|
| 14 dBm, precal off | 202 / 240 / 197 | 192 / 226 / 233 | 142 / 175 / 173 |
| 23 dBm, precal off | 104 / 32 / 96 | 115 / 53 / 110 | 139 / 89 / 93 |

No meaningful change: same AP tx-retry counts (~40k per 8 s test), and the 23 dBm collapse is still there. On the DIR-853 R1 the dominant factor is TX power, not precal. Since 5 GHz is flagged ext-PA here (`0x037` = `0x20`, `TSSI_5G` off), power comes from the single `MT_EE_EXT_PA_5G_TARGET_POWER` byte (`0x0f3` = `0x28`), which is the same path as #1129. That looks like the place to look for ext-PA MT7615 boards: on the A3 it caps power at ~9 dBm, while here it seems to drive the PA past its linear range.

Workaround for now: `wireless.radioX.txpower='14'` on 5 GHz. Restoring the original factory partition.

baracudaz, 2026-10-11:
Update for the DIR-853 R1: the precal patch does not help there (see above). Root cause turned out to be 5 GHz PA compression via the ext-PA target power path; RSSI sweep, an experiment changing `0x0f3`, and measurements are in https://github.com/openwrt/mt76/issues/1129#issuecomment-6111580249. Workaround: `txpower` 14 dBm on 5 GHz.