openwrt/mt76 #564: Mesh network wpa3 problems for MT7622BV + MT7915E

spamtree, 2021-08-09:
Posted this information on [Belkin RT3200/Linksys E8450 WiFi AX discussion](https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/275?u=alphahere)

It was recommend by forum member "daniel" to post this information as a formal bug report for the MT7622BV + MT7915E.

Below is my previous post. Both devices used for the mesh network were Belkin RT3200 (in the post used the word belkin for Belkin RT3200)

Started with a fresh snapshot about 4-6 days old now. Decided not to use batman but keep it simple and use 802.11s. 

No encryption mesh worked nicely with 2.4Ghz and 5Ghz frequencies. 

No success with wpa3 encryption and 2.4Ghz. The 5Ghz radio worked perfectly with wpa3 encryption, but on first connection had kernel oops and is identical on both belkin. However, mesh works nicely after initial  kernel oops. The 5Ghz radio the mesh network only works with channel 36, no other channel will work with mesh network on the 5Ghz radio and channel width does not change anything. All error logs are posted below. 

Package versions and kernel modules

```
OpenWrt SNAPSHOT, r17247-022d1fe80f

Linux test 5.10.54 #0 SMP Sun Aug 1 19:13:00 2021 aarch64 GNU/Linux

wpad-mesh-wolfssl - 2021-05-22-b102f19b-29

hostapd v2.10-devel

wpa_supplicant v2.10-devel

module:		/lib/modules/5.10.54/mt76.ko
license:	Dual BSD/GPL
depends:	mac80211,cfg80211
name:		mt76
vermagic:	5.10.54 SMP mod_unload aarch64

module:		/lib/modules/5.10.54/mt7615e.ko
license:	Dual BSD/GPL
depends:	mt7615-common,mt76,mac80211,mt76-connac-lib
name:		mt7615e
vermagic:	5.10.54 SMP mod_unload aarch64

module:		/lib/modules/5.10.54/mt76-connac-lib.ko
license:	Dual BSD/GPL
depends:	cfg80211,mac80211,mt76
name:		mt76_connac_lib
vermagic:	5.10.54 SMP mod_unload aarch64

module:		/lib/modules/5.10.54/mac80211.ko
license:	GPL
depends:	cfg80211,compat
name:		mac80211
vermagic:	5.10.54 SMP mod_unload aarch64

module:		/lib/modules/5.10.54/cfg80211.ko
alias:		net-pf-16-proto-16-family-nl80211
license:	GPL
depends:	compat
name:		cfg80211
vermagic:	5.10.54 SMP mod_unload aarch64

```

working wireless config for both belkins with radio1. No working mesh with radio0 and wpa3

```
config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/18000000.wmac'
	option channel '1'
	option band '2g'
	option htmode 'HT20'
	option cell_density '0'

config wifi-device 'radio1'
[... 121 lines cut ...]
```

Mesh connection wifi and ethernet nodes. Mesh devices were about 40 cm apart.

```
$ iperf -c wifi_mesh_node.lan -w 512K
------------------------------------------------------------
Client connecting to wifi_mesh_node.lan, TCP port 5001
TCP window size:  416 KByte (WARNING: requested  512 KByte)
------------------------------------------------------------
[  3] local 192.168.20.20 port 37788 connected with 192.168.20.40 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec   713 MBytes   598 Mbits/sec

$ iperf -c ethernet_mesh_node.lan -w 512K
------------------------------------------------------------
Client connecting to ethernet_mesh_node.lan, TCP port 5001
TCP window size:  416 KByte (WARNING: requested  512 KByte)
------------------------------------------------------------
[  3] local 192.168.20.20 port 54116 connected with 192.168.20.41 port 5001
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  1.08 GBytes   927 Mbits/sec
```

Any other info needed or wanted please let me know!

nemesisdesign, 2021-08-14:
Can you try with `cell_density '1'` on OpenWrt 21.04 RC and let us know how it goes?

See https://github.com/libremesh/lime-packages/issues/837#issuecomment-898807227

lynxthecat, 2021-10-02:
Here is what I experienced with 17648-16e83a7491 on 5Ghz with encryption:
```
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.548775] ------------[ cut here ]------------
Thu Sep 30 10:03:58 2021 kern.warn kernel: [   21.553407] WARNING: CPU: 1 PID: 1196 at __skb_flow_dissect+0x1a0/0x13cc
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.560099] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard pppox ppp_generic nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc sch_cake poly1305_neon nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libchacha libblake2s_generic iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables hwmon crc_ccitt compat fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.560280]  x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel veth vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 seqiv uas usb_storage leds_gpio xhci_plat_hcd gpio_button_hotplug
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.664497] CPU: 1 PID: 1196 Comm: napi/phy1-7 Tainted: G S                5.10.64 #0
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.672317] Hardware name: Linksys E8450 (UBI) (DT)
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.677186] pstate: 00000005 (nzcv daif -PAN -UAO -TCO BTYPE=--)
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.683188] pc : __skb_flow_dissect+0x1a0/0x13cc
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.687796] lr : __skb_get_hash+0x6c/0x130
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.691883] sp : ffffffc010db3990
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.695190] x29: ffffffc010db3990 x28: ffffff8001bf1280
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.700496] x27: 0000000000000000 x26: 0000000000000008
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.705801] x25: 0000000000000000 x24: ffffff800365d418
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.711105] x23: 000000000000004a x22: ffffffc010db3b00
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.716409] x21: 000000000000ffba x20: ffffffc010a89580
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.721713] x19: ffffff80037aa300 x18: 00069452b66d9eec
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.727018] x17: 0000000000000015 x16: ffffffc0107e83e0
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.732324] x15: 0000000000000001 x14: 0000000000000070
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.737628] x13: 0000000000000080 x12: ffffffc010867488
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.742932] x11: 0000000000000001 x10: 0000000000000001
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.748237] x9 : 0000000000000021 x8 : ffffff8000bdc200
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.753541] x7 : 0000000000000000 x6 : 0000000000000000
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.758845] x5 : 0000000000000000 x4 : 0000000000000000
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.764149] x3 : ffffffc010db3b00 x2 : 0000000000000000
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.769453] x1 : 0000000000000000 x0 : 0000000000000000
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.774757] Call trace:
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.777199]  __skb_flow_dissect+0x1a0/0x13cc
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.781460]  __skb_get_hash+0x6c/0x130
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.785232]  ieee80211_schedule_txq+0x784/0x9cc [mac80211]
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.790731]  ieee80211_schedule_txq+0x964/0x9cc [mac80211]
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.796229]  ieee80211_tx_pending+0xe0/0x240 [mac80211]
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.801452]  tasklet_action_common.constprop.0+0x164/0x17c
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.806929]  tasklet_action+0x24/0x30
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.810583]  _stext+0x124/0x28c
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.813716]  do_softirq+0x74/0x80
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.817021]  __local_bh_enable_ip+0x88/0x90
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.821197]  napi_threaded_poll+0x94/0xf0
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.825200]  kthread+0x120/0x124
Thu Sep 30 10:03:58 2021 kern.debug kernel: [   21.828419]  ret_from_fork+0x10/0x18
Thu Sep 30 10:03:58 2021 kern.warn kernel: [   21.831985] ---[ end trace df02c2995fdd6ba6 ]---
```
spamtree has your 5Ghz been stable? I also see erros like:
```
Fri Oct  1 22:03:55 2021 kern.err kernel: [   29.325310] mt7915e 0000:01:00.0: Message 000021ed (seq 2) timeout
Fri Oct  1 22:04:35 2021 daemon.err hostapd: nl80211 driver initialization failed.
```
and my device is unusable until power off and power on.

spamtree, 2021-10-02:
@lynxthecat 
  
5GHz works fine  only on channel 36 after reboot. Mesh locks up with other channels. I am using the latest snapshot using the config in the previous post.  Kernel ops appears to be the same as yours and only occurs when mesh is connecting once connected no more kernel ops. I am not seeing  'nl80211 driver initialization failed'.  Currently mesh is not in production setup and mesh is used only for testing mesh but appears stable. 
```

[   15.710358] ------------[ cut here ]------------
[   15.714999] WARNING: CPU: 1 PID: 1093 at __skb_flow_dissect+0x19c/0x1180
[   15.721694] Modules linked in: iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 seqiv leds_gpio xhci_plat_hcd gpio_button_hotplug
[   15.772588] CPU: 1 PID: 1093 Comm: napi/phy1-7 Tainted: G S                5.10.64 #0
[   15.780408] Hardware name: Linksys E8450 (UBI) (DT)
[   15.785277] pstate: 40000005 (nZcv daif -PAN -UAO -TCO BTYPE=--)
[   15.791280] pc : __skb_flow_dissect+0x19c/0x1180
[   15.795889] lr : __skb_get_hash+0x70/0x134
[   15.799975] sp : ffffffc010e33980
[   15.803280] x29: ffffffc010e33980 x28: 0000000000000008 
[   15.808586] x27: ffffff8002110fc0 x26: 0000000000000000 
[   15.813890] x25: ffffff800142ae18 x24: 0000000000000046 
[   15.819194] x23: ffffffc010e33b00 x22: 000000000000ffc2 
[   15.824498] x21: ffffffc010a99580 x20: ffffff800271f200 
[   15.829801] x19: 0000000000000000 x18: ffffff80021b07f0 
[   15.835105] x17: 0000000000000080 x16: 000000000000143c 
[   15.840409] x15: 0000000000000000 x14: 00007e1af81e41c4 
[   15.845713] x13: ff7f000000000003 x12: 4242000004f61e00 
[   15.851018] x11: 01204e90801af81e x10: 41c4801af81e41c4 
[   15.856322] x9 : ffffff80028728c0 x8 : ffffff8000c25800 
[   15.861626] x7 : 0000000000000000 x6 : 0000000000000000 
[   15.866929] x5 : 0000000000000000 x4 : 0000000000000000 
[   15.872234] x3 : ffffffc010e33b00 x2 : 000000000000ffda 
[   15.877537] x1 : 0000000000000000 x0 : 0000000000000000 
[   15.882841] Call trace:
[   15.885282]  __skb_flow_dissect+0x19c/0x1180
[   15.889544]  __skb_get_hash+0x70/0x134
[   15.893309]  ieee80211_schedule_txq+0x794/0x9dc [mac80211]
[   15.898796]  ieee80211_schedule_txq+0x974/0x9dc [mac80211]
[   15.904283]  ieee80211_tx_pending+0xdc/0x240 [mac80211]
[   15.909502]  tasklet_action_common.constprop.0+0x15c/0x16c
[   15.914978]  tasklet_action+0x24/0x30
[   15.918632]  _stext+0x124/0x28c
[   15.921765]  do_softirq+0x74/0x80
[   15.925071]  __local_bh_enable_ip+0x88/0x90
[   15.929247]  napi_threaded_poll+0x94/0xf0
[   15.933250]  kthread+0x120/0x124
[   15.936469]  ret_from_fork+0x10/0x18
[   15.940035] ---[ end trace c818a9e7a926faeb ]---
[   33.246355] ------------[ cut here ]------------
[   33.251010] WARNING: CPU: 0 PID: 2554 at airtime_link_metric_get+0xd0/0x710 [mac80211]
[   33.258916] Modules linked in: iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT nf_nat nf_flow_table nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 seqiv leds_gpio xhci_plat_hcd gpio_button_hotplug
[   33.309797] CPU: 0 PID: 2554 Comm: kworker/u4:5 Tainted: G S      W         5.10.64 #0
[   33.317702] Hardware name: Linksys E8450 (UBI) (DT)
[   33.322584] Workqueue: phy1 ieee80211_ibss_leave [mac80211]
[   33.328150] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO BTYPE=--)
[   33.334156] pc : airtime_link_metric_get+0xd0/0x710 [mac80211]
[   33.339989] lr : airtime_link_metric_get+0x3c/0x710 [mac80211]
[   33.345811] sp : ffffffc0114a39e0
[   33.349116] x29: ffffffc0114a39e0 x28: 0000000000000000 
[   33.354420] x27: ffffff800201c042 x26: ffffffc0089b4dd0 
[   33.359724] x25: 0000000000000000 x24: ffffffc0089a2e0c 
[   33.365028] x23: ffffff8002110fc0 x22: ffffff800201c038 
[   33.370333] x21: ffffff80028728c0 x20: 0000000000000000 
[   33.375637] x19: ffffff80017fe000 x18: ffffff80021b1810 
[   33.380941] x17: 0000000000000008 x16: 000000b5a2354998 
[   33.386245] x15: 0002dfedcbde758c x14: ffffffff000000d0 
[   33.391549] x13: 0000000000000243 x12: 51a425b8017ab42c 
[   33.396853] x11: 0000000000000000 x10: 0000000000000820 
[   33.402156] x9 : 0000000000000000 x8 : 00000000037e2002 
[   33.407461] x7 : 00000000a2a62801 x6 : 0000000000000000 
[   33.412765] x5 : 0000000000000000 x4 : 0000000000000080 
[   33.418069] x3 : 0000000000000000 x2 : ffffff8002110fc0 
[   33.423372] x1 : 0000000000000000 x0 : 0000000000000000 
[   33.428677] Call trace:
[   33.431125]  airtime_link_metric_get+0xd0/0x710 [mac80211]
[   33.436611]  airtime_link_metric_get+0x134/0x710 [mac80211]
[   33.442185]  mesh_rx_path_sel_frame+0xbc/0xab0 [mac80211]
[   33.447583]  ieee80211_mesh_rx_queued_mgmt+0xec/0x110 [mac80211]
[   33.453589]  ieee80211_ibss_leave+0xb00/0x1a60 [mac80211]
[   33.458981]  process_one_work+0x1d4/0x370
[   33.462981]  worker_thread+0x178/0x4e0
[   33.466722]  kthread+0x120/0x124
[   33.469942]  ret_from_fork+0x10/0x18
[   33.473507] ---[ end trace c818a9e7a926faec ]---

```