openwrt/mt76 #565: Kernel oops on Linksys E8450 (MT7622 + MT7915E)

gongtao0607, 2021-08-16:
/sys/fs/pstore/dmesg-ramoops-0:
```
Oops#1 Part1
<4>[128659.036653] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[128796.156968] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[128910.490268] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[129035.622986] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[129172.535291] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[129226.069120] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[129475.819308] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[129518.136433] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[129735.016656] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[130035.868157] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[130190.101241] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[130246.498251] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[130385.708547] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[131000.353555] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[131317.348100] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[131892.383481] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[131974.099289] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[132171.901826] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[132303.497368] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[132526.439882] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[132671.848467] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[132717.730026] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[132762.994666] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[132818.695945] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[132902.869040] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[133008.679624] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[133093.351962] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[133138.901571] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[133179.554892] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[133259.419902] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[133339.293674] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[133398.424739] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[133474.301225] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[133665.066387] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[133676.013494] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[134108.226124] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[134227.721927] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[134300.830328] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[134405.285322] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[134538.194713] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[134630.770466] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[135486.499666] Rekeying PTK for STA 36:ab:d6:cc:30:68 but driver can't safely do that.
<4>[169959.475842] Rekeying PTK for STA 36:ab:d6:cc:30:68 but driver can't safely do that.
<4>[170792.040041] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[171238.348830] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[171242.365969] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[171264.398098] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[174203.109465] Rekeying PTK for STA 36:ab:d6:cc:30:68 but driver can't safely do that.
<4>[174943.165249] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[178324.239546] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[178638.260226] Rekeying PTK for STA 36:ab:d6:cc:30:68 but driver can't safely do that.
<4>[178848.617618] Rekeying PTK for STA 6e:51:2b:82:b6:b8 but driver can't safely do that.
<4>[180020.763213] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[180546.382399] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[180795.214377] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[181460.098018] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[182013.263480] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[182388.992543] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[182595.407974] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[184479.085091] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[187106.260369] Rekeying PTK for STA aa:70:7f:2d:48:8b but driver can't safely do that.
<4>[187320.772544] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[187363.507512] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<4>[187457.313711] Rekeying PTK for STA 6e:7f:8d:1f:a6:e9 but driver can't safely do that.
<1>[187468.714971] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000053
<1>[187468.731598] Mem abort info:
<1>[187468.734487]   ESR = 0x96000005
[... 133 lines cut ...]
<7>[187468.911480] x13: 0000000000001d70 x12: 0000000000000000
<7>[187468.916871] x11: 0000000000000000 x10: ffffffc010a79570
<7>[187468.922263] x9 : ffffff801fed4530 x8 : 0000000003000000
<7>[187468.927656] x7 : ffffff80021abbc0 x6 : 00000001011d9793
<7>[187468.933048] x5 : 00000001011d996c x4 : 0000000000000000
<7>[187468.938440] x3 : ffffffc010dbbd18 x2 : 0000000000000002
<7>[187468.943832] x1 : 0000000000000000 x0 : ffffff8002162800
<7>[187468.949225] Call trace:
<7>[187468.951753]  mt76_tx_status_skb_add+0x1b0/0x230 [mt76]
<7>[187468.956974]  mt7915_queue_rx_skb+0xbcc/0x1598 [mt7915e]
<7>[187468.962279]  mt76_dma_rx_poll+0x2bc/0xc98 [mt76]
<7>[187468.966980]  __napi_poll+0x38/0x170
<7>[187468.970548]  napi_threaded_poll+0x80/0xe8
<7>[187468.974640]  kthread+0x120/0x128
<7>[187468.977948]  ret_from_fork+0x10/0x18
<0>[187468.981606] Code: a94153f3 a8c27bfd d65f03c0 d503201f (39414c24)
<4>[187468.987778] ---[ end trace 2cccb2822f78d8ce ]---
<0>[187469.004566] Kernel panic - not syncing: Oops: Fatal exception in interrupt
<2>[187469.011525] SMP: stopping secondary CPUs
<0>[187469.015530] Kernel Offset: disabled
<0>[187469.019098] CPU features: 0x0000002,04002004
<0>[187469.023444] Memory Limit: none
```

Strangely, I have two exact same models on the same nightly version (r17276, kernel 5.10.54) and almost identical configs (AFAIK only channels are different), but only 1 is crashing. The crashing one is under lighter usage but crashes almost every 4 days or so. The other one that has many wlan clients is solid stable like a rock.

gongtao0607, 2021-09-02:
After some diggings with disassembly and source code, I located this issue. It's in `mt7915_mac_add_txs_skb`:

```
	skb = mt76_tx_status_skb_get(mdev, wcid, pid, &list);
	if (!skb)
		goto out;

    ....

    out:
	mt76_tx_status_skb_done(mdev, skb, &list);
	mt76_tx_status_unlock(mdev, &list);
```

if `skb` is returned null, `__mt76_tx_status_skb_done` accesses `struct mt76_tx_cb *cb = mt76_tx_skb_cb(skb); cb->flag`; which is exactly 0x53

https://github.com/openwrt/mt76/blame/master/mt7915/mac.c#L1336

(Sorry I don't have a openwrt compiling/debugging setup. Hope these help.)

LorenzoBianconi, 2021-09-02:
>
> After some diggings with disassembly and source code, I located this issue. It's in mt7915_mac_add_txs_skb:
>
> skb = mt76_tx_status_skb_get(mdev, wcid, pid, &list);
> if (!skb)
> goto out;
>
>     ....

this issue has been already fixed here:
***@***.***/

but I think the patch is not merged yet.

Regards,
Lorenzo

> out:
> mt76_tx_status_skb_done(mdev, skb, &list);
> mt76_tx_status_unlock(mdev, &list);
>
> if skb is returned null, __mt76_tx_status_skb_done accesses struct mt76_tx_cb *cb = mt76_tx_skb_cb(skb); [...]

waxhell, 2021-09-03:
I encountered what looks like the same thing on a new snapshot pulled on 8/31/21. Looks like https://github.com/openwrt/mt76/issues/565#issuecomment-911269392 has a patch generated, but I am just including my pstore logs for additional reference.

`Oops#1 Part1
<6>[ 5370.776151] br-lan: port 5(wlan0) entered blocking state
<6>[ 5370.781555] br-lan: port 5(wlan0) entered forwarding state
<6>[ 5371.366785] br-lan: port 5(wlan0) entered disabled state
<6>[ 5372.130752] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
<6>[ 5372.137289] br-lan: port 5(wlan0) entered blocking state
<6>[ 5372.142602] br-lan: port 5(wlan0) entered forwarding state
<6>[118488.919642] device wlan0 left promiscuous mode
<6>[118488.924348] br-lan: port 5(wlan0) entered disabled state
<6>[118488.954141] device wlan1 left promiscuous mode
<6>[118488.958808] br-lan: port 6(wlan1) entered disabled state
<6>[118489.542443] br-lan: port 5(wlan0) entered blocking state
<6>[118489.547927] br-lan: port 5(wlan0) entered disabled state
<6>[118489.553594] device wlan0 entered promiscuous mode
<6>[118489.558536] br-lan: port 5(wlan0) entered blocking state
<6>[118489.563995] br-lan: port 5(wlan0) entered forwarding state
<6>[118489.766174] br-lan: port 6(wlan1) entered blocking state
<6>[118489.771715] br-lan: port 6(wlan1) entered disabled state
<6>[118489.777489] device wlan1 entered promiscuous mode
<6>[118489.782431] br-lan: port 6(wlan1) entered blocking state
<6>[118489.787930] br-lan: port 6(wlan1) entered forwarding state
<6>[118490.235668] br-lan: port 5(wlan0) entered disabled state
<6>[118490.253180] br-lan: port 6(wlan1) entered disabled state
<6>[118490.540228] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
<6>[118490.546863] br-lan: port 6(wlan1) entered blocking state
<6>[118490.552273] br-lan: port 6(wlan1) entered forwarding state
<6>[118490.944491] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
<6>[118490.951075] br-lan: port 5(wlan0) entered blocking state
<6>[118490.956562] br-lan: port 5(wlan0) entered forwarding state
<6>[119774.042645] mt7530 mdio-bus:00 lan1: Link is Down
<6>[119774.047630] br-lan: port 1(lan1) entered disabled state
<6>[119781.324603] mt7530 mdio-bus:00 lan1: Link is Up - 1Gbps/Full - flow control rx/tx
<6>[119781.332232] br-lan: port 1(lan1) entered blocking state
<6>[119781.337547] br-lan: port 1(lan1) entered forwarding state
<4>[120205.955517] ------------[ cut here ]------------
<6>[120205.960228] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
<4>[120205.967311] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x2a8/0x2b0
<7>[120205.973221] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table nf_conntrack_netlink nf_conntrack mt7915e mt7615e mt7615_common mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nfnetlink nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 seqiv uas usb_storage leds_gpio xhci_plat_hcd gpio_button_hotplug
<7>[120206.036900] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G S                5.10.61 #0
<7>[120206.044370] Hardware name: Linksys E8450 (UBI) (DT)
<7>[120206.049326] pstate: 40000005 (nZcv daif -PAN -UAO -TCO BTYPE=--)
<7>[120206.055410] pc : dev_watchdog+0x2a8/0x2b0
<7>[120206.059497] lr : dev_watchdog+0x2a8/0x2b0
<7>[120206.063584] sp : ffffffc010bcbd80
<7>[120206.066975] x29: ffffffc010bcbd80 x28: ffffffc010a7d580
<7>[120206.072366] x27: 0000000000000140 x26: 00000000ffffffff
<7>[120206.077756] x25: 0000000000000000 x24: 0000000000000000
<7>[120206.083147] x23: ffffff8000c55480 x22: 0000000000000001
<7>[120206.088538] x21: ffffffc010a96000 x20: ffffff8000c55000
<7>[120206.093928] x19: 0000000000000000 x18: 000000000000018f
<7>[120206.099318] x17: 0000000000000000 x16: 0000000000000000
<7>[120206.104709] x15: ffffffc010aa6a60 x14: 00000000000004ad
<7>[120206.110100] x13: 000000000000018f x12: ffffffc010bcba58
<7>[120206.115491] x11: ffffffc010afea60 x10: 00000000fffff000
<7>[120206.120881] x9 : ffffffc010afea60 x8 : 0000000000000000
<7>[120206.126272] x7 : 80000000fffff000 x6 : 0000000000000001
<7>[120206.131662] x5 : 0000000000000000 x4 : 0000000000000000
<7>[120206.137052] x3 : 0000000000000000 x2 : ffffff801febb028
<7>[120206.142443] x1 : ffffffc00f437000 x0 : 000000000000003f
<7>[120206.147833] Call trace:
<7>[120206.150358]  dev_watchdog+0x2a8/0x2b0
<7>[120206.154101]  call_timer_fn.constprop.0+0x20/0x80
<7>[120206.158796]  expire_timers+0xa8/0x110
<7>[120206.162536]  run_timer_softirq+0x110/0x290
<7>[120206.166710]  _stext+0x124/0x28c
<7>[120206.169931]  irq_exit+0xd8/0xf4
<7>[120206.173151]  __handle_domain_irq+0x7c/0xdc
<7>[120206.177327]  gic_handle_irq+0x68/0x8c
[... 163 lines cut ...]
<7>[283994.064360] x21: ffffff8002717bb8 x20: ffffff8002068828
<7>[283994.069750] x19: ffffff800222a800 x18: ffffff80019543b0
<7>[283994.075141] x17: 00000000ffffffff x16: 0000000000000000
<7>[283994.080531] x15: 0000000000000004 x14: ffffff800222c918
<7>[283994.085922] x13: 0000000000000000 x12: ffffff8001fc0c48
<7>[283994.091313] x11: ffffff800222c800 x10: ffffffc010a96000
[... cut ...]
<7>[283994.139495]  napi_threaded_poll+0x84/0xf0
<7>[283994.143585]  kthread+0x120/0x124
<7>[283994.146891]  ret_from_fork+0x10/0x18
<0>[283994.150548] Code: 54fffa41 d2800000 d65f03c0 d503201f (39414c24)
<4>[283994.156718] ---[ end trace 6f84a83193b17c4b ]---
<0>[283994.169736] Kernel panic - not syncing: Oops: Fatal exception in interrupt
<2>[283994.176692] SMP: stopping secondary CPUs
<0>[283994.180697] Kernel Offset: disabled
<0>[283994.184264] CPU features: 0x0000002,04002004
<0>[283994.188610] Memory Limit: none`

narenniranjan, 2021-09-21:
is there an update on this or any suggestions for resolution? Still getting this if I leave my 5 GHz band on.

Blaok, 2021-09-30:
Looks like this is fixed by 8044311f5de569dcfb6f8ef39bb68d073abad1bf.

gongtao0607, 2021-11-12:
Updated and have been stably up for 30 days. Cheers!

closed 2021-11-12 by gongtao0607