openwrt/mt76 #569: mt7628 transmits old frames

gnif, 2021-08-25:
When using `iperf3` to a client periodically the throughput will drop to `0` and `iwlwifi` will report the following:

```
reached 10 old SN frames from 44:01:bb:8a:d4:2a on queue 9, stopping BA session on TID 0
```

According to the `iwlwifi` source, this is caused by a bad access point (see: https://www.spinics.net/lists/linux-wireless/msg189740.html), as such the blame is pointed at the mt72 module.

Possibly unrelated, but if high throughput traffic (80-90Mbit/s) continues for more than an hour eventually the `mt76_wmac` just stops working. Attempting to restart the wifi (`wifi up` on OpenWrt) causes the following:

```
mt76_wmac 10300000.wmac: MCU message 08 (seq 8) timed out
```

Recovery is only possible by rebooting the hardware.

gnif, 2021-08-25:
I am not familiar enough with the structs/standards here but from what I gather the following looks wrong:

https://github.com/openwrt/mt76/blob/8b3d8e2de0842553935a809e549c00e2bfdcccdc/tx.c#L35

Shouldn't this be:

```
mtxq->agg_ssn = IEEE80211_SEQ_TO_SN(le16_to_cpu(hdr->seq_ctrl)); 
```

?

easyteacher, 2021-09-16:
The temp workaround is to set htmode to legacy to disable SMPS.

nbd168, 2021-09-16:
@gnif: the existing code is correct. Other places that set agg_ssn use IEEE80211_SN_TO_SEQ, and ieee80211_send_bar expects that format as well.

gnif, 2021-09-17:
Sorry, I forgot to come back and update this. Yes, I determined the same, the code is correct here.

@easyteacher is this on the station or the ap? Disabling it on the station is not a viable workaround.

easyteacher, 2021-09-17:
> Sorry, I forgot to come back and update this. Yes, I determined the same, the code is correct here.
> 
> @easyteacher is this on the station or the ap? Disabling it on the station is not a viable workaround. I tested this also on a station and this made no difference.

On the AP or the router side.

gnif, 2021-09-17:
Not sure how that issue #576 relates to this one, there is no kernel oops generated by either the repeat of old frames or the eventual device failure at high throughputs.

easyteacher, 2021-09-17:
To disable SMPS while not to sacrifice speed, this patch may help.

```
diff --git a/mt7603/main.c b/mt7603/main.c
index 8edea1e7..c72264ef 100644
--- a/mt7603/main.c
+++ b/mt7603/main.c
@@ -633,7 +633,7 @@ mt7603_sta_rate_tbl_update(struct ieee80211_hw *hw, struct ieee80211_vif *vif,
 	mt7603_wtbl_set_rates(dev, msta, NULL, msta->rates);
 	msta->rate_probe = false;
 	mt7603_wtbl_set_smps(dev, msta,
-			     sta->smps_mode == IEEE80211_SMPS_DYNAMIC);
+			     1 == 0);
 	spin_unlock_bh(&dev->mt76.lock);
 }
```

But if a connected client switches PS mode from on to off, it will lose connection.

valga, 2021-11-16:
Got the same issue with OpenWrt 21.02.1 running on TP-Link TL-WR841N v13 (MT7628), solved it by downgrading to 19.07.2. According to changelog, mt76 driver has been updated in 19.07.3 and there is at least one [report](https://forum.openwrt.org/t/tl-wr841n-v13-wifi-stability/79673) with WiFi stability issue on 19.07.4.

Symptoms: all wireless clients loose connection, access point disappears from available networks. Sometime it appears back after 10-20 seconds, sometimes mt76 driver just hangs and stops responding to commands (e.g. you can't even restart wireless). All wired clients have no issues at the same time.

Changing frequency from n to b/g/n partially helps: wireless clients doesn't loose connection, but they have temporary hiccups (about 5-10 seconds) in traffic.

Problem appears to be related with wireless traffic speed: I never had it with ADSL as uplink, but have it constantly with GPON.

sanitariu, 2021-11-16:
Do you know which changes broke this ? I am running r18070-49f615022c and problem is still here :(

valga, 2021-11-16:
@sanitariu Not yet.

If downgrading to 19.07.2 solves the issue for your, we can trace to changes introduced between 19.07.2 and 19.07.4/19.07.7.

DragonBluep, 2021-12-14:
> @sanitariu Not yet.
> 
> If downgrading to 19.07.2 solves the issue for your, we can trace to changes introduced between 19.07.2 and 19.07.4/19.07.7.

I remember this problem already existed in branch 18.06, probably last September. When I run a network speed test, the wi-fi will disappear within one minute.

myapple, 2022-01-04:
@DragonBluep   I wonder where this firmware came from, and what has been modified

DragonBluep, 2022-01-04:
@myapple Just collected from internet. Basically, I can confirm that it is officially released. You can get it from [here](https://github.com/DragonBluep/mtk-openwrt-sdk-v4.1.0.4/tree/master/wifi_drv)
By the way, can anyone tell me if it is legal to collect these codes? I am a little worried.

DragonBluep, 2022-01-04:
It looks like the internet will go down as usual, but it will come back up quickly so we won't feel it in most scenarios. My guess is that the core overheating triggered the temperature protection and the new firmware added an interrupt handler. Just a guess, in fact I don't know anything about mt76 drivers.
[image]

gnif, 2022-01-05:
Why are you guys still posting to this issue? I have an EXACT error reported of duplicated frames! Not just some random dropout under high loads. Unless you're seeing the exact same error I originally reported your issue is UNRELATED.

>> reached 10 old SN frames from 44:01:bb:8a:d4:2a on queue 9, stopping BA session on TID 0

> According to the iwlwifi source, this is caused by a bad access point (see: https://www.spinics.net/lists/linux-wireless/msg189740.html), as such the blame is pointed at the mt72 module.

I have since moved away from this hardware as there is NO solution to the re-transmission bug at this time, which is what this github issue is about! If this issue keeps going further and further off topic I will simply close it.

webysther, 2023-04-15:
Same problem with OpenWrt 22.03.4 r20123-38ccc47687 and MT7621 (Xiaomi Mi Router 4A Gigabit Edition).

Another find, when connecting with ac devices/old works but when using with ax intel on windows crash wlan without any kernel/system log relevant information.

brandsimon, 2023-05-21:
The same happens on my RE200 (MT7628AN) on OpenWRT 22.03, but it also happens on the stock firmware. Does this probably indicate a hardware/firmware issue instead of a driver issue?
As @webysther noted, I only have the problem with ax devices, not with ac devices.

@nbd168 
Is your comment above correct? It says the change to `IEEE80211_SN_TO_SEQ` is wrong, but it also says other places use `IEEE80211_SN_TO_SEQ`, which is correct.

dfateyev, 2023-09-04:
I can reproduce this on MT7621 with MT7603E on OpenWrt 23.05 branch (with all recent improvements on MT7603 / MT7628 provided). Once a 2.4GHz client with `iperf` loads AP to max, the messages about duplicate SN frames appear in logs and the WLAN connection speed drops for a while.

dfateyev, 2024-06-01:
> I can reproduce this on MT7621 with MT7603E on OpenWrt 23.05 branch (with all recent improvements on MT7603 / MT7628 provided).

Cannot reproduce the issue anymore, using the same hardware with `openwrt-master`  snapshot [7514d9bb6f](https://github.com/openwrt/openwrt/commit/7514d9bb6f7239fc5c962590b31bd3b6cd0ea7c5) (dated on May 27, with mt76 [2024-05-17](https://github.com/openwrt/openwrt/commit/ed23d4d00dcc26f69da6a2fbb89f21d99f3e25d9)). Looks like fixed, but will retest with an `openwrt-24.xx` release once it is available.

federicobriata, 2024-12-23:
> Cannot reproduce the issue anymore, using the same hardware with `openwrt-master` snapshot [7514d9bb6f](https://github.com/openwrt/openwrt/commit/7514d9bb6f7239fc5c962590b31bd3b6cd0ea7c5) (dated on [...]

The release 23.05.5 dated 2024-09- 24 it's the first release to include this fix and solve this problem for me. thanks @dfateyev for find this out!